davidjo/snd_hda_macbookpro #2: Document where you've forked these files

meeuw, 2019-08-18:
Could you please add a reference from which kernel/branch/sha you've forked your work?

It would be very practical for porting your changes to another kernel version, at this moment we'll have to guess what sources you've used as a base for patch_cirrus.c etc.

Nice work though! :confetti_ball:

davidjo, 2019-08-23:
All this has been done using a Ubuntu 18.04 LTS system.
I just used the kernel source dpkg of this distribution.
The initial kernel was 4.15.0-54.
Im not sure how you figure the original kernel source commit from this.
patch_cirrus.c is a bit more complicated as I think I added some initial patches for 8409 from the base kernel source as found on the various audio on MB 14,3 etc bug lists - although those patches are now basically ignored - apart from adding the 8409 device id.

meeuw, 2019-08-24:
Thanks for the information!

I've done some digging and found:

https://launchpad.net/ubuntu/+source/linux/4.15.0-54.58

Which references to:
[linux_4.15.0.orig.tar.gz](https://launchpad.net/ubuntu/+archive/primary/+sourcefiles/linux/4.15.0-54.58/linux_4.15.0.orig.tar.gz)
[linux_4.15.0-54.58.diff.gz](https://launchpad.net/ubuntu/+archive/primary/+sourcefiles/linux/4.15.0-54.58/linux_4.15.0-54.58.diff.gz)
[linux_4.15.0-54.58.dsc](https://launchpad.net/ubuntu/+archive/primary/+sourcefiles/linux/4.15.0-54.58/linux_4.15.0-54.58.dsc)

`linux_4.15.0.orig.tar.gz` is equal to `linux-4.15.tar.gz`

`linux_4.15.0-54.58.diff.gz` doesn't seem include patches on `patch_cirrus.c`, the other files are additions.

Reference for `patch_cirrus.c` at [git.kernel.org](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/sound/pci/hda/patch_cirrus.c?h=v4.15)

closed 2019-08-24 by meeuw