davidjo/snd_hda_macbookpro #5: Support for MBP 14,2

Mydlear, 2019-09-17:
Hello,
I would have to request the support for MBP 14,2 if you mind.
I managed to make my speakers work by swapping the device id's with the MBP14,1 in the files you provided. 
So I think the support for MBP 14,2 should be added. 

Regards,
Thomas.

leifliddy, 2019-09-17:
Just to clarify the statement above, the device of the MBP14,2 is 0x106b3600
Changing the device id in the following lines from 0x106b3300 to 0x106b3600 resulted in working speakers on the MBP14,2
```
patch_cirrus.c:       SND_PCI_QUIRK(0x106b, 0x3300, "MacBookPro 13,1", CS8409_MBP131),
patch_cirrus_mb141_data84.h:        retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_SUBSYSTEM_ID, 0x00000000, 0x106b3300, 8); // 0x001f2000
patch_cirrus_mb141_data84.h:        retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_SUBSYSTEM_ID, 0x00000000, 0x106b3300, 22); // 0x001f2000
patch_cirrus_new84.h:// 14,1 0x106b3300
patch_cirrus_new84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_new84.h:   else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_new84.h:   else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
```
Perhaps the conditional statements could be modified to include the 0x106b3300 device id. 
**I believe the MBP14,2 has an additonal microphone (3 total), so support for that may need to be added in the future. 

https://github.com/leifliddy/macbook12-audio-driver/issues/2

davidjo, 2019-10-07:
Sorry for the delay. Ive been working on a big update to handle headset playing and plugin/unplug which turned out to be quite complicated involving concurrency issues which I have hacked around. Will be updating the code shortly. I only have a 14,3 machine so cannot test any other machines. So it would seem if your above patch works the 14,2 has the same SSM3515 speaker chips as the 14,1 plus the same Cirrus 8409 chip - although have not found confirmation on the Web yet. Ive moved the 14,1 code into the main code base in the new update so someone will need to test and see if it still works with the new code.

davidjo, 2019-10-07:
PS Sorry I also thought Id get eg email notified when issues were opened but this doesnt appear to be happening - not sure why as I think my settings should give them.

Mydlear, 2019-10-12:
I can confirm that the MBP 14,2 (at least mine) has the Cirrus CS8409 chip, but i didn't found the real name of the speaker in hardinfo.
Concerning the update, i tried with changing every 0x106b3300 string to 0x106b3600 or 0x3300 to 0x3600 :
`patch_cirrus.c:       SND_PCI_QUIRK(0x106b, 0x3300, "MacBookPro 13,1", CS8409_MBP131),
patch_cirrus_mb141_data84.h:        retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_SUBSYSTEM_ID, 0x00000000, 0x106b3300, 8); // 0x001f2000
patch_cirrus_mb141_data84.h:        retval = snd_hda_codec_read_check(codec, codec->core.afg, 0, AC_VERB_GET_SUBSYSTEM_ID, 0x00000000, 0x106b3300, 22); // 0x001f2000
patch_cirrus_new84.h:// 14,1 0x106b3300
patch_cirrus_new84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_new84.h:   else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_new84.h:   else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_new84.h:   else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_new84.h:   else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {
patch_cirrus_real84.h:        else if (codec->core.subsystem_id == 0x106b3300) {`
And it works very well, both the speakers and the jack.

davidjo, 2019-10-14:
Thanks for the update - Ill create a new commit which should handle both 14,1 and 14,2 machines.

davidjo, 2019-10-15:
I have done an update so should handle both 14,1 and 14,2 if you would like to check it.
Only codes in patch_cirrus_new84.h and patch_cirrus_real84.h need to be fixed.

leifliddy, 2019-10-15:
doesn't this block also need to be modified in patch_cirrus.c?
```
static const struct snd_pci_quirk cs8409_fixup_tbl[] = {
       SND_PCI_QUIRK(0x106b, 0x3300, "MacBookPro 13,1", CS8409_MBP131),
       SND_PCI_QUIRK(0x106b, 0x3900, "MacBookPro 14,3", CS8409_MBP143),
       {} /* terminator */
};
```

davidjo, 2019-10-15:
No - that was from a previous patch from the sound bugs but I dont use it -
it became rapidly clear that the quirk feature was not going to be able
to handle the changes needed - if you look the line //snd_hda_pick_fixup(codec,
cs8409_models, cs8409_fixup_tbl,...
is commented in routine patch_cs8409.
I have noticed that the 0x106b, 0x3300 13,1 seems to be the same as the
14,1 model - not clear this is just an error in the original
patch.

Mydlear, 2019-10-16:
I just tested and it works fine by modifying : 
```
patch_cirrus_new84.h: else if (codec->core.subsystem_id == 0x106b3300 || codec->core.subsystem_id == 0x106b3600) {
patch_cirrus_new84.h: else if (codec->core.subsystem_id == 0x106b3300 || codec->core.subsystem_id == 0x106b3600) {
patch_cirrus_new84.h: else if (codec->core.subsystem_id == 0x106b3300 || codec->core.subsystem_id == 0x106b3600) {
patch_cirrus_new84.h: else if (codec->core.subsystem_id == 0x106b3300 || codec->core.subsystem_id == 0x106b3600) {
patch_cirrus_new84.h: else if (codec->core.subsystem_id == 0x106b3300 || codec->core.subsystem_id == 0x106b3600) {
patch_cirrus_real84.h: else if (codec->core.subsystem_id == 0x106b3300 || codec->core.subsystem_id == 0x106b3600) {
patch_cirrus_real84.h: else if (codec->core.subsystem_id == 0x106b3300 || codec->core.subsystem_id == 0x106b3600) {
patch_cirrus_real84.h: else if (codec->core.subsystem_id == 0x106b3300 || codec->core.subsystem_id == 0x106b3600) {
patch_cirrus_real84.h: else if (codec->core.subsystem_id == 0x106b3300 || codec->core.subsystem_id == 0x106b3600) {
patch_cirrus_real84.h: else if (codec->core.subsystem_id == 0x106b3300 || codec->core.subsystem_id == 0x106b3600) {
patch_cirrus_real84.h: else if (codec->core.subsystem_id == 0x106b3300 || codec->core.subsystem_id == 0x106b3600) {
patch_cirrus_real84.h: else if (codec->core.subsystem_id == 0x106b3300 || codec->core.subsystem_id == 0x106b3600) {
patch_cirrus_real84.h: else if (codec->core.subsystem_id == 0x106b3300 || codec->core.subsystem_id == 0x106b3600) {
```
Thanks again for the work.

closed 2019-10-23 by davidjo