davidjo/snd_hda_macbookpro #6: MacBook Pro 14,1

Matt-HJ-Bailey, 2019-10-24:
Thanks for working on this. I have successfully installed this patch on my MacBook Pro 14,1, based off kernel 5.3.7 and running Ubuntu 19.10. I've done a little bit of testing, and found that:

- Headphones aren't correctly identified. The OS detects headphones whether any are plugged in or not, and sound comes out the main speaker.

- Only the "front-left" speaker works, but this appears to come out centrally.

What other testing do you think needs to be done?

davidjo, 2019-10-27:
Yes - Ive had this problem - problem is this is a combination of factors and I dont really know the internals of the full Linux audio path ie pulseaudio, alsa and kernel module.
Its not that the OS is physically detecting headphones (that is separate) its that pulse (I think) is labelling the output as a headphone because its confused by the labelling of the speaker nodes (ie HDA nodes 0x02 and 0x03). If you actually plugin headphones it should turn off the speakers and play thro the headphones - at least thats what I observed on my machine - and if you unplug it will start playing thro speakers again - even tho labelled as headphones!!
I did play with pactl and saw that it had assigned a 2 channel sink (which I think is why its labelled headphones) as the default. You should have a sink containing the word surround - which I think is what it labels the 4 channel kernel input - I set my default sink to this surround source - although this seemed to be an unlabelled source so Im still not sure what the right thing to do is.
Again I dont really know how these various components are interacting and what the right solution is - maybe try on the linux audio lists.

Im assuming by front-left you mean the speaker test of ubuntu - must admit havent tried this for a while - although I definitely know that both left and right speakers are being used by the kernel when sending 4 channel audio. Sounds like its duplicating sound on left and right channels.

risen, 2019-10-27:
@Matt-HJ-Bailey did you compile on vanilla 5.3.7? I seem to be missing some header files…

Matt-HJ-Bailey, 2019-10-28:
> @Matt-HJ-Bailey did you compile on vanilla 5.3.7? I seem to be missing some header files…

I also had a few header files that I was missing. They just appear to have been relocated -- I think it was hda_codec.h that was missing for me, which I copied from /include/sound/hda_codec.h to /sound/pci/hda/hda_codec.h 

There is almost certainly a better way to do it, by changing #include "hda_codec.h" to #include <sound/hda_codec.h>, but I haven't gotten around to it yet.

@davidjo I've had a listen and both left and right speakers are playing in the speaker test. I think the Ubuntu speaker test matches up to the front-left and front-right channels used by pulseaudio, but I also don't know how these things talk to one another. I'll let you know if I make any progress.

arturasb, 2019-11-07:
> I also had a few header files that I was missing. They just appear to have been relocated -- I think it was hda_codec.h that was missing for me, which I copied from /include/sound/hda_codec.h to [...]

Could you please share the instruction how to build and apply/install this patch ?

noispirit, 2019-11-08:
> Could you please share the instruction how to build and apply/install this patch ?

You'll need to compile the whole kernel from source

Basically, you grab the kernel tarball at kernel.org, cd to the linux folder, throw the files you downloaded inside /sound/pci/hda/ (It will ask you to replace patch_cirrus.c, do it), copy hda_codec.h from /include/sound/ and paste it to the same location from patch_cirrus.c and compile the kernel as you normally do, or follow a guide specific to your linux flavor on how to compile a kernel.

Currently it doesn't support suspend/wake or the integrated mic.

leifliddy, 2019-12-27:
@noispirit You don't need to build an entire kernel. You just need to build the ```snd-hda-codec-cirrus.ko``` module, copy it to ```/lib/modules/$(shell uname -r)/updates/```, run ```depmod -a```, then ```reboot```. 

Take a look at this project:
https://github.com/leifliddy/macbook12-audio-driver

You could easily modify the ```install.cirrus.driver.sh``` script  to support this project.

arturasb, 2019-12-27:
> @noispirit You don't need to build an entire kernel.
> You just need to:
> 
>     1. build the `snd-hda-codec-cirrus.ko` module
> 
>     2. copy it to `/lib/modules/$(shell uname -r)/updates/`
> 
>     3. run `depmod -a`
> 
>     4. `reboot`
> 
> 
> Take a look at this project:
> https://github.com/leifliddy/macbook12-audio-driver
> 
> You could easily modify the `install.cirrus.driver.sh` script to support this project.

@noispirit , can you please give me more details on how to build & install only cirrus driver ?

leifliddy, 2019-12-27:
I just threw this together 
https://github.com/leifliddy/snd_hda_macbookpro

see if it works for you...

arturasb, 2019-12-27:
> I just threw this together
> https://github.com/leifliddy/snd_hda_macbookpro
> 
> see if it works for you...

Holly crap ! I mean Spirit !!!! You know... You made my day... No, whole 2019!!! It works. I got sound from my MBP's speakers, then I connected headphones - it worked, sound was there. Then I disconnected head phones - sound came out of laptop's speakers again !!!!
What a Holiday gift for a user wanting fully working post-2019 MBP ! Now such MBP is usable for content consumption too :)
I can't express enough how this "quick'n'dirty" hack made me happy :)  

I wish you best for coming 2020 and I wish this patch to land to kernel for next release :)

devalanche, 2019-12-28:
> > I just threw this together
> > https://github.com/leifliddy/snd_hda_macbookpro
> > see if it works for you...
> 
> Holly crap ! I mean Spirit !!!! You know... You made my day... No, whole 2019!!! It works. I got sound from my MBP's speakers, then I connected headphones - it worked, sound was there. Then I disconnected head phones - sound came out of laptop's speakers again !!!!
> What a Holiday gift for a user wanting fully working post-2019 MBP ! Now such MBP is usable for content consumption too :)
> I can't express enough how this "quick'n'dirty" hack made me happy :)
> 
> I wish you best for coming 2020 and I wish this patch to land to kernel for next release :)
> 
> P.S.: forgot to mention that volume control works for both speakers & headphones.

what kernel version do you have this working on?

arturasb, 2019-12-28:
> what kernel version do you have this working on?

```
$ uname -a

Linux pop-os-macbook-pro 5.3.0-7625-generic #27~1576337002~19.10~bc3488b-Ubuntu SMP Sat Dec 14 18:31:03 UTC  x86_64 x86_64 x86_64 GNU/Linux
```

devalanche, 2019-12-28:
whoa thank you very much!!
looking forward to try this on my 14,3 machine when the family days are passed

devalanche, 2019-12-30:
> I just threw this together
> https://github.com/leifliddy/snd_hda_macbookpro
> 
> see if it works for you...

So many thanks dude, sooo many thanks!!
much respect and love to all of you sharing and helping so much!! <3

devalanche, 2019-12-31:
is there any hope to get the microphone working as well?

mealissa, 2020-01-03:
anything working for 13.3?

fiowro, 2020-01-23:
@leifliddy  I also have MacBookPro14,1, and I tried https://github.com/leifliddy/snd_hda_macbookpro to install the audio driver. Auto-switching between headphones and speakers works. Headphones also work. But speakers are somehow misplaced. I check sound in gnome-control-center (because of the nice speaker test interface). When I click the "front left" button in it, I hear sound from the right speaker. When I click any other buttons ("front right", etc.) I get no sound. 

In macOS both left and right speakers are working, so that's not faulty hardware.

I'll be happy to provide any possible help with debugging this.  Any logs, output of commands, patching, etc.

P.S. At first I tried macbook12-audio-driver (just didn't understand it is only suitable for 12-inch macbooks) - hope this made no effect (I noticed some pulseaudio conf files in the repository, so I removed /etc/pulse and reinstalled pulseaudio to be sure).

P.P.S. Great thanks for you and everyone who paid their skills and knowledge to make this driver possible!

fiowro, 2020-01-23:
BTW, I use Archlinux, and the kernel is 5.4.13.

noispirit, 2020-01-23:
This seems to be an issue with PulseAudio rather than the audio driver. Looking at the patch everything seems to be fine (Except for the internal mic).

kswzr, 2020-02-28:
Currently built on linux kernel 5.3.0-40-generic for Macbook13,1 (13" without Touchbar) () and It works basically. Expecting following issues:
* fading to the right channel affects no sound - maybe only playing left channel? 
* sound quality quite bad - no bass

davidjo, 2020-03-03:
Im waiting for next Ubuntu LTS before doing more changes (plus working on other things).

As for the mike - I believe the HDA is setup as per OSX - what I have not looked at is how to connect that to PulseAudio etc.

What you need to do is ignore PulseAudio and go direct to Alsa  - look for the input channels and use arecord I believe direct on the input hardware device it lists (I have never used recording).

Once you can record using this then can figure how to connect to PulseAudio.

julian1059, 2020-03-27:
Hello and thanks for putting the drivers together for the macbookpro. I recently picked up a 2017 21.5” iMac and am running Ubuntu 19.10 with the 5.3.0-42-generic kernel installed. Obviously sound is not working. It looks like the audio cards on the devices are identical. Has anyone tested this driver on an iMac? When I went ahead and installed via the shell script provided by @leifliddy I  am not seeing any results. Messed with alsa, default sound settings, etc .... Still no luck. Confirmed the driver did indeed build and the .ko is in /lib/modules/(kernel version)/updates. The iMac is showing an option to use Analog Stereo, Stereo 2.1 and Stereo 4.0 Outputs. Any further configs that need to take place? Appreciate the help and am willing to help test any iMac related dev or issues.

davidjo, 2020-05-09:
So Ive updated to Ubuntu 20.04 and looking at these issues.
I have discovered the problem with mike sound input - it turns out the cirrus 8409 chip doesnt fit the auto config heuristics - which only expect a max of 8 ADCs but the 8409 has 18 - so it only stores the 1st 8 it finds and of course the ADCs needed are after the 1st 8.
I have done a fixup for this and can report I have been able to record sound with the internal mike - now looking at a headset mike.

DiegoMagdaleno, 2020-05-19:
@davidjo so do you mean the microphone should work now? and what about the dual speakers?, thank you for your hard work on this driver, I am looking into it for a personal project.

davidjo, 2020-05-28:
I can confirm the results from the Ubuntu speaker tests on 20.04 - sound only appears when Front Left activated - but I have checked and sound is coming from both/all 4 speakers.
This is because pulse is incorrectly interpreting the 4 channel output sink as a surround sink when it is not so the test is not working correctly.
Yes the internal mike/headset mikes are now delivering sound - just low volume.
(I used Audacity to normalize a recording using arecord to get a reasonable volume).
This needs to be fixed in the pulse audio setup - as does interpreting the speaker output correctly.

milan475, 2020-07-17:
First of all thanks for the great work you've done on the audio driver so far! I've done some testing, but I don't think pulse is at fault for not interpreting the 4 channel PCM correctly. I can confirm the sound coming from all speakers however when I created a test 4 channel S32_LE wav file even when playing it directly to hw0,0 with aplay by using 

`aplay -D hw:0,0 -M StereoTest32.wav `

You'l only hear the sound from the first channel. Interesting enough it seems to replicate the sound of this first channel over all speakers though. Any idea whats going on?

https://www.dropbox.com/s/9br71wfotmm30vm/StereoTest32.wav?dl=0

davidjo, 2020-07-20:
Ill try this out over next day or so - but I know the alsa module will duplicate a 2 channel input to 4 channels if the audio format requires 4 channels - not tried a mono source.
Have been looking at the channel ordering and whereas I assumed we had 2 stereo pairs of left/right tweeter, left/right woofer it may actually be left tweeter, left woofer, right tweeter right woofer.

milan475, 2020-07-20:
Allright, I'm curious what you'll find. I found no way to get any output from anything else then the first PCM channel. The 3rd and 4rd channel of my sample wav file where completely silent too.

I think this is also one of the reasons why the sound under linux is so 'thin' now. The sound information of the right channel is completely discarded.

Btw I also did play around win the allowed sound formats by changing

`hinfo->formats = SNDRV_PCM_FMTBIT_S32_LE | SNDRV_PCM_FMTBIT_S24_LE;`

S24_LE and S24_3L sound both crippled. I was able to get correct sound from S16_LE actually. So that one can be enabled too.

davidjo, 2020-07-23:
So Ive tried your file and am hearing similar results - yet looking at my old stuff it looks like I have such files and I seem to remember I definitely got sound from either left or right speakers - trying to resurrect the old 18.04 and kernel 4.15 to see whats going on.
Now thats interesting - in the initial stages I had issues with figuring the byte packing/format - and if you got the byte packing/format mis-aligned thats how you blew your ears - so its really interesting that S16_LE seems to be working.

davidjo, 2020-07-23:
By the way did you notice that if you play StereoTest32.wav on OSX eg Quicktime you only hear the 1st 2 tracks (correctly in the left then right speakers) - silence for tracks 3 and 4.

davidjo, 2020-07-23:
Here  are some thoughts - I know the PCI bus cannot do 24 bit transfers - but thats what S24_3LE is - and that is definitely the format Apple sets the HDA nids 0x02 and 0x03 to
- now Ive been working with S24_LE - which is 24 bits packed in 32 - or S32_LE - because they seemed to deliver sound without byte packing issues - but maybe they still have packing issues - just instead of blowing your ears we are getting consisent null bytes.
Ive also noticed that Apples 4 channel 24 bit format is 96 bits or 3 32 PCI bits ie it could be sent in 3 PCI 32 bit packets and maybe the PCI side of the Linux driver is messing things up somehow.
In the current driver the format of nids 0x02/0x03 is NOT Apples format (0x4031) - I just allowed S24_LE/S32_LE (x4041)
At one stage I did manage to do S24_3LE - but had to create a special raw format file file for this - almost all programs create S24_LE ie 32 bits - and I allowed S24_3LE format in the driver.
(Id also note that the format returned by the internal mike seems to be S24_LE and that is still set to Apples format ie S24_3LE or 0x4031).

davidjo, 2020-07-24:
So thats annoying - I have gotten an 18.04/kernel 4.15.0-52 working with an old version of the kernel module and StereoTest32.wav plays perfectly - sound from all 4 channels. It also confirms the channel order for macbook pro 14,3 is left tweeter, left woofer, right tweeter, right woofer (I can actually hear the difference in sound between the tweeter and woofers).
Now question is have I messed up something with all the updates or is it a kernel version difference.

milan475, 2020-07-26:
Well that's actually good news right! Then we just have to figure out what changed what causes this issue. Do you still have the source code for the old kernel module so we can compile that against a new kernel and see if that works?

davidjo, 2020-07-26:
Finally sorted - yes its in a git repo - of course I messed up - I missed that each amp needs a specific configuration to pick out the single channel it should use - so when I went to a single function to do amp setup I should have added a channel argument - and I didnt - so now fixed (it does mean every amp was working - just they all used the single 1st channel ie front left - as you suspected the right channel was not used at all).
The speaker order is definitely confirmed as front-left,rear-left,front-right,rear-right in pulse terms.
I think Ive got a pulse fixup for this using its remap sink - well it seems to be working for me - but pulse is very confusing.
Obviously we need something like this either in Alsa or pulse otherwise a standard stereo input will just play on the left speakers and be duplicated on the right. If you can do a custom alsa config that would be great.
Ill push an update over the next day.

milan475, 2020-07-26:
Awesome! 🎉 Looking forward to the commit. I just put a fresh Fedora on my machine so I can check it out. I'll see if I can write an ALSA Config. Definitely not an expert, put I played a bit with it the other day, and it seemed pretty straightforward. I think this page https://alsa.opensrc.org/Low-pass_filter_for_subwoofer_channel_(HOWTO) basically has the information we'll need.

milan475, 2020-07-26:
Alright, I couldn't wait, so I managed to figure out what you meant (have to say that wasn't too easy without any C experience). I haven't done anything with ALSA yet, but in the driver I configured it so that the amps for the left speaker always use the first PCM channel and and the amps for the right speaker always use the second. I also played around with the volume levels so the base is a bit increased.

If you then set pulse to just use the stereo sink and not the surround one the channels will be mapped correctly and it actually sounds pretty good. Definitely on par with sound quality in Bootcamp for example. When I have the time i'll check if I can use the high and low pass filters in ALSA to make it even better.

For those who can't wait here's a repo with the above changes: https://github.com/milan475/snd_hda_macbookpro

davidjo, 2020-07-27:
Thanks for your test audio file by the way - I would like to include your file in the repo as a test waveform - let me know if this is OK.

milan475, 2020-07-27:
Of course! Thank you again for all the work you've done so far. I've been digging a bit through the code yesterday. And it seems like a lot of work! How did you ever manage to extract the logic from the Apple drivers?

davidjo, 2020-07-27:
I didnt really extract the exact logic but just used the driver c++ vtable functions as a guide to split up the OSX logs - just old style hacking - and guessing patterns from the HDA command logs.

mfauvain, 2020-09-17:
Hi, thx for the work on this, I have a 14,1 and would like to test the driver, shall I use the one at https://github.com/milan475/snd_hda_macbookpro ?
with this driver, is the mic working?
thx

mfauvain, 2020-11-17:
Just tried https://github.com/milan475/snd_hda_macbookpro, and sound works on speaker and headphones, also internal mic works but headset mic does not.
Thx for the great work.

gogoprog, 2021-03-22:
Hello, I have a 14,1 and the sound only works on the speakers.

As soon as I plug my headphones I can see this:
```[ 68.134565] snd_hda_codec_cirrus hdaudioC0D0: cs_8409_interrupt_action - plug in 
[ 70.224558] snd_hda_codec_cirrus hdaudioC0D0: cs_8409_interrupt_action - headset detected 
[ 70.230534] snd_hda_codec_cirrus hdaudioC0D0: cs_8409_headset_type_detect_event headset does not have mike!! 
[ 70.507694] snd_hda_codec_cirrus hdaudioC0D0: cs_8409_headset_type_detect JACK DISCONNECT NOT IMPLEMENTED!! 
[ 78.286062] snd_hda_codec_cirrus hdaudioC0D0: cs_8409_read_status_and_clear_interrupt - ERROR - max count exceeded
```

And then the sound stops working, even on speaker.

Any clue?

davidjo, 2021-03-23:
JACK DISCONNECT is listed if the code thinks you have unplugged the headphones - in this case before it has finished the plugin events - which leads to the resulting behaviour.

So first ensure that you do things slowly - it takes something like up to 5 secs after headphone plugin for the audio switch to actually occur (this delay also happens on OSX) (unlike previous OSX models where it was near instantaneous).
Also leave a good amount of time between audio events eg start playing, plugin or plugin then start playing.

(By the had you started playing audio then plugged in or plugged in before start playing.)

(Although the timing difference 70.2 to 70.5 suggests not - unless you had a bad jack contact eg it didnt plugin in cleanly).

I havent used this in a while so might have to recheck it.

From this info cant really get anymore - would need to re build module using the debug logging - under the hda-... directory under build edit the Makefile and comment the normal build flags and uncomment the debug build flags.

Probably want to save the current modules (under /lib/modules/kernelversion/updates ubuntu) before doing next.

Clean, rebuild and reinstall then reboot.

Note with logging the time for audio system events gets even longer so again do things slowly - leave a good 30 secs before any audio event - eg start playing, plugin headset.
 
Probably want to restore the non-debug modules (remember to do a depmod after copying files to /lib/modules) as soon as possible.

Calea-Designs, 2021-10-21:
Feel bad reviving this, but I cannot get this functioning on my 14,1 via any of the instructions or suggestions I've read so far on the latest Kubuntu 21.10 release. Kernel is 5.13.0-20. After installing I get no change in the system tray audio manager. Bringing up PulseAudio, it only shows me 5 unavailable HDMI outputs. Alsamixer recognizes the CS8409 but only displays 5 <S/PDIF> entries that I can't do anything with. No tinkering that I've done in either has made any progress. This could be user error as this is my first time back in linux in ~5 years, but the driver's .ko file is in the build directory referenced in the script like it should be, so I think I did it right. USB audio works fine, haven't been able to test Bluetooth or HDMI as of yet.

davidjo, 2021-10-21:
Well first, Bluetooth/HDMI has nothing to do with this module - this is purely for the internal speakers, internal mikes and headphone socket - as far as I know Bluetooth/HDMI should work out of the box - at least that whats the various MacBookPro linux audio sites say.

Second is to make sure this module is loaded - check for snd_hda_codec/snd_hda_intel lines in the linux system log.
In particular look for snd_hda_codec_cirrus - there should be a lot of them if this module is loaded.

Calea-Designs, 2021-10-22:
Yes! That was it!

Removing snd-hda-codec-cs8409 from /lib/modules/5.13.0-20-generic/kernel/sound/pci/hda seems to let your module take over again. After a little tweaking I have everything working as expected. Thank you!

As a less important note, regarding the tweaking: Pulse needed to be set to the plain "Analog Stereo Output" profile, otherwise there was *terrible* loud popping whenever adjusting the volume and occasionally even just randomly. There's still slight popping when adjusting the volume on that profile but it's quiet and 100% usable. Sound quality is great otherwise.

TabulateJarl8, 2021-11-22:
So I seem to have gotten the driver working on a Macbook 14,1 after adding `CONFIG_MODULE_SIG=n` to the top of the Makefile and rebuilding. It worked fine until about a day later when it just seem to have randomly stopped working. I'm not entirely sure what the issue could be caused by, but here's some information if this helps you guys spot a pattern.

```
System:
  Kernel: 5.11.0-40-generic x86_64 bits: 64 compiler: N/A 
  Desktop: Gnome 3.36.9 Distro: Ubuntu 20.04.3 LTS (Focal Fossa) 
Machine:
  Type: Laptop System: Apple product: MacBookPro14,1 v: 1.0 serial: <filter> 
  Mobo: Apple model: Mac-B4831CEBD52A0C4C v: MacBookPro14,1 serial: <filter> 
  UEFI: Apple v: 429.140.8.0.0 date: 06/13/2021 
Audio:
  Device-1: Intel Sunrise Point-LP HD Audio driver: snd_hda_intel v: kernel 
  bus ID: 00:1f.3 
  Device-2: Broadcom and subsidiaries 720p FaceTime HD Camera driver: N/A 
  bus ID: 03:00.0 
  Sound Server: ALSA v: k5.11.0-40-generic 
```
```
# dmesg | grep cirrus
[    3.848560] snd_hda_codec_cirrus: loading out-of-tree module taints kernel.
[    3.848629] snd_hda_codec_cirrus: module verification failed: signature aclearnd/or required key missing - tainting kernel
[    3.849669] snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS8409: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
[    3.849674] snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    3.849676] snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
[    3.849678] snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
[    3.849680] snd_hda_codec_cirrus hdaudioC0D0:    inputs:
[    3.849681] snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x44
[    3.849683] snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x3c
[    4.325755] snd_hda_codec_cirrus hdaudioC0D0: out of range cmd 0:ffff:f0d:0
[    7.107363]  cs_8409_init+0x1e/0x480 [snd_hda_codec_cirrus]
```
Ubuntu Sound menu is only displaying "Dummy Device" in the list of output devices

closed 2022-05-10 by davidjo

lightitrd, 2023-02-16:
> I just threw this together https://github.com/leifliddy/snd_hda_macbookpro
> 
> see if it works for you...

worked for me too
linux mint 21.1 cinnamon on macbook pro A1708
but it was necessary to change kernel version to 6.1 

```
sudo apt install linux-oem-22.04c && reboot  
uname -a  
sudo ./install.cirrus.driver.sh && reboot
```

1notea, 2025-07-26:
Hello everyone, I recently decided to install Ubuntu 22.04.5 LTS (GNU/Linux 6.8.0‑64‑generic x86_64) on my MacBookPro14,1, but the repositories aren’t working. Does anyone have a solution for the audio issues on the latest builds?