davidjo/snd_hda_macbookpro #8: MacBook Pro 13,1

kaloz, 2020-04-14:
Hey,

Wanted to confirm the driver works on the 13,1 with one major issue: it makes (at least) wifi go crazy. The symptom and the fixup is similar to https://bugzilla.kernel.org/show_bug.cgi?id=196019 . Without the sound driver wifi works all the time.

davidjo, 2020-04-23:
Curious - at what point does this happen - looking at the referenced bug that seems to involve sleep - which is not handled at all in the driver.

jaxjexjox, 2020-12-16:
@kaloz I'm also using 13,1 and the subwoofer, isn't working for me, so audio is 'tinny'
which distro did you use?

Information here, perhaps it'll save David some time.
https://github.com/davidjo/snd_hda_macbookpro/issues/23

nerdyness, 2021-03-26:
I've just installed Ubuntu budgie on my MBP 13.1" and this did the trick to get my sound to work! I'm not sure if it's louder under Mac OS or not, I just wanted to say thanks for the work :+1: