davidjo/snd_hda_macbookpro #9: Can't make it work on Macbook Pro 13,3

cristianmiranda, 2020-04-20:
Hi guys, first I want to thank you for the amazing work you've been doing so far, much appreciated!

I have Ubuntu running on a Macbook Pro 2017 (13,3) with touchbar. Since I don't have any experience compiling kernel modules and patches I tried out https://github.com/leifliddy/snd_hda_macbookpro instructions and got the following output:

Am I doing something wrong here? I'm on kernel 5.6 I think

Thanks in advance!

```
root@Linux-MBP:~/snd_hda_macbookpro# ./install.cirrus.driver.sh
--2020-04-20 18:18:58--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.6.0.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.217.176, 2a04:4e42:33::432
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.217.176|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-04-20 18:18:59 ERROR 404: Not Found.

--2020-04-20 18:18:59--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.6.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.217.176, 2a04:4e42:33::432
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.217.176|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c:120:18: error: field ‘last_play_time’ has incomplete type
  120 |  struct timespec last_play_time;
      |                  ^~~~~~~~~~~~~~
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c:125:18: error: field ‘first_play_time’ has incomplete type
  125 |  struct timespec first_play_time;
      |                  ^~~~~~~~~~~~~~~
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c: In function ‘cs_8409_hda_jack_detect_enable_callback’:
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c:1434:6: warning: unused variable ‘err’ [-Wunused-variable]
 1434 |  int err;
      |      ^~~
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c: In function ‘cs_8409_init’:
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c:1495:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1495 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c:1477:6: warning: unused variable ‘ret_unsol_enable’ [-Wunused-variable]
 1477 |  int ret_unsol_enable = 0;
      |      ^~~~~~~~~~~~~~~~
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c: In function ‘cs_8409_cs42l83_callback’:
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c:1844:18: warning: unused variable ‘spec’ [-Wunused-variable]
 1844 |  struct cs_spec *spec = codec->spec;
      |                  ^~~~
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c: In function ‘cs_8409_automute’:
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c:1885:18: warning: unused variable ‘spec’ [-Wunused-variable]
 1885 |  struct cs_spec *spec = codec->spec;
      |                  ^~~~
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c: In function ‘patch_cs8409’:
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c:2133:8: error: implicit declaration of function ‘getnstimeofday’ [-Werror=implicit-function-declaration]
 2133 |        getnstimeofday(&(spec->last_play_time));
      |        ^~~~~~~~~~~~~~
In file included from /root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c:2150:
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus_new84.h: In function ‘cs_8409_save_and_clear_stream_format’:
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus_new84.h:697:13: warning: unused variable ‘format_id_sv’ [-Wunused-variable]
  697 |         int format_id_sv;
      |             ^~~~~~~~~~~~
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus_new84.h:696:13: warning: unused variable ‘channel_id_sv’ [-Wunused-variable]
  696 |         int channel_id_sv;
      |             ^~~~~~~~~~~~~
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus_new84.h:695:13: warning: unused variable ‘stream_tag_sv’ [-Wunused-variable]
  695 |         u32 stream_tag_sv;
      |             ^~~~~~~~~~~~~
In file included from /root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus_new84.h:863,
                 from /root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.c:2150:
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus_boot84.h: In function ‘cs42l83_device_id’:
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus_boot84.h:2002:39: warning: ‘<<’ in boolean context, did you mean ‘<’ ? [-Wint-in-bool-context]
 2002 |         ret_id = ((retval_dev & 0xff) << 16) || (retval_rev & 0xff);
      |                  ~~~~~~~~~~~~~~~~~~~~~^~~~~~
/root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus_boot84.h: In function ‘cs42l83_tip_sense’:
[... 230 lines cut ...]
linux-5.6/sound/pci/hda/hda_tegra.c
linux-5.6/sound/pci/hda/hp_x360_helper.c
linux-5.6/sound/pci/hda/patch_analog.c
linux-5.6/sound/pci/hda/patch_ca0110.c
linux-5.6/sound/pci/hda/patch_ca0132.c
linux-5.6/sound/pci/hda/patch_cirrus.c
linux-5.6/sound/pci/hda/patch_cmedia.c
linux-5.6/sound/pci/hda/patch_conexant.c
linux-5.6/sound/pci/hda/patch_hdmi.c
linux-5.6/sound/pci/hda/patch_realtek.c
linux-5.6/sound/pci/hda/patch_si3054.c
linux-5.6/sound/pci/hda/patch_sigmatel.c
linux-5.6/sound/pci/hda/patch_via.c
linux-5.6/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/5.6.0-1007-oem/build M=/root/snd_hda_macbookpro/build/hda-5.6.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.6.0-1007-oem'
  CC [M]  /root/snd_hda_macbookpro/build/hda-5.6.0/patch_cirrus.o
make[1]: Leaving directory '/usr/src/linux-headers-5.6.0-1007-oem'
cp snd-hda-codec-cirrus.ko /lib/modules/5.6.0-1007-oem/updates

contents of /lib/modules/5.6.0-1007-oem/updates
total 4
drwxr-xr-x 2 root root 4096 abr 20 14:36 dkms

```

noispirit, 2020-04-20:
@cristianmiranda Probably it needs some rework to make it work for kernel 5.6. I'd suggest to rollback your kernel to 5.4 or less for now, but maybe someone might have another idea.

cristianmiranda, 2020-04-20:
@noispirit that did the trick! I rolled back to 5.4 and was able to compile the driver. Thank you for the advices! I guess I won't upgrade my kernel any time soon, Or now that I have the .ko file compiled, maybe I can upgrade and reuse that?

closed 2020-04-20 by cristianmiranda

leifliddy, 2020-04-21:
@cristianmiranda I looks like some functions + data structures got renamed with kernel 5.6
I've just updated https://github.com/leifliddy/snd_hda_macbookpro 
Can you try compiling this module again for kernel 5.6 and let me know if it works?

cristianmiranda, 2020-04-21:
@leifliddy Nice! Works perfectly on 5.6. Thank you!
Quick question, do you know if this works for macbooks with T2 chip as well? I have a 2018 macbook pro 15".