davidjo/snd_hda_macbookpro #10: No audio after suspend

cristianmiranda, 2020-04-21:
Hi again folks! I've noticed that after sleeping and resuming my macbook (closing the lid) I get no audio. Maybe some modules need to be reloaded?

davidjo, 2020-04-23:
Im afraid Ive not attempted to fix sleep/resume so not surprising - there seem to be many issues with this generally running linux on all sorts of Macs.
It does appear Apple resets the audio totally after a sleep so may be as simple as detecting the wake then re-sending all the i2c setup commands.

cristianmiranda, 2020-04-23:
No worries. I'm willing to experiment a little bit, maybe you can point me in the right direction. What do you mean by re-sending all the "i2c" setup commands? By looking at the code I see that the .ko files are copied over, but nothing else. Right?

davidjo, 2020-05-01:
I mean that all the commands in the function cs_8409_boot_setup (and anything it calls) need to be redone - looks as tho its going to need a resume function defining (there is a suspend in the original patch_cirrus.c) from quick scan of other codecs.
(Its also likely some parts of cs_8409_boot_setup wont need to be done and redoing may cause issues).
This is a module execution problem - not a module install problem - the copy of the .ko files is a module install stage.

cristianmiranda, 2020-07-23:
I understand. I'm able to try stuff if you could point me on the right direction (I have no clue about what to do in order for this to work). Let me know if you come up with any ideas. Thanks again!.

Booligoosh, 2022-05-20:
Hey @davidjo, I was taking a cursory look through this to see if it was something that I (as a developer with some rusty basic C experience but no kernel/os experience) would be able to implement or not, and while looking through the code, I saw a bunch of suspend/resume code behind a `CONFIG_PM` flag. I tried compiling with this flag and nothing seemed to change.

I was just wondering if this feature is already a work-in-progress, and if not, whether the code behind those flags affects where I'd need to start if trying to implement it. Thanks!

davidjo, 2022-05-20:
I think the reason nothing changed is because as far as I can tell now the CONFIG_PM flag is set on by default in the main kernel build.
I have added some functions (cs_8409_apple_resume and cs_8409_apple_suspend) mainly to see if they were called  - which they are - which indicates CONFIG_PM is active (you need to update the Makefile and uncomment the KBUILD_EXTRA_CFLAGS labelled debug (and comment the KBUILD_EXTRA_CFLAGS line labelled normal) re-build etc (you will get a lot of kernel messages).
So far it appears the codec suspend function is called after finishing streaming - and then the resume function obviously on starting streaming.
What little code analysis Ive done shows there are suspend/resume functions in hda_codec.c which are installed into a hda_codec_driver_pm struct (what I dont understand is the difference between the functions labelled runtime and those without).
If you look at the main patch_cs8409.c you will see suspend/resume functions for the Dell version for the CS42L42.
The big question is what needs to be re-done on the CS42L83 chip (which is different from the CS42L42) which is undocumented - plus for Apple also have to do something about the amps - all these subchips are permanently powered on at the moment. Still not sure about power on/off of the CS8409 either altho the Dell version should help here.
There may be some info from the CS4208_xx.inf files as found on the Bootcamp Windows install image which lists the commands needed on power on/off for the CS42L83.

megabitsenmzq, 2023-10-16:
Do we have any update on this?