davidjo/snd_hda_macbookpro #11: iMac 19.1 sound issue

aydar-y, 2020-05-07:
I am trying to make sound work on iMac 19.1. It looks like I have the same sound hardware as most latest MBPs. Everything I tried never worked. Here is the result of inxi -SMA:

System:
  Host: aydar-iMac Kernel: 5.6.11-050611-generic x86_64 bits: 64 
  Desktop: Gnome 3.36.1 Distro: Ubuntu 20.04 LTS (Focal Fossa) 
Machine:
  Type: Laptop System: Apple product: iMac19,1 v: 1.0 serial: C02ZD2YBJV3N 
  Mobo: Apple model: Mac-AA95B1DDAB278B95 v: iMac19,1 
  serial: C02936600JNKDMX2S UEFI: Apple v: 1037.100.345.0.0 date: 02/17/2020 
Audio:
  Device-1: Intel Cannon Lake PCH cAVS driver: snd_hda_intel 
  Device-2: AMD Ellesmere HDMI Audio [Radeon RX 470/480 / 570/580/590] 
  driver: snd_hda_intel 
  Sound Server: ALSA v: k5.6.11-050611-generic

aydar-y, 2020-05-07:
The result of  aplay -L is below:

default
    Playback/recording through the PulseAudio sound server
surround21
    2.1 Surround output to Front and Subwoofer speakers
surround40
    4.0 Surround output to Front and Rear speakers
surround41
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50
    5.0 Surround output to Front, Center and Rear speakers
surround51
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
null
    Discard all samples (playback) or generate zero samples (capture)
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
hdmi:CARD=HDMI,DEV=0
    HDA ATI HDMI, HDMI 0
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=1
    HDA ATI HDMI, HDMI 1
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=2
    HDA ATI HDMI, HDMI 2
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 3
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=4
    HDA ATI HDMI, HDMI 4
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=5
    HDA ATI HDMI, HDMI 5
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=6
    HDA ATI HDMI, HDMI 6
    HDMI Audio Output
dmix:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 0
    Direct sample mixing device
dmix:CARD=HDMI,DEV=7
    HDA ATI HDMI, HDMI 1
    Direct sample mixing device
dmix:CARD=HDMI,DEV=8
    HDA ATI HDMI, HDMI 2
    Direct sample mixing device
dmix:CARD=HDMI,DEV=9
    HDA ATI HDMI, HDMI 3
    Direct sample mixing device
dmix:CARD=HDMI,DEV=10
    HDA ATI HDMI, HDMI 4
    Direct sample mixing device
dmix:CARD=HDMI,DEV=11
    HDA ATI HDMI, HDMI 5
    Direct sample mixing device
dmix:CARD=HDMI,DEV=12
    HDA ATI HDMI, HDMI 6
    Direct sample mixing device
dsnoop:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 0
    Direct sample snooping device
dsnoop:CARD=HDMI,DEV=7
    HDA ATI HDMI, HDMI 1
    Direct sample snooping device
dsnoop:CARD=HDMI,DEV=8
    HDA ATI HDMI, HDMI 2
    Direct sample snooping device
dsnoop:CARD=HDMI,DEV=9
    HDA ATI HDMI, HDMI 3
    Direct sample snooping device
dsnoop:CARD=HDMI,DEV=10
    HDA ATI HDMI, HDMI 4
    Direct sample snooping device
dsnoop:CARD=HDMI,DEV=11
    HDA ATI HDMI, HDMI 5
    Direct sample snooping device
dsnoop:CARD=HDMI,DEV=12
    HDA ATI HDMI, HDMI 6
    Direct sample snooping device
hw:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 0
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=7
    HDA ATI HDMI, HDMI 1
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=8
    HDA ATI HDMI, HDMI 2
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=9
    HDA ATI HDMI, HDMI 3
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=10
    HDA ATI HDMI, HDMI 4
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=11
    HDA ATI HDMI, HDMI 5
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=12
    HDA ATI HDMI, HDMI 6
    Direct hardware device without any conversions
plughw:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 0
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=7
    HDA ATI HDMI, HDMI 1
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=8
    HDA ATI HDMI, HDMI 2
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=9
    HDA ATI HDMI, HDMI 3
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=10
    HDA ATI HDMI, HDMI 4
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=11
    HDA ATI HDMI, HDMI 5
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=12
    HDA ATI HDMI, HDMI 6
    Hardware device with all software conversions
usbstream:CARD=HDMI
    HDA ATI HDMI
    USB Stream Output
sysdefault:CARD=PCH
    HDA Intel PCH, Generic Analog
    Default Audio Device
front:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    Front speakers
surround21:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    4.0 Surround output to Front and Rear speakers
surround41:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
dmix:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    Direct sample mixing device
dsnoop:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    Direct sample snooping device
hw:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    Direct hardware device without any conversions
plughw:CARD=PCH,DEV=0
    HDA Intel PCH, Generic Analog
    Hardware device with all software conversions
usbstream:CARD=PCH
    HDA Intel PCH
    USB Stream Output

aydar-y, 2020-05-07:
**The result of systool -m snd_hda_intel -v is below:**

Module = "snd_hda_intel"

  Attributes:
    coresize            = "53248"
    initsize            = "0"
    initstate           = "live"
    refcnt              = "3"
    srcversion          = "A1E50883834174BF30093EF"
    taint               = ""
    uevent              = <store method only>

  Parameters:
    align_buffer_size   = "-1"
    bdl_pos_adj         = "-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1"
    beep_mode           = "N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N,N"
    dmic_detect         = "Y"
    enable_msi          = "-1"
    enable              = "Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y,Y"
    id                  = "(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null)"
    index               = "1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1"
    jackpoll_ms         = "0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0"
    model               = "imac27,(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null)"
    patch               = "(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null),(null)"
    pm_blacklist        = "Y"
    position_fix        = "-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1"
    power_save          = "1"
    power_save_controller= "Y"
    probe_mask          = "-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1"
    probe_only          = "0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0"
    single_cmd          = "-1"
    snoop               = "-1"

  Sections:

aydar-y, 2020-05-07:
**Here's pacmd list-cards result:**

2 card(s) available.
    index: 0
	name: <alsa_card.pci-0000_01_00.1>
	driver: <module-alsa-card.c>
	owner module: 7
	properties:
		alsa.card = "0"
		alsa.card_name = "HDA ATI HDMI"
		alsa.long_card_name = "HDA ATI HDMI at 0x81760000 irq 32"
		alsa.driver_name = "snd_hda_intel"
		device.bus_path = "pci-0000:01:00.1"
		sysfs.path = "/devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card0"
		device.bus = "pci"
		device.vendor.id = "1002"
		device.vendor.name = "Advanced Micro Devices, Inc. [AMD/ATI]"
		device.product.id = "aaf0"
		device.product.name = "Ellesmere HDMI Audio [Radeon RX 470/480 / 570/580/590]"
		device.string = "0"
		device.description = "Ellesmere HDMI Audio [Radeon RX 470/480 / 570/580/590]"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-pci"
	profiles:
		output:hdmi-stereo: Digital Stereo (HDMI) Output (priority 900, available: no)
		output:hdmi-surround: Digital Surround 5.1 (HDMI) Output (priority 800, available: no)
		output:hdmi-surround71: Digital Surround 7.1 (HDMI) Output (priority 800, available: no)
		output:hdmi-stereo-extra1: Digital Stereo (HDMI 2) Output (priority 700, available: no)
		output:hdmi-surround-extra1: Digital Surround 5.1 (HDMI 2) Output (priority 600, available: no)
		output:hdmi-surround71-extra1: Digital Surround 7.1 (HDMI 2) Output (priority 600, available: no)
		output:hdmi-stereo-extra2: Digital Stereo (HDMI 3) Output (priority 700, available: no)
		output:hdmi-surround-extra2: Digital Surround 5.1 (HDMI 3) Output (priority 600, available: no)
		output:hdmi-surround71-extra2: Digital Surround 7.1 (HDMI 3) Output (priority 600, available: no)
		output:hdmi-stereo-extra3: Digital Stereo (HDMI 4) Output (priority 700, available: no)
		output:hdmi-surround-extra3: Digital Surround 5.1 (HDMI 4) Output (priority 600, available: no)
		output:hdmi-surround71-extra3: Digital Surround 7.1 (HDMI 4) Output (priority 600, available: no)
		output:hdmi-stereo-extra4: Digital Stereo (HDMI 5) Output (priority 700, available: no)
		output:hdmi-surround-extra4: Digital Surround 5.1 (HDMI 5) Output (priority 600, available: no)
		output:hdmi-surround71-extra4: Digital Surround 7.1 (HDMI 5) Output (priority 600, available: no)
		output:hdmi-stereo-extra5: Digital Stereo (HDMI 6) Output (priority 700, available: no)
		output:hdmi-surround-extra5: Digital Surround 5.1 (HDMI 6) Output (priority 600, available: no)
		output:hdmi-surround71-extra5: Digital Surround 7.1 (HDMI 6) Output (priority 600, available: no)
		output:hdmi-stereo-extra6: Digital Stereo (HDMI 7) Output (priority 700, available: no)
		output:hdmi-surround-extra6: Digital Surround 5.1 (HDMI 7) Output (priority 600, available: no)
		output:hdmi-surround71-extra6: Digital Surround 7.1 (HDMI 7) Output (priority 600, available: no)
		off: Off (priority 0, available: unknown)
	active profile: <off>
	ports:
		hdmi-output-0: HDMI / DisplayPort (priority 5900, latency offset 0 usec, available: no)
			properties:
				device.icon_name = "video-display"
		hdmi-output-1: HDMI / DisplayPort 2 (priority 5800, latency offset 0 usec, available: no)
			properties:
				device.icon_name = "video-display"
		hdmi-output-2: HDMI / DisplayPort 3 (priority 5700, latency offset 0 usec, available: no)
			properties:
				device.icon_name = "video-display"
		hdmi-output-3: HDMI / DisplayPort 4 (priority 5600, latency offset 0 usec, available: no)
			properties:
				device.icon_name = "video-display"
		hdmi-output-4: HDMI / DisplayPort 5 (priority 5500, latency offset 0 usec, available: no)
			properties:
				device.icon_name = "video-display"
		hdmi-output-5: HDMI / DisplayPort 6 (priority 5400, latency offset 0 usec, available: no)
			properties:
				device.icon_name = "video-display"
		hdmi-output-6: HDMI / DisplayPort 7 (priority 5300, latency offset 0 usec, available: no)
			properties:
				device.icon_name = "video-display"
    index: 1
[... 22 lines cut ...]
		output:analog-stereo: Analog Stereo Output (priority 1500, available: unknown)
		output:analog-stereo+input:analog-stereo: Analog Stereo Duplex (priority 1515, available: unknown)
		output:analog-surround-21: Analog Surround 2.1 Output (priority 1300, available: unknown)
		output:analog-surround-21+input:analog-stereo: Analog Surround 2.1 Output + Analog Stereo Input (priority 1315, available: unknown)
		output:analog-surround-40: Analog Surround 4.0 Output (priority 4200, available: unknown)
		output:analog-surround-40+input:analog-stereo: Analog Surround 4.0 Output + Analog Stereo Input (priority 4215, available: unknown)
		off: Off (priority 0, available: unknown)
	active profile: <output:analog-surround-21>
	sinks:
		alsa_output.pci-0000_00_1f.3.analog-surround-21/#0: Built-in Audio Analog Surround 2.1
	sources:
		alsa_output.pci-0000_00_1f.3.analog-surround-21.monitor/#0: Monitor of Built-in Audio Analog Surround 2.1
	ports:
		analog-input-internal-mic: Internal Microphone (priority 8900, latency offset 0 usec, available: yes)
			properties:
				device.icon_name = "audio-input-microphone"
		analog-input-mic: Microphone (priority 8700, latency offset 0 usec, available: yes)
			properties:
				device.icon_name = "audio-input-microphone"
		analog-output-speaker: Speakers (priority 10000, latency offset 0 usec, available: yes)
			properties:
				device.icon_name = "audio-speakers"
		analog-output-headphones: Headphones (priority 9900, latency offset 0 usec, available: yes)
			properties:
				device.icon_name = "audio-headphones"

aydar-y, 2020-05-07:
**And here's pacmd list-sinks result:**

1 sink(s) available.
  * index: 0
	name: <alsa_output.pci-0000_00_1f.3.analog-surround-21>
	driver: <module-alsa-card.c>
	flags: HARDWARE HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY DYNAMIC_LATENCY
	state: RUNNING
	suspend cause: (none)
	priority: 9039
	volume: front-left: 45876 /  70% / -9.29 dB,   front-right: 45876 /  70% / -9.29 dB,   lfe: 45876 /  70% / -9.29 dB
	        balance 0.00
	base volume: 65536 / 100% / 0.00 dB
	volume steps: 65537
	muted: no
	current latency: 23.82 ms
	max request: 6 KiB
	max rewind: 6 KiB
	monitor source: 0
	sample spec: s16le 3ch 44100Hz
	channel map: front-left,front-right,lfe
	used by: 1
	linked by: 1
	configured latency: 23.22 ms; range is 0.50 .. 2000.00 ms
	card: 1 <alsa_card.pci-0000_00_1f.3>
	module: 8
	properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "Generic Analog"
		alsa.id = "Generic Analog"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "1"
		alsa.card_name = "HDA Intel PCH"
		alsa.long_card_name = "HDA Intel PCH at 0x81810000 irq 31"
		alsa.driver_name = "snd_hda_intel"
		device.bus_path = "pci-0000:00:1f.3"
		sysfs.path = "/devices/pci0000:00/0000:00:1f.3/sound/card1"
		device.bus = "pci"
		device.vendor.id = "8086"
		device.vendor.name = "Intel Corporation"
		device.product.id = "a348"
		device.product.name = "Cannon Lake PCH cAVS"
		device.form_factor = "internal"
		device.string = "surround21:1"
		device.buffering.buffer_size = "529200"
		device.buffering.fragment_size = "264600"
		device.access_mode = "mmap+timer"
		device.profile.name = "analog-surround-21"
		device.profile.description = "Analog Surround 2.1"
		device.description = "Built-in Audio Analog Surround 2.1"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-pci"
	ports:
		analog-output-speaker: Speakers (priority 10000, latency offset 0 usec, available: yes)
			properties:
				device.icon_name = "audio-speakers"
	active port: <analog-output-speaker>

aydar-y, 2020-05-07:
**The line in alsa-base.conf is set to:**

options snd-hda-intel model=imac27

I tried most of the solutions out there and I would really appreciate your help on this.

davidjo, 2020-05-09:
Unfortunately it doesnt appear the above logs have the information needed to determine the chip to see if this module is going to be useful.
Look under /sys/class/sound you should see something like hwC0D0 (I believe for card 0 device 0) (there maybe others eg hwC1D0) which should be a linked directory - go into that directory and cat the following entries vendor_id, vendor_name, subsystem_id.
do this for all similar entries and post them here.
yes looking at the above aplay log you should have at least 2 - from the internal sound chip plus the hdmi sound for your graphics card - dont really need the hdmi entries

julian1059, 2020-05-11:
Thank you for opening up this issue @aydar-y 

I have the same problem on my 2017 iMac 21.5" that I upgraded to Ubuntu 20.04 (previously 19.10). Spent quite a while installing these patches and attempting to fix the sound issue, but to no avail. Went ahead and grabbed the info requested by @davidjo 

Here is my output. 
hwC0D0:
vendor_id: 0x10138409
vendor_name: Cirrus Logic

hwC2D0:
vendor_id: 0x1002aa01
vendor_name: ATI

Let me know if I can help.

davidjo, 2020-05-12:
So indeed for hwC0D0 you have the Cirrus 8409 HDA chip (so this module is what is needed) - but I dont see the subsystem_id - there are specific checks in the code for the subsystem_id so if its different things wont work - one feature the subsystem_id detects is which amplifier chips you have (the HDA sends digital audio to the amplifier chips which have the DACs to convert to analog and send to the speakers).
(hwCW2D0 is your graphics card)

davidjo, 2020-05-14:
Another command to perform is:
ioreg -l -a >'some file name'.xml

This dumps your IORegistry as an xml file.
You then need to look for strings like AppleHDATDMAmpxxxxxxx eg AppleHDATDMAmpMAX98706 - there is a number associated with these fields.
The one that is non-zero defines which amplifiers you have (currently this module only works
for MAX98706 and SSM3515 amps).
Other amps defined are TAS5758L or TAS5764L - which are not implemented yet and need access to a machine with those chips to dump an OSX log file before can begin to work on them.

aydar-y, 2020-05-25:
Thanks for responding! 
Here's my info:

Here is my output.
hwC0D0:
vendor_id: 0x1002aa01
vendor_name: ATI

hwC1D0:
vendor_id: 0x10138409
vendor_name: Cirrus Logic

I will provide the ioreg -l -a >'some file name'.xml output a bit later

aydar-y, 2020-05-25:
Here's the output from IORegistry:

AppleHDATDMAmpMAX98706 0 
AppleHDATDMAmpSSM3515 0 
AppleHDATDMAmpTAS5758L 0 
AppleHDATDMAmpTAS5764L 4

aydar-y, 2020-05-25:
Or if the relevant numbers come in front:
3 AppleHDATDMAmpMAX98706 
0 AppleHDATDMAmpSSM3515 
0 AppleHDATDMAmpTAS5758L 
0 AppleHDATDMAmpTAS5764L

davidjo, 2020-05-28:
No Im afraid its the one after - so you have 4 of the TAS5764L amps which I dont have logs for.
I dont have access to an imac to get OSX logs.
In order to get logs we would need an Imac with High Sierra installed (because thats the OS I have a logging kext driver for - could be booted from external drive) and someone would need to replace the current AppleHDA.kext with the logging version and then play audio thro the internal speakers and dump the log.
Also will really need the subsystem_id  for the audio card - the same one with the vendor_id of 0x10138409 for your system - it should be in same place as vendor_id.

CuriousTommy, 2020-06-20:
Hi @davidjo,

I don't think the iMac 19.1 is able to run High Serria. I believe it came with Mojave pre-installed.

davidjo, 2020-06-20:
ah - didnt realise it was that later a model
so unfortunately I dont have any Mojave install or plans to move to Mojave (Im still wary of APFS  and when I started there was no way to read it under linux) - although theres a good chance I may be forced to if programs I use begin to require Mojave
- unless of course those chips are used on older imacs which could run High Sierra - I think most of the machines with MAX/SSM amps are macbooks so could be an imac/macbook issue - maybe they are more powerful amps than you would want for a laptop but OK for a desktop (because those amps are in my High Sierra system as they appear in my own ioreg dump).

doctordbx, 2020-06-28:
Hi there, I have iMac 18,3 with the same configuration:

From ioreg
<key>AppleHDATDMAmpMAX98706</key>
<integer>0</integer>
<key>AppleHDATDMAmpSSM3515</key>
<integer>0</integer>
<key>AppleHDATDMAmpTAS5758L</key>
<integer>0</integer>
<key>AppleHDATDMAmpTAS5764L</key>
<integer>4</integer>

I'm happy to do what you need to get support for this device since I'm also keen to get Mint Linux working with my 18,3 iMac.

Let me know and I'll do what I can.

davidjo, 2020-06-28:
Can you run High Sierra - eg install on spare SSD drive

doctordbx, 2020-06-28:
I surely can. I'll install it today.

davidjo, 2020-06-29:
So next depends on what you are comfortable with.
We need to replace the AppleHDA.kext with a logging version.
This involves a number of stages:
First, we replace the AppleHDA exec (/System/Library/Extensions/AppleHDA.kext/Contents/MacOS/AppleHDA) with an unsigned version - I use the github unsign program to do this.
This lets us modify the kext (although I always disable SIP and gatekeeper before doing this).
Second, we replace the IOHDAFamily exec (/System/Library/Extensions/AppleHDA.kext/Contents/PlugIns/IOHDAFamily.kext/Contents/MacOS/IOHDAFamily)
with my logging version.
I can send you the assembler source and build makefile if you are able to compile programs or I can send you the exec.
Ill send my README files which reminds me how to activate the updated kernel extension.
We boot the updated kernel (this takes a while because of the logging), play some sound and grep out the HDA log lines which you then send to me so I can figure out the amp programming.

doctordbx, 2020-06-29:
Might be best just to send me the execs, it's a pretty bare bones install and just for this exercise. you can email me directly on my username and gmail dot com. Everything else will be easy enough. It's 10.40 pm here now and I'm heading to bed, but if you can drop them to me overnight I'll get what you need.

davidjo, 2020-06-29:
OK will do - I was going to suggest emailing would be easier for this.

paniclabde, 2020-07-22:
Hi, I'm on an iMac 18,3 too looking to fix my sound problems for elementaryOS / Ubuntu Bionic.
Is there any progress on this issue or can I be of any assistance?

davidjo, 2020-07-23:
I have some preliminary fixups which have been working on with doctordbx which seem to get the amps working but not the mikes - at the moment trying to figure out some other general issues - I can attach the the current version here but not ready to push them to the repo yet.

aydar-y, 2020-07-28:
Hi David, do you have any progress with Catalina, or downgrading to High Sierra is the only way? Mohave?

davidjo, 2020-07-28:
Not sure what you mean here - you dont need to downgrade your OSX - if your machine has some new chips then would need to get logs but this could be done by booting from an external SSD. But currently no Im still on High Sierra.
First thing would be to do an ioreg -a -l and see exactly what chips you have. Look for eg AppleBusController..... or AppleHDATDMBusManager.... and AppleHDATDMAmp..... which have a non-zero value after them.

davidjo, 2020-07-29:
I forgot - I need the subsystem id of the 19,1 imac  - from the ioreg -l -a look for IOHDACodecFunctionSubsystemID it should report a big integer number after it then can update the subsystem_id ifs.

ndoroshenko, 2020-10-03:
> I forgot - I need the subsystem id of the 19,1 imac - from the ioreg -l -a look for IOHDACodecFunctionSubsystemID it should report a big integer number after it then can update the subsystem_id ifs.

Hi, @davidjo! I'm on iMac 18,3 (Retina 5K, 27-inch, 2017).

Seems that I have TAS5764L amplifier:
```
<key>AppleHDATDMAmpMAX98706</key>
<integer>0</integer>
<key>AppleHDATDMAmpSSM3515</key>
<integer>0</integer>
<key>AppleHDATDMAmpTAS5758L</key>
<integer>0</integer>
<key>AppleHDATDMAmpTAS5764L</key>
<integer>4</integer>
```

Perhaps,
```
<key>IOHDACodecFunctionSubsystemID</key>
<integer>275451904</integer>
```
might be the relevant value.

Let me know if I can help you with anything.

CuriousTommy, 2020-10-03:
@davidjo I own an iMac19,1. I can confirm that the number is the same as @ndoroshenko 

```
<key>IOHDACodecFunctionSubsystemID</key>
<integer>275451904</integer>
```

davidjo, 2020-10-03:
Thanks - that indeed defines the subsystem ID as 0x106b1000 (275451904 in hex)

czosel, 2020-11-13:
Hi @davidjo You mentioned above that you have some preliminary fixes to get the amps working - could you share them? I'm using an iMac 18,3 as well, and can confirm all the info posted by @ndoroshenko. 

I'd be also happy to help out if you need anything - your work on this is greatly appreciated.

davidjo, 2020-11-22:
Here is a patch which should add codes for the tas5764 amps - this should work for both 18,3 and 19,1 as according to the above the subsystem ids are the same.
I think this plays audio but there are unresolved issues with recording.

[patch_imacs.diff.txt](https://github.com/davidjo/snd_hda_macbookpro/files/5579333/patch_imacs.diff.txt)

egorenar, 2020-11-22:
> Here is a patch which should add codes for the tas5764 amps - this should work for both 18,3 and 19,1 as according to the above the subsystem ids are the same.
> I think this plays audio but there are [...]

Unfortunately, the patch doesn't apply cleanly to master.
Regards

CuriousTommy, 2020-11-24:
Had to modify the install script, but I was able to compile @egorenar branch on Fedora 33. I can confirm that audio is working on my iMac19,1!

aydar-y, 2021-01-04:
Didn't work for me on TAS5764. Is it still limited to High Sierra?

aydar-y, 2021-01-04:
BTW, got this warning when installing the latest patch by @egorenar: 
warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1610 |         hinfo = spec->gen.stream_analog_playback;

davidjo, 2021-01-04:
For OSX logs yes I havent tried to update to a later OS - but the linux driver can work on later machines if there havent been major changes to the audio hardware.

The compile error is expected - it stems from some debugging code I added early on - but it was only for logging.

Do you get any sound at all?
What do you see on the Sound settings dialog for available outputs?
Have you tried running the speaker test?

Have you tried:
aplay -D hw:0,0 StereoTest32_reduced_m24dB.wav ?
(Very important to use the reduced version).

mani47, 2021-01-09:
Hi, I'm trying to apply the same patch from @egorenar branch to my iMac mid 2017 model and I'm getting this error, am I doing sth wrong?
```
make[1]: Entering directory '/usr/src/linux-headers-5.10.5-051005-generic'
  CC [M]  /home/agent47/projects/snd_hda_macbookpro/build/hda-5.10.5/patch_cirrus.o
/home/agent47/projects/snd_hda_macbookpro/build/hda-5.10.5/patch_cirrus.c: In function ‘cs_8409_init’:
/home/agent47/projects/snd_hda_macbookpro/build/hda-5.10.5/patch_cirrus.c:1617:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1617 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
  LD [M]  /home/agent47/projects/snd_hda_macbookpro/build/hda-5.10.5/snd-hda-codec-cirrus.o
  MODPOST /home/agent47/projects/snd_hda_macbookpro/build/hda-5.10.5/Module.symvers
  CC [M]  /home/agent47/projects/snd_hda_macbookpro/build/hda-5.10.5/snd-hda-codec-cirrus.mod.o
make[3]: *** No rule to make target 'scripts/module.lds', needed by '/home/agent47/projects/snd_hda_macbookpro/build/hda-5.10.5/snd-hda-codec-cirrus.ko'.  Stop.
make[2]: *** [scripts/Makefile.modpost:117: __modpost] Error 2
make[1]: *** [Makefile:1709: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.5-051005-generic'
make: *** [Makefile:17: all] Error 2
cp snd-hda-codec-cirrus.ko /lib/modules/5.10.5-051005-generic/updates
cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory
make: *** [Makefile:22: install] Error 1

contents of /lib/modules/5.10.5-051005-generic/updates
total 0
```

davidjo, 2021-01-09:
So I think things are sort of OK - the build is compiling but failing at the link stage.
Im guessing because its listing 5.10.5 you are using kernel 5.10 so what I think has happened is that there has been a change to the build system for out of kernel module building for kernel 5.10 (the No rule to make target is the key indicator of this - scripts/module.lds is a component of the main kernel build system). This has happened before but I dont have a 5.10 system.

davidjo, 2021-01-10:
It seems this is an Ubuntu bug - 1906131 in launchpad.
A suggested quick fix in that bug is:

If anyone come across this looking for a workaround, here's a quick fix I've found on a GitHub repo for RTL8192EU

https://github.com/Mange/rtl8192eu-linux-driver/issues/205#issuecomment-736366347

sudo wget https://raw.githubusercontent.com/torvalds/linux/master/scripts/module.lds.S -O /usr/src/linux-headers-5.10.2-051002-generic/scripts/module.lds

sudo sed -i '$ d' /usr/src/linux-headers-5.10.2-051002-generic/scripts/module.lds

davidjo, 2021-01-10:
Im trying to consolidate a number of the fixups.
I have corrected my error of failing to add the imac subsystem id in patch_cirrus_new84.h which @egorenar  fixed.
This patch should now apply to current master.
Ive slightly modified the code organization compared to @egorenar.

[patch_imacs.diff.txt](https://github.com/davidjo/snd_hda_macbookpro/files/5792770/patch_imacs.diff.txt)

czosel, 2021-01-10:
Thanks for your work on this @davidjo!

The patch indeed applies to the current master, and the installation also worked on my iMac 18,3 (Ubuntu 20.04 (5.4.0 Kernel) / Catalina) machine. However, no sound. The system dialog "Sound" shows three configurations for "Speakers - Built-in Audio":
- Analog Stereo Output
- Analog Surround 2.1 Output
- Analog Surround 4.0 Output

I wasn't able to run the command you mentioned above, `aplay -D hw:0,0 StereoTest32_reduced_m24dB.wav` because I don't have the wav file.

mani47, 2021-01-10:
Thanks so much for your work @davidjo, With your suggestion and using the patch you linked to, patch was successful but after reboot I still hear no sound. I tested with both speaker test and `aplay` (@czosel wav file is under test folder of Github repo):

[image]

czosel, 2021-01-10:
Thanks @mani47 - for me, there's no sound with the `aplay` command as well.

mani47, 2021-01-10:
The output of `sudo lspci -v` for me:

```
00:1f.3 Audio device: Intel Corporation 100 Series/C230 Series Chipset Family HD Audio Controller (rev 31)
	Subsystem: Intel Corporation 100 Series/C230 Series Chipset Family HD Audio Controller
	Flags: bus master, fast devsel, latency 64, IRQ 57
	Memory at 92720000 (64-bit, non-prefetchable) [size=16K]
	Memory at 4000000000 (64-bit, non-prefetchable) [size=64K]
	Capabilities: [50] Power Management version 3
	Capabilities: [60] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Kernel driver in use: snd_hda_intel

01:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Baffin HDMI/DP Audio [Radeon RX 550 640SP / RX 560/560X]
	Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] Baffin HDMI/DP Audio [Radeon RX 550 640SP / RX 560/560X]
	Flags: bus master, fast devsel, latency 0, IRQ 58
	Memory at 92660000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [48] Vendor Specific Information: Len=08 <?>
	Capabilities: [50] Power Management version 3
	Capabilities: [58] Express Legacy Endpoint, MSI 00
	Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Kernel driver in use: snd_hda_intel
```

davidjo, 2021-01-10:
Have you tried the @egorenar version?
Could you try it and see if that works - Im debugging blind here because I have no imac.

Also can you look at your syslog (or kern.log) and see if you get any snd_hda_intel: UNKNOWN subsystem id 0x....
lines.

mani47, 2021-01-10:
Yes I tried @egorenar version and same result:

```bash
sudo dmesg | grep snd_hda_intel
[    2.216539] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
[    2.216600] snd_hda_intel 0000:01:00.1: Force to non-snoop mode
[    2.298737] snd_hda_intel: UNKNOWN subsystem id 0x106b0f00
[    2.301061] snd_hda_intel 0000:01:00.1: bound 0000:01:00.0 (ops amdgpu_dm_audio_component_bind_ops [amdgpu])
[    2.310216] snd_hda_intel: UNKNOWN subsystem id 0x106b0f00
```

I also tried adding this option to the end of `/etc/modprobe.d/alsa-base.conf` which solved the issue for some people according to this SO thread: https://askubuntu.com/a/1264125

```
options snd-hda-intel model=generic
```

after doing this and reboot dmesg output is different but still no sound:

```
[    2.282241] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
[    2.282340] snd_hda_intel 0000:01:00.1: Force to non-snoop mode
[    2.456854] snd_hda_intel 0000:01:00.1: bound 0000:01:00.0 (ops amdgpu_dm_audio_component_bind_ops [amdgpu])
```

davidjo, 2021-01-10:
This is concerning - UNKNOWN subsystem  id 0x106b0f00 - this is not one of the known  subsystem ids - I think this message can only have come from the updated module.

However, there are a quite a few kernel prints / dev_info with this message.

So if you could go thro patch_cirrus_new84.h and patch_cirrus_real84.h and add some eg unique number to each UNKNOWN print eg printk("snd_hda_intel: UNKNOWN 1 subsystem id 0x%08x",codec->core.subsystem_id); changing the number for each print (or dev_info print).
Then rebuild etc and check the log again for UNKNOWN.

Also, can you look at the ioreg -l -a for your machine and search for 275451648 (0x106b0f00 in decimal) and see what item its associated with.

davidjo, 2021-01-10:
So having looked around 0x106b0f00 is associated with a Sigmatel STAC922xx - which is implemented by the patch_sigmatel.c driver which already has fixups for macs.

What I dont understand is why such a machine would be loading the cirrus driver at all.

davidjo, 2021-01-10:
Ah! So it appears the imac 18,2 does indeed use the 8409 HDA driver and must use a Sigmatel chip as the subsystem (which is different from the direct patch_sigmatel.c driver.

(The alsa-info.sh script would provide all the information).

Unfortunately this driver cant handle this - this would need an OSX log of the HDA commands from such a machine.

mani47, 2021-01-13:
Thanks @davidjo for looking into this, can I get that info in anyway without installing macOS?

davidjo, 2021-01-13:
No - the 8409 is an Apple designed chip (altho built by Cirrus) and we have no available documentation - in addition the commands we need are vendor commands which again could do anything and there is no available docs.

If you dont want to mess with your setup you could boot High Sierra on an external disk - see doctordbx items above.

PS I found the details about 18,2 machines online - there is an alsa-info.sh script (alsa-utils package) dump if you google.

rogers06, 2021-01-15:
Thanks for all the work on this issue.  I've been trying to get audio to work on my iMac 19,1 for the past couple weeks after a new install of Ubuntu 20.04.1.

The most recent version of `patch_imacs.diff.txt` from @davidjo above applied and built with no issues.  After a reboot, I was able to play system sounds and hear audio on youtube.  The mic doesn't seem to be functioning right now, but I haven't dug into that to rule out any other config problems yet.

davidjo, 2021-01-15:
Thanks for the update - Ill commit this to the main repo.
Have you tried headphones?
There is a known issue with mikes on these imacs - the HDA nids associated with the internal mike seems to be swapped on the imacs cf. the macbook pros - I changed the coding but there seems to be something else missing.
PS. I am also going to update the amp setup see issue #23 - the current setup is not mapping the speakers correctly.

rogers06, 2021-01-16:
The headphone jack was not functional.  In Settings->Sound I selected "Headphones - Built-In Audio" as the output device and got no joy.  The sound continued to be routed to the iMac's main speakers.  The mic on my earbuds did not work either after switching the input device between "Internal Microphone - Built-In Audio" or "Microphone - Built-In Audio."

Thanks for pointing out the amp issue.  I can confirm that the channel mixup happens with this model of iMac, too.  I'll take a look at the channel mapping patch soon and post any results over there.  I'm currently down a rabbit hole with AMD video drivers failing to install and don't want to confound the two issues.

davidjo, 2021-01-16:
Sorry - have you physically tried using headphones - Im still unsure how the Setttings dialog is supposed to interact with the driver. Now that you mention it I think just changing the dialog might not switch the output - as output switching is currently driven by headphone plug events.
If you do I would suggest just holding the headphones relatively to your ears initially - do not put them in just in case.

rogers06, 2021-01-17:
Sorry for being unclear about the test. I plugged in a pair of known-good earbuds using the headphone jack (not USB or Bluetooth, though I should check those at some point).  When playing system sounds through the Settings app and video in YouTube, I did not hear any output from the earbuds and the sounds continued to play through the iMac speakers.  Hope that's a better explanation of the test and results.

davidjo, 2021-01-17:
Ive confirmed switching the dialog does not mute the speakers as I suspected (probably because Ive not fixed the automuting yet).
This driver only deals with the internal audio system, USB/Bluetooth should just work.
So if plugging in headphones does not mute the speakers its suggesting the unsolicited response system isnt working right.
To try and fix this I would need the snd_hda log from your machine (ie grep for snd_hda in dmesg/syslog).

rogers06, 2021-01-21:
Here's the output from bootup thru plugging in and then unplugging a set of headphones.  For clarity, there is a mic on this set of headphones.  Hope it helps and glad to send more as needed.
```
rogers06@iMac-Ubuntu:~$ dmesg | grep snd_hda
[    5.537473] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    5.537754] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
[    5.537789] snd_hda_intel 0000:01:00.1: Force to non-snoop mode
[    5.571607] snd_hda_codec_cirrus: loading out-of-tree module taints kernel.
[    5.571641] snd_hda_codec_cirrus: module verification failed: signature and/or required key missing - tainting kernel
[    5.572367] snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS8409: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
[    5.572368] snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    5.572369] snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
[    5.572369] snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
[    5.572370] snd_hda_codec_cirrus hdaudioC0D0:    inputs:
[    5.572370] snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x45
[    5.572371] snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x3c
[    5.674965] Modules linked in: mei_hdcp snd_sof_pci snd_sof_intel_byt snd_sof_intel_ipc snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_xtensa_dsp snd_sof_intel_hda snd_sof snd_hda_ext_core snd_soc_acpi_intel_match intel_rapl_common snd_soc_acpi x86_pkg_temp_thermal snd_soc_core intel_powerclamp applesmc(+) coretemp snd_compress ac97_bus snd_pcm_dmaengine kvm_intel kvm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_hda_codec_cirrus(OE+) crct10dif_pclmul snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi ghash_clmulni_intel brcmfmac snd_hda_intel snd_seq snd_intel_dspcfg aesni_intel snd_hda_codec hci_uart crypto_simd cryptd snd_hda_core glue_helper brcmutil snd_hwdep btqca rapl intel_cstate efi_pstore btrtl snd_pcm btbcm 8250_dw ee1004 snd_seq_device btintel cfg80211 apple_mfi_fastcharge(+) snd_timer bluetooth mei_me snd acpi_als kfifo_buf industrialio acpi_tad ecdh_generic ecc mei soundcore apple_bl intel_pch_thermal mac_hid sch_fq_codel parport_pc ppdev lp parport drm ip_tables
[    5.675059] Modules linked in: mei_hdcp snd_sof_pci snd_sof_intel_byt snd_sof_intel_ipc snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_xtensa_dsp snd_sof_intel_hda snd_sof snd_hda_ext_core snd_soc_acpi_intel_match intel_rapl_common snd_soc_acpi x86_pkg_temp_thermal snd_soc_core intel_powerclamp applesmc(+) coretemp snd_compress ac97_bus snd_pcm_dmaengine kvm_intel kvm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_hda_codec_cirrus(OE+) crct10dif_pclmul snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi ghash_clmulni_intel brcmfmac snd_hda_intel snd_seq snd_intel_dspcfg aesni_intel snd_hda_codec hci_uart crypto_simd cryptd snd_hda_core glue_helper brcmutil snd_hwdep btqca rapl intel_cstate efi_pstore btrtl snd_pcm btbcm 8250_dw ee1004 snd_seq_device btintel cfg80211 apple_mfi_fastcharge(+) snd_timer bluetooth mei_me snd acpi_als kfifo_buf industrialio acpi_tad ecdh_generic ecc mei soundcore apple_bl intel_pch_thermal mac_hid sch_fq_codel parport_pc ppdev lp parport drm ip_tables
[    5.944428] snd_hda_codec_cirrus hdaudioC0D0: cs_8409_boot_setup_real headphone ALREADY PLUGGED IN UNIMPLEMENTED!!
[ 2420.828323] snd_hda_codec_cirrus hdaudioC0D0: cs_8409_interrupt_action - plug in
[ 2422.784783] snd_hda_codec_cirrus hdaudioC0D0: cs_8409_interrupt_action - headset detected
[ 2422.793471] snd_hda_codec_cirrus hdaudioC0D0: cs_8409_headset_type_detect_event headset does not have mike!!
```

davidjo, 2021-01-21:
On the other hand - youve hit something else I cant try - I saw from Apples driver there are 3 headphone types detected, Apple with mike, non-Apple headphone only and non-Apple with mike - couldnt test non-Apple with mike because dont have such headphone!!
Ill see if this needs fixing to at least allow headphones to work - although I thought it should just skip mike in this case.

jboger, 2021-05-13:
Hi,
Just thought I'd add some more info here. I am on Mac Pro 6.1 (late 2013 - AKA the trash can). 
My info is as follows:
[root@fedora hwC0D0]# cat vendor_id 
0x10134208
[root@fedora hwC0D0]# cat vendor_name 
Cirrus Logic
[root@fedora hwC0D0]# cat subsystem_id 
0x106b7800
[root@fedora hwC0D0]# 

I have tried the patch but it does not work. I assume it's the wrong vendor_id. Would be glad to help in any way to get you the correct info you need from macOS running Big Sur on a Mac Pro 6.1. For the record, I got sound working by pairing airpods over bluetooth and using that instead. 

Thanks a lot for great work!

[49 more replies]