davidjo/snd_hda_macbookpro #12: Mic not working on 14,3

treymerkley, 2020-05-19:
I'm using Leifliddy's fork. Speakers work great, but the internal microphone still only gives static on the other end. What should I do to fix?

davidjo, 2020-05-28:
Mikes (internal and headset) now working partially.
The sound volume is very low but this seems to be the value returned by OSX kernel driver.
Further amplication at eg the pulse audio level is needed but dont know how to do this yet although PulseEffects seems to be the appropriate feature to be using.

treymerkley, 2020-05-29:
Is it disabled by default, or is there a way to see that the mic is on via CLI? I'm playing with PulseEffects and Pavucontrol but I'm not getting any internal mic activity.

davidjo, 2020-05-31:
arecord -L should show  the 2 capture streams

davidjo, 2020-05-31:
I tested using arecord  in raw mode with 2 channels, 44100 freq then imported into audacity and used the normalize effect to get audible output

treymerkley, 2020-06-05:
I don't _think_ I see it here, is this what it's supposed to look like?
```
[trey@citron ~]$ arecord -L
surround21
    2.1 Surround output to Front and Subwoofer speakers
surround40
    4.0 Surround output to Front and Rear speakers
surround41
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50
    5.0 Surround output to Front, Center and Rear speakers
surround51
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
null
    Discard all samples (playback) or generate zero samples (capture)
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
speex
    Plugin using Speex DSP (resample, agc, denoise, echo, dereverb)
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
default
    Default ALSA Output (currently PulseAudio Sound Server)
usbstream:CARD=PCH
    HDA Intel PCH
    USB Stream Output
usbstream:CARD=HDMI
    HDA ATI HDMI
    USB Stream Output
```

MrAsimov, 2020-06-08:
I can confirm the internal microphone is working with kernel v.5.5.19 on MacBookPro13,3

davidjo, 2020-06-10:
Glad its "working" for you - dont know why OSX has such a low volume - having looked at the raw dump file Im suspicious Apple has a 16 bit ADC (usual for HDA chips) that is swapped into the low 16 bits of the 24 bit stream format hence the low volume (multiply by 127 gives reasonable volume).
The gain can be set on the intmike widget (0x22) using cs_8409_intmike_volume_setup but havent yet felt comfortable changing this value but you could experiment with it.

treymerkley, 2020-06-27:
Here's the output of `arecord -l`:
```
[trey@citron ~]$ arecord -l
**** List of CAPTURE Hardware Devices ****
```
I'm on kernel 5.4.44-1, so let me try on 5.6.16 and see if that makes the difference.

MrAsimov, 2020-06-28:
@treymerkley if your internal audio input is having 'hiccups' try by switching between the built-in audio profiles and see if its recognizing the background sound with help of the pavucontrol command, I had that problem with MBP 13,3 model.

treymerkley, 2020-06-29:
That fixed it! Looks like for most users it's just a matter of their distro releasing a stable of a 5.5+ kernel. Thank you!

closed 2020-06-29 by treymerkley