davidjo/snd_hda_macbookpro #13: Volume is pretty weird

Bicet, 2020-05-31:
Hi there!
Thanks for the fantastic work!
Everything works like a charm on MBPr 14,3. 
I've a mixed feeling about the volume, I don't get if some bass are missing or the volume is very low (or both), at least compare to MacOs.
Is there a way I can help you out by providing logs or whatever?

davidjo, 2020-05-31:
Sorry - dont get which volume you are talking about - if its the play volume that should be OK - are you saying the max play volume is much lower than OSX?
Or are you saying the bass doesnt seem as loud as OSX - this has been commented on before.
My best guess is Apple is doing some DSP processing creating a woofer/tweeter combination at the CoreAudio level - this is nothing to do with the kernel driver. The nearest equivalent for linux seems to be the PulseEffects module of pulse which has DSP filter effects - but I have no knowledge of pulse and it seems to be a big black box.
By the way this is all about perceived volume - you boost the bass and the audio sounds "louder" even tho you keep the overall sound power the same.
Note that all these pulse effects can be done by the user creating the appropriate user pulse config file (the PulseEffects module is an installable package).

Bicet, 2020-06-01:
Sorry if I didn't explaing well.
So here're 2 videso recorded at max volume both on macos (catalina) and arch.

https://www.youtube.com/watch?v=Xa41IRTDCM8

vs

https://www.youtube.com/watch?v=Q9_La0QgalI

The recording is pretty shitty, I know, but I feel like you can feel the difference.

In the meanwhile I will give pulseeffects a try.

davidjo, 2020-06-01:
My play testing was primarily done with aplay - have you tried that?
Be careful - DONT use the hw:0,0  device - just dont specify a device - the touch bar volume controls worked for me with aplay.
Im trying to store the audio so I can load into audacity and see the actual wave form difference.
I can hear a difference in "volume" but as I said there are psychoaural  issues.
One possibility is that playing 2 channel audio thro pulse is only going to the tweeter speakers - I created some test audio files eg in audacity - you can duplicate a stereo track to  give  the 4 channel audio track the driver is really expecting and see if that makes a difference.

davidjo, 2020-06-02:
So it is looking as tho thro pulse only the tweeter speakers are active - see the spectrum from audacity for the same section cut from your 2 videos - note massive loss of bass plus overall sound is lower - although > 9 kHz seems very similar
[image]
[image]
So having just seen this from some googling it appears for pulse we likely need to use upmix to convert 2 channel to 4 channel (although that seems to be about creating a surround sound - which is not quite what wanted here - just need a plain duplication).

Bicet, 2020-06-03:
Sorry if I didn't reply before but I took a couple of day of detoxing.
Your last reply clearly visualise what I was trying to report.
I WOULD SAY IT AGAIN. Thanks a lot for your work, these are smaller issue.
Please tell me if is there something I can do to help you out 

^_^

davidjo, 2020-06-03:
Did you try using aplay? Did it sound louder?
Its not difficult to make a 4 channel file - just import a stereo eg .wav file into audacity, select all the track,  click duplicate then export the 4 channel file.
Does this 4 channel sound better than the 2 channel?
From the googling Ive done (try pulse and upmix) there does appear to be a difference between alsa and pulse - alsa does copy 2 channel audio to the other 2 channnels whereas pulse does not upmix.
There are some pulse configs listed which add an upmix solution to a sink you might want to try.

Bicet, 2020-06-04:
I've tried the trick with audacity, but I cannot see any difference in volume, maybe I've the wrong track :/

davidjo, 2020-06-04:
If you mean you tried plotting the spectrum in audacity if you do a plot then select another audio section you MUST tell plot spectrum to replot - otherwise it doesnt update the plot - this caught me out for a while.

davidjo, 2020-06-05:
OK so think Ive sorted this - somehow I left in the debug volume set (-24dB) for the amps - now updated to full volume value.

Bicet, 2020-06-06:
Holy Shit, what a difference. Just updated kernel and reinstalled the drivers. 
THANKS A LOT!
I think the issue can now be closed, but I feel like it's up to you.
Again, thanks so muh

davidjo, 2020-06-06:
Glad that helped. Im pretty certain we still have the effects issue because if you look the plots above have  a different shape - gain should just scale each frequency equally.
Can you still detect a tonal difference between OSX and linux?
Does the bass seem weaker under linux?
However this is all about messing with pulse config files now - and Im as knowledgeable as you about this.

Bicet, 2020-06-06:
> Glad that helped. Im pretty certain we still have the effects issue because if you look the plots above have a different shape - gain should just scale each frequency equally.
> Can you still detect a [...]

So the absolute volume seems pretty similar now, no distorsion at all. Still the bass seems a bit weaker, but there's a huge difference. I'm using a pulse profile that seems to effectively mitigate this issue. https://github.com/JackHack96/PulseEffects-Presets

davidjo, 2020-06-06:
Thanks for the update - thats useful  - Ill go ahead and close this now

closed 2020-06-06 by davidjo