davidjo/snd_hda_macbookpro #16: cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory

jso8910, 2020-09-13:
When I run the install script, I get this error
```
make -C /lib/modules/5.8.6-1-MANJARO/build CFLAGS_MODULE="-DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/jason/snd_hda_macbookpro/build/hda-5.8.6 modules
make[1]: Entering directory '/usr/lib/modules/5.8.6-1-MANJARO/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.8.6-1-MANJARO/build'
make: *** [Makefile:17: all] Error 2
cp snd-hda-codec-cirrus.ko /lib/modules/5.8.6-1-MANJARO/updates
cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory
make: *** [Makefile:22: install] Error 1
```

What I did was run the pacman install and then reboot, then git clone, cd, and install, but I get this error. I am using Manjaro KDE LinuX

davidjo, 2020-09-20:
Looks like maybe the kernel build system has changed for 5.8 - Im still using 5.4 (Ubuntu 20.04) and believe it works for 5.6 - the modules target should exist in the kernel build system - its not something Ive added

ITelligencia, 2020-09-24:
I am using ubuntu 20.4, kernel 5.4, on macbookpro13,3 but getting similar issue:

Building modules, stage 2.
  MODPOST 1 modules
ERROR: "snd_hda_jack_detect_enable_callback" [/tmp/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
ERROR: "snd_hda_jack_detect_state" [/tmp/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
ERROR: "snd_hda_jack_tbl_get" [/tmp/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[1]: *** [Makefile:1647: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-48-generic'
make: *** [Makefile:17: all] Error 2
cp snd-hda-codec-cirrus.ko /lib/modules/5.4.0-48-generic/updates
cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory.

I originally used lubuntu 20.04, whereby the cirrus patch did install smoothly.

LapisLazulis, 2020-09-26:
Got it to work on kernel 5.8 by installing the linux-headers packages for my kernel (so linux58-headers), redoing everything and putting a chmod +x (don't know if it changes anything but still did it) and then launching it in sudo mode just to be sure, it seems to work that way.

ITelligencia, 2020-09-29:
Thank you LapisLazulis, it was the sudo mode that was indeed missing in my attempt!

TanyaMio, 2020-09-29:
Same problem.
Ubuntu 20.04 kernel 5.4.0-48-lowlatency
Tried both with sudo and under root (after running sudo su)

Building modules, stage 2.
  MODPOST 1 modules
ERROR: "snd_hda_jack_detect_enable_callback" [/home/miya/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
ERROR: "snd_hda_jack_detect_state" [/home/miya/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
ERROR: "snd_hda_jack_tbl_get" [/home/miya/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[1]: *** [Makefile:1647: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-48-lowlatency'
make: *** [Makefile:17: all] Error 2
cp snd-hda-codec-cirrus.ko /lib/modules/5.4.0-48-lowlatency/updates
cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory
make: *** [Makefile:22: install] Error 1

It seems like it can't find the snd-hda-codec-cirrus.ko in the hda-5.4.0 directory. However I don't understand how it should appear there... I tried copying it from an earlier headers folder, but it seems like the directory is overwritten every time I start the script... I don't know what to do...

ITelligencia, 2020-09-29:
For me it didn't work as root, only when I used sudo: 

23 cd /tmp 
24 sudo apt-get install wget make gcc linux-headers-generic 
25 git clone https://github.com/davidjo/snd_hda_macbookpro 
26 sudo apt install git 
27 git clone https://github.com/davidjo/snd_hda_macbookpro 
28 cd snd_hda_macbookpro/ 
29 sudo ./install.cirrus.driver.sh 
30 sudo reboot 

Ruben Rouw 
Elastix Certified Engineer 

780.604.6041 

IT Consulting | Design | Implementation 
Email is the best way to reach me. 
To get my immediate attention, if I don't respond to text, try WhatsApp.

TanyaMio, 2020-09-30:
Thank you for the reply. Unfortunately this didn't work for me... I still
get the same error.

history :

875  cd /tmp
876  ls
877  git clone https://github.com/davidjo/snd_hda_macbookpro
878  cd snd_hda_macbookpro/
879  sudo ./install.cirrus.driver.sh

end of output for sudo ./install.cirrus.driver.sh :

Building modules, stage 2.
  MODPOST 1 modules
ERROR: "snd_hda_jack_detect_enable_callback"
[/tmp/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
ERROR: "snd_hda_jack_detect_state"
[/tmp/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
ERROR: "snd_hda_jack_tbl_get"
[/tmp/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[1]: *** [Makefile:1647: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-48-lowlatency'
make: *** [Makefile:17: all] Error 2
cp snd-hda-codec-cirrus.ko /lib/modules/5.4.0-48-lowlatency/updates
cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory
make: *** [Makefile:22: install] Error 1

davidjo, 2020-09-30:
So the cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory is a secondary error - because the module has failed to build.

The primary errors are:   ERROR: "snd_hda_jack_detect_enable_callback"
[/tmp/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
These are module load errors because the kernel build system is unable to find the symbol snd_hda_jack_detect_enable_callback when linking the module.

There are 3 such missing symbols snd_hda_jack_detect_enable_callback, snd_hda_jack_detect_state, snd_hda_jack_tbl_get.
These are all functions that should be provided by the base snd_hda_intel code.

One issue is that all testing I have has been with the generic kernel - I have never tried the lowlatency version - have you tried the generic kernel?

If you look at the script all it is doing is downloading a kernel source into a directory build (in /tmp/snd_hda_macbookpro in your case) then extracting out  the specific hda directory into an hda-{kernelversion} directory under build - inside that hda-x directory it builds the module. you can try out building in there - just doing make performs the build.

The symbols snd_hda_jack_detect_enable_callback etc should be provided by the snd-hda-codec.ko kernel module found under /lib/modules/kernelversion etc.

TanyaMio, 2020-09-30:
I don't know if it is useful for the solution in my case, but I have run in terminal:
sudo cat /usr/lib/modules/5.4.0-48-lowlatency/modules.symbols | grep "snd_hda_jack"

And got this output:
alias symbol:snd_hda_jack_unsol_event snd_hda_codec
alias symbol:snd_hda_jack_set_dirty_all snd_hda_codec
alias symbol:snd_hda_jack_add_kctl_mst snd_hda_codec
alias symbol:snd_hda_jack_poll_all snd_hda_codec
alias symbol:snd_hda_jack_detect_enable_callback_mst snd_hda_codec
alias symbol:snd_hda_jack_pin_sense snd_hda_codec
alias symbol:snd_hda_jack_report_sync snd_hda_codec
alias symbol:snd_hda_jack_set_gating_jack snd_hda_codec
alias symbol:snd_hda_jack_detect_state_mst snd_hda_codec
alias symbol:snd_hda_jack_add_kctls snd_hda_codec
alias symbol:snd_hda_jack_tbl_get_mst snd_hda_codec
alias symbol:snd_hda_jack_detect_enable snd_hda_codec
alias symbol:snd_hda_jack_tbl_get_from_tag snd_hda_codec

I noticed that all 3 missing symbols are modified with _mst in the end:
alias symbol:snd_hda_jack_detect_enable_callback_mst snd_hda_codec
alias symbol:snd_hda_jack_detect_state_mst snd_hda_codec
alias symbol:snd_hda_jack_tbl_get_mst snd_hda_codec

Is there anything I can do to fix it?

Zozerbox, 2020-10-01:
Hello, 

First off thank you for making this patch/script.  I have been using it for weeks without issue, however upon a fresh installation of Ubuntu 20.04 5.4.0-48 generic (same version i was using before) I am running into the same issue OP is experiencing.  I have read through the thread, and tried running the make command in the /build/hda-5.4.0-48 folder and still recieve the same error pertaining to the snd_hda_jack_detect_enable_callback.  Im not sure what would have changed since my last install, however admittedly i am a bit of a noob at this.  Here is my output: 

 Building modules, stage 2.
  MODPOST 1 modules
ERROR: "snd_hda_jack_detect_enable_callback" [/home/ben/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
ERROR: "snd_hda_jack_detect_state" [/home/ben/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
ERROR: "snd_hda_jack_tbl_get" [/home/ben/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[1]: *** [Makefile:1647: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-48-generic'
make: *** [Makefile:17: all] Error 2
cp snd-hda-codec-cirrus.ko /lib/modules/5.4.0-48-generic/updates
cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory
make: *** [Makefile:22: install] Error 1

contents of /lib/modules/5.4.0-48-generic/updates
total 4
drwxr-xr-x 2 root root 4096 Sep 30 16:12 dkms

Any help is GREATLY APPRECIATED!!  And again, thank you for your time and effort =)

Zozerbox

delsuc, 2020-10-02:
same problem here than @TanyaMio 
however I am using (I beleive) the regular 5.4.0 kernel on my MacBookPro 2017

Ven0mstrike42, 2020-10-03:
same issues as stated above. I am running Linux Mint 20 with 5.4.0-48-generic kernel on a MacBookPro 14,1

davidjo, 2020-10-03:
So first interesting difference is TanyaMio has modules under /usr/lib/modules whereas the Makefile expects modules under /lib/modules (look at the lines defining KDIR). 

So the issue maybe this mst - the name change seems to be associated with a kernel patch for DP-MST - maybe theres some kernel config mismatch.

The last 5.4 kernel Ive built with is -45  - this maybe an issue with the -48 update  - maybe the config now activates DP-MST and a fix is needed - given that most people reporting errors seem to be using -48.
But thats all I know for the moment.

davidjo, 2020-10-03:
hey ho - I can confirm the issue is with the subversion 48 of the 5.4 kernel - just updated to 5.4-48 and got the same build issues. Going to have to determine exactly what has changed.

davidjo, 2020-10-04:
It appears in order to fix something with HDMI cable plugin there has been an argument change to a lot of the sound jack routines - so cant just simply add _mst - apparently now a jack can have multiple devices associated with it - need to figure out what this device argument needs to be.

Unfortunately this is currently part of the kernel patches debian/Ubuntu applies - not part of the mainline kernel sources -so the install script also needs fixing to somehow handle this as currently it downloads the mainline kernel source to help building and not the debian/Ubuntu patched version.

davidjo, 2020-10-04:
So I now think the only thing needed to be done is to download the patched Ubuntu kernel source and then extract the parts needed (updated versions of the original symbols are defined as inline function calls in the patched source which map to the new _mst function calls hence I dont think changes to my source code are needed).
However this seems to be complicated - the simplest solution seems to be for the moment to require a pre install using eg apt install of the linux-source package. 
Theres also the issue of what about non-Ubuntu systems??

davidjo, 2020-10-04:
So it appears that 5.5 change has been backported to the 5.4 kernel from 5.4.0-48.
Ive attached an updated script which seems to fix the problem.
If you have not installed the 5.4 kernel source package (Ubuntu) you will need to install it for the new install script to work for 5.4 (but not earlier/later kernel versions).
[install.cirrus.driver.sh.txt](https://github.com/davidjo/snd_hda_macbookpro/files/5323357/install.cirrus.driver.sh.txt)
Please see how this works.
(note need to remove the .txt extension added in order to be able to upload)

Ven0mstrike42, 2020-10-04:
New patch worked perfectly. thank you.

TanyaMio, 2020-10-05:
The new patch worked! Thank you!

delsuc, 2020-10-05:
works perfectly here !   Thanks @davidjo

davidjo, 2020-10-05:
Thanks for the updates - Ill commit this update to the main repo.

Zozerbox, 2020-10-06:
Im still getting the same error code as above... what am i doing wrong?  Any help is greatly appreciated as im a beginner to this.  

make -C /lib/modules/5.4.0-48-generic/build CFLAGS_MODULE="-DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/ben/snd_hda_macbookpro/build/hda-5.4.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-48-generic'
  CC [M]  /home/ben/snd_hda_macbookpro/build/hda-5.4.0/patch_cirrus.o
/home/ben/snd_hda_macbookpro/build/hda-5.4.0/patch_cirrus.c: In function ‘cs_8409_init’:
/home/ben/snd_hda_macbookpro/build/hda-5.4.0/patch_cirrus.c:1610:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1610 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
  LD [M]  /home/ben/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.o
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "snd_hda_jack_detect_enable_callback" [/home/ben/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
ERROR: "snd_hda_jack_detect_state" [/home/ben/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
ERROR: "snd_hda_jack_tbl_get" [/home/ben/snd_hda_macbookpro/build/hda-5.4.0/snd-hda-codec-cirrus.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[1]: *** [Makefile:1647: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-48-generic'
make: *** [Makefile:17: all] Error 2
cp snd-hda-codec-cirrus.ko /lib/modules/5.4.0-48-generic/updates
cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory
make: *** [Makefile:22: install] Error 1

contents of /lib/modules/5.4.0-48-generic/updates
total 4
drwxr-xr-x 2 root root 4096 Sep 30 16:12 dkms