davidjo/snd_hda_macbookpro #18: Getting an error on Manjaro 5.8.16

sansyrox, 2020-11-12:
```
make -C /lib/modules/5.8.16-2-MANJARO/build CFLAGS_MODULE="-DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/bruh/snd_hda_macbookpro/build/hda-5.8.16 modules
make[1]: Entering directory '/usr/lib/modules/5.8.16-2-MANJARO/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.8.16-2-MANJARO/build'
make: *** [Makefile:17: all] Error 2
cp snd-hda-codec-cirrus.ko /lib/modules/5.8.16-2-MANJARO/updates
cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory
make: *** [Makefile:22: install] Error 1

contents of /lib/modules/5.8.16-2-MANJARO/updates
total 0
```

Hi, I am getting this error on Manjaro kernel 5.8.16.

Can you please suggest a fix? 

Thanks in advance :smile:

sansyrox, 2020-11-12:
I created a symlink to fix this issue

```
linux kernel headers not found in /usr/src: /usr/src/linux-headers-5.8.16-2-MANJARO
assuming the linux kernel headers package is not installed
please install the appropriate linux kernel headers package:
sudo apt install linux-headers-MANJARO
```

as linux headers are found in `
/lib/modules/$(uname -r)/
`

sansyrox, 2020-11-12:
From what I can understand is that it is a directory issue. I'll be happy to make a PR for the solution if found. :)

sansyrox, 2020-11-13:
For the time being, I've downgraded my kernel to 5.4.x and it works. Thanks!

leifliddy, 2020-11-18:
Did you see this issue?
https://github.com/davidjo/snd_hda_macbookpro/issues/16
It looks like you might just need to install the ```linux58-headers``` package

leifliddy, 2020-11-18:
https://github.com/Dunedan/mbp-2016-linux/issues/29#issuecomment-581821331

Artawower, 2020-12-22:
I have installed linux-headers and other tools, but i get 
```
linux kernel headers not found in /usr/src:
Debian (eg Ubuntu): /usr/src/linux-headers-5.7.19-2-MANJARO
Fedora: /usr/src/kernels/5.7.19-2-MANJARO
assuming the linux kernel headers package is not installed
please install the appropriate linux kernel headers package:
sudo apt install linux-headers-MANJARO
```
What i did wrong?

davidjo, 2020-12-23:
Based on the comments of sansyrox it seems Manjaro does not put the linux headers under /usr/src which the install script assumes.

One solution is his suggestion to create a symlink in /usr/src to the correct place.

Artawower, 2020-12-23:
I tried to made symlink like above: /lib/modules/$(uname -r)/ but script doesn't see this file, where is correct place? I also tried use find for 'linux-headers*' but without luck

davidjo, 2020-12-24:
If you would like could you suggest a patch to the install script install.cirrus.driver.sh - mainly the distro ifs right at the beginning - something using /lib/modules/$(uname -r) as the check  - similar to the Fedora Based Distro check.
Ive never used a Manjaro system so cant test it.
(Its only the initial check where its significant - the Makefile uses the kernel  module building setup so should know where it is. Only place its also used is the specific 5.4.0-48 version fixup - but ignore this.)

davidjo, 2022-05-10:
Arch test added for new version which should fix this.

closed 2022-05-10 by davidjo