davidjo/snd_hda_macbookpro #19: Compile issue on Linux 4.15

ColPaulR, 2020-11-18:
I am having trouble compiling. Below is the output. Please help!

--2020-11-17 20:56:25--  https://cdn.kernel.org/pub/linux/kernel/v4.x/linux-4.15.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.49.176, 2a04:4e42:c::432
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.49.176|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 102181404 (97M) [application/x-xz]
Saving to: ‘build/linux-4.15.tar.xz’

linux-4.15.tar.xz   100%[===================>]  97.45M  7.03MB/s    in 14s     

2020-11-17 20:56:39 (6.86 MB/s) - ‘build/linux-4.15.tar.xz’ saved [102181404/102181404]

linux-4.15/sound/pci/hda/
linux-4.15/sound/pci/hda/Kconfig
linux-4.15/sound/pci/hda/Makefile
linux-4.15/sound/pci/hda/ca0132_regs.h
linux-4.15/sound/pci/hda/dell_wmi_helper.c
linux-4.15/sound/pci/hda/hda_auto_parser.c
linux-4.15/sound/pci/hda/hda_auto_parser.h
linux-4.15/sound/pci/hda/hda_beep.c
linux-4.15/sound/pci/hda/hda_beep.h
linux-4.15/sound/pci/hda/hda_bind.c
linux-4.15/sound/pci/hda/hda_codec.c
linux-4.15/sound/pci/hda/hda_codec.h
linux-4.15/sound/pci/hda/hda_controller.c
linux-4.15/sound/pci/hda/hda_controller.h
linux-4.15/sound/pci/hda/hda_controller_trace.h
linux-4.15/sound/pci/hda/hda_eld.c
linux-4.15/sound/pci/hda/hda_generic.c
linux-4.15/sound/pci/hda/hda_generic.h
linux-4.15/sound/pci/hda/hda_hwdep.c
linux-4.15/sound/pci/hda/hda_intel.c
linux-4.15/sound/pci/hda/hda_intel.h
linux-4.15/sound/pci/hda/hda_intel_trace.h
linux-4.15/sound/pci/hda/hda_jack.c
linux-4.15/sound/pci/hda/hda_jack.h
linux-4.15/sound/pci/hda/hda_local.h
linux-4.15/sound/pci/hda/hda_proc.c
linux-4.15/sound/pci/hda/hda_sysfs.c
linux-4.15/sound/pci/hda/hda_tegra.c
linux-4.15/sound/pci/hda/local.h
linux-4.15/sound/pci/hda/patch_analog.c
linux-4.15/sound/pci/hda/patch_ca0110.c
linux-4.15/sound/pci/hda/patch_ca0132.c
linux-4.15/sound/pci/hda/patch_cirrus.c
linux-4.15/sound/pci/hda/patch_cmedia.c
linux-4.15/sound/pci/hda/patch_conexant.c
linux-4.15/sound/pci/hda/patch_hdmi.c
linux-4.15/sound/pci/hda/patch_realtek.c
linux-4.15/sound/pci/hda/patch_si3054.c
linux-4.15/sound/pci/hda/patch_sigmatel.c
linux-4.15/sound/pci/hda/patch_via.c
linux-4.15/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/4.15.0-124-generic/build CFLAGS_MODULE="-DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/tmp/snd_hda_macbookpro/build/hda-4.15.0 modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-124-generic'
  CC [M]  /tmp/snd_hda_macbookpro/build/hda-4.15.0/patch_cirrus.o
/tmp/snd_hda_macbookpro/build/hda-4.15.0/patch_cirrus.c: In function ‘cs_8409_init’:
/tmp/snd_hda_macbookpro/build/hda-4.15.0/patch_cirrus.c:1610:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
         hinfo = spec->gen.stream_analog_playback;
               ^
In file included from /tmp/snd_hda_macbookpro/build/hda-4.15.0/patch_cirrus.c:1944:0:
/tmp/snd_hda_macbookpro/build/hda-4.15.0/patch_cirrus_hda_generic_copy.h: In function ‘is_ctl_used’:
/tmp/snd_hda_macbookpro/build/hda-4.15.0/patch_cirrus_hda_generic_copy.h:273:2: error: implicit declaration of function ‘snd_array_for_each’; did you mean ‘snd_array_free’? [-Werror=implicit-function-declaration]
  snd_array_for_each(&spec->paths, i, path) {
  ^~~~~~~~~~~~~~~~~~
  snd_array_free
/tmp/snd_hda_macbookpro/build/hda-4.15.0/patch_cirrus_hda_generic_copy.h:273:44: error: expected ‘;’ before ‘{’ token
  snd_array_for_each(&spec->paths, i, path) {
                                            ^
/tmp/snd_hda_macbookpro/build/hda-4.15.0/patch_cirrus_hda_generic_copy.h:278:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
cc1: some warnings being treated as errors
scripts/Makefile.build:330: recipe for target '/tmp/snd_hda_macbookpro/build/hda-4.15.0/patch_cirrus.o' failed
make[2]: *** [/tmp/snd_hda_macbookpro/build/hda-4.15.0/patch_cirrus.o] Error 1
Makefile:1585: recipe for target '_module_/tmp/snd_hda_macbookpro/build/hda-4.15.0' failed
make[1]: *** [_module_/tmp/snd_hda_macbookpro/build/hda-4.15.0] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-124-generic'
Makefile:17: recipe for target 'all' failed
make: *** [all] Error 2
cp snd-hda-codec-cirrus.ko /lib/modules/4.15.0-124-generic/updates
cp: cannot stat 'snd-hda-codec-cirrus.ko': No such file or directory
Makefile:22: recipe for target 'install' failed
make: *** [install] Error 1

contents of /lib/modules/4.15.0-124-generic/updates
total 4
drwxr-xr-x 2 root root 4096 Nov 17 20:46 dkms

davidjo, 2020-11-21:
Unfortunately I dont have a linux  version 4 kernel running any more.

Im guessing according to the error the issue is a macro defined in version 5 and not in 4 - snd_array_for_each.
This is occurring in patch_cirrus_hda_generic_copy.h which is a copy of local functions defined in hda_generic.h/hda_generic.c 
which I needed for the way I handled too many adcs (more than AUTO_CFG_MAX_INS).

You could try replacing the is_ctl_out function in patch_cirrus_hda_generic_copy.h with your version for kernel 4.15 in the kernel file hda_generic.c.

ColPaulR, 2020-11-21:
Thanks for the reply and pointer to a potential fix. I’ll give it a try and let you know the results.  

Paul

ColPaulR, 2020-11-22:
I googled the macro and then copied the below in front of the is_ctl_out function:
#ifndef snd_array_for_each
/* a helper macro to iterate for each snd_array element */	
#define snd_array_for_each(array, idx, ptr) \
	for ((idx) = 0, (ptr) = (array)->list; (idx) < (array)->used; \
		     (ptr) = snd_array_elem(array, ++(idx)))
#endif /* __SOUND_HDAUDIO_H */

It compiled and installed. Works like a charm! Thank you so much for providing sound for my 2016 MacBook Pro 15” running Ubuntu 18.04 (actually triple boot with OS X and Windows 10).

Paul

davidjo, 2020-11-22:
Glad I could help - was worried there may have been other 4/5 compatibility issues.

closed 2020-11-22 by davidjo