davidjo/snd_hda_macbookpro #21: Sound pops when coming from more than one audio source

frayment, 2020-12-06:
When sound is played from more than one source at once (such as music and a system sound together), a very loud popping sound is emitted on the MacBookPro13,1. (Ubuntu 20.04.1 LTS, 5.4.0-56-generic). This occurs via the headphone jack and to a lesser degree, the speakers.

This does _not_ happen every time two sounds are emitted, but it rather seems like when a waveform is at its peak and is mixed with another sound, the resulting waveform forces the speakers to jump. This is easy to test - play a YouTube video and turn the sound up and down or cause the beep sound in a terminal and every second time it will pop.

I'm going to assume this is a common problem so I'll use the rest of this issue to say that I'm happy to test or contribute to the driver in any way possible. I don't normally work with Linux drivers but I'm going to look through the code and see if I can understand any of it.

davidjo, 2020-12-06:
Unfortunately probably cant help on this - I dont  know at all the ins and outs of the linux audio path - I have noticed issues during testing when starting/stopping streaming depending on whats left over in the buffers - but dont know what the fix would be or eg how to clear the buffers or delay output. It may be a pulse issue (Im assuming thats involved because dont think direct alsa handles multiple sources). I think Apple might do some ramping up/down - theres a lot of processing going on in Apples higher audio layer CoreAudio - their driver just streams. Might want to try the alsa/pulse lists. Even on standard linux PCs Ive noticed some buffering issues for audio.

frayment, 2020-12-06:
Thanks for the reply. I don't know if this is related but I've tried audio recording via. audacity too and a similar phenomenon is occurring - when starting recordings, every second time a large ramp will appear at the beginning (and make a pop).

This happens when using the "default" input source which I'm assuming is ALSA, and then it cuts out after about 4 seconds (and no more is recorded). Pulse is even worse, it simply doesn't record at all.

[image]

Attached image contains the smooth ramp to 0 (the small noises in between are just me clicking my fingers). Note also that audio only comes through one channel but the ramp is present in both.

If these issues don't seem familiar to you, I'll go ask the pulse/alsa teams. In saying that, I only just performed a stock install of Ubuntu yesterday, so it would be unlikely that I've accidentally changed a configuration.

frayment, 2020-12-06:
I've had some more time to test the headphone jack and it is fairly stable when I'm using one audio source however I've just come across an issue where a deafening noise, and I'm talking like +100dB starts playing and only stops when one hurriedly removes the jack.

I don't know how to get this to happen again (and frankly don't want to), but it occurred when I opened my settings to the audio tab. I'm starting to wonder if the problem is indeed some configuration on my end and not the driver?

davidjo, 2020-12-06:
Yes - I did see something similar - but dont have the time to explore it in detail.
I think part of it is initially I didnt want to diverge too much from the Apple HDA command order just in case.
I dont know enough about the streaming part of the linux kernel drivers to know how to eg delay processing/clear buffers or how it interacts with sending HDA commands.

frayment, 2020-12-06:
That is fair. I applaud you nonetheless for being able to develop the driver!

davidjo, 2020-12-06:
That sounds like the initial sound I ever got - be careful when plugging in/out - it cant handle it being done rapidly - leave a good few seconds before changing state eg start playing - it should handle plugin while playing without issue.
There are thousands of HDA commands sent which does take some time maybe near seconds.
Check the source for cs_8409_setup_amps12/cs_8409_setup_amps34 - you can set a permanent volume reduction - larger numbers lower volume - once I realised this I lowered it for debugging.

frayment, 2020-12-07:
> That sounds like the initial sound I ever got - be careful when plugging in/out - it cant handle it being done rapidly

Are you talking about the deafening sine wave? Plugging in/out just gives me small popping sounds.
That wave (it has happened a second time now so its somewhat reproducible) occurs randomly when I open the Settings application but only if the first page it automatically opens is the sound settings.

If I can get it to reproduce by toggling as you describe, I'll be curious to see what happens if I try to inline cs_8409_setup_amps12/cs_8409_setup_amps34 with the list of verbs that are sent, with volume set to a max of 0 to try to "mute" the screech.

davidjo, 2020-12-07:
yes - Im talking the deafening sine? wave - curious if it is a real sine wave - other issue that caused very loud sounds was mismatch between audio format being sent and programmed into the HDA - this was when start playing and just the amps.
My thought is something is being buffered somewhere (either HDA or amp chips) which is not cleared out properly.
Also it may be the unmute needs to be delayed - it could be Apple is assuming certain delays in the higher audio stuff which isnt true for the linux audio path.

frayment, 2020-12-07:
I still can't get it to play just by inserting headphones but using my Settings method I've managed to record it along with some other interesting things. The driver doesn't like me inserting/removing headphones while recording so I've had to use a phone so the capture is crap.

> yes - Im talking the deafening sine? wave - curious if it is a real sine wave

If you zoom in and squint hard enough, it definitely looks like one. I don't have a proper microphone setup so this is coming from my earphones as they jiggle around near the phone mic hence why the pitch is so unsteady.

[image]

Sometimes when inserting the headphones, audio will play but sound very digital, like half of the sound is missing. Checking the spectrogram shows that I'm right:

[image]

The very far left that is red everywhere is music playing normally, then the second block sounds incorrect, and is missing everything roughly below 3.5kHz. Third block the music goes quiet but the fourth, the large full block can be seen once again after removing and inserting headphones, returning to normal.

On the far right you can also see the "sine" wave. It is almost exactly 5.5kHz. Does this mean anything to you?
Close up for what its worth:

[image]

> My thought is something is being buffered somewhere (either HDA or amp chips) which is not cleared out properly.

Can you buffer the removal of half of the sound like that?

I've also managed to "fix" headphones already being plugged in on boot. I'll upload a PR later if my tinkering doesn't yield any more results.