davidjo/snd_hda_macbookpro #23: MacBook13,1 (A1708) audio quality is not ideal.

jaxjexjox, 2020-12-14:
Hi there,

Firstly, thanks for the hard work!
We've installed your driver on our MacBook13,1 under Ubuntu 20.04 and found that the audio is particularly weak / poor.

Based on a little bit of googling, I got the impression, some MacBooks have multiple sets of speakers and the volume needs to be adjusted for the larger bass speaker, or something to that effect.   Based on someone elses posts elsewhere.

However, upon opening the alsamixer tool, it does appear only one speaker has the ability to have the volume changed, remaning devices were locked at 0 and listed as s/pdif input / outputs of some kind.

Is this standard behaviour as the driver is incomplete, or is it possible to get 'good quality' sound, out of this model of Mac?

Audio device:
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
	Subsystem: Intel Corporation Sunrise Point-LP HD Audio
	Flags: bus master, fast devsel, latency 64, IRQ 59
	Memory at 92820000 (64-bit, non-prefetchable) [size=16K]
	Memory at 90000000 (64-bit, non-prefetchable) [size=64K]
	Capabilities: [50] Power Management version 3
	Capabilities: [60] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel

Please let me know what we should be doing, if anything.   The sound works but it's not ideal for listening to music or even really video conferencing.

davidjo, 2020-12-14:
Can you run ioreg -l -a >'some file name'.xml
This dumps your IORegistry as an xml file.
Then look for strings  like AppleHDATDMAmpxxxxxxx eg AppleHDATDMAmpSSM3515  which should be followed by an integer string - most will be 0 but one (for 13,1 I think its AppleHDATDMAmpSSM3515) should have a non-zero count and let me know.
Also look for the string IOHDACodecFunctionSubsystemID which should be followed by a large integer number also report that.
I have never had a 13,1 machine so all information I have is from other users so just want to confirm these values.

jaxjexjox, 2020-12-15:
Hello,

Sorry I had no idea that was a Mac command! I waited all day to finish work, to boot Ubuntu, to realise it's for Mac.
Here's some output.
AppleHDATDMAmpMAX98706
AppleHDATDMAmpSSM3515

<key>IOHDACodecFunctionSubsystemID</key
<integer>275460864</integer>

davidjo, 2020-12-15:
so the 275460864  in hex is 0x106b3300 - which you can see in the driver code.
Whats the integer number for AppleHDATDMAmpMAX98706 and AppleHDATDMAmpSSM3515 - this tells you how many amps and hence speakers you have (only one should be non-zero)
You should see something similar to
 <key>AppleHDATDMAmpMAX98706</key>
 <integer>4</integer>
which is for my MacBook Pro 14,3

davidjo, 2020-12-15:
Also what is the value for
`                      <key>IORegistryEntryName</key>
                        <string>MacBookPro14,3</string>
`
again thats for my machine.

jaxjexjox, 2020-12-15:
First post:

<key>AppleHDATDMAmpMAX98706</key>
			<integer>0</integer><key>AppleHDATDMAmpSSM3515</key>
			<integer>4</integer>

Your second post, hopefully this is helpful?:

<key>IORegistryEntryID</key>
			<integer>4294967575</integer>
			<key>IORegistryEntryName</key>
			<string>MacBookPro13,1</string>
			<key>IOServiceBusyState</key>
			<integer>0</integer>

davidjo, 2020-12-16:
Thanks - this confirms what I thought.
So your Macbook pro 13,1 has 4 SSM3515 speaker amps thus 4 speakers and subsystem id is 0x106b3300 (which is the same as a Macbook pro 14,1).

Now we have to start looking at the Alsa and Pulse audio layers to see what needs configuring.
When you say audio weak/poor can you be more specific - eg is it a low volume, does it lack bass.
If you play something with no bass on OSX and linux what are the audio differences you hear.

jaxjexjox, 2020-12-16:
You'll note, that alsamixer is showing a single speaker.
Don't quote me, but I've seen someone else, with a similar fault as described here - with a shot of multiple speakers with multiple control.

https://i.imgur.com/CxIB3Sx.png

jaxjexjox, 2020-12-16:
Ok so, first things first - when I logged this initial fault, we'd only tested 20.04 Ubuntu.  (apologies!)
I have now put the driver on 20.10.

The sound control panel does better than it did, however I still don't seem to have the sub being driven.
Alsamixer still only shows a single speaker to be messed with.

The 2.1 speaker test page:
https://i.imgur.com/Ge9Q6Ds.png
Front left and front right buttons, without question, test the same speaker. (what I believe to be the left speaker)
The Subwoofer button, appears to be driving, the right speaker.
The woofer, does not appear to be getting driven.

The 4.1 speaker test page:
https://i.imgur.com/wPP8a3a.png
Front left button appears to test, the left.
The rear left button, appears to actually output the right

Nothing appears to drive the woofer. (I double checked my ears, booted MacOS and spotify, same song, distinctly better woofer performance)

jaxjexjox, 2020-12-16:
Finally, I'm messing with pavucontrol and sound profiles, linux seems quite complicated in how it addresses audio devices.

https://i.imgur.com/ofamRl6.png

A large combination of options on the left window and right, trying 2.0 / 2.1 / 4.1 in an attempt to get the driver to push 'something' at the woofer but at this point, no such luck.

I could be doing something, entirely incorrect, as I'm not that experienced, apologies.
Let me know what else I could do, if anything.

davidjo, 2020-12-16:
As I said there are 4 amps and hence 4 speakers (for me 14,3 its clear there are 2 speakers on each side of the laptop - if you look at the teardowns its one speaker facing up and one speaker facing down - Im guessing the upward facing one is a tweeter and the bottom facing one is a woofer - so there are 2 "woofers").

Have you got the latest version?
There are 2 test waveforms under tests - these play a sound on each channel separately.
You need to identify which speaker they are playing from - yes I know its difficult to tell if coming from the top or bottom speaker.
Note that if you play these sounds you need to set the sound source in the Sound dialog of the Setttings app to surround 4.0.

What we now have to do is figure out how to map our input sounds (Im assuming in general this is 2.0 ie stereo) to this 4 channel system  (note that pulse/alsa will map 2.1 or 4.1 or any other format by downmixing/upmixing to the output format).

Now comes the first issue - a "normal" 4 channel surround system should be 2 stereo pairs but it seems Apple in its infinite wisdom decided that the 1st 2 channels are right/left (cant remember which one now) and the 2nd 2 channels do left/right - but you should be able to figure this from the test waveform - well I think that is what was found last time I checked.
(Try installing audacity - you can load the test waveform and play it muting all but one channel).

So if you confirm we have 2 left channels followed by 2 right channels we need to some channel mapping - this can be done in either alsa or pulse - my preference is to do this in alsa.
My assumption is that Apple is also performing some digital filtering which effectively enhances the bass for the 2 woofer channels. Again we need to do this in pulse/alsa (there is a pulse effects module which seems to be able to do this).

But thats as far as I have got - mainly by googling about downmix/upmix, channel duplication and channel mapping.
Note that if you have a 2.1/4.1 etc source you need to figure a mixing protocol to send the woofer sound to both woofer speakers (maybe with some contribution from the tweeter source as well as these woofers are not pure low frequencies).
If you have a 4.x, 6.x etc source you need to mix all pair channels down to a single stereo pair.

davidjo, 2020-12-16:
PS I should have said all the above upmix/downmix are config file changes for pulse/alsa (plus installing extra modules like pulse effects or alsa modules) - no driver changes are needed.

jaxjexjox, 2020-12-16:
Thanks for help David, will do in a few hours when my other half leaves and I can get my hands on it.
Shocked at how difficult this is, on a 2017 laptop that sells to hundreds of thousands (millions?) of people each year.

I'd have thought there would've been a bigger focus on this from many other developers helping out.

One question, once I identify and fix this, will we be able to roll those improvements, back into your driver for other users?

davidjo, 2020-12-17:
Sure we can add sample config files eg .asoundrc (alsa) or default.pa (pulse).
(some alsa  ideas can be seen in eg https://gist.github.com/thanley11/100754cc911442901867)

I think the goal is to create a "dummy" pcm channel which is 2.0 stereo - which we set as the sound source in the Sound section in Settings. Then all sound sources get downmixed to this 2.0 channel which then duplicates left and right channels and adds appropriate digitial filtering before sending to the  raw surround 4.0 pcm channel which sends to the driver.

But the linux audio system is very complex and havent found really good examples.

jaxjexjox, 2020-12-17:
I'm using the test file now (one is too quiet) and I've got to be honest, all 4 independent channels appear to be coming from a single speaker. (??)

I may need to adjust something in Audacity, as Ubuntu can definitely talk to more than 1 speaker, as the sound control panel, is allowing me send test audio to multiple speakers.

jaxjexjox, 2020-12-17:
Ok so I messed for it for about 15 minutes.
Colour me confused!
Firstly there's no subwoofer (as such) physically inside the device.
There are, infact, 4 speakers and they are infact 'front / back
https://www.youtube.com/watch?v=FTS7lTA8c7g

I can confirm this, by putting my ear darn close to the unit (A1708 remember) and by choosing "rear left / rear right"  it will cycle back and forth between right front and right rear https://imgur.com/wPP8a3a  (as mentioned, the labelling is wrong)

HOWEVER Audacity is 'useless' (?) or I'm doing something wrong.  The 4 channel test audio file, is definitely, all coming out, of the right speaker.

jaxjexjox, 2020-12-17:
I have now listened quite intently and it seems like this.

The default sound tool built in, the rear right speaker is correct.
The front left speaker is correct.

Pressing the button for rear left comes out on the other side.
Pressing the button for front right comes out on the other side.
I suspect those 2 are reversed.

davidjo, 2020-12-17:
Well  - I just tried audacity and it doesnt seem to be working right - I think its mixing down to mono and then just sending to all speakers - even using the supposed HW device (again use the StereoTest32_reduced_m24dB.wav if you use the HW device).
What should work is this
aplay StereoTest32.wav
or (and its very important to use the low volume version)
aplay -D hw:0,0 StereoTest32_reduced_m24dB.wav
- this sends audio direct to the hardware with no high level messing.

jaxjexjox, 2020-12-17:
Ok look, I'm fairly confident it's going like this

1 FRONTLEF
2 REARLEF
3 FRONTRIGHT
4 REARRIGHT

which honestly feels like that's what's intended? Despite the week sound which initiated my initial support query.

Does that match with your data?

(Regardless, labelling in the sound app, is wrong)

jaxjexjox, 2020-12-17:
Sorry if that's not clear, the 1,2,3,4 is based on the order things play in the file.
Note, I did use both versions of the file, with the same results.

davidjo, 2020-12-17:
My current thought is that the audio system on these laptops  is essentially a stereo speaker system (ie 2 channel) where each speaker has a tweeter and woofer.
However, because the driver is requesting 4 channels of audio  pulse/alsa incorrectly identifies this as a 4 channel surround setup.
We need to create a slave/virtual pcm channel which is a stereo/2 channel setup and use that as the high level device for alsa or as the pulse default sink (so that eg true surround inputs will be downmixed to 2 channels). We can then use the .asoundrc to perform the operations to duplicate the left, right streams, apply signal processing and remap to the channel order you have seen.
(Actually I have just re-looked at OSXs layout files (under /System/Library/Extensions/AppleHDA.kext/Contents/Resources) - your 13,1 machine is I think layout54.xml.zlib) and it  specifies this order of left tweeter, left woofer, right tweeter, right woofer).
I think the first stage is to setup an .asoundrc which creates this slave/virtual pcm device, duplicates the channels and correctly maps them. Once this is working can figure out how to do the DSP processing.

jaxjexjox, 2020-12-19:
Any tips on precisely what I need to do going forward? Even A tutorial.

davidjo, 2020-12-19:
Sorry - Im no linux audio expert so all Ive been doing is googling etc (and came to the conclusion its quite confusing - there the pulse/alsa interface and the higher level alsa/kernel alsa hardware driver interface).
I think Im beginning to get an inkling of what to do but thats all it is so far.
One thought I had is to try the alsa (or pulse) mailing lists and see if someone could help.

frayment, 2020-12-20:
Since I've been working on headphones for so long, I only just realised yesterday that indeed my 13,1 only has left speakers working too so I can confirm this isn't isolated. (At least now I know my right speaker isn't busted)

frayment, 2020-12-20:
@davidjo Can I please confirm that your channel mapping for the sound tests are as follows:

1: Left tweeter
2: Right tweeter
3: Left woofer
4: Right woofer

I'm writing a channel mapper in ALSA and need to know what to translate each channel to.

frayment, 2020-12-20:
@jaxjexjox Can you please try this configuration on your system?

Place into `~/.asoundrc` the following:

```
pcm.apple_hw {
	type hw
	card 0
	device 0
	channels 4
	rate 44100
	hint {
		show on
		description "Macbook13 sound card"
	}
}

pcm.apple_mixer {
	type route
	slave
	{
		#pcm "hw:0,0"
		pcm "apple_hw"
	}
	ttable {
		0.0 1.0 # FL
		1.3 5.0 # FR - underpowered 5 times?
		2.1 0.5 # RL
		3.2 0.5 # RR
	}
	hint {
		show on
		description "MacbookPro13,1 audio mixer"
	}
}
```

And run `aplay -D apple_mixer StereoTest32_reduced_m24dB.wav` as described by David (that is, use the REDUCED test).

jaxjexjox, 2020-12-20:
Thank you so much, I'll give this a try.   I really hope when we figure this out we can "roll this back in" somehow.

I don't know a whole heap about being allowed to contribute to Linux or Ubuntu but I'm suprised this hasn't been taken care of already by others much smarter.

Perhaps when we're finished, this could at least be made into a simple script?

I'll try over the next few days (I'm on leave by the laptop is with me)

frayment, 2020-12-20:
> I don't know a whole heap about being allowed to contribute to Linux or Ubuntu but I'm suprised this hasn't been taken care of already by others much smarter.

The sound driver is still in alpha and is entirely David's baby. Not many people want to contribute (which is why I'm trying even though I completely uninstalled macOS...)

> Perhaps when we're finished, this could at least be made into a simple script?

This is an ALSA configuration and is intended to be loaded on top of the systems default setup for sound. Placing it in `~/.asoundrc` works temporarily for a sound test but putting it in `/etc/asound.conf` _looks_ like it is made available at boot.

I've been messing around with the configuration and it looks like by default on Ubuntu 20.04, ALSA defaults to routing all output through PulseAudio so this cannot be easily integrated. If a custom sink can be made with the remapping done in pulse then it might work but my preliminary attempts have so far failed.

davidjo, 2020-12-21:
I believe what you need to do for pulse based systems like Ubuntu is try out the pactl command for controlling pulse and set the pulse default sink to the new pcm  - configs supposedly can be stored in the ~/.config/pulse/default.pa file.

This is my attempt at remapping in pulse - but yes havent managed to get this to work yet:
load-module module-remap-sink sink_name=remap-analog-surround-40 master=alsa_output.pci-0000_00_1f.3.analog-surround-40  channels=4 channel_map=front-left,front-right,rear-left,rear-right master_channel_map=front-left,rear-left,front-right,rear-right remix=yes

frayment, 2020-12-21:
> load-module module-remap-sink sink_name=remap-analog-surround-40 master=alsa_output.pci-0000_00_1f.3.analog-surround-40 channels=4 channel_map=front-left,front-right,rear-left,rear-right master_channel_map=front-left,rear-left,front-right,rear-right remix=yes

Funny you write this, I have tried verbatim the exact same command and it doesn't work. At least we're on the same page. Pulse doesn't seem to care for the sink via the config _and_ `pacmd`...

davidjo, 2020-12-21:
Nice - also using ttable in .asoundrc is what I was thinking of as well - for signal processing I think we need to look at the ladspa alsa library.

I think there should be a way I can specifically name the PCM stream in the driver - but havent found good example of that yet.

frayment, 2020-12-21:
What signal processing are you trying to apply?

You can also use `bindings` for the remapping but I had to use `ttable` for attenuation for the right tweeter.
At 5x power I swear I can hear it clipping at the very high ends which the left tweeter doesn't do. But it is a start.

frayment, 2020-12-21:
This ALSA configuration should also work for you now David, the only difference being your right speaker won't sound awful (please comment out my `ttable` section and use the other in `pcm.apple_converter`)

```
# define the apple hardware device
pcm.apple_hw {
	type hw
	card 0
	device 0
	# default rate and channel count but not necessarily respected
	channels 4
	rate 44100
	hint {
		show on
		description "Cirrus 8409 sound card"
	}
}

# plugin 1: create volume software controller
pcm.apple_softvol {
	type softvol
	slave {
		pcm "apple_hw"
	}
	control {
		name "Apple Cirrus 8409"
		card 0
		count 2 # stereo control
	}
	min_dB -51.0
	max_dB 0.0
	resolution 256
	hint {
		show on
		description "Apple software volume control"
	}
}

# plugin 2: map audio channels, convert stream and apply attenuation
pcm.apple_converter {
	type route
	slave {
		#pcm "hw:0,0"
		#pcm "apple_hw"
		pcm "apple_softvol"
		# convert audio to 4 channel, 32-bit little endian
		channels 4
		format "S32_LE"
	}

	# this is required for ONLY MacbookPro13,1
	ttable {
		0.0 1.0
		1.3 6.0 # 6x boost sounds more equal
		2.1 0.5
		3.2 0.5
		# added this to stream some of the front audio through the rear
		# so the sound doesn't sound flat on two channel
		0.1 0.25
		1.2 0.6
	}
	# for all others
	#ttable {
	#	0.0 1.0
	#	1.1 1.0
	#	2.2 0.5
	#	3.3 0.5
	#	# added this to stream some of the front audio through the rear
	#	# so the sound doesn't sound flat on two channel
	#	0.2 0.25
	#	1.3 0.25
	#}
	hint {
		show on
		description "Apple channel converter"
	}
}

# plugin 3: apply a rate converter to force 44.1kHz
pcm.apple_rate {
	type rate
	slave {
		pcm "apple_converter"
		rate 44100
	}
	hint {
		show on
		description "Apple rate converter"
	}
}

# plugin 4: apply mixing effects
#pcm.apple_mixer {
#	hint {
#		show on
#		description "Apple mixer"
#	}
#}
```

I'm going to use `apple_mixer` for "signal processing".
When testing this, please go to alsamixer and turn down the new volume control (that is not the PCM one).
Then you should be able to test with any audio and not only the reduced file.

@jaxjexjox You can ignore that last test, please use this one as it is current.

`aplay -D apple_rate any_audio_file.wav`

davidjo, 2020-12-21:
Thanks - Ill try this.

If you look at the layout..xml.zlib files (by the way I think for 13,1 its layout51.xml) there are some interesting Dsp definitions (which we have no idea how to interpret as this is nothing to do with the low level driver) eg DspEqualization32, DspMozartCompressor, DspLoudness, DspBuzzKill, DspFIRdirect_2chIn_4chOut (Im guessing FIR means its a classic Finite Impulse Response digital filter), DspControlFreak4ch, DspThermalSpeakerProtection4ch.
These may explain the difference in audio "quality" you can hear.

jaxjexjox, 2020-12-21:
I'm still on holiday but I suspect a much better chance to to this tomorrow.

If you don't mind could you list all the command I might need? I'm kinda Linux fluent but I've not messed with alsa or pulse audio at all yet

frayment, 2020-12-21:
I've just asked on the alsa mailing list, hopefully they may have some insight.

> DspLoudness, DspBuzzKill, DspThermalSpeakerProtection4ch

Well that certainly does sound like a few plugins that may solve my high pitched screeching issue too :stuck_out_tongue:

frayment, 2020-12-21:
@jaxjexjox 

1. Copy that second script I wrote and put it in `/etc/asound.conf`.
2. Open `alsamixer` and scroll to the right until you find the audio control for "Apple Cirrus 8409"
3. Turn it down to around 30. If it is too low when you test, you can turn it up. Better to be safe than sorry.
4. Run `aplay -D apple_rate any_audio_file.wav` with any sound file you have.
5. Confirm that you hear front left, front right, rear left, rear right.

jaxjexjox, 2020-12-24:
> @jaxjexjox
> 
> 1. Copy that second script I wrote and put it in `/etc/asound.conf`.
> 2. Open `alsamixer` and scroll to the right until you find the audio control for "Apple Cirrus 8409"
> 3. Turn it down to around 30. If it is too low when you test, you can turn it up. Better to be safe than sorry.
> 4. Run `aplay -D apple_rate any_audio_file.wav` with any sound file you have.
> 5. When running the test wav, confirm that you hear front left, front right, rear left, rear right.

Sorry to bug you, finally got free time on hols!
To clarify, it's Ubuntu 20.10 now, not 20.04  (dunno if it changes anything)

I've copied verbatim your text above into /etc/asound.conf (didn't change any permissions on the file)
Rebooted the system, because I'm a Windows guy and that's what we do, to be sure!

opening alsamixer the devices, appear unchanged to me in title? It's still referencing 
"HDA Intel PCH" as first entry with the chip listed as "Cirrus Logic CS8409"
Would this imply I've not applied the Apple 'profile' to the soundcard?

Opening the stock "sound" menu within Ubuntu still only produces headphones / speakers (built in) and 3 configurations of Analog 4.0 / 2.1 and Stereo

Clicking the buttons in the control panel itself, still produces incorrect sound.
Finally, using the aplay -D apple_rate <file> command, produces the error

"aplay: main:830: audio open error: Device or resource busy"
I speculate, I've done something wrong here.

jaxjexjox, 2020-12-24:
Note: I have installed, pulseaudio in an attempt to mess with stuff historically on this machine to try and diagnose, I'm unaware if this might break stuff!

frayment, 2020-12-24:
> Ok so I've removed pulseaudio from the system, rebooted, this has resulted in some improvements.
> arecord -L is confirming, that the custom PCM devices are indeed on the system, asound.conf working

I don't know if you're supposed to remove pulse - it comes preinstalled in Ubuntu and still shows for me but as long as you have the custom devices...

> I am unsure if your command, is accidentally, incorrect? But if I type aplay -D **apple_hw** (file) it seems to be work.

`apple_hw` is the direct hardware device which you shouldn't use (the channels don't get swapped there). Use `apple_rate` to have all of the proper effects everything applied to the audio.

> Left Front, Left Rear, Right Front, Right Rear (?? I think that's what we're after)

That's what we already had. It won't change if you use `apple_hw`.

> If I type aplay -D **apple_rate** however, rather than HW - the result is significantly more quiet and a different order.
> Left FRont, Right FRont, Left REar, Right REar (??)

Okay perfect, that's what I was testing. If you run alsamixer and change the "Apple Cirrus 8409" bar, you will be able change the sound level. Does the right front sound as loud as the left front for you?

> Also of note, the volume slider, now appears, unable to change volume.
> The 'normal sound control panel' (I do not know the term) appears to be lacking the device

That's probably because you uninstalled pulse which at least on my computer, alsa defaults to using. The PCM devices I created don't route through the default volume control, they use "Apple Cirrus 8409" so that you can test the devices separately from your system sound level.

> Is that a fixable thing?

Try installing pulse again.

> While I've got you - the trackpad, has anyone written some code to disable it while typing?? My poor wife needs to use this system, I'm trying to make ubuntu as usable as possible for her.

Works out of the box for me? Try [p2rkw/xf86-input-mtrack](https://github.com/p2rkw/xf86-input-mtrack).

PS: In your original post you wrote that you kept getting "Device busy". Just keep running the command until it works. Happens because you already have a sound playing in the background or one just ended a few seconds ago.

davidjo, 2020-12-24:
My problem with the touch pad I think is palm detection - the plain Ubuntu system should be disabling touchpad (I have gnome-tweaks installed which shows this setting). I find if I keep the palm high its reasonable.
Im using the defaul libinput but you can tweak this - you can put some eg pressure tweaks in config files in /etc/udev/hwdb.d.
you can try things out with eg sudo libinput measure touchpad-pressure --touch-thresholds=10:8 --palm-threshold=20.
This needs the libinput package installed - cant remember if this is default or not.
Thanks for the suggestion of mtrack - I have come across this so will have to try it out.

jaxjexjox, 2020-12-25:
> I don't know if you're supposed to remove pulse - it comes preinstalled in Ubuntu and still shows for me but as long as you have the custom devices...

Removing pulse fixed my issues.  I will re-install and test based on your reply.
I'm /almost/ positive it was not installed as default in my 20.10 install?

> `apple_hw` is the direct hardware device which you shouldn't use (the channels don't get swapped there). Use `apple_rate` to have all of the proper effects everything applied to the audio.
>
>
> That's [...]

Using aplay -D *apple_rate* produces an atrociously crackly mess, on the right side of the laptop, for the first part of test.
It goes like this, I **think**. 
Left REAR?? medium level volume
Right REAR?? high level, crackly mess noisy thing
Left Front, fairly quiet
Right Front, fairly quiet.

Don't quote me on that, sadly - but it seems, the quieter speakers are the front ones (??) and they appear to be the ones, that play in the latter half of the test?

> > Also of note, the volume slider, now appears, unable to change volume.
> > The 'normal sound control panel' (I do not know the term) appears to be lacking the device
> 
> That's probably because you uninstalled pulse which at least on my computer, alsa defaults to using. The PCM devices I created don't route through the default volume control, they use "Apple Cirrus 8409" so that you can test the devices separately from your system sound level. As for showing it in the control panel, I haven't gotten that far. Probably be easier to just use the PCM in the sound driver.
> 

Happy to wait if we end up solving it!

> > Is that a fixable thing?
> 
> Try installing pulse again.
> 

Appears to have fixed volume, hasn't impacted ability to test.
Note, should there be any difference, besides volume on the 2 included test files?
The 'reduced' file doesn't clip and mash on the right hand side (first play) however, it is by far, the loudest speaker of all 4.  You'd think they'd all be identical or 2 pairs?

> > While I've got you - the trackpad, has anyone written some code to disable it while typing?? My poor wife needs to use this system, I'm trying to make ubuntu as usable as possible for her.
> 
> Works out of the box for me? Try [p2rkw/xf86-input-mtrack](https://github.com/p2rkw/xf86-input-mtrack).
> 
> PS: In your original post you wrote that you kept getting "Device busy". Just keep running the command until it works. Happens because you already have a sound playing in the background or one just ended a few seconds ago.
> 
> PPS: `apple_mixer` should not be showing up? I commented it out (and it does nothing)
> 
> PPPS: Ensure no headphones are plugged in and none are plugged in at boot.

Do I need to re-import, a new version of your asound.conf file? I def see apple_mixer in my arecord -L list.
No headphones, no dock - we're on holiday, it's laptop on its own.

No more device busy, not sure why, phew?

jaxjexjox, 2020-12-25:
> My problem with the touch pad I think is palm detection - the plain Ubuntu system should be disabling touchpad when typing (I have gnome-tweaks installed which shows this setting). I find if I keep the palm high its reasonable.
> Im using the defaul libinput but you can tweak this - you can put some eg pressure tweaks in config files in /etc/udev/hwdb.d.
> you can try things out with eg sudo libinput measure touchpad-pressure --touch-thresholds=10:8 --palm-threshold=20.
> This needs the libinput package installed - cant remember if this is default or not.
> Thanks for the suggestion of mtrack - I have come across this so will have to try it out.

`sudo libinput measure touchpad-pressure --touch-thresholds=10:8 --palm-threshold=20
Using Apple SPI Touchpad: /dev/input/event5

This device does not have the capabilities for pressure-based touch detection.
Details: Device does not have ABS_PRESSURE or ABS_MT_PRESSURE
`

Maybe I've done something wrong? I just outright want it to stop detecting touches for about .5 seconds, after typing at all.

frayment, 2020-12-25:
> Using aplay -D _apple_rate_ produces an atrociously crackly mess, on the right side of the laptop, for the first part of test.
> It goes like this, I **think**.
> Left REAR?? medium level volume
> Right REAR?? high level, crackly mess noisy thing
> Left Front, fairly quiet
> Right Front, fairly quiet.

It should be FL-FR-RL-RR, I didn't make any changes to the first output so it shouldn't be rear left. As for the right side, I've played the actual front rear at an increased volume and it sounds even worse than what you're experiencing with the front right. As for why our front right is a higher level (and a crackly mess) than our front left, I don't know. I suppose any one of these are possible:

1. My front right speaker is partially blown (unlikely, was fine before I installed Linux)
2. Your front right speaker has a problem (also unlikely)
3. Right side isn't receiving the right amount of power
4. Right side isn't initialised properly

You can go to the config file and turn down the `6.0` in `1.3 6.0 # 6x boost sounds more equal` to something like `5.0` or `4.0` to get it equalised. For mine, I had to boost up six times to get it to the same volume as the front left. Good to hear the rear left and rear right are both quiet though.

> Don't quote me on that, sadly - but it seems, the quieter speakers are the front ones (??) and they appear to be the ones, that play in the latter half of the test?

Boost the sound from `0.5` to something like `3.0` in the `ttable` and you will see that what you think are the front speakers are indeed the rear. They will be clipping even more. You may also want to comment out the last two lines which I explain are to mix the front channels slightly into the rear ones to give more depth, otherwise you will get confused what you are actually hearing.

> Note, should there be any difference, besides volume on the 2 included test files?
> The 'reduced' file doesn't clip and mash on the right hand side (first play) however, it is by far, the loudest speaker of all 4. You'd think they'd all be identical or 2 pairs?

The front left and front right _should_ be identical but they're not. Hence why I boosted mine 6x to get a rough estimate. I didn't do anything scientific and your ears will inevitably be different from mine.

> Do I need to re-import, a new version of your asound.conf file? I def see apple_mixer in my arecord -L list.
> No headphones, no dock - we're on holiday, it's laptop on its own.
> 
> No more device busy, not sure why, phew?

Don't worry about the busy devices, it's not an error. You just wait a few seconds if it shows. Check your `/etc/asound.conf` and `~/asoundrc`. You shouldn't have both unless you have a reason to and if `apple_mixer` is showing up, you probably still have the old config somewhere. Delete both and make sure you are only using the newest in **only one** of these two places.

Merry Christmas.

jaxjexjox, 2020-12-26:
I'll try some more of your reply when I get home from leave.

I take it that this is the only known driver for this peripheral for Linux?  I'm very happy with this driver, for free of course, but if someone else is also working on it, I'll obviously investigate.

Thought canonical might prioritise this but seemingly not

davidjo, 2020-12-26:
> `sudo libinput measure touchpad-pressure --touch-thresholds=10:8 --palm-threshold=20
> Using Apple SPI Touchpad: /dev/input/event5
> 
> This device does not have the capabilities for pressure-based touch detection.
> Details: Device does not have ABS_PRESSURE or ABS_MT_PRESSURE
> `

It works on my 14,3 machine so maybe the 13,1 touchpad doesnt do pressure - might only be using area based detection.

davidjo, 2020-12-26:
Just in case  - you are using the roadrunner2 macbook12-spi-driver?

davidjo, 2020-12-28:
> It should be FL-FR-RL-RR, I didn't make any changes to the first output so it shouldn't be rear left. As for the right side, I've played the actual front rear at an increased volume and it sounds [...]

> My front right speaker is partially blown (unlikely, was fine before I installed Linux)Your front right speaker has a problem (also unlikely)
Right side isn't receiving the right amount of power
Right side isn't initialised properly
You can go to the config file and turn down the 6.0 in 1.3 6.0 # 6x boost sounds more equal to something like 5.0 or 4.0 to get it equalised. For mine, I had to boost up six times to get it to the same volume as the front left. Good to hear the rear left and rear right are both quiet though. 

I have been listening to the test audio waveform (using aplay) on my 14,3 machine (no .asoundrc) and I cant say I hear any of these problems - the volume of the left/right tweeters sounds very similar - as does the woofers - there is an obvious difference between the tweeters/woofers sound without any additional processing. There is no crackling of any kind - my initial guess would be the amps may be clipping because of the boosted volumes. Try setting the ttable 2nd number to 1.0 for all 4 channels in the .asoundrc.

davidjo, 2020-12-28:
> Left REAR?? medium level volume
Right REAR?? high level, crackly mess noisy thing
Left Front, fairly quiet
Right Front, fairly quiet.

So I think this is consistent with what I hear - the "front" speakers are the tweeters - they sound quiet because like normal tweeters their response drops off for the low frequencies (psychoacoustically bass "sounds" louder than higher frequencies).
The left rear ie woofer sounds stronger because of better bass response - and medium level because the ttable volume is 1.0.
The right rear ie woofer is clipping because of the volume increase due to the large ttable volume of 6.0.

davidjo, 2020-12-28:
I may have made progress in how to setup pulse audio for this driver.

I have been able to get the pulse remap sink command to work - and also to understand why it doesnt seem to work if put into a default.pa file.

frayment, 2020-12-31:
> I may have made progress in how to setup pulse audio for this driver.
> 
> I have been able to get the pulse remap sink command to work - and also to understand why it doesnt seem to work if put into a default.pa file.

That's good. I've been away for a few days, but will start experimenting again too.

> my initial guess would be the amps may be clipping because of the boosted volumes.

This is why I'm thinking that there is a difference in the amp setup on our computers. I *still* cannot test this yet without my cable but I'm going to go over all of the `snd_hda_codec_write` calls when I do.

Perhaps the right side is being sent less power due to the wrong speaker order and is acting like a bass speaker, whereas the bass on the left is as loud as the tweeter because it's being fed too much - only time will tell...

> Try setting the ttable 2nd number to 1.0 for all 4 channels in the .asoundrc.

This is basically the "original" selection by default and is where all these issues are coming from. FL and RL are equally loud, FR is extremely quiet (but notably not clipping) and RR is basically non existent.

davidjo, 2020-12-31:
Well I keep going forwards/backwards - discovering all sorts of undocumented issues with pulseaudio/alsa so not clear Im any further along!!

As to the volume issue - this seems to be different between jaxjexjox experience and your experience.
Using aplay -D hw:0,0 StereoTest32_reduced_m24dB.wav do you hear the difference described above between FR and RR?
(I was suggesting using 1.0 for jaxjexjox because my reading of his reports is that the 6.0 value is causing clipping).

davidjo, 2020-12-31:
This is my current attempt at setting up audio - there are still issues.
I think its working for stereo media but its not doing what I thought for multi channel media.
It should be duplicating the channels onto the tweeters/woofers.

Im attaching my versions of .asoundrc, ~/.config/pulse/default.pa and a trysetup2 file.

[dot.asoundrc.txt](https://github.com/davidjo/snd_hda_macbookpro/files/5757857/dot.asoundrc.txt)

[default.pa.txt](https://github.com/davidjo/snd_hda_macbookpro/files/5757858/default.pa.txt)

[trysetup2.txt](https://github.com/davidjo/snd_hda_macbookpro/files/5757859/trysetup2.txt)

Remove the .txt from all, rename dot.asoundrc to .asoundrc and place in $HOME, mv default.pa to ~/.config/pulse

If the pulse initialization fails - you see Dummy Output in Sound section of Settings dialog, run trysetup2 as bash -x trysetup2.

OK so there seems to be a big issue with using default.pa files.
Currently at least for Ubuntu the pulse audio system is driven by the /etc/pulse/default.pa (which can be overridden by the same file in ~/.config/pulse).
It is setup to use module-udev-detect - this uses linux udev system to find audio devices (mainly for pluggable audio devices but works for the primary hardware device even tho is permanent in this case).

module-udev-detect finds cards then loads them using module-alsa-card and sets up sinks/sources.
But it first checks for access to the sound device /dev/snd/controlC0 etc.
If the sound device is busy the module load fails - this means any commands in default.pa referencing sinks created by this device fail. Now it apparently continues polling the device until its not busy and then loads it BUT this is past default.pa commands so none of those setup commands get run. This for me has been primary reason for Dummy Output - or missing the remap device from the module-remap-sink command mentioned above.

Why is the device busy??
Well the GUI login process first uses gdm to present the login screen then switches to gnome-shell after user login (Ubuntu).
Currently Ubuntu defaults to user based pulseaudio daemons (rather than a system wide daemon).
So the gdm process itself fires up a pulseaudio daemon - for login sounds. This then switches to the user separately forked daemon. I think because of the large number of HDA commands sent by the Apple driver these 2 daemons overlap so when the user daemon fires up the sound device is busy.
(Curiously both playing and capturing seem to be activated.)
I have a possible fixup involving a script in PostLogin (/etc/gdm3/PostLogin/Default) but this is work in progress - basically kill the current user daemon and restart it when know the gdm daemon has stopped.

frayment, 2020-12-31:
@davidjo I was about to respond to this, but I think I may have just fixed it! I suggest you look quickly at this file: `/usr/share/pulseaudio/alsa-mixer/profile-sets/default.conf`

I've managed to get a sink setup and it actually works with the `apple_rate` pcm just fine. You can even use your hotkeys to turn the sound up and down.

1. Copy my little config below to the bottom of the above `default.conf` file.
2. You need to go into alsamixer and turn the `Apple Cirrus` control right up to 100% or you won't hear it. 
3. Go to PulseAudio Volume Control and set your sink to Apple Cirrus.
4. Turn your sound up and down, you should be able to hear it!

Now on my end, there is this strange clipping on the right side, but that may be because I used `left,rear-left,rear-right,right`. I'm testing it now but please try the below for me without any of your configuration and tell me if it works.

```
[Mapping apple-output]
description = Apple Cirrus 8409 sink
device-strings = apple_rate
channel-map = left,right,rear-left,rear-right
priority = 1
direction = output
```

frayment, 2020-12-31:
Here is my updated `asound.conf` file:

```
# define the apple hardware device
pcm.apple_hw {
	type hw
	card 0
	device 0
	# default rate and channel count but not necessarily respected
	channels 4
	rate 44100
	hint {
		show on
		description "Cirrus 8409 sound card"
	}
}

# plugin 1: map audio channels, convert stream and apply attenuation
pcm.apple_converter {
	type route
	slave {
		#pcm "hw:0,0"
		pcm "apple_hw"
		# convert audio to 4 channel, 32-bit little endian
		channels 4
		format "S32_LE"
	}

	# this is required for ONLY MacbookPro13,1
	ttable {
		0.0 1.0
		1.3 6.0 # 6x boost sounds more equal
		2.1 0.5
		3.2 0.5
		# added this to stream some of the front audio through the rear
		# so the sound doesn't sound flat on two channel
		0.1 0.25
		1.2 0.6
	}
	# for all others
	#ttable {
	#	0.0 1.0
	#	1.1 1.0
	#	2.2 0.5
	#	3.3 0.5
	#	# added this to stream some of the front audio through the rear
	#	# so the sound doesn't sound flat on two channel
	#	0.2 0.25
	#	1.3 0.25
	#}
	hint {
		show on
		description "Apple channel converter"
	}
}

# plugin 2: apply a rate converter to force 44.1kHz
pcm.apple_rate {
	type rate
	slave {
		pcm "apple_converter"
		rate 44100
	}
	hint {
		show on
		description "Apple rate converter"
	}
}

#ctl.!default {
#   type hw
#   card 0
#}
```

Once you use this, you will lose the `Apple Cirrus` bar in alsamixer because it's no longer needed.

For you @davidjo, you will obviously want to put your old ttable settings back to how you like them.

davidjo, 2020-12-31:
I did think of messing with profiles - but was trying to avoid having to update a system file.

Actually 14,3 is the same as 13,1 - speaker order is left tweeter, left woofer, right tweeter, right woofer (see layout57.xml)

frayment, 2020-12-31:
> Actually 14,3 is the same as 13,1 - speaker order is left tweeter, left woofer, right tweeter, right woofer (see layout57.xml).

Ah, was wondering what went wrong there. Thanks.

> but was trying to avoid having to update a system file.

That file is simply a profile. We can most likely create our own elsewhere and use some command in `default.pa` to set it and thus eliminating any system changes but that is something I am yet to solve.

jaxjexjox, 2021-01-01:
Sorry it's been a few days as I'm back at physically working in an office.

So where are we at this point, does the clipping continue on the right side?

What should I do from here?

frayment, 2021-01-01:
@jaxjexjox For me personally I don't really have much clipping as I do a right side that just sounds like it's missing half of its range. Given your right side also sounds a lot louder at `6.0` than mine does means there's something that isn't the same between our two computers.

What you can do for now is try the new configs written above and report back. This new method will let you actually use the audio device properly with all the audio on your computer but I'm still trying to get it to work with headphones as the above configs will stop them working (at least for me).

I'm writing an installation script in another branch on my fork that contains these configs and installs them in places that don't override system defaults but you won't be able to see my solution until I push the commit which I won't do until probably the weekend or sooner when I figure out how to get headphones working with it.

davidjo, 2021-01-02:
I think Ive found a template to follow - this is for recent MBPs with T2 chip (2018) - so we need to modify it - but it creates Profiles, adds in .asoundrc updates etc.
(I think the reason headphones stop working with just the Mapping is because it doesnt allow any other output - a Profile key with multiple output paths is needed).

https://gist.github.com/kevineinarsson/8e5e92664f97508277fefef1b8015fba

(Although Ive been trying to get udev rules to work and failing).

frayment, 2021-01-02:
> (Although Ive been trying to get udev rules to work and failing).

Have a look at my `pulse` branch in my fork at [frayment/snd_hda_macbookpro](https://github.com/frayment/snd_hda_macbookpro/tree/pulse).

My udev rule works fine for the profile. Note that I'm not done yet. The install script can be considered ready but the profile still has issues.

Whatever it is that I've done, I can't have a mapping for quadraphonic, stereo and mono at the same time or my outputs clip into each other and the speakers sound awful. Just mapping the quadraphonic speaker works fine for now though.

[30 more replies]