davidjo/snd_hda_macbookpro #25: Microphone hang after plug in

noellimx, 2021-02-15:
Hi, hope this is the right place-

i have ubuntu 20.04 on macbook 13,1 and ran the cirrus patch. i am using at pulseaudio for surface diagnostics. I have a 3.5mm jack headphone with mic.

On start up with plugged in headphone, input works but output doesn't. (im assuming the input is receiving from internal mic based on the next behavior). after plugging out the headphone input continues to work and output (internal speaker) also works. plugging in again, output is transferred to headphone but the microphone bar hangs.

I reboot to try with no headphone, I/O works perfectly until headpphones plugged in again (same behavior as above)

Is there a solution for this? I can make do with permanent internal mic. Thanks

davidjo, 2021-02-16:
Booting in with headphones was not something I implemented initially - theres been some work (see issue #22) but its not resolved yet for the 13,1 (which I dont have) - however my updates to attempt to handle plugged in headset are not in the main code yet (see patch_boot.diff.txt in issue #22).

By the, what sort of headset/headphone are you using, generic or Apple??
Apples OSX driver distinguishes between 3 headset types, generic no mike, generic with mike and Apple with mike - generic with mike is also not implemented.

noellimx, 2021-02-17:
Thanks for ur reply. I booted with headphones plugged in just for troubleshooting purposes, I need the mic working for my zoom sessions. My earpiece is Sony (generic ?) with mic. 

Sry this is quite technical to me, what do you mean by OSX driver? Isn't the driver support for Linux compatibility with cirrus hardware?

davidjo, 2021-02-17:
Im afraid this driver is not at the install and go stage - because there is no documentation either from Apple or Cirrus etc about the chips involved - it is based on reverse engineering the OSX driver.

Yes - a Sony mike is likely generic with mike - this is not fully implemented - because I dont have such a headset.
At the moment either try to get an Apple headset or use a bluetooth/usb headset - they should work as they do not use this driver.

noellimx, 2021-02-17:
I understand now, seems like tough work. Thx anw, the patch has been a great help for a start.

snake0x65, 2021-03-07:
Hi Guys, 
We have just released support for CS8409 and CS42L42 for Dell laptops, I guess older Mac models are using same HW configuration. I just thought ouy might be interested in these series of patches.  
https://git.kernel.org/pub/scm/linux/kernel/git/tiwai/sound.git/log/?h=for-next