davidjo/snd_hda_macbookpro #27: Mainline?

garritfra, 2021-04-05:
Hi,

first off, thanks for making this driver! It works great.
My question would be if this could be mainlined into the kernel, since there is probably some demand for sound on newer MacBooks. It took me quite some time to discover this driver, since it's not yet mentioned in many of the forums.

I understand that this is still in very early stages, but mainlining it could kickstart more development work and significantly help the Linux-on-Mac ecosystem.

Let me know if this was already a topic being discussed.

Cheers!

davidjo, 2021-04-08:
Unfortunately I dont think thats going to happen because this is a reverse engineered driver with no documentation for the Apple custom version chips (apparently there is now a driver for a very similar system on Dell laptops CS8409/CS42L42 - at least for the head phones - but these are documented chips #25).
I seem to remember some comments on other kernel/driver lists which suggested reverse engineered drivers dont get into the kernel.
It is now mentioned on the dunedan linux on macbook pros github - which is where I have looked to see generally how to boot linux

garritfra, 2021-04-08:
I see, thanks for your explanation.

Just out of curiosity: are the chips on the older MacBooks documented? I'm able to get sound working on a 2011 MBP, but I frankly don't know if apple back then used off-the-shelf hardware or if the maintainers just had more time to get the chips working on Linux.

closed 2021-04-18 by garritfra