davidjo/snd_hda_macbookpro #28: gentoo support

maxmadzz, 2021-06-13:
It seems that this shell script is to download a vanilla kernel source code and then modify the file for module compilation. Can you release it as a patch file or support the operation of the existing kernel source code version or support the compilation of Gentoo distribution?

davidjo, 2021-06-13:
Dont quite get what you need here - what existing source code version? patch file to what?
I dont use Gentoo so someone with such a system needs to work with it and suggest an update.

maxmadzz, 2021-06-14:
I see it in the shell script

`# attempt to download linux-x.x.x.tar.xz kernel`

Can it be used to use the existing kernel source code? Instead of downloading vanilla kernel source code.

`sed -i 's`
Can you change the form of "sed replace text" to "patch/diff"?

davidjo, 2021-06-14:
You can see how to use your systems kernel sources if you look at the fixup for kernel version 5.4-48.
The tar command below the sudo apt install linux-source comment (which may not be the command you need to get kernel sources on Gentoo) shows how to extract the hda source folder from kernel sources.
The choice of using sed was just easier - and saves having to maintain multiple patch scripts possibly dependent on OS.
Dont quite understand why you want patch/diff - the changes were some global name changes in the mainline source plus additional argument to snd_hda_jack_tbl_get_from_tag.
Everything in the script can be done manually so you can generate your own patch/diff scripts if you want them.
(note that this does not change the system kernel sources - the edits only affect the snd_hda_macbookpro files).

maxmadzz, 2021-07-29:
All Gentoo packages are automatically built from source code.Before the package configuration is compiled, the Gentoo package manager accepts the patch file and applies it to the package.

davidjo, 2021-07-29:
Just comment everything from cd $hda_dir in the install script then run it - you now have the modified files under build/hda-{kernel version} in the directory you ran the script in.
diff -ruN your original gentoo kernel source with those files to create your patch file.
(This is all a guess - someone with a Gentoo system needs to be doing this).