davidjo/snd_hda_macbookpro #29: built in microphone stops working when plugging in a headphones

MightyPrince1, 2021-07-10:
And once it's taken out that doesn't fix it
in the settings I made sure, that it used the built in microphone

(the headset does have a mic, but a) it sucks b) it doesn't work if I use it as well)

Model: Macbookpro 14,3
Kernel: 5.4.130-1-MANJARO

MightyPrince1, 2021-07-10:
Love the work you're doing btw

davidjo, 2021-07-11:
Cant help with this - all the code currently is based on logging actions under OSX  - dont even know if you can do this in OSX.
Mike usage isnt something I looked at sorting out so its very untested - the current behaviour is the default for OSX on headphone plugin - and at the moment any linux user side setup will be ignored on headset plugin - dont really understand the interaction between user audio setup and the kernel driver on linux as the kernel drivers dont seem to use any user input.
Some important functions are cs_8409_unplug_handle_disconnect and cs_8409_intmike_linein_resetup.
(usb/bluetooth headphones/mic work on linux and are nothing to do with this driver).

MightyPrince1, 2021-07-11:
@davidjo is there anything I could do to disable automatic mic switching(aka when plugging in a headphone it will simply not do anything to the mic)?

davidjo, 2021-07-11:
look in cs_8409_plugin_event_continued - this does the main work of switching outputs/inputs on plugin in particular cs_8409_intmike_linein_disable as its name implies.

Also switch_input_src - this switches the inputs (and displayed as input options to the user) the kernel driver has found via snd_hda_activate_path (a base kernel function) calls.

ExternPointer, 2026-09-05:
**MacBookPro14,3, kernel 7.1.9: internal mic unusable with wired headphones that have a mic — patch attached**

Hardware/software: MacBook Pro 15" 2017 (MacBookPro14,3, A1707, Touch Bar / T1), codec CS8409/CS42L83 (`HDA:10138409,106b3900`), Arch Linux (Omarchy), kernel 7.1.9-arch1-2, driver built from current master via DKMS. Speakers and internal mic work; the problem appears only with wired headphones that have a microphone.

With such headphones plugged in (`headset has mike!!` in dmesg) the internal mic cannot be used:

- `cs_8409_pcm_capture_pre_prepare_hook` sets up only the headset mic path when `jack_present && have_mike`; `cs_8409_capture_setup()` is never called. The headset mic did not work either (no data from the capture stream), so no microphone was available at all.
- Plugging the headphones during a capture powers off the internal-mic ADC (`PLUGIN WHILE CAPTURING UNIMPLEMENTED!!`) and the capture stalls.

The attached patch makes the driver always use the internal mic: the capture hooks always call `cs_8409_capture_setup()` / `cs_8409_capture_cleanup()`, the boot/plug/unplug handlers no longer disable the internal mic or switch the input paths, and they do not power off the internal-mic ADC while a capture is running.

Tested on 14,3 with headphones (with mic): internal mic records, plugging/unplugging during a capture no longer breaks it. The headset mic is not used at all; it did not work before the patch either.

[snd_hda_macbookpro-capture-src.patch](https://github.com/user-attachments/files/31862160/snd_hda_macbookpro-capture-src.patch)

dnlbtz, 2026-11-09:
**MacBookPro14,3: headset mic and internal mic both usable, independent of the jack**

MacBookPro14,3 (15" 2017, subsystem 0x106b3900), CachyOS, kernel 7.3, PipeWire 1.6, driver from the `test_unsol` branch (4526085).

With a four-pole headset in, the type detect works ("headset has mike!!"), but there was no usable microphone at all, as reported above for the same model. Four causes in the driver:

1. `cs_8409_cs42l83_mark_jack` marks only the headphone pin 0x24 dirty. `snd_hda_jack_report_sync` therefore never re-reads the mic pin 0x3c, so "Mic Jack" stays off and PipeWire never offers the headset mic.
2. "Capture Source" cannot be set to "Mic". The generic parser sets `dyn_adc_idx[]` to positions in the full list of 18 ADCs, then `cs_8409_apple_check_dyn_adc_switch` shrinks `adc_nids`/`input_paths` to the two connected ones without remapping it. `get_input_path()` returns NULL and `mux_select()` refuses every change.
3. `cs_8409_capture_pcm_prepare` sets the stream up on `hinfo->nid`, which with `dyn_adc_switch` is always `adc_nids[0]` = 0x22, while the hook routes the CS42L83 into 0x1a. The capture never runs (`arecord`: Input/output error).
4. The capture hooks pick the headset side from `jack_present && have_mike`, not from the selected input. The plug and unplug handlers call `switch_input_src` and disable the internal mike, even during a running capture. So the internal mike is unusable while a headset is in, which is this issue.

Fix: https://github.com/dnlbtz/snd_hda_macbookpro/tree/headsetMic
- `f21c28b` reports the Mic Jack and remaps `dyn_adc_idx` (1, 2)
- `a01406b` sets the stream up on the selected ADC (3)
- `1c94ee9` makes the codec side follow the Capture Source, adds a `cap_sync_hook` for switching during a capture, and leaves the input paths to the mux. `INTERNAL_MIKE_ONLY` is no longer needed, so its Makefile line is gone (4)

The bottom commit `a7fce12` is the resume-after-hibernation fix from #198, ported onto `test_unsol`. The three mic commits also apply without it, but they haven't been tested that way.

Tested on the 14,3: with the headset in, Mic Jack goes on and PipeWire switches to the headset mic, which records. The internal mic can still be chosen with the headset in (Capture Source / `pactl set-source-port`), also during a running capture. Unplugging and plugging the headset while recording from the internal mic causes no gap. Builds with the Makefile's normal flags and with the debug ones (`MYSOUNDDEBUGFULL`). Not tested: three-pole headphones, and kernels before 6.17 (the `patch_cirrus_apple.h` path is changed the same way but not built).

Outside the driver: the headset mic arrives about 35 dB quieter than the internal mics (speech around -72 dBFS). macOS applies that gain in software. Layout 57 has `ExtMic/HeadsetMic_dBV` = 35, AppleHDA passes it to `DspFuncManager::setMicPreAmpGain`, and `applyMicPreAmpGain` multiplies the input by 10^(35/20). On Linux it belongs in userspace too; here a PipeWire filter-chain source applies that gain on the headset port.

davidjo, 2026-11-09:
I dont know if you saw this but this is a comment I made under pull #219.

However - I did figure something out - prior to kernel 5.13 (I think) the auto config had a dimension for the number of adc nids which the 8409 exceeds (I did do some fixups to handle this but without changing this dimension) - even though only 2 adc nids actually do something useful (internal mike, headset mike).
At 5.13 mainline (now with the Dell module so now hda_generic.c has to handle the actual number of adc nids) this dimension was increased,
What I did not realise is this lead to 2 different code paths in input handling.
Prior to 5.13 the dyn_adc_switch path was not used - the Alt Analog path was used - because the pre Dell path only looked at the 1st 8 adc nids (the old auto config dimension) and none of these are connected.
Post the Dell module the dyn_adc_switch path is now followed. I did not notice this at the time.
So Im still sorting out what this means.

So yes my implementation was based on following the Alt Analog path which does shrink the input nids - and is the code that switch_input_src  and cs_8409_init_input_src are based on (see eg check_dyn_adc_switch in hda_generic.c).
(Although I didnt think the input paths need fixing - thats a separate list from the per nid 18 nid dimensioned lists used under the dyn_adc_switch path).

Prior to test_unsol I added code to disable the dyn_adc_switch (ie set it to 0) before calling build_pcms - which should have changed it back to the Alt Analog version - but thats as far as I got - and I think I disabled the setting to 0 while doing the mutex fixup (just in case things might have worked).

Sorry - input_paths is dimensioned 18 - I was thinking about the input_mux list.

dnlbtz, 2026-11-10:
@davidjo, your own comment in `cs_8409_apple_check_dyn_adc_switch` describes what I listed as cause 2: with `dyn_adc_switch` set, the shrink moves the ADCs but `dyn_adc_idx` keeps the old positions. So there are two ways out: clear `dyn_adc_switch` again as you did before `test_unsol`, or keep it and fix the indices. I tried both on the 14,3 (kernel 7.3, PipeWire 1.6) with a four-pole headset in.

With your reset enabled again (the commented block in `cs_8409_apple_parse_auto_config`, nothing else changed on `test_unsol`), `snd_hda_gen_build_pcms` creates a second capture device, `hw:0,2` "Alt Analog", for the second ADC. With the headset in, recording straight from it gets the same signal as device 0 with "Capture Source" on "Mic" on my branch: mono, mostly zero samples, and the headphone signal crossing over into it at the same level (about -125 dBFS at 1 kHz with a -10 dBFS tone playing, whether the mike touched an earcup or not), while the internal mike looks nothing like that. Without the headset it delivers nothing. So device 2 is the CS42L83 input, and your capture prepare, which goes by `hinfo->nid`, sets it up without changes. But PipeWire made no source for device 2, only the usual one on device 0, where the headset mike port stayed "not available" and "Mic Jack" stayed off. So the desktop never offers it. To use it, someone would have to point an application at `hw:0,2` by hand or write their own ALSA/UCM config.

With `dyn_adc_switch` left on there is one capture device and "Capture Source" moves the running stream to the other ADC (`dyn_adc_pcm_resetup`). The generic parser turns it on by itself whenever no single ADC reaches every input (`check_dyn_adc_switch` in generic.c), and on the 8409 the internal mike is only on 0x22 and the headset mike only on 0x1a, so since 5.13 this is the path the kernel expects for this chip. PipeWire then shows "Internal Microphone" and "Microphone" as two ports of the same source and follows the jack. On plain `test_unsol` that path doesn't work yet: with the headset in, a capture on device 0 runs but delivers no frames at all, because the stream is set up on `hinfo->nid`, which is always `adc_nids[0]` = 0x22.

So I'd keep `dyn_adc_switch`. It needs three things, all on my [`headsetMic`](https://github.com/dnlbtz/snd_hda_macbookpro/tree/headsetMic) branch: `dyn_adc_idx` remapped after the shrink, the capture stream set up on `adc_nids[dyn_adc_idx[cur_mux]]` instead of `hinfo->nid`, and the mike pin 0x3c marked dirty so "Mic Jack" follows the headset. With those, the headset mike shows up in PipeWire, "Capture Source" switches, and the internal mike stays usable with a headset in. The macOS layout also applies about 35 dB of gain to the headset mike in software (`HeadsetMic_dBV`), so it is very quiet without that, whichever path is used.

Two corrections to my first comment above. The "Mic Jack" fix (cause 1) and the `!spec->capturing` guards are also in #218 by Francisco-Beltran-Millalen, posted a month earlier. And `headsetMic` is now rebuilt on top of my resume branch (see #198), so the commit hashes I listed above are out of date; the branch link is current. It also has a fix from today: a capture from the headset mike stopped as soon as a sound on the headphones ended, because the headphone cleanup powers the whole CS42L83 down. That cleanup now waits until the capture is over. acmccubbin ran into the same thing on a 13,3 and suggested the same condition in #198.

davidjo, 2026-11-10:
Thanks for the updates.

Yes - if you look at the code in cs_8409_apple_check_dyn_adc_switch I have copied the code from the hda_generic.c for check_dyn_adc_switch as comments - if you follow the dyn_adc_switch path then no shrinking is done - that was done because the original code at 5.4 kernel followed the Alt Analog path.
So yes Ive come to the conclusion that the shrink coding probably should be removed.
The mux_select function shows what needs to be done to setup the mike switch.

I looked at the had_generic.c/hda_codec.c etc code and came away with the following.
Two struct hda_pcm are setup - each struct hda_pcm has 2 pcm streams (struct hda_pcm_stream) a playback stream and a capture stream (this is before dyn_adc_switch is created).
The first struct hda_pcm has the internal speakers/internal mike, the second the headphone speakers, headset mike.
Note that the headset mike nid path exists whether the headset has a mike or not.
(This is dumped in the big data block from the auto config).
As I said I did not clear dyn_adc_switch before - in kernel 5.4 the dyn_adc_switch path was never followed - this did lead to two entries in the ubuntu settings page for sound - 2 entries for speakers and 2 entries for mikes - which you could manually select (this was before I had user sync working) - this of course is pre-Pipewire.

The low volume is expected - see comments in cs_8409_capture_pcm_prepare about 24/16 bit - and there are no amps on the 8409 to fix this - has to be a Pipewire fix.

I have just added some additional log prints to my base resume function - maybe you want to try these out and see what values are being returned when you are doing your hibernation tests etc.
I found this code in patch_realtek.c where the init function changes what is being done based on the S3/S4 test.

The other thing is whether to follow the Dell suspend/resume ideas for the cs42L42 - in particular it uses the reset_gpio pin to hold the CS42L42 in reset during suspend. Im sort of assuming that the CS42L83 probably follows this as its likely a main feature of the CS42 series chips - the main differences I found with the CS42L83 is that the registers dont seem to map to the CS42L42 registers - some have similar content but some have different content.

One more thing - if you look at the Dell suspend/resume setup there are a number of sleeps being done which probably we should be following as well - and may explain some of the strange data seen.

hanscnelson, 2026-11-10:
Tested `headsetMic` at `07f76f7` on a MacBookPro14,2 (subsystem `0x106b3600`), Omarchy, kernel `7.2.15-1-omarchy`, PipeWire 1.6.9. The loaded module srcversion is `591CAE4E1DF062DE3A23F28`.

With a four-pole headset plugged in:

- Headphone Jack and Mic Jack are both on.
- Capture Source is `Mic`.
- The PipeWire source port is `analog-input-mic`. The internal-mic port is unavailable.
- The recording level follows the headset microphone, not the built-in microphone.

The built-in microphone was not retested with the headset unplugged, and a three-pole plug was not tested.

dnlbtz, 2026-11-10:
@davidjo, I tried both.

Your power event lines, on the 14,3 with a debug build of `headsetMic` (which sits on your `e008285`): four runtime resumes with `power_save=1` each logged `cs_8409_apple_resume pm event 0 power is on`, and a hibernation restore logged `pm event 64 restore` (PM_EVENT_RESTORE). I can't test S3 on this machine; the 13,3 and 14,2 people could show what a deep sleep gives, presumably 16 (PM_EVENT_RESUME). So the event does tell a runtime resume from a system one. The GPIO check doesn't need it, but it would be a cheap extra condition if you'd rather only rerun the boot setup after a system sleep.

Removing the shrink: I built `test_unsol` + `headsetMic` with the shrink only on the Alt Analog path (not when `dyn_adc_switch` is set, as the generic `check_dyn_adc_switch` does) and without my `dyn_adc_idx` line. Then `num_adc_nids` stays 18 and `dyn_adc_idx` is right on its own (16 and 8, so 0x22 and 0x1a). But `adc_nids[0]` is now 0x12, which isn't connected to anything, and `snd_hda_gen_build_pcms` sets the capture stream up on `adc_nids[0]`. The result was no capture at all: `arecord` on `hw:0,0` got an Input/output error and PipeWire had nothing to record from. So the shrink is still needed to put a connected ADC first; with it, the only thing to fix is `dyn_adc_idx`, which is the one line in `cs_8409_apple_check_dyn_adc_switch` on my branch (`spec->dyn_adc_idx[i] = nums;` where the paths are moved down).

On the sleeps: for the plug-in click I tried the waits Apple uses in `AppleHDAMikeyInternalCS8409::powerHSBIAS` (40 ms after switching the bias on, 20 ms after clearing HSBIAS_REF). They made no audible difference to the click, which macOS has too.