davidjo/snd_hda_macbookpro #33: Kernel module not loaded:module verification failed: signature and/or required key missing - tainting kernel

chespinoza, 2021-11-02:
Hi guys, I've just tried using your driver but it seems it needs some signature? I'm using Ubuntu Impish.

```
[    3.154641] snd_hda_codec_cirrus: loading out-of-tree module taints kernel.
[    3.154782] snd_hda_codec_cirrus: module verification failed: signature and/or required key missing - tainting kernel

```
Thanks.

davidjo, 2021-11-02:
This is a known feature of newer linux kernels (last 5-10 years)- however the module is loaded and works.
The "taint" kernel seems to mean that eg submitting kernel bugs with a "tainted" kernel are not likely to be looked at by the kernel developers.
There is some process to sign kernel modules but Ive never figured out how to do it with the out of kernel builds.

TabulateJarl8, 2021-11-21:
I seem to have fixed the issue by adding `CONFIG_MODULE_SIG=n` to the top of the Makefile in the `patch_cirrus` folder.

davidjo, 2021-11-21:
Thanks thats very useful to know - kernel development is not something I want to get too deep into - just enough build/update modules for specific tasks I need - will add it - probably around April next year when new Ubuntu LTS should be available.

closed 2021-11-30 by davidjo