davidjo/snd_hda_macbookpro #34: Doesn't work in kernel 5.13 (Ubuntu 21.10) for MBP13,1

yerke, 2021-11-07:
Hello.
Thank you very much for working on this. I just tried it on kernel 5.13 (Ubuntu 21.10) for MBP13,1, and it didn't work unfortunately.
It seems to install correctly, but then I don't see any audio devices after rebooting.
Would it be possible to fix it? Thanks a lot!

davidjo, 2021-11-07:
This unfortunately is a known issue with 5.13 (and Im guessing upwards) (see comments at end of issue #6).
Apparently some Dell laptops have an implementation of 8409/cs42l42 headphone jack similar to Apples which is now apparently a valid kernel module as of 5.13 - so my current guess is this gets loaded and it detects Apples as a Dell laptop but then nothing else is valid so just doesnt do anything. (It doesnt look like the  Dells  use Apples system of driving the amps from the 8409).
Im way behind on kernels - still on 21.04 LTS so 5.4/5.8.
It should still work on 5.12 I think.

Usaamah99, 2021-11-11:
It's working for me on Manjaro Gnome Kernel Version 5.13.19

yerke, 2021-11-12:
@Usaamah99 Interesting. Which device do you have?

Usaamah99, 2021-11-12:
MacBook Pro 14,3

egorenar, 2021-11-16:
I have a working solution for 5.12, 5.14 and 5.15 but only tested on iMac27 5k.
In recent kernel versions Cirrus driver was heavily modified and it needed some rework.

https://github.com/egorenar/snd-hda-codec-cs8409

jim1982ha, 2021-11-21:
Same issue on MacBook Pro 14,1

OS: Ubuntu 21.10 x86_64 
Host: MacBookPro14,1 1.0 
Kernel: 5.13.0-21-generic 
CPU: Intel i5-7360U (4) @ 3.600GHz 
GPU: Intel Iris Plus Graphics 640 

No sound  after applying https://github.com/milan475/snd_hda_macbookpro patch:

$ cat /var/log/kern* | grep -i snd_hda
Nov 16 16:50:57 jm-MacBookPro kernel: [    3.040696] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Nov 16 16:50:57 jm-MacBookPro kernel: [    3.147870] snd_hda_codec_cs8409 hdaudioC0D0: autoconfig for CS8409: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
Nov 16 16:50:57 jm-MacBookPro kernel: [    3.147874] snd_hda_codec_cs8409 hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Nov 16 16:50:57 jm-MacBookPro kernel: [    3.147875] snd_hda_codec_cs8409 hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
Nov 16 16:50:57 jm-MacBookPro kernel: [    3.147877] snd_hda_codec_cs8409 hdaudioC0D0:    mono: mono_out=0x0
Nov 16 16:50:57 jm-MacBookPro kernel: [    3.147878] snd_hda_codec_cs8409 hdaudioC0D0:    inputs:
Nov 16 16:50:57 jm-MacBookPro kernel: [    3.147879] snd_hda_codec_cs8409 hdaudioC0D0:      Internal Mic=0x44
Nov 16 16:50:57 jm-MacBookPro kernel: [    3.147880] snd_hda_codec_cs8409 hdaudioC0D0:      Mic=0x3c

$ sudo  dmesg | grep snd_hda_intel
[    3.222029] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])

No device detected neither in alsamixer, gnome-control-center nor pavucontrol...

[image]

I read about https://github.com/egorenar/snd-hda-codec-cs8409 but don't have the right knowledge to try / assess it. Would that make sense to try on MacBook Pro 14,1 too ?

davidjo, 2021-11-21:
As far as Im aware there is very little difference between 14,1 and 14,3 - think they changed the actual amp chips but otherwise similar - so yes this would make sense

jim1982ha, 2021-11-21:
OK, nice. Next question is... how can I do it? (ie: need the details of command to execute to apply the patch, etc..).
Sorry about that..

davidjo, 2021-11-21:
I cant really help - I dont have a 5.13 system and not going to upgrade yet and its clear something has changed relating to the snd_hda_cirrus module at 5.13 which messes this patch up.

Also check end of issue 6.

jim1982ha, 2021-11-22:
OK, thanks for the feedback (and the work!) 
I will have a look and update here if useful.

mikkelhegn, 2021-12-01:
On a Macbook pro 13,1" from 2015 (A1706) - the driver from @egorenar made the sound work on 15.5: https://github.com/egorenar/snd-hda-codec-cs8409.

This is PopOS

```
> uname -a
Linux my-pop-os 5.15.5-76051505-generic #202111250933~1638201579~21.04~09f1aa7-Ubuntu SMP Tue Nov 30 02: x86_64 x86_64 x86_64 GNU/Linux
```

jim1982ha, 2021-12-02:
I applied the driver using "make && sudo make install":

```
~/snd-hda-codec-cs8409$ make
make -C /lib/modules/5.13.0-21-generic/build CFLAGS_MODULE="-DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/jm/snd-hda-codec-cs8409 modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-21-generic'
  CC [M]  /home/jm/snd-hda-codec-cs8409/patch_cs8409.o
/home/jm/snd-hda-codec-cs8409/patch_cs8409.c: In function ‘cs_8409_init’:
/home/jm/snd-hda-codec-cs8409/patch_cs8409.c:1539:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1539 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
  CC [M]  /home/jm/snd-hda-codec-cs8409/patch_cs8409-tables.o
  LD [M]  /home/jm/snd-hda-codec-cs8409/snd-hda-codec-cs8409.o
  MODPOST /home/jm/snd-hda-codec-cs8409/Module.symvers
  CC [M]  /home/jm/snd-hda-codec-cs8409/snd-hda-codec-cs8409.mod.o
  LD [M]  /home/jm/snd-hda-codec-cs8409/snd-hda-codec-cs8409.ko
  BTF [M] /home/jm/snd-hda-codec-cs8409/snd-hda-codec-cs8409.ko
Skipping BTF generation for /home/jm/snd-hda-codec-cs8409/snd-hda-codec-cs8409.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-21-generic'

~/snd-hda-codec-cs8409$ sudo make install
mkdir -p /lib/modules/5.13.0-21-generic/updates/
cp snd-hda-codec-cs8409.ko /lib/modules/5.13.0-21-generic/updates/
depmod -a
```

But we are not there yet: still nothing in the Settings +

```
$ cat /var/log/kern* | grep -i snd_hda
grep: (standard input): binary file matches
$ sudo dmesg | grep snd_hda_intel
[sudo] password for jm: 
[    3.190625] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
```

mikkelhegn, 2021-12-02:
@jim1982ha - for what it's worth, I installed the snd_hda_macbookpro driver first. I have very little insight as to how these work, but it did the trick for me.

jim1982ha, 2021-12-02:
I managed to make it work!
Thank you for your precious feedbacks!

To help others who may face same case than me:
What i did is "simply" changed any references to "MacBookPro 14,3" by "MacBookPro 14,1" in @egorenar gitbub repo downloaded files:
1. Modify all "MacBookPro 14,3" by "MacBookPro 14,1" 
2. Modify all "MBP143" by "MBP141" 

after that, did another "sudo make && sudo make install" on the modified files + restarted, and VOILA!

```
$ make
make -C /lib/modules/5.13.0-22-generic/build CFLAGS_MODULE="-DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/jm/snd-hda-codec-cs8409-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-22-generic'
  CC [M]  /home/jm/snd-hda-codec-cs8409-master/patch_cs8409.o
/home/jm/snd-hda-codec-cs8409-master/patch_cs8409.c: In function ‘cs_8409_init’:
/home/jm/snd-hda-codec-cs8409-master/patch_cs8409.c:1539:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1539 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
  CC [M]  /home/jm/snd-hda-codec-cs8409-master/patch_cs8409-tables.o
  LD [M]  /home/jm/snd-hda-codec-cs8409-master/snd-hda-codec-cs8409.o
  MODPOST /home/jm/snd-hda-codec-cs8409-master/Module.symvers
  CC [M]  /home/jm/snd-hda-codec-cs8409-master/snd-hda-codec-cs8409.mod.o
  LD [M]  /home/jm/snd-hda-codec-cs8409-master/snd-hda-codec-cs8409.ko
  BTF [M] /home/jm/snd-hda-codec-cs8409-master/snd-hda-codec-cs8409.ko
Skipping BTF generation for /home/jm/snd-hda-codec-cs8409-master/snd-hda-codec-cs8409.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-22-generic'

~/snd-hda-codec-cs8409-master$ sudo make
[sudo] password for jm: 
make -C /lib/modules/5.13.0-22-generic/build CFLAGS_MODULE="-DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/jm/snd-hda-codec-cs8409-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-22-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-22-generic'

~/snd-hda-codec-cs8409-master$ sudo make install 
mkdir -p /lib/modules/5.13.0-22-generic/updates/
cp snd-hda-codec-cs8409.ko /lib/modules/5.13.0-22-generic/updates/
depmod -a

~/snd-hda-codec-cs8409-master$ sudo reboot
```

Sound card is detected now and I can hear sound coming from "Speakers - Build-in Audio" source:

[image]
 
Though, from the tests i just did now:  I noticed that sounds volume is quite low.
will check this, if anything can be improved?

mikkelhegn, 2021-12-02:
Interestingly, as mine identify as MacBookPro14,2, and don't see any references to that in the code.

```
> sudo dmidecode -s system-product-name
MacBookPro14,2
```

egorenar, 2021-12-02:
> @jim1982ha - for what it's worth, I installed the snd_hda_macbookpro driver first. I have very little insight as to how these work, but it did the trick for me.
> 
> @egorenar - can you help clarify the steps to get your patched version installed? Also be aware that the headphones might not work, I plugged mine in and 5 seconds later I had a very painful high pitch tone, so had to rip them out immediately! :-)

Sure.
First, clone the repository and then depending on your target kernel version, switch to the right branch.
Currently, my repo supports kernels 5.12, 5.14 and 5.15. 5.13 might work as well but didn't test. I'm using 5.15.6 at the moment.

After choosing the right branch, just issue `make` and then as root or with sudo `make install`.

Then i usually remove all drivers with
```
rmmod snd-hda-intel snd-hda-codec-cirrus snd-hda-codec-cs8409 snd-hda-codec-generic
```
And reload it again with
```
modprobe snd-hda-intel
```

I assume that you installed linux kernel headers. This depends on your distro. I'm using Void Linux.

Regards

sebastianhaberey, 2022-04-15:
@egorenar I followed your [instructions](https://github.com/davidjo/snd_hda_macbookpro/issues/34#issuecomment-984577563) on my MacBook Pro 14,3 Ubuntu 21.10 (5.13.0.22-generic) and the "Built-In Audio" works now. Great work, thanks!

Just as a side note: the balance doesn't seem to function (the slider has no effect) but stereo is fine and I never use balance anyways.

sebastianhaberey, 2022-04-23:
Sadly, after the update to _5.13.0.39-generic_ this has stopped working. The device "Speakers - Built-in Audio" is still available in the Ubuntu system settings, but there is no sound.

davidjo, 2022-05-10:
Try the new version - closing old issues.

closed 2022-05-10 by davidjo