davidjo/snd_hda_macbookpro #35: not working anymore on kernel 5.15+

slynobody, 2021-11-22:
tested: debian 5.15 trunk, 5.15.0-1, xanmod 5.15.4
no sound at all (mbp 2016)

the chip should be included in 5.15 though out of the box: https://git.kernel.org/pub/scm/linux/kernel/git/tiwai/sound.git/log/?h=for-next&qt=grep&q=cs8409

how to activate it in 5.15+?

davidjo, 2021-11-22:
Im afraid this is the 5.13 issue (see other issues end #34 and end #6).

Unfortunately the solution added in 5.13 (and Im assuming in all later kernels such as 5.15)  is specific to certain Dell laptops and has nothing to do with Apples chips - its not even clear the 8409 chips are the same as Apples is likely a custom version of the 8409. But the chip IDs are the same so the new kernel module takes over yet is totally incapable of handling Apples chips.

slynobody, 2021-11-22:
thank you for your answer!

as far i understood the links you gave, https://github.com/egorenar/snd-hda-codec-cs8409 got the chip to work under 5.15 (imac) - and you got the according speaker-driver (mbp-only, or the same?)?
how could someone join the two approaches with, say, 5.15.0-1 on debian (or xanmod, i darkly remember your patch worked in 5.14)? how could it be possible to 'update' your installer with the egorenar-code for 5.15+?

davidjo, 2021-11-22:
I am not running 5.13 kernel yet so cant answer any specific questions.
What little Ive looked at is that egorenar is combining my updates with the 5.13 cs8409 kernel updates - so just try using that - just replace the contents of my patch_cirrus folder from a clone with the egorenar files

slynobody, 2021-11-23:
Thank you very much for your kind help.

The method to integrate the newer egorenar patch works - even for a xanmod-kernel!

But it seems to work only for 5.13 and 5.14 - and not 5.15, as well as for 5.16 (which seems at least to also have a slightly different directory-structure making script-changes nec {got at least the script working with some minor changes, esp. some changed directories and including a pahole-flags.sh from another archive while it is not included in the gz anymore}

hopefully someone will merge egorenar and your effort scriptwise as well as porting the solution afterwards also to 5.15+ (or finding a way to make the new kernel-codes in 5.13 and 5.15 for the chip useable under real-world conditions)!

chadady, 2022-02-20:
Thanks for your work on this.

I got Ubuntu 5.13.0-30 to work by blacklisting snd_hda_codec_cs8409

Add 'blacklist snd_hda_codec_cs8409'  to /etc/modprobe.d/blacklist.conf

davidjo, 2022-02-21:
Nice idea - hadnt thought about that.

babstar99, 2022-03-06:
> Thanks for your work on this.
> 
> I got Ubuntu 5.13.0-30 to work by blacklisting snd_hda_codec_cs8409
> 
> Add 'blacklist snd_hda_codec_cs8409' to /etc/modprobe.d/blacklist.conf

Confirmed, also working on MBP 2017,  5.13.0-27-generic Linux Mint 20.3

davidjo, 2022-05-10:
Use new version now

closed 2022-05-10 by davidjo