davidjo/snd_hda_macbookpro #37: "linux kernel headers not found in /usr/src"

Cyanide-07, 2021-12-17:
when running "sudo ./install.cirrus.driver.sh" i get the following messages:
linux kernel headers not found in /usr/src:
Debian (eg Ubuntu): /usr/src/linux-headers-5.15.8-arch1-1
Fedora: /usr/src/kernels/5.15.8-arch1-1
assuming the linux kernel headers package is not installed
please install the appropriate linux kernel headers package:
sudo apt install linux-headers-1
--
which is kind of annoying cuz i've definitely installed linux-headers
any help would be appreciated

davidjo, 2021-12-17:
This is likely a distribution issue - which distribution are you using?
Each distribution seems to have its own idea where to place headers.
However if you have installed the headers just edit install.cirrus.driver.sh and remove the exit 1 after the above messages.
The resulting build commands use the kernel makefiles which should know where the headers are.

Cyanide-07, 2021-12-18:
distro:arch
removing the exit 1 seems to have worked and everything downloaded now it recognizes my speakers and stuff which is huge progress, however audio seems to not be functioning trying to troubleshoot myself but so far nothing

Cyanide-07, 2021-12-19:
fixed myself now works brilliantly thanks a ton this has saved me so much headache and hastle :))

closed 2021-12-19 by Cyanide-07

Cyanide-07, 2022-04-30:
> Yes, this is an issue on Arch. Considering Arch and it's derivatives are rolling releases that use the newest kernel, [you're better off using this fork that works on that latest kernel anyways](https://github.com/egorenar/snd-hda-codec-cs8409)

Hey I saw this a few days ago and funnily enough I've just done like a full reinstall on my laptop due to some issues I was having and I was just wondering if you genuinely got the fork to work for you as I'm having a few issues.

Cyanide-07, 2022-04-30:
idrk if ur gonna read that one anyways but if u do nvm thnx for sending that works like a charm for some reason it wasn't working when I first installed it.

anyways cheers for that one mate saved me a lot of headache :)).

davidjo, 2022-05-10:
Try the new version - closing old issues.