davidjo/snd_hda_macbookpro #38: Zorin OS (aka Ubuntu) Macbook Pro 14,1 no touchbar has no audio

EmeraldLoc, 2022-02-03:
I am using Zorin OS with Gnome on a macbook pro 14,1 no touchpad. if i have sound play, the bar moves up like audio is playing. The sound settings shows three audio devices, Speakers, Headphones, and HDMI / Display Port built in audio. I have installed the drivers with the package, it worked in mint, however it does not seem to be working with zorin, and arch. just to make it clear, audio works in both Windows and Macosx. Thanks in advance.

davidjo, 2022-02-03:
First qustion is what kernel versions are being used?
There are a number of issues with kernels depending on the kernel version.

EmeraldLoc, 2022-02-03:
5.13.0-28-generic

EmeraldLoc, 2022-02-04:
I have just tried booting into kernel 5.11.0-38, and it is the same.

davidjo, 2022-02-04:
As you seem to have discovered - kernels 5.13 and above dont work without fixups (this will be fixed after new Ubuntu LTS is released soon).

you did rebuild the module for 5.11? - it needs rebuilding for each kernel version
check if the module is loaded and which version is loaded

can you compare boot kernel logs from  mint and the other ssystems - looking for snd_hda_intel and cirrus?

other issue is how Zorin is assigning the audio path - does it use pulse audio?

can you get sound if you use aplay and the plughw: device from aplay -L?

can you get sound if you use aplay and the hardware device (the hw:... from aplay -L)
BE VERY CAREFULL - TRY AND USE A VOLUME REDUCED SOUND WAVE - THERE IS NO VOLUME CONTROL - IT WILL BE LOUD!!

EmeraldLoc, 2022-02-04:
so for your kernel questions, no I didn't rebuild anything. I booted into grub and picked the older firmware. I did uname -r to check the kernal version, and it said 5.11. 

It will take me a while to do the kernel log test. 

Zorin's audio path is, I dont know. Currently pipewire and pulseaudio are a procces when i tab in killall, how can i check which one is being used. I can get rid of one of them.

There was no sound

I had 2 devices: CARD=PCH,DEV=0 and CARD=PCH,DEV=3. When i tried to do DEV=0 it said it was busy, and when I did DEV=3 no sound came out. DEV=3 is HDA Intel PCH, Generic Digital and Dev=0 is HDA Intel PCH, Generic Digital

davidjo, 2022-02-04:
You need to rebuild the kernel module for each kernel version ie you need to re-run install.cirrus.driver.sh for each kernel version (probably a good idea to do this in a new directory).
Check the system log after boot and make sure there is something like cs_8409_init in the log.
If there are no messages with cs_8409 in the log then the module is not loaded.

Yes Ive had great difficulty figuring out exactly how the kernel audio devices map into eg pulse audio (never heard of pipewire).

If the devices are coming from aplay -L then I think the module isnt loaded - you should get one which is Generic Analog (at least thats what get on my 14,3 machine).

Have you run the alsa-info.sh  script (its available on the web) - its a script from the alsa developers to give lots of details about the audio hardware.

At the moment I would say the module is loaded in 5.13 but it doesnt work (as expected) and is not loaded in 5.11 so you also wont get audio output as the original kernels module cannot handle MacBook Pros.

Generic Digital tends to mean its the device for HMDI audio.

EmeraldLoc, 2022-02-04:
[image]
This is what I see from aplay. I am having to do something right now, I will get the other data in about 10 min maybe

EmeraldLoc, 2022-02-04:
I ran alsa info, the audio drivers for snd-hda-macbookpro were not loaded. snd-hda-intel was loaded. Should i reinstall the drivers?

davidjo, 2022-02-04:
So DEV=0 is the analog device, DEV=3 the HDMI device.
the module snd_hda_intel is always loaded - its snd_hda_codec_cirrus you need to look for - there is no snd_hda_macbookpro module - snd-hda-macbookpro is just the overall package name.

I cant remember about Cirrus Logic Generic now - think its still in the code - again need to look for cs_8409 in the system log.

EmeraldLoc, 2022-02-05:
cs_8409 does not exist in kernel log kern.log. snd_hda_codec_cirrus does exist in alsa info script.

davidjo, 2022-02-05:
So Im pretty certain this means its not loading the module - which is snd-hda-codec-cirrus.ko - so not suprising theres no audio. (It still exists as snd_hda_macbookpro provides an updated version of the original kernel snd-hda-codec-cirrus.ko module).

Check under the build directory of where you cloned snd_hda_macbookpro and see if snd-hda-codec-cirrus.ko exists.
After building the install.cirrus.driver.sh script copies this to the actual kernel module folder - which for ubuntu is stored under /lib/modules/kernel-version/ directory in the updates sub-directory - where kernel version is whatever kernel version you are booting.

Ensure that under the build directory in the cloning directory there is an hda-5.11.0 directory - otherwise its not built the module for the 5.11 kernel version - the actual build files are under this hda-kernel-version directory and is where snd-hda-codec-cirrus.ko should initially exist.

EmeraldLoc, 2022-02-05:
First Thing. snd-hda-codec-cirrus does exist in the hda-5.11.0 build directory under build under snd_hda_macbookpro which was cloned to Home. I cannot, however, find it in the kernel. Second Thing. When I installed it, i remember it downloading linux-5.11 which is my kernel version, and that file exists in the build directory. Am I supposed to clone this file elsewhere?

davidjo, 2022-02-05:
If you look at the end of the install.cirrus.driver.sh theres a cd and 2 make commands.
the cd goes into the build/hda-5.11.0 directory, the 1st make compiles and creates the module and the 2nd make install is what copies the created module .ko file to the primary kernel updates directory.
This command needs to be run as root.
if you say that snd-hda-codec-cirrus.ko  exists in build/hda-5.11.0 then cding into build/hda-5.11.0
and running sudo make install should do the copy - it should print that it is doing the copy.
You need to be booted into the 5.11.0 kernel when you do this (as it picks up from the running kernel where to put the module).
Check again if the module is under the correct /lib/modules...
Reboot again after doing this before testing the sound.

EmeraldLoc, 2022-02-06:
I found the module that the install file said it was in, and it was there in the kernel I was booted into. In the update folder. However, audio still is not working. Do I have to rebuild mkincipo in order for the kernel to read it?

davidjo, 2022-02-06:
I dont think so (but mkinccpio seems to be arch specific so dont have it).

under the clone snd_hda_macboopro you should see a directory tests - there are 2 test waves and a README.

try aplay -D plughw StereoTest32_reduced_m24dB.wav

this should bypass pulseaudio and play directly through alsa

As it happens just noticed my linux boot on my macbookpro 14,3 is using 5.11 because I was trying something else.
So I cloned my snd_hda_macbookpro repo, built, installed, rebooted and did the above command and got sound (Ubuntu).

if you are getting lines with cs_8409 in the syslog file and the above works then the issue (as usual) is a user level issue of how the raw audio devices are mapped to the user settings.

EmeraldLoc, 2022-02-06:
cs_8409 does not exist in syslog (Im using nano btw) The aplay command did not work, so it seems that is due to the kernel module not loading.

davidjo, 2022-02-06:
Well next thing is to get dangerous and move the original snd-hda-codec-cirrus.ko out of /lib/modules/kernel-version/sound/pci/hda/ to somewhere else (dont delete) - you need to do this as root.
Check with lsmod if snd_hda_codec_cirrus is loaded in both cases (yes the module name is snd-hda-codec-cirrus.ko but the module symbol name in lsmod is snd_hda_codec_cirrus.

EmeraldLoc, 2022-02-06:
That directory does not exist, the snd-hda-codec-cirrus.ko is in the updates folder. Does this matter?

davidjo, 2022-02-07:
sorry it should be /lib/modules/kernel-version/kernel/sound/pci/hda/

what about the lsmod?

also check the system log for a line which contains alsactl and Found hardware - what does that contain?

EmeraldLoc, 2022-02-07:
snd_hda_codec_cirrus does exist when running command lsmod.

alsactl does not exist in the syslog (I searched for it using nano with CNTRL+W) found exists because of my appimage launcher that I have. Perhaps this is the issue.

davidjo, 2022-02-07:
yes it looks as tho Zorin may have a different audio stack above the kernel driver.
(Ive found this a big problem with the various linux distributions that higher level audio functions seem to be handled quite differently).

lets try and confirm which snd_hda_codec_cirrus module is loaded.

look for lines similar to the following (this is from my 14,3 machine with a 5.11 kernel):

Feb  6 08:09:34 david-MacBookPro kernel: [    3.189151] snd_hda_codec_cirrus: loading out-of-tree module taints kernel.
Feb  6 08:09:34 david-MacBookPro kernel: [    3.189246] snd_hda_codec_cirrus: module verification failed: signature and/or required key missing - tainting kernel
Feb  6 08:09:34 david-MacBookPro kernel: [    3.192076] snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS8409: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
Feb  6 08:09:34 david-MacBookPro kernel: [    3.192080] snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Feb  6 08:09:34 david-MacBookPro kernel: [    3.192083] snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
Feb  6 08:09:34 david-MacBookPro kernel: [    3.192085] snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
Feb  6 08:09:34 david-MacBookPro kernel: [    3.192086] snd_hda_codec_cirrus hdaudioC0D0:    inputs:
Feb  6 08:09:34 david-MacBookPro kernel: [    3.192088] snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x44
Feb  6 08:09:34 david-MacBookPro kernel: [    3.192090] snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x3c

in particular check for out-of-tree module and taints kernel
you should also find the CS8409 line

EmeraldLoc, 2022-02-07:
Hmmmm. I seem to only have the first two in syslog.

davidjo, 2022-02-08:
So this may be another Zorin configuration difference - the above messages (except the taint lines) come from dev_info kernel log calls - now linux can map the various types of kernel log msgs to different files - maybe theres another log file in /var/log - also see what the output of dmesg is right after booting (dmesg shows the kernel log buffer and I think contains every msg).

Another thing to try is to edit the Makefile in the build/hda-... directory and at the top you can see a debug build comment and a line under it which is commented - comment the line below normal build and uncomment this line.

make clean

make 

sudo make install

This will replace the module with a debug build which does LOTS of logging - you probably want to either save the previous non-debug module from /lib/modules 1st or undo the above edits and rebuild and install the module once things have been sorted.

EmeraldLoc, 2022-02-08:
make[2]: *** [scripts/Makefile.build:288: /home/calebelm/snd_hda_macbookpro/build/hda-5.11.0/patch_cirrus.o] Error 1
make[1]: *** [Makefile:1849: /home/calebelm/snd_hda_macbookpro/build/hda-5.11.0] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-38-generic'
make: *** [Makefile:17: all] Error 2

This is what shows up when i try to change Makefile and run commands you showed me in snd_hda_macbookpro/build/hda

EmeraldLoc, 2022-02-08:
I've decided I am going to move over to Ubuntu 20.04 LTS. For anyone who was having the same problem, try to follow all the steps davidjo gave me. Thank you for your hard work!

closed 2022-02-08 by EmeraldLoc