davidjo/snd_hda_macbookpro #40: Problems with Imac 18.3 2017 27" 5K

Schmeling-M, 2022-02-14:
I have tried installing this patch on the above running Linux mint, but with no success.

As far as I can see, the Cirrus soundchip is the same, but the amp chip maby not so.

Does anyone know whether this patch can work for an Imac 18.3 2017 27" 5K  or if there are other ways of getting working audio on this with linux?

davidjo, 2022-02-14:
You need the egorenar version of this repo - this includes the fixups for imac 5Ks plus newer kernels.

Schmeling-M, 2022-02-15:
Thank you,  I'll see if I can get that to work :-)

closed 2022-02-15 by Schmeling-M

davidjo, 2022-02-17:
Did you check out the imac branch?
(I didnt realise that you need to do this for the egorenar repo before running the install script - the base branch is just my version).

(snd-hda-codec-cs8409 is just the patch_cirrus folder component so has to be copied into an existing clone).

Schmeling-M, 2022-02-17:
Actually I did, but when I do this:

`git clone https://github.com/egorenar/snd_hda_macbookpro.git
cd snd_hda_macbookpro/
#run the following command as root or with sudo
./install.cirrus.driver.sh
reboot`

I'm not really sure which I'm getting. How do I make sure, I get the right one?

davidjo, 2022-02-17:
That gives you the default wrong one.
You need:

git clone https://github.com/egorenar/snd_hda_macbookpro.git
cd snd_hda_macbookpro/
git checkout -b imac27-5k origin/imac27-5k
sudo ./install.cirrus.driver.sh

Schmeling-M, 2022-02-22:
After a complete reinstall of Linux Mint, and the installation of the driver, I now have sound, but there are several issues:

Issue 1:
Sound is speaker only. Plugging in headphones does nothing. No switching and never any sound from the headphones.

Issue 2: 
Channels are messed up. The Imac speakers, as I can see from photos at Ifixit, have 4 wires, and two speakers (trebble and bass) for each speaker. Using a sound pressure meter, I can measure which speakers correspond to the audiochannels. 

For the speakers there are 3 configurations:
2CH analogue:
(Linux audio channel - Speaker)
L - Right Trebble
R - Right Bass

2.1CH Surround
(Linux audio channel - Speaker)
L - Left Trebble
R - Left Bass
Sub - Right Bass and perhaps trebble

4.0CH Surround
(Linux audio channel - Speaker)
FrontL - Left Trebblele
FrontR - Left Bass
RearL - Right Trebble
RearR - Right Bass

For the headphone configuration there is only one configuration:
2CH analogue:
(Linux audio channel - Speaker)
L - Right Trebble
R - Left or Right or perhaps both Bass

Both issues are kind of critical to me, but I'm guessing that the configuration is the most important.

Is there any way to remap the channels? Perhaps in the drivercode?

Are there any fixes for the headphone issue?

davidjo, 2022-02-22:
The headphone is a known imac issue - there is some difference between the imac devices and macbook pro devices which means the driver code doesnt seem to work for imacs. To fix this would need someone with an imac (I dont have one) to load my logging OSX kernel module to try and determine the difference (remember there is no documentation for any of this and the chips are special Apple versions).

The channel problem is the egorenar version does not seem to have my patch_channel_map.diff.txt update which seems to be the best fixup. How to apply this fixup has been recently discussed at the end of issue #23.

Channels can be remapped either in user pulse audio configs or user alsa configs - the other issues document several attempts at this.

The correct audio output device for the apple speaker setup is only analog-stereo ie 2CH analogue.

I will be updating my repo with all these fixes after the new Ubuntu LTS comes out (April) and we know which kernel is being used - because of the many incompatible kernel audio source changes with minor kernel version for major kernel version 5.

Schmeling-M, 2022-02-25:
Does this mean, that you will make an updated driver, that will work for Retina Imacs, like the egorenar version but with the correct channel mapping, but you need someone with an Imac to figure out the headphone jack issue?

If this is the case, I think I can help. I have an Imac, with Linux Mint, but I can wipe and install OSX and run your module on it. I need an Imac running Linux but with sound, and right now I have no way of getting there, so where do we start? :-)

davidjo, 2022-02-25:
Yes I will be  merging all these fixups into a new version.

To get the headphone jack to work requires a log - but you could use usb or bluetooth headphones - those drivers exist and work.

Schmeling-M, 2022-02-26:
Certainly yes, but I really like my BeyerDynamic T5p's :-D and they are analogue. Besides right now, the machine is not a production machine. So it  is only a matter of installing OSX, and running your tool, and I think this could help both me and many others.

I sounds awesome with the merging of the fixes. There must be many people waiting to get their Imac runing with linux + sound ;-)

If you can give me the instructions on what to do, to get the info you need, then I'll see, the I'll see if I can get this info before I start installing on it.

davidjo, 2022-02-26:
I dont suggest you do this with your good headphones - any cheap headset - but have you tried playing music (speakers - 2ch audio) then plugging in headphones - dont have the headphones in just close to ear (I got some very loud noises when doing the initial work) - others have said they got some sound from headphones with imacs but the noise was extremely distorted/feedback type and loud.

First issue is can you install High Sierra on your imac - Im not sure what the timeline is for imac models and lowest working OSX version.

Im still on HIgh Sierra because that was last one bootable on HFS as I understand - the later versions switch to APFS - which Im waiting for a reasonable linux APFS module for reading - there seems to be one now so might upgrade my OSX - but not to latest version.

If necessary Ill boot another version on a usb disk - but try and determine the lowest version OSX you can boot.

The process is non trivial - you need to replace a kernel module (this is all OSX) in the AppleHDA.kext (/Systems/Library/Extensions) with my logging version - currently only for High Sierra.
This requires turning off all Apple protections Gatekeeper and SIP in recovery boot mode plus unsigning the AppleHDA.kext - github has a program source to do this - unsign (this also needs building).
The replacing just involves swapping the kernel module component with my version ie moving out the old version and copying in my version.

I can send you the binary kernel module if you are OK with it - or Ill send the source and you can build it yourself - you will need the Apple developer tools to do this.

Schmeling-M, 2022-02-27:
Thanks for your advice regarding headphones ;-) will follow.

After a really frustrating and annoying process, I have High Sierra running on my Imac. 

On a newer mac, I couldn't download HS from the appstore, and when entering recovery, only option was Monteray. All kinds of attempts of creating a HS installer on Windows 10 failed. Only reason it succeed was, that I had a Big Sur installer lying around. I managed to install from that, where-after I was able to download High Sierra, and create an installmedia. If I didn't know better, I would say that Apple was actively trying to prevent me from installing High Sierra.

Anyway. I'm now running High Sierra.

Just to make things as easy for my self as possible, I'll go with the binary. I have no idea on how to turn the protections. I can google my way of course, but if you have any instructions, they will be most welcome :-)

davidjo, 2022-02-27:
Thats good - I suspected it should work as my Macbook Pro is 2017 vintage.

Yes Apple seems to be trying to out Microsoft in making a totally locked system - which is why Im wary of any upgrades.

I suggest you look thro the imac 19,1 issue which is where I did the same process to get the initial imac updates working.

From that I got a base log of booting an imac which is what the current imac driver is based off.
We are going to need the process of booting OSX on the imac, start sound playing then plugging in headphones then unplug while logging - because thats whats missing currently.

Ill collect things and sort out the todo list.

Schmeling-M, 2022-02-28:
I'll familiarize myself with that issue. As I can see, email may needed. Mine is max at and then my last name from my username (username without "-M") and then dot dk

davidjo, 2022-02-28:
Thats good - so I hope I should have emailed you what I sent to doctordbx for logging.

phileagleson, 2022-04-18:
Hi David, First of all, thank you so much for your work on this. I wanted to see if you ever got any information for the headphone jack? I have an iMac 19,1 I've been trying to get audio working on. I've had luck using egoranar's fork to get the speakers working but no luck with headphones. I'd be happy to get any required logs from macOS  if needed but the oldest MacOS my machine will run appears to be Mojave. Any chance your logger would work on that version?

davidjo, 2022-04-20:
I think we have a fix for the headphone jack (apparently imacs use an inverted jack detection signal cf macbook pros).
Im working on major update for the new Ubuntu 22.04 LTS release (which will be based on the new separate CS8409 linux kernel module) which Im looking to make available in the next week or so.
(Unfortunately the logger will only work on High Sierra - would have to  re-assemble the driver from Mojave).