davidjo/snd_hda_macbookpro #44: Doesn't appear to be working on kernel 5.16.19

Booligoosh, 2022-04-21:
I upgraded to kernel 5.16.19 from (I think) 5.16.15, and this no longer appears to work. Sound does sometimes work if I plug headphones in via the USB port, but even that seems on-and-off.

I'm using Pop!_OS 22.04 beta which uses Pipewire, but prior to the kernel update, it was working fine.

If it's broken for anyone else, that would be helpful to know if it's just an issue with my system or an incompatibility with the kernel itself.

davidjo, 2022-04-21:
What kernel was it working on - is it this version or egorenar  - this version does not work on kernels > 5.13.
Working on getting it work on 5.15 for Ubuntu 22.04 LTS.

EruEri, 2022-04-21:
I manage to get it work on kernel 5.16.20 on Fedora
And also huge thank you @davidjo  for the driver

Booligoosh, 2022-04-22:
Ah, my bad @davidjo, I was using leifliddy's driver on 5.16.15 before, and then switched over to your one a few days before updating the kernel as I'd seen you planned to work in those new audio quality patches. I thought I remembered your one working on 5.16.15 too, but I might be remembering wrong.

Awesome to hear you're working on a version for 22.04, many thanks for creating the driver!

closed 2022-04-22 by Booligoosh