davidjo/snd_hda_macbookpro #45: 22.04 ubuntu - steps not working?

otuk, 2022-04-28:
I was using this fix in my 20.04 ubuntu mbp, I just installed 22.04 on the same mbp and the same steps did not work.
I noticed some 404s, in compilation I only see warnings pls help.
Note I have sound from Bluetooth/but nothing from wired headphones or speakers.
Thanks

$ sudo ./install.cirrus.driver.sh 
--2022-04-28 17:25:09--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.15.0.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.193.176, 151.101.129.176, 151.101.1.176, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.193.176|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-04-28 17:25:09 ERROR 404: Not Found.

--2022-04-28 17:25:09--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.15.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.193.176, 151.101.1.176, 151.101.65.176, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.193.176|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

linux-5.15/sound/pci/hda/
linux-5.15/sound/pci/hda/Kconfig
linux-5.15/sound/pci/hda/Makefile
linux-5.15/sound/pci/hda/ca0132_regs.h
linux-5.15/sound/pci/hda/hda_auto_parser.c
linux-5.15/sound/pci/hda/hda_auto_parser.h
linux-5.15/sound/pci/hda/hda_beep.c
linux-5.15/sound/pci/hda/hda_beep.h
linux-5.15/sound/pci/hda/hda_bind.c
linux-5.15/sound/pci/hda/hda_codec.c
linux-5.15/sound/pci/hda/hda_controller.c
linux-5.15/sound/pci/hda/hda_controller.h
linux-5.15/sound/pci/hda/hda_controller_trace.h
linux-5.15/sound/pci/hda/hda_eld.c
linux-5.15/sound/pci/hda/hda_generic.c
linux-5.15/sound/pci/hda/hda_generic.h
linux-5.15/sound/pci/hda/hda_hwdep.c
linux-5.15/sound/pci/hda/hda_intel.c
linux-5.15/sound/pci/hda/hda_intel.h
linux-5.15/sound/pci/hda/hda_intel_trace.h
linux-5.15/sound/pci/hda/hda_jack.c
linux-5.15/sound/pci/hda/hda_jack.h
linux-5.15/sound/pci/hda/hda_local.h
linux-5.15/sound/pci/hda/hda_proc.c
linux-5.15/sound/pci/hda/hda_sysfs.c
linux-5.15/sound/pci/hda/hda_tegra.c
linux-5.15/sound/pci/hda/hp_x360_helper.c
linux-5.15/sound/pci/hda/ideapad_s740_helper.c
linux-5.15/sound/pci/hda/patch_analog.c
linux-5.15/sound/pci/hda/patch_ca0110.c
linux-5.15/sound/pci/hda/patch_ca0132.c
linux-5.15/sound/pci/hda/patch_cirrus.c
linux-5.15/sound/pci/hda/patch_cmedia.c
linux-5.15/sound/pci/hda/patch_conexant.c
linux-5.15/sound/pci/hda/patch_cs8409-tables.c
linux-5.15/sound/pci/hda/patch_cs8409.c
linux-5.15/sound/pci/hda/patch_cs8409.h
linux-5.15/sound/pci/hda/patch_hdmi.c
linux-5.15/sound/pci/hda/patch_realtek.c
linux-5.15/sound/pci/hda/patch_si3054.c
linux-5.15/sound/pci/hda/patch_sigmatel.c
linux-5.15/sound/pci/hda/patch_via.c
linux-5.15/sound/pci/hda/thinkpad_helper.c
make -C /lib/modules/5.15.0-27-generic/build CFLAGS_MODULE="-DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/otuk/snd_hda_macbookpro/build/hda-5.15.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-27-generic'
  CC [M]  /home/otuk/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus.o
/home/otuk/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus.c: In function ‘cs_8409_init’:
/home/otuk/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus.c:1610:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1610 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
  LD [M]  /home/otuk/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cirrus.o
  MODPOST /home/otuk/snd_hda_macbookpro/build/hda-5.15.0/Module.symvers
  CC [M]  /home/otuk/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cirrus.mod.o
  LD [M]  /home/otuk/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cirrus.ko
  BTF [M] /home/otuk/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cirrus.ko
Skipping BTF generation for /home/otuk/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cirrus.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-27-generic'
cp snd-hda-codec-cirrus.ko /lib/modules/5.15.0-27-generic/updates
depmod -a

contents of /lib/modules/5.15.0-27-generic/updates
total 1772
-rw-r--r-- 1 root root 1812384 Apr 28 17:25 snd-hda-codec-cirrus.ko

otuk, 2022-04-28:
Also: The sound card  is  CS8409 from Cirrus Logic

davidjo, 2022-04-29:
This is expected - this module is not yet fixed to work with the 22.04 LTS kernel (5.15) which has major changes.

I am working on a new version for 22.04 which should be available in next week or so.

otuk, 2022-04-29:
@davidjo Thank you for the quick response.  I tried what you suggested above as temporary fix, I rebooted and yes it does work!  Thank you so much.  Will be waiting to install the newer version when  is available, let me know if you need a tester for it, I can help that way.

ozhankaraman, 2022-05-02:
Ubuntu 22.04 support will be great, thanks for all your contribution (+1)

Alimjoo, 2022-05-03:
I just Installed Ubuntu 22.04 on my Macbook Pro 14,1, 
**Add 'blacklist snd_hda_codec_cs8409' to /etc/modprobe.d/blacklist.conf**
This works for me. Thank you.

davidjo, 2022-05-03:
Have to say this blacklist idea was originally suggested by [chadady] in #35

Booligoosh, 2022-05-10:
Can confirm that the new version works on 22.04 for me - Thanks so much for the hard work @davidjo! :tada:

davidjo, 2022-05-10:
Thanks for update - closing old issues.

closed 2022-05-10 by davidjo