davidjo/snd_hda_macbookpro #46: MacBook Pro 11,1 - Ubuntu 20.04 - cs4208

kash-raman, 2022-05-04:
Hi,
Thank you for putting this repo together. 
I'm on a macbook pro 11,1 and having issue with mic 
cat /proc/asound/card*/codec* | grep Codec
Codec: Intel Haswell HDMI
Codec: Cirrus Logic CS4208 

followed the instruction here
https://askubuntu.com/questions/984239/no-microphone-picked-up-on-ubuntu-16-04-on-macbook-pro and added

options snd-hda-intel model=mbp11

Unable to get the mic to work. Any recommendations?

davidjo, 2022-05-05:
Sorry dont think I can help here.
This repo is specifically for 8409 chips - which it turns out are very unique as an HDA device.
The sort of good news is that I believe the CS4208 is a much more standard HDA chip so should be able to make it work.
Looking at the patch_cirrus.c file for cs4208_mac_fixup_tbl your model is not handled - which is why setting model=mbp11 is needed - and the fact it doesnt seem to work says there is some difference between the 11,1 models and the 11,2 models.
The hex digits for the SND_PCI_QUIRK are from the Subsystem ID from  /proc/asound/card*/codec#0 listing. 

All I can suggest is to google and use the linux tools for analyzing HDA chips (which is how I started having no linux audio knowledge).
In particular from the quirks you need to look at updating pincfgs and maybe gpio fixups - these are typical suspects.
I did see a lot about trying to get HDAs to work on Macs (try looking for alsa mailing lists) - but turned out was totally irrelevant for me with the 8409 chip.

One other suggestion is to find the cs4208_xx.inf (eg cs4208_39.inf) file  from eg Windows boot camp installer which lists hda commands Windows would send to HDA chips which includes Macs for booting Windows on Macs.
You will need the subsystem id to determine which actions to look for.

kash-raman, 2022-05-05:
Thanks David. 

I was able to navigate thru some of your comments.  
Here is the listing subsystem id:  https://pastebin.com/Cq0Zd2Lh 

I tried the windows boot camp installer but I'm unsuccessful in extracting the inf. https://www.driverscape.com/download/cirrus-logic-cs4208-%28ab-93%29 

I'm bit novice with the drivers. Could you help me familiarize analyzing HDA chips and updating pincfgs and maybe gpio fixups.

davidjo, 2022-05-06:
First ignore the Intel Haswell HDMI Codec - thats for video - you just need the Cirrus Logic CS408 entry.

Look at Nodes  labelled Mic - these are the nids/nodes/widgets associated with the microphone.

I think what I did was I had the Windows installer on a USB disk and I just plain mounted that and it was somewhere on there.

You need to look at the syslog file and see what the auto parser is finding for this codec - look
for something like:

snd_hda_codec_cirrus hdaudioC0D0: autoconfig for CS8409: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
snd_hda_codec_cirrus hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
snd_hda_codec_cirrus hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
snd_hda_codec_cirrus hdaudioC0D0:    mono: mono_out=0x0
snd_hda_codec_cirrus hdaudioC0D0:    inputs:
snd_hda_codec_cirrus hdaudioC0D0:      Internal Mic=0x44
snd_hda_codec_cirrus hdaudioC0D0:      Mic=0x3c

If you look in patch_cirrus.c you can see a number of static tables eg static const struct hda_pintbl mba6_pincfgs
- these replace the pincfgs from the chip - which is how the auto parser determines the above output.
(But dont see a pincfg  fixup table for the mbp11 currently).

You probably need to download the Intel HDA specification document to see what the various bits mean.

You also probably need to setup a kernel module build to make fixups similar to the way this repo does it.

Amr3zzat, 2022-05-13:
I have the same issue with Cirrus Logic CS4208, I hope anyone can help soon, I tried many suggestions without any beneift

teutat3s, 2024-01-17:
Chiming in here because my search results lead me here (linux Macbook 11,5 cs4208 no working headset microphone).

I was able to extract the driver linked by @kash-raman (thanks!) and found a file named `cs4208_13.inf`, and one suspicious looking line in the section `Settings4208CommonFixes`:
```
HKR,cs420x,n18WidgetCaps, ... 0x0040048B      ; HS: override widget caps: +CCL (HS In mono mic fix)
...
```

Looking for that address `0x0040048B` in the Subsystem output (`cat /proc/asound/card*/codec\#*`) I found 
```
Node 0x17 [Pin Complex] wcaps 0x40048b: Stereo Amp-In
  Amp-In caps: ofs=0x00, nsteps=0x03, stepsize=0x27, mute=0
  Amp-In vals:  [0x00 0x00]
  Pincap 0x00000024: IN Detect
  Pin Default 0x400000f0: [N/A] Line Out at Ext N/A
    Conn = Unknown, Color = Unknown
    DefAssociation = 0xf, Sequence = 0x0
  Pin-ctls: 0x00:
  Unsolicited: tag=00, enabled=0
  Power states:  D0 D3 EPSS
  Power: setting=D3, actual=D3
```

Now I'm not sure what to do with this information :shrug: 

I believe that the current wiring assumes that 0x18 is the pink mic, could that be what needs fixing here?
```
Node 0x18 [Pin Complex] wcaps 0x40048a: Mono Amp-In
  Control: name="Mic Boost Volume", index=0, device=0
    ControlAmp: chs=3, dir=In, idx=0, ofs=0
  Amp-In caps: ofs=0x00, nsteps=0x03, stepsize=0x27, mute=0
  Amp-In vals:  [0x00]
  Pincap 0x00001724: IN Detect
    Vref caps: HIZ 50 GRD 80
  Pin Default 0x00ab9040: [Jack] Mic at Ext N/A
    Conn = Comb, Color = Pink
    DefAssociation = 0x4, Sequence = 0x0
  Pin-ctls: 0x24: IN VREF_80
  Unsolicited: tag=02, enabled=1
  Power states:  D0 D3 EPSS
  Power: setting=D3, actual=D3
```

davidjo, 2024-01-18:
So just to make clear the 0x0040048B isnt an "address" its a 32 bit bit field - the capability items (wcaps, pincap) are all bit fields - as is Pin Default.
You need the Intel HDA specification document (download from the web) and it will tell you what all the bit fields mean.

There is an hda verb sender program which allows you to manually send HDA verbs to an HDA device - cant remember off hand the specific package - so you can try updating the capabilities manually and see if you can get the  mike to work.
Unfortunately there may be other problems that cant be fixed this way.
The Pink value is from the Pin Default bit field if I remember right.
What looks different is the 0x0040048a current value for the node wcaps versus the value Windows would set 0x0040048B - but cant remember what bit distinguishes mono and stereo.