davidjo/snd_hda_macbookpro #48: MBP 13,1

thefudgeishot, 2022-05-17:
```
➜  ~ uname -r
5.15.32-1-MANJARO
```
I installed the latest version and see the built in outputs and inputs in pavucontrol. However there is no sound output.

davidjo, 2022-05-17:
Could you get some more information please.
Look for  Subsystem ID from /proc/asound/card*/codec#0
(not sure which card eg card0, card1 will be the CS8409 - but thats the one you need - not the HDMI card).
Also grep your system log  under /var/log (possibly syslog?? - dont know Manjaro) for snd_hda and post that file here.

davidjo, 2022-05-17:
Well 0x106b3300 is the id I expected so should be working.
Sounds like arch doesnt create a syslog - its a file called syslog in that folder that has boot.log etc on Ubuntu.
OK so Arch apparently doesnt install a syslog logger by default - you are supposed to use journalctl I think.
Theres obviously stuff on the net about this (its how I found out).
Seems like you might be able to install eg rsyslog as a logger to get it or use journalctl somehow.
Im looking for lines like:
`snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)`
`snd_hda_intel: Primary patch_cs8409 NOT FOUND trying APPLE`
`snd_hda_codec_cs8409 hdaudioC0D0: autoconfig for CS8409: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker`

thefudgeishot, 2022-05-18:
Perfect, I managed to get rsyslog running found a couple instances in the logs. I found -
`snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])`
`snd_hda_codec_generic hdaudioC0D0: autoconfig for Generic: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker`
`snd_hda_intel 0000:00:1f.3: azx_get_response timeout, switching to polling mode: last cmd=0x20270503`
`snd_hda_intel 0000:00:1f.3: No response from codec, disabling MSI: last cmd=0x20270503`
`snd_hda_intel 0000:00:1f.3: azx_get_response timeout, switching to single_cmd mode: last cmd=0x20270503`
I'll upload the whole syslog in case I missed something.
[syslog.txt](https://github.com/davidjo/snd_hda_macbookpro/files/8712766/syslog.txt)

davidjo, 2022-05-18:
I think I see what the problem is - essentially my module is not being loaded - I dont see Primary patch_cs8409 NOT FOUND trying APPLE and I do see no codecs found!.
I also notice you are using kernel 5.16 - now it appears there are issues with this kernel (there was another report it doesnt work with 5.16) - which suggested there have been some kernel file structure changes.
Also it appears the install script needs some updating to work with Arch - which stores modules under /usr/lib/modules rather than /lib/modules of Ubuntu.
Can you do an ls of the /usr/lib/modules/..kernel version../ directory (eg mine is /lib/modules/5.15.0-23-generic).
Im looking for a directory updates and it should contain my module eg updates/snd-hda-codec-cs8409.ko.
The script command is make install in the build/hda... directory in the git clone main directory - if you go into build/hda.. you can manually run it with sudo make install - ensure there are no errors.

thefudgeishot, 2022-05-18:
That's strange because I dont usually use 5.16 turns out when I ran a full system upgrade yesterday after I opened the issue. I'm gonna get the syslog again when running a 5.15 kernel that does have `/usr/lib/modules/{kernel}/updates/snd-hda-codec-cs8409.ko`. Sorry about that.

thefudgeishot, 2022-05-18:
Hmm, I think we are getting closer I just may not have done this right. I opened the `~/snd_hda_macbookpro/build/{kernel}/` and from the make file I assumed I had to move `snd-hda-codec-cs8409.ko` to `/lib/modules/{kernel}/updates/`. After that I ran `sudo depmod -a` and rebooted. After I booted I checked syslog and couldn't find anything relating to `snd_hda_cs8409` or any path. I also double checked that I moved everything to the right kernel and booted with that kernel. But never the less as expected still no audio.

thefudgeishot, 2022-05-18:
Sorry incase you need it to troubleshoot.
[syslog2.TXT](https://github.com/davidjo/snd_hda_macbookpro/files/8715398/syslog2.TXT)

davidjo, 2022-05-18:
Also sorry - I think the script does work because I think someone said on Arch /lib is a symbolic link to /usr/lib - so current version should work - this is also for running sudo make install in the build/hda... directory (also for moving to /lib/modules)
Had a look at the new syslog2.txt - the module still doesnt look  to be installed.
So check /lib is such a symbolic link and check under /usr/lib/modules/...kernver../updates.
also run a lsmod and grep for snd  - this should show the loaded modules associated with sound.

thefudgeishot, 2022-05-18:
```
➜  ~ lsmod | grep snd
snd_hda_codec_hdmi     86016  0
snd_hda_codec_cs8409   151552  0
snd_soc_skl           221184  0
snd_soc_hdac_hda       28672  1 snd_soc_skl
snd_hda_ext_core       36864  2 snd_soc_hdac_hda,snd_soc_skl
snd_soc_sst_ipc        20480  1 snd_soc_skl
snd_soc_sst_dsp        40960  1 snd_soc_skl
snd_soc_acpi_intel_match    57344  1 snd_soc_skl
snd_soc_acpi           16384  2 snd_soc_acpi_intel_match,snd_soc_skl
snd_soc_core          393216  2 snd_soc_hdac_hda,snd_soc_skl
snd_compress           28672  1 snd_soc_core
ac97_bus               16384  1 snd_soc_core
snd_hda_codec_generic    98304  3 snd_hda_codec_cs8409
snd_pcm_dmaengine      16384  1 snd_soc_core
ledtrig_audio          16384  1 snd_hda_codec_generic
snd_hda_intel          61440  3
snd_intel_dspcfg       32768  2 snd_hda_intel,snd_soc_skl
snd_intel_sdw_acpi     20480  1 snd_intel_dspcfg
snd_hda_codec         184320  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_codec_cs8409,snd_hda_intel,snd_soc_hdac_hda
snd_hda_core          118784  8 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_codec_cs8409,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_soc_hdac_hda,snd_soc_skl
snd_hwdep              16384  1 snd_hda_codec
snd_pcm               163840  8 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_compress,snd_soc_core,snd_soc_skl,snd_hda_core,snd_pcm_dmaengine
snd_timer              49152  1 snd_pcm
snd                   126976  15 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
soundcore              16384  1 snd
```
Hugh data dump, there is no symlink under `/usr/lib/modules/...kernver../updates` so that could be a problem.
```
➜  updates ls -l
total 1948
drwxr-xr-x 2 root root    4096 May 17 13:50 dkms
-rw-r--r-- 1 root root 1989904 May 18 17:55 snd-hda-codec-cs8409.ko
```
I'm not sure what else should be symlinked but I'm happy to data dump a folder.

davidjo, 2022-05-18:
Well the lsmod says snd_hda_codec_cs8409 is definitely loaded - but this could be the default module so  no audio would be expected.
You have the module in updates - I thought the idea of updates is you can put a module of same name in it and it will override the default module but maybe thats not happening.
The default module is in kernel/sound/pci/hda.

See if you have the command dmesg (or install it??) - as soon as possible after boot run this command and save the output.
(dmesg dumps the raw kernel message buffer but its a cyclic buffer so will eventually loose the initial boot msgs as more kernel messages are sent).

The next thing is a bit more dangerous (all done as superuser) - move the default module under kernel/sound/pci/hda to somewhere safe so can put it back after the test.
depmod -a and reboot - now the only snd_hda_codec_cs8409  module should be the updated version - check lsmod/dmesg as before.
Again, Im looking to see Primary patch_cs8409 NOT FOUND trying APPLE.

thefudgeishot, 2022-05-18:
[image]
not sure which kernel you want me to remove.

davidjo, 2022-05-18:
Now thats interesting - another Arch difference - it looks as though it stores compressed modules (I think thats what the .xz prefix indicates) (yep some googling says compressed kernel modules are a thing).
The module we are talking about is snd-hda-codec-cs8409 - under Ubuntu this is snd-hda-codec-cs8409.ko but the above says for Arch its snd-hda-codec-cs8409.ko.xz.

The dmesg command looks to be working.

We absolutely need snd-hda-intel and snd-hda-generic.

So will Arch load uncompressed modules - not clear - try creating a compressed module of snd-hda-codec-cs8409.ko under updates - apparently xz is command needed - supposedly in xz-utils
maybe xz snd-hda-codec-cs8409.ko - you should end up with a file snd-hda-codec-cs8409.ko.xz.

davidjo, 2022-05-18:
By the way Ive updated the repo with better version - but nothing to solve your problem.

thefudgeishot, 2022-05-18:
that xz command seemed to have worked and I managed to get `snd-hda-codec-cs8409.ko.xz`. I then ran `depmod -a` and rebooted. upon reboot I ran the dmesg command. Fingers crossed we are getting closer.
[dmesglog.txt](https://github.com/davidjo/snd_hda_macbookpro/files/8718003/dmesglog.txt)

davidjo, 2022-05-18:
Theres still something wrong - and Ive realized we DONT want
snd-hda-codec-generic (Im confused because the source code uses a lot of
hda_generic functions but this is a codec name - we want the specific codec
snd-hda-codec-cs8409).
(We MUST have snd-hda-intel).

So it looks like theres something wrong with codec detection - you should
at least get the default snd-hda-codec-cs8409 module.

Did you do anything previously like blacklisting modules or any other
messing with the module config files in /etc/ eg /etc/modules, added
entries in /etc/modprobe.d/blacklist... files??

thefudgeishot, 2022-05-19:
Hmm I doubt it I've never heard of those directory's and and they look fairly normal after inspection. Either they don't exist like the blacklist files or I have no history of editing those files. The only directories I have changed are the ones you've suggested and the experiment directory which has since been reset back to its default.

As another attempt I removed the `snd-hda-generic` and kept `snd-hda-intel`.  I ran the dmesglog again on boot. 
[dmesglog.txt](https://github.com/davidjo/snd_hda_macbookpro/files/8721807/dmesglog.txt)
```
➜  ~ sudo dmesg > dmesglog.txt && lsmod | grep snd
snd_hda_codec_hdmi     86016  0
snd_soc_skl           221184  0
snd_soc_hdac_hda       28672  1 snd_soc_skl
snd_hda_ext_core       36864  2 snd_soc_hdac_hda,snd_soc_skl
snd_soc_sst_ipc        20480  1 snd_soc_skl
snd_soc_sst_dsp        40960  1 snd_soc_skl
snd_soc_acpi_intel_match    57344  1 snd_soc_skl
snd_soc_acpi           16384  2 snd_soc_acpi_intel_match,snd_soc_skl
snd_soc_core          393216  2 snd_soc_hdac_hda,snd_soc_skl
snd_compress           28672  1 snd_soc_core
ac97_bus               16384  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_hda_intel          61440  0
snd_intel_dspcfg       32768  2 snd_hda_intel,snd_soc_skl
snd_intel_sdw_acpi     20480  1 snd_intel_dspcfg
snd_hda_codec         184320  3 snd_hda_codec_hdmi,snd_hda_intel,snd_soc_hdac_hda
snd_hda_core          118784  6 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_soc_hdac_hda,snd_soc_skl
snd_hwdep              16384  1 snd_hda_codec
snd_pcm               163840  8 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_compress,snd_soc_core,snd_soc_skl,snd_hda_core,snd_pcm_dmaengine
snd_timer              49152  1 snd_pcm
snd                   126976  8 snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
soundcore              16384  1 snd
```

davidjo, 2022-05-19:
Can you dump /proc/asound/card0/codec#0 (and codec#2) and attach the full dumps.

At the moment for some reason your machine is not detecting the sound card - certainly not the cs8409 card.
snd-hda-codec-generic is used as a last resort when no specific module fits - this is detected using the Vendor Id (just above the the Subsystem Id in the codec#0 dump).

davidjo, 2022-05-19:
Make sure you restored snd-hda-codec-generic.ko.xz - this is the codec of last resort and its the codecs which deliver the cdec#0 data.

thefudgeishot, 2022-05-19:
I thought that was the issue and I had removed it entirely, so I just installed your recent update as per the instruction and built the drivers again so that should have restored generic.

davidjo, 2022-05-19:
Sorry - this repo wont re-create snd-hda-codec-generic.ko.xz - I should have said this needs to be saved like snd-hda-intel.ko.xz.
You need to restore it from the original kernel module installation package (at least on ubuntu the modules are in a separate package something like linux-modules.....).
Also I think another issue is after any messing with kernel modules you must redo depmod - so sudo depmod -a should be done after any moving of modules - it doesnt matter if you do the command more than once.

thefudgeishot, 2022-05-19:
Oh I meant I reran this command `pacman -S wget make gcc linux-headers` to fix the kernels. And of course I reran `sudo depmod -a`

thefudgeishot, 2022-05-19:
I suppose for reference of whats going on I ran `dmesg` and `lsmod | grep snd`
[dmesglog.txt](https://github.com/davidjo/snd_hda_macbookpro/files/8728889/dmesglog.txt)
```
snd_hda_codec_hdmi     86016  0
snd_soc_skl           221184  0
snd_soc_hdac_hda       28672  1 snd_soc_skl
snd_hda_ext_core       36864  2 snd_soc_hdac_hda,snd_soc_skl
snd_soc_sst_ipc        20480  1 snd_soc_skl
snd_soc_sst_dsp        40960  1 snd_soc_skl
snd_soc_acpi_intel_match    57344  1 snd_soc_skl
snd_soc_acpi           16384  2 snd_soc_acpi_intel_match,snd_soc_skl
snd_soc_core          393216  2 snd_soc_hdac_hda,snd_soc_skl
snd_compress           28672  1 snd_soc_core
ac97_bus               16384  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_hda_intel          61440  0
snd_intel_dspcfg       32768  2 snd_hda_intel,snd_soc_skl
snd_intel_sdw_acpi     20480  1 snd_intel_dspcfg
snd_hda_codec         184320  3 snd_hda_codec_hdmi,snd_hda_intel,snd_soc_hdac_hda
snd_hda_core          118784  6 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_soc_hdac_hda,snd_soc_skl
snd_hwdep              16384  1 snd_hda_codec
snd_pcm               163840  8 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_compress,snd_soc_core,snd_soc_skl,snd_hda_core,snd_pcm_dmaengine
snd_timer              49152  1 snd_pcm
snd                   126976  8 snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
soundcore              16384  1 snd
```

davidjo, 2022-05-19:
Well it looks like snd-hda-codec-generic.ko has been lost - I think you need the linux package to get back the primary kernel modules - linux-headers is just the kernel headers I think.
Do you still have a snd-hda-codec-cs8409.ko(.xz) somewhere - if you do dont understand why its not loading that?

thefudgeishot, 2022-05-19:
Just a thought maybe its the kernel version? I could possibly downgrade to the kernel version you use and is confirmed working.

davidjo, 2022-05-19:
I dont think so - Ubuntu 22.04 is 5.15.0-23 so dont think 32 should be that different.
Unfortunately as Ive discovered during this project each distribution makes its own changes cf mainline - and in particular Ubuntu - so eg 5.15.0-23 doesnt correspond to the mainline kernel 5.15.23.
Can you boot OSX?
If so boot OSX and run ioreg -a -l -x - this should create an .xml file on stdout which you can redirect to a file.
Then I need to see the values for the keys IOHDACodecVendorID and IOHDACodecFunctionSubsystemID.
Otherwise I think something has gotten corrupted so would suggest a re-install of Linux.
Looking back the first log we got shows snd-hda-codec-generic was loaded which doesnt make sense even then.