davidjo/snd_hda_macbookpro #51: Ubuntu 22.04 not work while building from source

orklann, 2022-06-09:
Hi, Thank you for the great driver.

I just installed Ubuntu 22.04 on my iMac 2017, and follow the steps to install the driver, but got building errors.

```
sudo ./install.cirrus.driver.sh 
```

```shell
linux-source-5.15.0/sound/pci/hda/
linux-source-5.15.0/sound/pci/hda/patch_ca0110.c
linux-source-5.15.0/sound/pci/hda/hda_intel.c
linux-source-5.15.0/sound/pci/hda/patch_cirrus.c
linux-source-5.15.0/sound/pci/hda/hda_bind.c
linux-source-5.15.0/sound/pci/hda/Kconfig
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_spi.c
linux-source-5.15.0/sound/pci/hda/ideapad_s740_helper.c
linux-source-5.15.0/sound/pci/hda/patch_realtek.c
linux-source-5.15.0/sound/pci/hda/patch_analog.c
linux-source-5.15.0/sound/pci/hda/hp_x360_helper.c
linux-source-5.15.0/sound/pci/hda/hda_intel_trace.h
linux-source-5.15.0/sound/pci/hda/patch_cs8409.c
linux-source-5.15.0/sound/pci/hda/hda_controller.c
linux-source-5.15.0/sound/pci/hda/hda_intel.h
linux-source-5.15.0/sound/pci/hda/patch_hdmi.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.c
linux-source-5.15.0/sound/pci/hda/hda_eld.c
linux-source-5.15.0/sound/pci/hda/hda_local.h
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-source-5.15.0/sound/pci/hda/ca0132_regs.h
linux-source-5.15.0/sound/pci/hda/hda_codec.c
linux-source-5.15.0/sound/pci/hda/hda_generic.h
linux-source-5.15.0/sound/pci/hda/hda_hwdep.c
linux-source-5.15.0/sound/pci/hda/hda_controller.h
linux-source-5.15.0/sound/pci/hda/hda_beep.h
linux-source-5.15.0/sound/pci/hda/patch_cs8409-tables.c
linux-source-5.15.0/sound/pci/hda/patch_sigmatel.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.h
linux-source-5.15.0/sound/pci/hda/patch_ca0132.c
linux-source-5.15.0/sound/pci/hda/hda_component.h
linux-source-5.15.0/sound/pci/hda/hda_beep.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.h
linux-source-5.15.0/sound/pci/hda/hda_tegra.c
linux-source-5.15.0/sound/pci/hda/hda_generic.c
linux-source-5.15.0/sound/pci/hda/hda_sysfs.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.h
linux-source-5.15.0/sound/pci/hda/hda_proc.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.c
linux-source-5.15.0/sound/pci/hda/patch_conexant.c
linux-source-5.15.0/sound/pci/hda/hda_controller_trace.h
linux-source-5.15.0/sound/pci/hda/Makefile
linux-source-5.15.0/sound/pci/hda/patch_si3054.c
linux-source-5.15.0/sound/pci/hda/hda_jack.h
linux-source-5.15.0/sound/pci/hda/patch_via.c
linux-source-5.15.0/sound/pci/hda/thinkpad_helper.c
linux-source-5.15.0/sound/pci/hda/hda_jack.c
linux-source-5.15.0/sound/pci/hda/patch_cmedia.c
make -C /lib/modules/5.15.0-37-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-37-generic'
  CC [M]  /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o
In file included from /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1314:
/home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:1192:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1192 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
  CC [M]  /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409-tables.o
In file included from ./include/sound/info.h:12,
                 from ./include/sound/hda_codec.h:13,
                 from /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.h:15,
                 from /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409-tables.c:11:
/home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409-tables.c:518:54: error: ‘CS8409_WARLOCK_MLK’ undeclared here (not in a function); did you mean ‘CS8409_WARLOCK’?
  518 |         SND_PCI_QUIRK(0x1028, 0x0B92, "Warlock MLK", CS8409_WARLOCK_MLK),
      |                                                      ^~~~~~~~~~~~~~~~~~
./include/sound/core.h:474:50: note: in definition of macro ‘SND_PCI_QUIRK’
  474 |         {_SND_PCI_QUIRK_ID(vend, dev), .value = (val)}
      |                                                  ^~~
/home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409-tables.c:519:63: error: ‘CS8409_WARLOCK_MLK_DUAL_MIC’ undeclared here (not in a function)
  519 |       SND_PCI_QUIRK(0x1028, 0x0B93, "Warlock MLK Dual Mic", CS8409_WARLOCK_MLK_DUAL_MIC),
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~

./include/sound/core.h:474:50: note: in definition of macro ‘SND_PCI_QUIRK’
  474 |         {_SND_PCI_QUIRK_ID(vend, dev), .value = (val)}
      |                                                  ^~~
/home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409-tables.c:567:10: error: array index in initializer not of integer type
  567 |         [CS8409_WARLOCK_MLK] = {
      |          ^~~~~~~~~~~~~~~~~~
/home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409-tables.c:567:10: note: (near initialization for ‘cs8409_fixups’)
/home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409-tables.c:573:10: error: array index in initializer not of integer type
  573 |         [CS8409_WARLOCK_MLK_DUAL_MIC] = {
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409-tables.c:573:10: note: (near initialization for ‘cs8409_fixups’)
make[2]: *** [scripts/Makefile.build:285: /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409-tables.o] Error 1
make[1]: *** [Makefile:1875: /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-37-generic'
make: *** [Makefile:22: all] Error 2
cp snd-hda-codec-cs8409.ko /lib/modules/5.15.0-37-generic/updates
cp: cannot stat 'snd-hda-codec-cs8409.ko': No such file or directory
make: *** [Makefile:30: install] Error 1

contents of /lib/modules/5.15.0-37-generic/updates
total 0

```

Is there any workaround or fix to this?

davidjo, 2022-06-09:
Sorry - this is what Ive discovered is a problem with linux audio kernel - it seems to update significantly every so often on sublevel kernel version updates (also Ubuntus definition of kernel versions is very different from  mainline kernel versions).
This looks like its similar to the other reported issue with kernel 5.18 - which means when I update my kernel to 5.15.0-37 should be able to see whats going on.
Im going to try and switch the install script to a patch system for the existing files which hopefully will reduce these issues.

orklann, 2022-06-09:
Thank you for your great work. Just take your time, I am now using a bluetooth headphone to get audio output, but a wire headphone would be good too.

orklann, 2022-06-14:
After `git pulll` and run `sudo ./install.cirrus.driver.sh `

```
linux-source-5.15.0/sound/pci/hda/
linux-source-5.15.0/sound/pci/hda/patch_ca0110.c
linux-source-5.15.0/sound/pci/hda/hda_intel.c
linux-source-5.15.0/sound/pci/hda/patch_cirrus.c
linux-source-5.15.0/sound/pci/hda/hda_bind.c
linux-source-5.15.0/sound/pci/hda/Kconfig
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_spi.c
linux-source-5.15.0/sound/pci/hda/ideapad_s740_helper.c
linux-source-5.15.0/sound/pci/hda/patch_realtek.c
linux-source-5.15.0/sound/pci/hda/patch_analog.c
linux-source-5.15.0/sound/pci/hda/hp_x360_helper.c
linux-source-5.15.0/sound/pci/hda/hda_intel_trace.h
linux-source-5.15.0/sound/pci/hda/patch_cs8409.c
linux-source-5.15.0/sound/pci/hda/hda_controller.c
linux-source-5.15.0/sound/pci/hda/hda_intel.h
linux-source-5.15.0/sound/pci/hda/patch_hdmi.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.c
linux-source-5.15.0/sound/pci/hda/hda_eld.c
linux-source-5.15.0/sound/pci/hda/hda_local.h
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-source-5.15.0/sound/pci/hda/ca0132_regs.h
linux-source-5.15.0/sound/pci/hda/hda_codec.c
linux-source-5.15.0/sound/pci/hda/hda_generic.h
linux-source-5.15.0/sound/pci/hda/hda_hwdep.c
linux-source-5.15.0/sound/pci/hda/hda_controller.h
linux-source-5.15.0/sound/pci/hda/hda_beep.h
linux-source-5.15.0/sound/pci/hda/patch_cs8409-tables.c
linux-source-5.15.0/sound/pci/hda/patch_sigmatel.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.h
linux-source-5.15.0/sound/pci/hda/patch_ca0132.c
linux-source-5.15.0/sound/pci/hda/hda_component.h
linux-source-5.15.0/sound/pci/hda/hda_beep.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.h
linux-source-5.15.0/sound/pci/hda/hda_tegra.c
linux-source-5.15.0/sound/pci/hda/hda_generic.c
linux-source-5.15.0/sound/pci/hda/hda_sysfs.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.h
linux-source-5.15.0/sound/pci/hda/hda_proc.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.c
linux-source-5.15.0/sound/pci/hda/patch_conexant.c
linux-source-5.15.0/sound/pci/hda/hda_controller_trace.h
linux-source-5.15.0/sound/pci/hda/Makefile
linux-source-5.15.0/sound/pci/hda/patch_si3054.c
linux-source-5.15.0/sound/pci/hda/hda_jack.h
linux-source-5.15.0/sound/pci/hda/patch_via.c
linux-source-5.15.0/sound/pci/hda/thinkpad_helper.c
linux-source-5.15.0/sound/pci/hda/hda_jack.c
linux-source-5.15.0/sound/pci/hda/patch_cmedia.c
patching file patch_cs8409.c
Hunk #1 succeeded at 1277 (offset 7 lines).
Hunk #2 succeeded at 1286 (offset 7 lines).
Hunk #3 succeeded at 1315 (offset 7 lines).
patching file patch_cs8409.h
Hunk #2 succeeded at 306 (offset 5 lines).
Hunk #3 succeeded at 333 (offset 5 lines).
Hunk #4 succeeded at 368 (offset 5 lines).
make -C /lib/modules/5.15.0-37-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-37-generic'
  CC [M]  /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o
In file included from /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1321:
/home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:1192:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1192 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
  CC [M]  /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409-tables.o
  LD [M]  /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cs8409.o
  MODPOST /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/Module.symvers
  CC [M]  /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cs8409.mod.o
  LD [M]  /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cs8409.ko
  BTF [M] /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cs8409.ko
Skipping BTF generation for /home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cs8409.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-37-generic'
#cp snd-hda-codec-cs8409.ko /lib/modules/5.15.0-37-generic/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.15.0-37-generic/build M=/home/rkt/drivers/snd_hda_macbookpro/build/hda-5.15.0 modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-37-generic'
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
  INSTALL /lib/modules/5.15.0-37-generic/updates/snd-hda-codec-cs8409.ko
  SIGN    /lib/modules/5.15.0-37-generic/updates/snd-hda-codec-cs8409.ko
At main.c:160:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: ../crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: ../crypto/bio/bss_file.c:75
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  /lib/modules/5.15.0-37-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-37-generic'
depmod -a

contents of /lib/modules/5.15.0-37-generic/updates
total 1464
-rw-r--r-- 1 root root 1495736 Jun 14 20:38 snd-hda-codec-cs8409.ko
```

But it seems my headphone and iMac speaker does not get sound output after reboot, what I did wrong?

carrelebo, 2022-06-14:
@orklann I've install the new driver version from github and use a gitclone, normally all after the installation work, else no use this command and relaunch with sudo the install driver file :
```
sudo apt install wget make gcc linux-headers-generic
```
don't worry about message during the driver installation, just reboot

orklann, 2022-06-15:
@orklann How to check UNKONW subsystem id in the kernel log?

Maybe my iMac is not supported by your driver.

carrelebo, 2022-06-15:
@orklann use "ioreg -l -x" command in OSX, copy all, make a txt file with it and add as a comment, après il pourra il pourra voir le problème

orklann, 2022-06-15:
@carrelebo I can not post the output of `ioreg -l -x` here, it's too long to post, so uploaded it to dropbox, and here is the link.

[ioreg.txt](https://www.dropbox.com/s/dlx2ylj1b2jkkgq/ioreg.txt?dl=0)

davidjo, 2022-06-15:
@orklann. Under Ubuntu the system log file is /var/log/syslog - this is a text file - grep for snd_hda and see if a line with UNKNOWN subsystem id shows.
Checked your ioreg file - you should be supported - IOHDACodecFunctionSubsystemID is 0x106b0e00 which is implemented.

orklann, 2022-06-15:
@davidjo Here is the output of `sudo cat  /var/log/syslog | grep snd_hda`

Not `UNKONWN` found.

```
Jun 12 00:18:33 rkt-ubuntu kernel: [112134.516611] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 00:18:33 rkt-ubuntu kernel: [112134.516706] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 05:48:19 rkt-ubuntu kernel: [131920.809021] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 05:48:19 rkt-ubuntu kernel: [131920.809116] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 06:09:00 rkt-ubuntu kernel: [133162.091767] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 06:09:00 rkt-ubuntu kernel: [133162.091863] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 07:26:19 rkt-ubuntu kernel: [137801.236545] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 07:26:19 rkt-ubuntu kernel: [137801.236642] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 07:28:58 rkt-ubuntu kernel: [137960.429096] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 07:28:58 rkt-ubuntu kernel: [137960.429191] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 07:30:41 rkt-ubuntu kernel: [138062.621393] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 07:30:41 rkt-ubuntu kernel: [138062.621487] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 07:32:25 rkt-ubuntu kernel: [138167.165622] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 07:32:25 rkt-ubuntu kernel: [138167.165716] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 07:52:23 rkt-ubuntu kernel: [139364.611593] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 07:52:23 rkt-ubuntu kernel: [139364.611698] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 08:03:04 rkt-ubuntu kernel: [140005.635636] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 08:03:04 rkt-ubuntu kernel: [140005.635730] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 08:03:49 rkt-ubuntu kernel: [140050.551776] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 08:03:49 rkt-ubuntu kernel: [140050.551872] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:12:42 rkt-ubuntu kernel: [151384.062614] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:12:42 rkt-ubuntu kernel: [151384.062710] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:13:38 rkt-ubuntu kernel: [151440.310682] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:13:38 rkt-ubuntu kernel: [151440.310776] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:26:34 rkt-ubuntu kernel: [152216.083916] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:26:34 rkt-ubuntu kernel: [152216.084014] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:29:21 rkt-ubuntu kernel: [152382.624054] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:29:21 rkt-ubuntu kernel: [152382.624149] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:34:50 rkt-ubuntu kernel: [152712.452519] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:34:50 rkt-ubuntu kernel: [152712.452629] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:38:14 rkt-ubuntu kernel: [152915.855176] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:38:14 rkt-ubuntu kernel: [152915.855273] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:56:39 rkt-ubuntu kernel: [154021.058045] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 11:56:39 rkt-ubuntu kernel: [154021.058140] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 12:47:32 rkt-ubuntu kernel: [157074.430858] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 12:47:32 rkt-ubuntu kernel: [157074.430953] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 13:07:40 rkt-ubuntu kernel: [158282.507937] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 13:07:40 rkt-ubuntu kernel: [158282.508032] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 13:44:17 rkt-ubuntu kernel: [160478.665232] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 13:44:17 rkt-ubuntu kernel: [160478.665329] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 14:18:17 rkt-ubuntu kernel: [162519.255030] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 14:18:17 rkt-ubuntu kernel: [162519.255125] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 15:38:05 rkt-ubuntu kernel: [167306.891522] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
[... cut ...]
Jun 12 17:36:46 rkt-ubuntu kernel: [ 3793.403981] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 17:36:46 rkt-ubuntu kernel: [ 3793.404075] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 17:40:28 rkt-ubuntu kernel: [ 4014.898971] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
Jun 12 17:40:28 rkt-ubuntu kernel: [ 4014.899065] snd_hda_codec_generic hdaudioC0D2: Unable to sync register 0x2f0d00. -5
grep: (standard input): binary file matches
```

davidjo, 2022-06-15:
Ignore hdaudioC0D2 - thats your HDMI or other video port - just add "| grep hdaudioC0D0"  to your command.
The issue is  snd_hda_codec_generic hdaudioC0D0: autoconfig for Generic: - this means no 8409 module is loaded at all just the generic driver hence snd_hda_codec_generic
- should be something like snd_hda_codec_cs8409: autoconfig for CS8409.

orklann, 2022-06-15:
Here is the new output with `cat  /var/log/syslog | grep hdaudioC0D0`

```
Jun 12 15:43:19 rkt-ubuntu kernel: [    7.125395] snd_hda_codec_generic hdaudioC0D0: autoconfig for Generic: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
Jun 12 15:43:19 rkt-ubuntu kernel: [    7.125409] snd_hda_codec_generic hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Jun 12 15:43:19 rkt-ubuntu kernel: [    7.125416] snd_hda_codec_generic hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
Jun 12 15:43:19 rkt-ubuntu kernel: [    7.125421] snd_hda_codec_generic hdaudioC0D0:    mono: mono_out=0x0
Jun 12 15:43:19 rkt-ubuntu kernel: [    7.125424] snd_hda_codec_generic hdaudioC0D0:    inputs:
Jun 12 15:43:19 rkt-ubuntu kernel: [    7.125428] snd_hda_codec_generic hdaudioC0D0:      Internal Mic=0x45
Jun 12 15:43:19 rkt-ubuntu kernel: [    7.125432] snd_hda_codec_generic hdaudioC0D0:      Mic=0x3c
Jun 12 15:43:24 rkt-ubuntu kernel: [   12.712485] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
Jun 12 15:43:24 rkt-ubuntu kernel: [   12.712581] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
Jun 12 15:50:37 rkt-ubuntu kernel: [    6.810647] snd_hda_codec_generic hdaudioC0D0: autoconfig for Generic: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
Jun 12 15:50:37 rkt-ubuntu kernel: [    6.810666] snd_hda_codec_generic hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Jun 12 15:50:37 rkt-ubuntu kernel: [    6.810674] snd_hda_codec_generic hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
Jun 12 15:50:37 rkt-ubuntu kernel: [    6.810681] snd_hda_codec_generic hdaudioC0D0:    mono: mono_out=0x0
Jun 12 15:50:37 rkt-ubuntu kernel: [    6.810686] snd_hda_codec_generic hdaudioC0D0:    inputs:
Jun 12 15:50:37 rkt-ubuntu kernel: [    6.810690] snd_hda_codec_generic hdaudioC0D0:      Internal Mic=0x45
Jun 12 15:50:37 rkt-ubuntu kernel: [    6.810696] snd_hda_codec_generic hdaudioC0D0:      Mic=0x3c
Jun 12 15:50:42 rkt-ubuntu kernel: [   12.418229] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
Jun 12 15:50:42 rkt-ubuntu kernel: [   12.418333] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
Jun 12 15:59:37 rkt-ubuntu kernel: [    7.381878] snd_hda_codec_generic hdaudioC0D0: autoconfig for Generic: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
Jun 12 15:59:37 rkt-ubuntu kernel: [    7.381900] snd_hda_codec_generic hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Jun 12 15:59:37 rkt-ubuntu kernel: [    7.381910] snd_hda_codec_generic hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
Jun 12 15:59:37 rkt-ubuntu kernel: [    7.381921] snd_hda_codec_generic hdaudioC0D0:    mono: mono_out=0x0
Jun 12 15:59:37 rkt-ubuntu kernel: [    7.381928] snd_hda_codec_generic hdaudioC0D0:    inputs:
Jun 12 15:59:37 rkt-ubuntu kernel: [    7.381935] snd_hda_codec_generic hdaudioC0D0:      Internal Mic=0x45
Jun 12 15:59:37 rkt-ubuntu kernel: [    7.381942] snd_hda_codec_generic hdaudioC0D0:      Mic=0x3c
Jun 12 15:59:42 rkt-ubuntu kernel: [   12.996736] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
Jun 12 15:59:42 rkt-ubuntu kernel: [   12.996833] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
Jun 12 16:02:52 rkt-ubuntu kernel: [    6.874191] snd_hda_codec_generic hdaudioC0D0: autoconfig for Generic: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
Jun 12 16:02:52 rkt-ubuntu kernel: [    6.874197] snd_hda_codec_generic hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Jun 12 16:02:52 rkt-ubuntu kernel: [    6.874199] snd_hda_codec_generic hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
Jun 12 16:02:52 rkt-ubuntu kernel: [    6.874201] snd_hda_codec_generic hdaudioC0D0:    mono: mono_out=0x0
Jun 12 16:02:52 rkt-ubuntu kernel: [    6.874202] snd_hda_codec_generic hdaudioC0D0:    inputs:
Jun 12 16:02:52 rkt-ubuntu kernel: [    6.874203] snd_hda_codec_generic hdaudioC0D0:      Internal Mic=0x45
Jun 12 16:02:52 rkt-ubuntu kernel: [    6.874205] snd_hda_codec_generic hdaudioC0D0:      Mic=0x3c
Jun 12 16:02:56 rkt-ubuntu kernel: [   12.468704] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
Jun 12 16:02:56 rkt-ubuntu kernel: [   12.468799] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
Jun 12 16:06:46 rkt-ubuntu kernel: [    7.179156] snd_hda_codec_generic hdaudioC0D0: autoconfig for Generic: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
Jun 12 16:06:46 rkt-ubuntu kernel: [    7.179162] snd_hda_codec_generic hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Jun 12 16:06:46 rkt-ubuntu kernel: [    7.179166] snd_hda_codec_generic hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
Jun 12 16:06:46 rkt-ubuntu kernel: [    7.179169] snd_hda_codec_generic hdaudioC0D0:    mono: mono_out=0x0
Jun 12 16:06:46 rkt-ubuntu kernel: [    7.179170] snd_hda_codec_generic hdaudioC0D0:    inputs:
Jun 12 16:06:46 rkt-ubuntu kernel: [    7.179173] snd_hda_codec_generic hdaudioC0D0:      Internal Mic=0x45
Jun 12 16:06:46 rkt-ubuntu kernel: [    7.179175] snd_hda_codec_generic hdaudioC0D0:      Mic=0x3c
Jun 12 16:06:51 rkt-ubuntu kernel: [   12.780776] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
Jun 12 16:06:51 rkt-ubuntu kernel: [   12.780873] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
[... cut ...]
Jun 12 16:33:45 rkt-ubuntu kernel: [   12.356441] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
Jun 12 16:33:45 rkt-ubuntu kernel: [   12.356536] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
Jun 12 17:25:49 rkt-ubuntu kernel: [ 3135.718376] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
Jun 12 17:25:49 rkt-ubuntu kernel: [ 3135.718472] snd_hda_codec_generic hdaudioC0D0: Unable to sync register 0x23b0000. -5
grep: (standard input): binary file matches
```

orklann, 2022-06-15:
@davidjo  How to load `8409` module instead of the generic one?

orklann, 2022-06-15:
@davidjo Is a fresh install around a week ago.

`sudo rmmod snd_hda_code_generic` gets a error:

`rmmod: ERROR: Module snd_hda_codec_generic is in use
`

orklann, 2022-06-15:
`sudo modprobe snd_hda_codec_cs8409` loads fine, after that, `lsmod | grep cs8409 ` shows:
```
lsmod | grep cs8409                                                20:18:15
snd_hda_codec_cs8409   114688  0
snd_hda_codec_generic   102400  3 snd_hda_codec_cs8409
snd_hda_codec         155648  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_codec_cs8409,snd_hda_intel,snd_soc_hdac_hda
snd_hda_core          110592  8 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_codec_cs8409,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_soc_hdac_hda,snd_soc_skl
```

davidjo, 2022-06-15:
Hmm - I dont like the 0 for snd_hda_codec_cs8409 - mine says 1 - suggesting its not fully loaded.
Did you do modprobe -v?
Also in a separate terminal window do tail -f /var/log/syslog then switch to previous window and do the rmmod/modprobe -v
- check the tail -f window and see what messages it prints.

Try a full rebuild - in your clone directory do sudo rm -fr build then redo the sudo install... comand.
check the module  in the /lib/modules/5.15.0-37-generic/updates/snd-hda-codec-cs8409.ko is same time or later than that under build/hda-5.15.0.

if you dont see any snd_hda_codec_cs8409 messages in the syslog then its still not loaded.

orklann, 2022-06-16:
@davidjo yes, I opened a terminal and keep this command running `tail -f /var/log/syslog`. And run `sudo modprobe snd_hda_codec_cs8409` in another terminal, as you guess, I did not see any `snd_hda_codec_cs8409 ` in the `tail -f /var/log/syslog` terminal window.

davidjo, 2022-06-16:
Not sure what I can say - something seems to be messing with the module loading  - there are files under /etc eg modules and modules-load.d which can be used to prevent module loading - which is why I asked about a fresh install - you never added a module blacklist?

orklann, 2022-08-29:
I just reinstalled Ubuntu 22.04 on my iMac 2017 recently, and after that, I tried this package again, and it works now. Thank you for the fix.

davidjo, 2022-08-29:
https://github.com/davidjo/snd_hda_macbookpro/issues/51#issuecomment-1172319970

The issue I think with the mike is that Ive left this as in the OSX driver and it seems to return a very low amplitude - I just recorded a bit under linux with Audacity and amplified it and it seemed OK.
My idea was that if you do want to use the mike you add an Alsa or Pulse user module to amplify the input - Im assuming on OSX the amplification is done by the higher level CoreAudio kernel subsystem.
The buttons are detected but are not mapped from hardware values to linux kernel values so wont change volume (at least for Apple headphone type with mike/buttons).