davidjo/snd_hda_macbookpro #52: Linux Mint with kernel 5.15.0.33

Ven0mstrike42, 2022-06-28:
attempting to build for Macbookpro14,1 using Linux Mint 20.3 with kernel 5.15.0.33. throwing build errors
not sure if dumb.
```
landon@Mintbook:~/snd_hda_macbookpro$ sudo ./install.cirrus.driver.sh 
--2022-06-28 17:46:01--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.15.0.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 2a04:4e42:72::432, 146.75.9.176
Connecting to cdn.kernel.org (cdn.kernel.org)|2a04:4e42:72::432|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-06-28 17:46:01 ERROR 404: Not Found.

Failed to download linux-5.15.0.tar.xz
Trying to download base kernel version linux-5.15.tar.xz
This may lead to build failures as too old
If this is an Ubuntu-based distribution this almost certainly will fail to build

--2022-06-28 17:46:01--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.15.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 2a04:4e42:72::432, 146.75.9.176
Connecting to cdn.kernel.org (cdn.kernel.org)|2a04:4e42:72::432|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

linux-5.15/sound/pci/hda/
linux-5.15/sound/pci/hda/Kconfig
linux-5.15/sound/pci/hda/Makefile
linux-5.15/sound/pci/hda/ca0132_regs.h
linux-5.15/sound/pci/hda/hda_auto_parser.c
linux-5.15/sound/pci/hda/hda_auto_parser.h
linux-5.15/sound/pci/hda/hda_beep.c
linux-5.15/sound/pci/hda/hda_beep.h
linux-5.15/sound/pci/hda/hda_bind.c
linux-5.15/sound/pci/hda/hda_codec.c
linux-5.15/sound/pci/hda/hda_controller.c
linux-5.15/sound/pci/hda/hda_controller.h
linux-5.15/sound/pci/hda/hda_controller_trace.h
linux-5.15/sound/pci/hda/hda_eld.c
linux-5.15/sound/pci/hda/hda_generic.c
linux-5.15/sound/pci/hda/hda_generic.h
linux-5.15/sound/pci/hda/hda_hwdep.c
linux-5.15/sound/pci/hda/hda_intel.c
linux-5.15/sound/pci/hda/hda_intel.h
linux-5.15/sound/pci/hda/hda_intel_trace.h
linux-5.15/sound/pci/hda/hda_jack.c
linux-5.15/sound/pci/hda/hda_jack.h
linux-5.15/sound/pci/hda/hda_local.h
linux-5.15/sound/pci/hda/hda_proc.c
linux-5.15/sound/pci/hda/hda_sysfs.c
linux-5.15/sound/pci/hda/hda_tegra.c
linux-5.15/sound/pci/hda/hp_x360_helper.c
linux-5.15/sound/pci/hda/ideapad_s740_helper.c
linux-5.15/sound/pci/hda/patch_analog.c
linux-5.15/sound/pci/hda/patch_ca0110.c
linux-5.15/sound/pci/hda/patch_ca0132.c
linux-5.15/sound/pci/hda/patch_cirrus.c
linux-5.15/sound/pci/hda/patch_cmedia.c
linux-5.15/sound/pci/hda/patch_conexant.c
linux-5.15/sound/pci/hda/patch_cs8409-tables.c
linux-5.15/sound/pci/hda/patch_cs8409.c
linux-5.15/sound/pci/hda/patch_cs8409.h
linux-5.15/sound/pci/hda/patch_hdmi.c
linux-5.15/sound/pci/hda/patch_realtek.c
linux-5.15/sound/pci/hda/patch_si3054.c
linux-5.15/sound/pci/hda/patch_sigmatel.c
linux-5.15/sound/pci/hda/patch_via.c
linux-5.15/sound/pci/hda/thinkpad_helper.c
patching file patch_cs8409.c
Hunk #1 succeeded at 1262 (offset -8 lines).
Hunk #2 succeeded at 1271 (offset -8 lines).
Hunk #3 succeeded at 1300 (offset -8 lines).
patching file patch_cs8409.h
Hunk #3 FAILED at 328.
Hunk #4 succeeded at 359 (offset -1 lines).
1 out of 4 hunks FAILED -- saving rejects to file patch_cs8409.h.rej
make -C /lib/modules/5.15.0-33-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/landon/snd_hda_macbookpro/build/hda-5.15.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-33-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
  You are using:           gcc (Ubuntu 9.4.0-5ubuntu1) 9.4.0
  CC [M]  /home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o
In file included from /home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1306:
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:157:10: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
  157 |         .linein_jack_in = 0,
      |          ^~~~~~~~~~~~~~
      |          mic_jack_in
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:158:10: error: ‘struct sub_codec’ has no member named ‘force_status_change’
  158 |         .force_status_change = 1,
      |          ^~~~~~~~~~~~~~~~~~~
In file included from /home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1306:
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:1192:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1192 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_macbook_exec_verb’:
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:2346:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2346 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_imac_exec_verb’:
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:2392:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2392 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
In file included from /home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:3024,
                 from /home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1306:
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_new84.h: In function ‘get_hda_cvt_setup_apple_8409’:
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_new84.h:659:3: error: label at end of compound statement
  659 |   default:
      |   ^~~~~~~
make[2]: *** [scripts/Makefile.build:285: /home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o] Error 1
make[1]: *** [Makefile:1875: /home/landon/snd_hda_macbookpro/build/hda-5.15.0] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-33-generic'
make: *** [Makefile:22: all] Error 2
#cp snd-hda-codec-cs8409.ko /lib/modules/5.15.0-33-generic/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.15.0-33-generic/build M=/home/landon/snd_hda_macbookpro/build/hda-5.15.0 modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-33-generic'
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
cat: /home/landon/snd_hda_macbookpro/build/hda-5.15.0/modules.order: No such file or directory
  DEPMOD  /lib/modules/5.15.0-33-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-33-generic'
depmod -a

contents of /lib/modules/5.15.0-33-generic/updates
total 0

```

davidjo, 2022-06-29:
The script is downloading the mainline kernel which is different from Ubuntu kernels - hence the patch fails and then the compile fails because variables are missing.
So I guess if you cat /etc/os-release it says Mint not Ubuntu?
If you look in the install script there is a line
# we need to handle Ubuntu based distributions eg Mint here
this is a hint this needs fixing for Mint.
copy the 3 lines from (and including) the if and change the Ubuntu name to whatever name you see in /etc/os-release.
This should switch it to use the Ubuntu kernel source and not mainline (ie isubuntu needs to be set to 1).

Ven0mstrike42, 2022-06-29:
I forced it to use the ubuntu selection by setting isubuntu to 1 manually. Still fails at patch_cirrus_new84.h:659. which segment do i need to remove?

Ven0mstrike42, 2022-06-29:
Latest output
```
landon@Mintbook:~/snd_hda_macbookpro$ sudo ./install.cirrus.driver.sh 
linux-source-5.15.0/sound/pci/hda/
linux-source-5.15.0/sound/pci/hda/hda_eld.c
linux-source-5.15.0/sound/pci/hda/patch_ca0110.c
linux-source-5.15.0/sound/pci/hda/hda_bind.c
linux-source-5.15.0/sound/pci/hda/hda_tegra.c
linux-source-5.15.0/sound/pci/hda/patch_si3054.c
linux-source-5.15.0/sound/pci/hda/hda_jack.c
linux-source-5.15.0/sound/pci/hda/patch_cirrus.c
linux-source-5.15.0/sound/pci/hda/patch_ca0132.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.c
linux-source-5.15.0/sound/pci/hda/hda_generic.h
linux-source-5.15.0/sound/pci/hda/hda_intel_trace.h
linux-source-5.15.0/sound/pci/hda/hda_local.h
linux-source-5.15.0/sound/pci/hda/hda_beep.h
linux-source-5.15.0/sound/pci/hda/patch_via.c
linux-source-5.15.0/sound/pci/hda/patch_conexant.c
linux-source-5.15.0/sound/pci/hda/hda_controller.h
linux-source-5.15.0/sound/pci/hda/hda_proc.c
linux-source-5.15.0/sound/pci/hda/patch_analog.c
linux-source-5.15.0/sound/pci/hda/hda_sysfs.c
linux-source-5.15.0/sound/pci/hda/hda_controller_trace.h
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.h
linux-source-5.15.0/sound/pci/hda/hda_codec.c
linux-source-5.15.0/sound/pci/hda/ideapad_s740_helper.c
linux-source-5.15.0/sound/pci/hda/hda_intel.c
linux-source-5.15.0/sound/pci/hda/ca0132_regs.h
linux-source-5.15.0/sound/pci/hda/patch_cs8409-tables.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.c
linux-source-5.15.0/sound/pci/hda/hda_beep.c
linux-source-5.15.0/sound/pci/hda/hda_jack.h
linux-source-5.15.0/sound/pci/hda/hda_intel.h
linux-source-5.15.0/sound/pci/hda/thinkpad_helper.c
linux-source-5.15.0/sound/pci/hda/patch_cmedia.c
linux-source-5.15.0/sound/pci/hda/patch_hdmi.c
linux-source-5.15.0/sound/pci/hda/hda_generic.c
linux-source-5.15.0/sound/pci/hda/hda_component.h
linux-source-5.15.0/sound/pci/hda/Makefile
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.h
linux-source-5.15.0/sound/pci/hda/Kconfig
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_spi.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.h
linux-source-5.15.0/sound/pci/hda/hp_x360_helper.c
linux-source-5.15.0/sound/pci/hda/hda_hwdep.c
linux-source-5.15.0/sound/pci/hda/patch_sigmatel.c
linux-source-5.15.0/sound/pci/hda/hda_controller.c
linux-source-5.15.0/sound/pci/hda/patch_realtek.c
patching file patch_cs8409.c
Hunk #1 succeeded at 1277 (offset 7 lines).
Hunk #2 succeeded at 1286 (offset 7 lines).
Hunk #3 succeeded at 1315 (offset 7 lines).
patching file patch_cs8409.h
Hunk #2 succeeded at 306 (offset 5 lines).
Hunk #3 succeeded at 333 (offset 5 lines).
Hunk #4 succeeded at 368 (offset 5 lines).
make -C /lib/modules/5.15.0-33-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/landon/snd_hda_macbookpro/build/hda-5.15.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-33-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
  You are using:           gcc (Ubuntu 9.4.0-5ubuntu1) 9.4.0
  CC [M]  /home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o
In file included from /home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1321:
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:1192:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1192 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
In file included from /home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:3024,
                 from /home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1321:
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_new84.h: In function ‘get_hda_cvt_setup_apple_8409’:
/home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_new84.h:659:3: error: label at end of compound statement
  659 |   default:
      |   ^~~~~~~
make[2]: *** [scripts/Makefile.build:285: /home/landon/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o] Error 1
make[1]: *** [Makefile:1875: /home/landon/snd_hda_macbookpro/build/hda-5.15.0] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-33-generic'
make: *** [Makefile:22: all] Error 2
#cp snd-hda-codec-cs8409.ko /lib/modules/5.15.0-33-generic/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.15.0-33-generic/build M=/home/landon/snd_hda_macbookpro/build/hda-5.15.0 modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-33-generic'
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
cat: /home/landon/snd_hda_macbookpro/build/hda-5.15.0/modules.order: No such file or directory
  DEPMOD  /lib/modules/5.15.0-33-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-33-generic'
depmod -a

contents of /lib/modules/5.15.0-33-generic/updates
total 0

Ven0mstrike42, 2022-06-29:
got it working. thanks for the help and all you do.

davidjo, 2022-06-30:
Can you list the NAME=  line in /etc/os-release for Mint.
Is it just Mint?
If so Ill fix the build script to handle this.

Ven0mstrike42, 2022-06-30:
```
NAME= "Linux Mint"

Ven0mstrike42, 2022-06-30:
I can't speak for other Ubuntu based distributions but the line ID_LIKE may be a useful identifier for setting isubuntu as well. 
```
NAME="Linux Mint"
VERSION="20.3 (Una)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 20.3"
VERSION_ID="20.3"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=una
UBUNTU_CODENAME=focal

davidjo, 2022-07-01:
Thanks for the update - yes using ID_LIKE is a good idea - of course just checked my plain Ubuntu system and its ID_LIKE is debian just to make life interesting.