davidjo/snd_hda_macbookpro #53: Macbook Pro 13.1 and various distros

thering1975TG, 2022-07-04:
Hi

Im mainly a linux server user rather than desktop, but seeing as Apple ending OS updates now for this model, i thought i would try linux in dual boot for now with the intention of fully installing and removing macos.

However regardless of the numerous distros i have tried and different mix of kernels, i cannot get the internal speakers to work. Im presuming its the newer distro releases or kernels that are giving the grief rather than the patch but this is my experience so far:

- Ubuntu 22.04 stock kernel - Audio Device shows as Dummy Output - Patch failed with folder location erros, only tried once
- Zorrin OS 16 Stock kernel and Liquorix - Audio device shows as Dummy Output Patch completed but no ko file in updates folder
- Deepin Latest Stock Kernel - This one showed Intel PCH Speakers as option but no sound, Patch completed but no ko file in updates
- Fedora 36 5.18 / 5.17 /5.16 kernel - Audio shows as dummy output patch completes fine and ko file located in update folder for each kernel tested.

Actually quite liking fedora 36 at the moment, however Bluetooth audio, usbc headphones and HDMI sound all work fine, however checking in terminal it only sees the HDMI outputs and no internal speakers.

TO be honest its fun playing and testing, be a shame to give up the macbook but worst case i will just pick up a linux compatible laptop instead

carrelebo, 2022-07-05:
@thering1975TG for the new update the driver work with the new 1.18 linux in kali linux (install the distribution with "weakly image" option "https://cdimage.kali.org/kali-images/kali-weekly/kali-linux-2022-W27-live-amd64.iso" and update it), enter good all command and before install the installation of the driver, install the dependences. 

if you're in multibooting systems install the distrib with these 3 parition :
- BTRFS or  Ext4 with "/" montage point
- swap 2-8 Gb
- and Fat32 partition with "/boot/efi" montage point

after the install you have any problems with other OS install on your system.

thering1975TG, 2022-07-05:
Thanks will try when I get home have used kali before so will try tonight

thering1975TG, 2022-07-05:
Okay So tried Kali and have to say gave up. I use Kali for pentesting so have it running in a VM anyway, however Kali was the most awful experience of all the distros so far. For example the installer was the only one which did not recognise the trackpad or keyboard, and i do not have a spare keyboard without dismantling my hackintosh wire management lol which im not going to do (took ages to hide everything) so no way of installing it.

So i tried Manjaro and this was the results:

-Manjaro with kernel 5.15 5.18 and 5.19 Audio hardware shows as dummy output, patch installs fine but no sound

- Manjaro with Kernel 5.10 Hardware shows as Intel speakers with ability to select 2.0, 2.1, 4.1 patch installs okay but still no sound

Thanks for the advice on trying Kali do appreciate it, just did not play ball for my machine, but i guess a good excuse to get a starlab machine which i know will work and just sell off the MBP

But it's been a fun journey all the same and loved trying all the different flavours so thats a positive

davidjo, 2022-07-06:
Ubuntu 22.04 stock kernel - Audio Device shows as Dummy Output - Patch failed with folder location erros, only tried once

This doesnt make sense - I run Ubuntu 22.04 and develop against this so I know it works on Ubuntu 22.04.
What specifically are the folder location errors.

thering1975TG, 2022-07-06:
I will install ubuntu 22.04 again tonight and log the patching issues

Last night on fedora I did log the error which on that distro was ssl errors.

Once I have reinstalled I would report back with info

thering1975TG, 2022-07-06:
Okay reinstalled ubuntu 22.04

The error when following the guide was not folder location but due to the fact the kernel source was not downloaded so got error

Ubuntu linux kernel source not found in /usr/

However downloaded kernel source and it completed fine. I should learn to read and not see the obvious lol

Anyways after a reboot it does indeed work on ubuntu takes about 1 min for the sound hardware to appear and then seems to work

davidjo, 2022-07-06:
Ah - yes you do need the kernel sources - which didnt really want to do auto on ubuntu.
I should add a msg to download sources in the script.

Glad it works.

carrelebo, 2022-07-06:
@thering1975TG 
You can switch the sound card with alsamixer in the case of there are the sound who not work

leifliddy, 2022-07-30:
the reason you're seeing these ssl errors in Fedora 
```
make INSTALL_MOD_DIR=updates -C /lib/modules/5.18.13-200.fc36.x86_64/build M=/root/snd_hda_macbookpro/build/hda-5.18.13 modules_install
make[1]: Entering directory '/usr/src/kernels/5.18.13-200.fc36.x86_64'
  INSTALL /lib/modules/5.18.13-200.fc36.x86_64/updates/snd-hda-codec-cs8409.ko
  SIGN    /lib/modules/5.18.13-200.fc36.x86_64/updates/snd-hda-codec-cs8409.ko
At main.c:160:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
sign-file: ./certs/signing_key.pem: No such file or directory
  DEPMOD  /lib/modules/5.18.13-200.fc36.x86_64
make[1]: Leaving directory '/usr/src/kernels/5.18.13-200.fc36.x86_64'
depmod -a
```
Is due to the use of ```modules_install``` in the Makefile and the ```CONFIG_MODULE_SIG_ALL=y``` kernel option. 
Anytime ```make modules_install``` is run it'll try to sign the module(s) with ```./certs/signing_key.pem``` which doesn't exist.