davidjo/snd_hda_macbookpro #55: add patch to package listings

leifliddy, 2022-07-30:
Please add the ```patch``` package to the install package instructions. 
ie
```dnf install wget make gcc kernel-devel patch```

Also it might be a good idea to use ```set -e``` at the top of your install script
```
#!/bin/bash

set -e

# NOTA BENE - this script should be run as root
```
So that if any errors are encountered -- like if you don't have the ```patch``` binary installed for example, then the script will exit upon that command hitting a non-zero exit. 

```
linux-5.18.13/sound/pci/hda/patch_sigmatel.c
linux-5.18.13/sound/pci/hda/patch_via.c
linux-5.18.13/sound/pci/hda/thinkpad_helper.c
./install.cirrus.driver.sh: line 142: patch: command not found
```

otherwise it'll keep right on going
```
linux-5.18.13/sound/pci/hda/patch_sigmatel.c
linux-5.18.13/sound/pci/hda/patch_via.c
linux-5.18.13/sound/pci/hda/thinkpad_helper.c
./install.cirrus.driver.sh: line 141: patch: command not found
./install.cirrus.driver.sh: line 143: patch: command not found
make -C /lib/modules/5.18.13-200.fc36.x86_64/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/root/snd_hda_macbookpro/build/hda-5.18.13 modules
make[1]: Entering directory '/usr/src/kernels/5.18.13-200.fc36.x86_64'
  CC [M]  /root/snd_hda_macbookpro/build/hda-5.18.13/patch_cs8409.o
```

davidjo, 2022-07-31:
Thanks for suggestion - I think Ill have to do this - never thought you wouldnt have patch if you were building kernel modules.

leifliddy, 2022-07-31:
Just a (minor) suggestion. I might use that with my project as well. Sounds works perfectly on my 14,1 MBP running Fedora 36 w/ kernel 5.18.x. Nice!!

closed 2022-07-31 by leifliddy