davidjo/snd_hda_macbookpro #56: How to avoid stereo duplication / improve audio quality by replicating CoreAudio filtering

JayFoxRox, 2022-08-04:
This commit talks about stereo duplication: https://github.com/davidjo/snd_hda_macbookpro/commit/063110b132b69ea8e6223d3c6870df47137ccacb and this is also mentioned in the README changes in https://github.com/davidjo/snd_hda_macbookpro/commit/6640c79c061b7596b01901cf3f581af94cb9a557

I wonder if this (current-state) is enough for good audio, or if it's possible to get full access to the speaker array to do something like https://github.com/chadmed/asahi-audio (the author also wants to find a more generic plug-and-play solution: https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/2210).

davidjo, 2022-08-04:
The auto duplication only occurs if you set the audio to the stereo output (for ubuntu in Audio Settings).
You should also see a 4 channel surround output - that requires a 4 channel stream if you set audio output to that.
All you have to do is create the 4 channel stream.
(Note that unlike the Apple way this is a left/right tweeter left/right woofer order).
(Apples is left tweeter/left woofer/right tweeter/right woofer).
If you look thro the issues here (possibly closed issues) various attempts at generating 4 channel streams from pulse or alsa have been tried ie actual user configuration scripts have been attempted - you shouldnt need to change the actual kernel driver.
(Unfortunately Ubuntu 22.04 is not using pipewire  - next non-LTS version will I believe).

slynobody, 2022-09-26:
I got working 4-channel sound on an Imac qualitywise near macos with these changes:

* /usr/share/alsa-card-profile/mixer/profile-sets/default.conf 
  change 'channel-map' under [Mapping analog-surround-40] 
  front-left,front-right,rear-left,rear-right > front-left,rear-left,rear-right,front-right
* /usr/share/pipewire/pipewire-pulse.conf
changes to stream.properties = {
    #node.latency          = 1024/48000
    node.autoconnect      = true
    resample.quality      = 4
    #channelmix.normalize  = true
    channelmix.mix-lfe    = true
    channelmix.upmix      = true
    channelmix.upmix-method = psd  # none, simple
    #channelmix.lfe-cutoff = 150
    #channelmix.fc-cutoff  = 12000
    channelmix.rear-delay = 12.0
    channelmix.stereo-widen = 0.2
    #channelmix.hilbert-taps = 0
    #dither.noise = 0
}
then restart according services or reboot and change audio to 'surround 4.0' 

it even works very nice with wine if you enable '4-channel-audio' in wine-settings (if the 4.0-device is visible / used)
----
what does not work yet is
* kernel 5.19-support (only through changes in scripts)
* kernel 6-support
* working headphone (very, very low quality; crash when headphone inserted at login-time

davidjo, 2022-09-27:
Nice work.
Headphones need to switch back to 2 channel - the headphone audio path is separate from the speaker path so any downmixing needs doing to be done in pipewire.