davidjo/snd_hda_macbookpro #59: Arch Linux with kernel 5.19.1-arch2-1 build errors

seyyidibrahimgulec, 2022-08-14:
Attempting to build is throwing errors. Is it a similar problem with #52?

*OS:* `ArchLinux x86_64`
*Host:* `MacBookPro14,1 1.0`
*Kernel:* `5.19.1-arch2-1`

```bash
λ  sudo ./install.cirrus.driver.sh
[sudo] password for ibrahim:
--2022-08-14 13:31:11--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.19.1.tar.xz
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving cdn.kernel.org (cdn.kernel.org)... 199.232.17.176, 2a04:4e42:1b::432
Connecting to cdn.kernel.org (cdn.kernel.org)|199.232.17.176|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

linux-5.19.1/sound/pci/hda/
linux-5.19.1/sound/pci/hda/Kconfig
linux-5.19.1/sound/pci/hda/Makefile
linux-5.19.1/sound/pci/hda/ca0132_regs.h
linux-5.19.1/sound/pci/hda/cs35l41_hda.c
linux-5.19.1/sound/pci/hda/cs35l41_hda.h
linux-5.19.1/sound/pci/hda/cs35l41_hda_i2c.c
linux-5.19.1/sound/pci/hda/cs35l41_hda_spi.c
linux-5.19.1/sound/pci/hda/hda_auto_parser.c
linux-5.19.1/sound/pci/hda/hda_auto_parser.h
linux-5.19.1/sound/pci/hda/hda_beep.c
linux-5.19.1/sound/pci/hda/hda_beep.h
linux-5.19.1/sound/pci/hda/hda_bind.c
linux-5.19.1/sound/pci/hda/hda_codec.c
linux-5.19.1/sound/pci/hda/hda_component.h
linux-5.19.1/sound/pci/hda/hda_controller.c
linux-5.19.1/sound/pci/hda/hda_controller.h
linux-5.19.1/sound/pci/hda/hda_controller_trace.h
linux-5.19.1/sound/pci/hda/hda_eld.c
linux-5.19.1/sound/pci/hda/hda_generic.c
linux-5.19.1/sound/pci/hda/hda_generic.h
linux-5.19.1/sound/pci/hda/hda_hwdep.c
linux-5.19.1/sound/pci/hda/hda_intel.c
linux-5.19.1/sound/pci/hda/hda_intel.h
linux-5.19.1/sound/pci/hda/hda_intel_trace.h
linux-5.19.1/sound/pci/hda/hda_jack.c
linux-5.19.1/sound/pci/hda/hda_jack.h
linux-5.19.1/sound/pci/hda/hda_local.h
linux-5.19.1/sound/pci/hda/hda_proc.c
linux-5.19.1/sound/pci/hda/hda_sysfs.c
linux-5.19.1/sound/pci/hda/hda_tegra.c
linux-5.19.1/sound/pci/hda/hp_x360_helper.c
linux-5.19.1/sound/pci/hda/ideapad_s740_helper.c
linux-5.19.1/sound/pci/hda/patch_analog.c
linux-5.19.1/sound/pci/hda/patch_ca0110.c
linux-5.19.1/sound/pci/hda/patch_ca0132.c
linux-5.19.1/sound/pci/hda/patch_cirrus.c
linux-5.19.1/sound/pci/hda/patch_cmedia.c
linux-5.19.1/sound/pci/hda/patch_conexant.c
linux-5.19.1/sound/pci/hda/patch_cs8409-tables.c
linux-5.19.1/sound/pci/hda/patch_cs8409.c
linux-5.19.1/sound/pci/hda/patch_cs8409.h
linux-5.19.1/sound/pci/hda/patch_hdmi.c
linux-5.19.1/sound/pci/hda/patch_realtek.c
linux-5.19.1/sound/pci/hda/patch_si3054.c
linux-5.19.1/sound/pci/hda/patch_sigmatel.c
linux-5.19.1/sound/pci/hda/patch_via.c
linux-5.19.1/sound/pci/hda/thinkpad_helper.c
patching file patch_cs8409.c
Hunk #1 succeeded at 1444 (offset 174 lines).
Hunk #2 succeeded at 1453 (offset 174 lines).
Hunk #3 succeeded at 1482 (offset 174 lines).
patching file patch_cs8409.h
Hunk #1 succeeded at 19 (offset 1 line).
Hunk #2 succeeded at 303 (offset 2 lines).
Hunk #3 FAILED at 328.
Hunk #4 succeeded at 363 (offset 3 lines).
1 out of 4 hunks FAILED -- saving rejects to file patch_cs8409.h.rej
make -C /lib/modules/5.19.1-arch2-1/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1 modules
make[1]: Entering directory '/usr/lib/modules/5.19.1-arch2-1/build'
  CC [M]  /home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1/patch_cs8409.o
In file included from /home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1/patch_cs8409.c:1488:
/home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h:157:10: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
  157 |         .linein_jack_in = 0,
      |          ^~~~~~~~~~~~~~
      |          mic_jack_in
/home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h:158:10: error: ‘struct sub_codec’ has no member named ‘force_status_change’
  158 |         .force_status_change = 1,
      |          ^~~~~~~~~~~~~~~~~~~
/home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h:1192:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1192 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
/home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_macbook_exec_verb’:
/home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h:2346:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2346 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
/home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_imac_exec_verb’:
/home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h:2392:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2392 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
make[2]: *** [scripts/Makefile.build:249: /home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1/patch_cs8409.o] Error 1
make[1]: *** [Makefile:1843: /home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.19.1-arch2-1/build'
make: *** [Makefile:22: all] Error 2
#cp snd-hda-codec-cs8409.ko /lib/modules/5.19.1-arch2-1/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.19.1-arch2-1/build M=/home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1 modules_install
make[1]: Entering directory '/usr/lib/modules/5.19.1-arch2-1/build'
cat: /home/ibrahim/snd_hda_macbookpro/build/hda-5.19.1/modules.order: No such file or directory
  DEPMOD  /lib/modules/5.19.1-arch2-1
make[1]: Leaving directory '/usr/lib/modules/5.19.1-arch2-1/build'
depmod -a

contents of /lib/modules/5.19.1-arch2-1/updates
total 712
-rw-r--r-- 1 root root 725768 Aug 14 08:45 hci_uart.ko
```

davidjo, 2022-08-14:
Not quite - I think Arch is a plain linux kernel distribution (unlike Ubuntu and Ubuntu based distributions such as Mint which uses a significantly modified kernel for similar kernel versions which is what 52 was about).
Yours seem to be a specific change in the base kernel source in 5.19 vs 5.15 in which the existing force_status_change variable seems to have been removed (this is not something I added).
However the current patch file patch_cs8409.h contains this variable as context in hunk 3 so the patch hunk fails.
Just add the linein_jack_in:1 line under the mic_jack_in:1 line should fix the problem.

seyyidibrahimgulec, 2022-08-14:
Thank you for your quick informative response. I will do as you say and let you know if that works.

seyyidibrahimgulec, 2022-08-15:
As you said, `force_status_change` variable deletion was causing them to patch failed. I removed the following lines from the code, and everything works fine now. Thank you again. 

https://github.com/davidjo/snd_hda_macbookpro/blob/bcc755abe10efa5852c465a4b50f84132c0431d1/patch_cirrus/patch_cirrus_apple.h#L158
https://github.com/davidjo/snd_hda_macbookpro/blob/bcc755abe10efa5852c465a4b50f84132c0431d1/patch_cirrus/patch_cs8409.h#L334
https://github.com/davidjo/snd_hda_macbookpro/blob/bcc755abe10efa5852c465a4b50f84132c0431d1/patch_patch_cs8409.h.diff#L430

slynobody, 2022-08-17:
described solution does not work here on debian (also 19.1)? could you supply a patch with the needed changes, maybe i am doing something wrong with this 4 steps?

effect (part):
patch_cirrus_new84.h:1236:72: error: ‘struct cs8409_spec’ has no member named ‘unsol_list’
 1236 |                 list_for_each_entry_safe(unsol_entry, unsol_temp, &spec->unsol_list, list)
      |                                                                        ^~
./include/linux/container_of.h:18:33: note: in definition of macro ‘container_of’
   18 |         void *__mptr = (void *)(ptr);                                   \
      |                                 ^~~
./include/linux/list.h:531:9: note: in expansion of macro ‘list_entry’
  531 |         list_entry((ptr)->next, type, member)
      |         ^~~~~~~~~~
./include/linux/list.h:761:20: note: in expansion of macro ‘list_first_entry’
  761 |         for (pos = list_first_entry(head, typeof(*pos), member),        \
      |                    ^~~~~~~~~~~~~~~~
/mnt/Data/current/bin/snd/hda/patch_cirrus_new84.h:1236:17: note: in expansion of macro ‘list_for_each_entry_safe’
 1236 |                 list_for_each_entry_safe(unsol_entry, unsol_temp, &spec->unsol_list, list)
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./include/linux/bits.h:22,
                 from ./include/linux/ratelimit_types.h:5,
                 from ./include/linux/printk.h:9,
                 from ./include/asm-generic/bug.h:22,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from ./include/linux/gfp.h:5:
/mnt/Data/current/bin/snd/hda/patch_cirrus_new84.h:1236:72: error: ‘struct cs8409_spec’ has no member named ‘unsol_list’
 1236 |                 list_for_each_entry_safe(unsol_entry, unsol_temp, &spec->unsol_list, list)
      |                                                                        ^~
(...)

before, without the changes mentioned here:

patch_cirrus_apple.h:157:10: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
  157 |         .linein_jack_in = 0,
      |          ^~~~~~~~~~~~~~
      |          mic_jack_in
snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h:158:10: error: ‘struct sub_codec’ has no member named ‘force_status_change’
  158 |         .force_status_change = 1,
      |          ^~~~~~~~~~~~~~~~~~~
snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h:1192:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1192 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_macbook_exec_verb’:
snd/snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h:2346:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2346 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_imac_exec_verb’:
snd/snd_hda_macbookpro/build/hda-5.19.1/patch_cirrus_apple.h:2392:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2392 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in

Thank you for your work!

francescodesu, 2022-08-21:
> described solution does not work here on debian (also 19.1)? could you supply a patch with the needed changes, maybe i am doing something wrong with this 4 steps?
>
> effect (part): [...]

Same on Manjaro!

EimaMei, 2022-08-21:
> described solution does not work here on debian (also 19.1)? could you supply a patch with the needed changes, maybe i am doing something wrong with this 4 steps?
>
> effect (part): [...]

I get the same error spam too after doing the steps. I am on the `5.19.1-3-MANJARO` kernel

davidjo, 2022-08-21:
Im afraid I dont chase kernels - I just run Ubuntu LTS. 

Im not sure whats going on but some patch updates seem to be being missed which  gives the error
has no member named ‘unsol_list’ (this is a variable I add).

you need to look at patch_patch_cs8409.h.diff
1st run the install script which will fail because the patch of patch_cs8409.h with patch_patch_cs8409.h.diff fails.
hunk3 will fail because of changes in patch_cs8409.h ie removal of original kernel variable focus_status change.
look at hunk3 (the patch part starting @@ -305,6 +328,9 @@ struct sub_codec {  in patch_patch_cs8409.h.diff)
manually edit the file  patch_cs8409.h and add the lines:
#ifdef APPLE_CODECS
       unsigned int linein_jack_in:1;
#endif
after the line unsigned int mic_jack_in:1;
for file patch_cs8409.h

do not rerun the install script
manually run the commands 
make (or sudo make if permission issues)
then if ok 
sudo make install

EimaMei, 2022-08-21:
> Im afraid I dont chase kernels - I just run Ubuntu LTS.
> 
> Im not sure whats going on but some patch updates seem to be being missed which gives the error has no member named ‘unsol_list’ (this is a variable I add).
> 
> you need to look at patch_patch_cs8409.h.diff 1st run the install script which will fail because the patch of patch_cs8409.h with patch_patch_cs8409.h.diff fails. hunk3 will fail because of changes in patch_cs8409.h ie removal of original kernel variable focus_status change. look at hunk3 (the patch part starting @@ -305,6 +328,9 @@ struct sub_codec { in patch_patch_cs8409.h.diff) manually edit the file patch_cs8409.h and add the lines: #ifdef APPLE_CODECS unsigned int linein_jack_in:1; #endif after the line unsigned int mic_jack_in:1; for file patch_cs8409.h
> 
> do not rerun the install script manually run the following commands in the build/hda-.... directory (where ... is part of kernel version) make (or sudo make if permission issues) then if ok sudo make install

I tried this out and audio works perfectly now! Thank you very much for the help

asutoshvariar, 2022-09-02:
Tried using the instructions on Fedora 36 (kernel 5.19.6) but didn't work. Edited both the patch_cs8409 files.
Error log (sorry for the size, not sure what's important):
[error.txt](https://github.com/davidjo/snd_hda_macbookpro/files/9479774/error.txt)

davidjo, 2022-09-03:
What seems to be happening is that the existing kernel file patch_cs8409.h has not been updated in your compile or maybe the -DAPPLE_CODECS is being ignored.
All the errors are variables I added to struct cs8409_spec not existing or other additions under APPLE_CODECS ifdef eg the missing struct hda_cvt_setup_apple definition.

Drezzler, 2022-09-03:
Having same problem in Ubuntu 20.04 after latest update. 

Have  tried the solution mentioned by @seyyidibrahimgulec , that seems to prop up even more errors.

Is their a way to solve this ?

noispirit, 2022-09-04:
> Having same problem in Ubuntu 20.04 after latest update.
> 
> Have tried the solution mentioned by @seyyidibrahimgulec , that seems to prop up even more errors.
> 
> @slynobody - did you find a solution ?

You must erase those lines inside ./build/hda-[kernel version] patch_cirrus.h file, also add manually what @davidjo said..

Although I have sound, mic is dead now. Seems like something went wrong with the last kernel update in Ubuntu 22.04.1