davidjo/snd_hda_macbookpro #62: Audio ouput not work after doing a Ubuntu update

orklann, 2022-09-05:
Hi, I can make this driver works after install a fresh Ubuntu 22.04.1, and just now, I apply a Ubuntu update, and after restarting Ubuntu, the audio output not work, there is only one output which is "Dummy Output", is there anything to fix it?

orklann, 2022-09-05:
I rebuild and install the driver, after update Ubuntu, and get errors.

`sudo ./install.cirrus.driver.sh `

```
linux-source-5.15.0/sound/pci/hda/
linux-source-5.15.0/sound/pci/hda/hda_intel.h
linux-source-5.15.0/sound/pci/hda/hda_component.h
linux-source-5.15.0/sound/pci/hda/hda_local.h
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.c
linux-source-5.15.0/sound/pci/hda/patch_conexant.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.c
linux-source-5.15.0/sound/pci/hda/hp_x360_helper.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.h
linux-source-5.15.0/sound/pci/hda/thinkpad_helper.c
linux-source-5.15.0/sound/pci/hda/hda_generic.c
linux-source-5.15.0/sound/pci/hda/ca0132_regs.h
linux-source-5.15.0/sound/pci/hda/hda_controller_trace.h
linux-source-5.15.0/sound/pci/hda/hda_codec.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409-tables.c
linux-source-5.15.0/sound/pci/hda/hda_jack.c
linux-source-5.15.0/sound/pci/hda/hda_controller.c
linux-source-5.15.0/sound/pci/hda/hda_controller.h
linux-source-5.15.0/sound/pci/hda/patch_cirrus.c
linux-source-5.15.0/sound/pci/hda/Kconfig
linux-source-5.15.0/sound/pci/hda/Makefile
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_spi.c
linux-source-5.15.0/sound/pci/hda/hda_sysfs.c
linux-source-5.15.0/sound/pci/hda/patch_si3054.c
linux-source-5.15.0/sound/pci/hda/ideapad_s740_helper.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.h
linux-source-5.15.0/sound/pci/hda/patch_cmedia.c
linux-source-5.15.0/sound/pci/hda/hda_eld.c
linux-source-5.15.0/sound/pci/hda/hda_intel.c
linux-source-5.15.0/sound/pci/hda/hda_generic.h
linux-source-5.15.0/sound/pci/hda/patch_hdmi.c
linux-source-5.15.0/sound/pci/hda/patch_ca0110.c
linux-source-5.15.0/sound/pci/hda/patch_sigmatel.c
linux-source-5.15.0/sound/pci/hda/patch_via.c
linux-source-5.15.0/sound/pci/hda/hda_hwdep.c
linux-source-5.15.0/sound/pci/hda/patch_ca0132.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-source-5.15.0/sound/pci/hda/hda_beep.h
linux-source-5.15.0/sound/pci/hda/hda_tegra.c
linux-source-5.15.0/sound/pci/hda/patch_realtek.c
linux-source-5.15.0/sound/pci/hda/hda_beep.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.c
linux-source-5.15.0/sound/pci/hda/hda_intel_trace.h
linux-source-5.15.0/sound/pci/hda/hda_bind.c
linux-source-5.15.0/sound/pci/hda/hda_jack.h
linux-source-5.15.0/sound/pci/hda/hda_proc.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.h
linux-source-5.15.0/sound/pci/hda/patch_analog.c
patching file patch_cs8409.c
Hunk #1 succeeded at 1442 (offset 172 lines).
Hunk #2 succeeded at 1451 (offset 172 lines).
Hunk #3 succeeded at 1480 (offset 172 lines).
patching file patch_cs8409.h
Hunk #1 succeeded at 19 (offset 1 line).
Hunk #2 succeeded at 302 (offset 1 line).
Hunk #3 FAILED at 328.
Hunk #4 succeeded at 362 (offset 2 lines).
1 out of 4 hunks FAILED -- saving rejects to file patch_cs8409.h.rej
make -C /lib/modules/5.15.0-47-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/rkt/projects/snd_hda_macbookpro/build/hda-5.15.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-47-generic'
  CC [M]  /home/rkt/projects/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o
In file included from /home/rkt/projects/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1486:
/home/rkt/projects/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:157:10: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
  157 |         .linein_jack_in = 0,
      |          ^~~~~~~~~~~~~~
[... 12 lines cut ...]
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
/home/rkt/projects/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_imac_exec_verb’:
/home/rkt/projects/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:2392:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2392 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
make[2]: *** [scripts/Makefile.build:297: /home/rkt/projects/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o] Error 1
make[1]: *** [Makefile:1881: /home/rkt/projects/snd_hda_macbookpro/build/hda-5.15.0] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-47-generic'
make: *** [Makefile:22: all] Error 2
#cp snd-hda-codec-cs8409.ko /lib/modules/5.15.0-47-generic/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.15.0-47-generic/build M=/home/rkt/projects/snd_hda_macbookpro/build/hda-5.15.0 modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-47-generic'
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
cat: /home/rkt/projects/snd_hda_macbookpro/build/hda-5.15.0/modules.order: No such file or directory
  DEPMOD  /lib/modules/5.15.0-47-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-47-generic'
depmod -a

contents of /lib/modules/5.15.0-47-generic/updates
total 0

``

orklann, 2022-09-05:
It seems the driver did not build due to the errors I have.

orklann, 2022-09-05:
I just followed your instruction in another issue [here](https://github.com/davidjo/snd_hda_macbookpro/issues/59#issuecomment-1221629846), which fixed the errors, and reinstall the driver, it now works fine.

orklann, 2022-09-05:
Thank you for your hard works.

closed 2022-09-05 by orklann

stephan1827, 2024-02-14:
I have the same issue and checked the fix but it seems like the code changes are already in there. Do I miss something here?

```
  unsigned int hp_jack_in:1;
  unsigned int mic_jack_in:1;
  #ifdef APPLE_CODECS
	  unsigned int linein_jack_in:1;
  #endif
  unsigned int suspended:1;
  unsigned int paged:1;
```

stephan1827, 2024-02-16:
I did a fresh install with mint and now it works