davidjo/snd_hda_macbookpro #63: Install script not working anymore

martroben, 2022-09-05:
I've been re-installing the driver periodically, since somehow my system always drops back to "dummy output" after apt upgrades. It has always worked well.
Now I got an unexpected "dummy output" just from connect headphones. (The same headphones have always worked fine before.)

I rebooted, did a fresh clone of the repository and attempted to re-install the driver. Alas, it doesn't seem to work anymore. Here's the output:

```
user@user-MacBookPro:/tmp/snd_hda_macbookpro$ sudo ./install.cirrus.driver.sh
[sudo] password for user: 
linux-source-5.15.0/sound/pci/hda/
linux-source-5.15.0/sound/pci/hda/hda_intel.h
linux-source-5.15.0/sound/pci/hda/hda_component.h
linux-source-5.15.0/sound/pci/hda/hda_local.h
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.c
linux-source-5.15.0/sound/pci/hda/patch_conexant.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.c
linux-source-5.15.0/sound/pci/hda/hp_x360_helper.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.h
linux-source-5.15.0/sound/pci/hda/thinkpad_helper.c
linux-source-5.15.0/sound/pci/hda/hda_generic.c
linux-source-5.15.0/sound/pci/hda/ca0132_regs.h
linux-source-5.15.0/sound/pci/hda/hda_controller_trace.h
linux-source-5.15.0/sound/pci/hda/hda_codec.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409-tables.c
linux-source-5.15.0/sound/pci/hda/hda_jack.c
linux-source-5.15.0/sound/pci/hda/hda_controller.c
linux-source-5.15.0/sound/pci/hda/hda_controller.h
linux-source-5.15.0/sound/pci/hda/patch_cirrus.c
linux-source-5.15.0/sound/pci/hda/Kconfig
linux-source-5.15.0/sound/pci/hda/Makefile
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_spi.c
linux-source-5.15.0/sound/pci/hda/hda_sysfs.c
linux-source-5.15.0/sound/pci/hda/patch_si3054.c
linux-source-5.15.0/sound/pci/hda/ideapad_s740_helper.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.h
linux-source-5.15.0/sound/pci/hda/patch_cmedia.c
linux-source-5.15.0/sound/pci/hda/hda_eld.c
linux-source-5.15.0/sound/pci/hda/hda_intel.c
linux-source-5.15.0/sound/pci/hda/hda_generic.h
linux-source-5.15.0/sound/pci/hda/patch_hdmi.c
linux-source-5.15.0/sound/pci/hda/patch_ca0110.c
linux-source-5.15.0/sound/pci/hda/patch_sigmatel.c
linux-source-5.15.0/sound/pci/hda/patch_via.c
linux-source-5.15.0/sound/pci/hda/hda_hwdep.c
linux-source-5.15.0/sound/pci/hda/patch_ca0132.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-source-5.15.0/sound/pci/hda/hda_beep.h
linux-source-5.15.0/sound/pci/hda/hda_tegra.c
linux-source-5.15.0/sound/pci/hda/patch_realtek.c
linux-source-5.15.0/sound/pci/hda/hda_beep.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.c
linux-source-5.15.0/sound/pci/hda/hda_intel_trace.h
linux-source-5.15.0/sound/pci/hda/hda_bind.c
linux-source-5.15.0/sound/pci/hda/hda_jack.h
linux-source-5.15.0/sound/pci/hda/hda_proc.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.h
linux-source-5.15.0/sound/pci/hda/patch_analog.c
patching file patch_cs8409.c
Hunk #1 succeeded at 1442 (offset 172 lines).
Hunk #2 succeeded at 1451 (offset 172 lines).
Hunk #3 succeeded at 1480 (offset 172 lines).
patching file patch_cs8409.h
Hunk #1 succeeded at 19 (offset 1 line).
Hunk #2 succeeded at 302 (offset 1 line).
Hunk #3 FAILED at 328.
Hunk #4 succeeded at 362 (offset 2 lines).
1 out of 4 hunks FAILED -- saving rejects to file patch_cs8409.h.rej
make -C /lib/modules/5.15.0-47-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/tmp/snd_hda_macbookpro/build/hda-5.15.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-47-generic'
  CC [M]  /tmp/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o
In file included from /tmp/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1486:
/tmp/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:157:10: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
  157 |         .linein_jack_in = 0,
      |          ^~~~~~~~~~~~~~
      |          mic_jack_in
/tmp/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:158:10: error: ‘struct sub_codec’ has no member named ‘force_status_change’
  158 |         .force_status_change = 1,
      |          ^~~~~~~~~~~~~~~~~~~
In file included from /tmp/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1486:
/tmp/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/tmp/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:1192:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1192 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
/tmp/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_macbook_exec_verb’:
/tmp/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:2346:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2346 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
/tmp/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_imac_exec_verb’:
/tmp/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:2392:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2392 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
make[2]: *** [scripts/Makefile.build:297: /tmp/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o] Error 1
make[1]: *** [Makefile:1881: /tmp/snd_hda_macbookpro/build/hda-5.15.0] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-47-generic'
make: *** [Makefile:22: all] Error 2
#cp snd-hda-codec-cs8409.ko /lib/modules/5.15.0-47-generic/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.15.0-47-generic/build M=/tmp/snd_hda_macbookpro/build/hda-5.15.0 modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-47-generic'
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
cat: /tmp/snd_hda_macbookpro/build/hda-5.15.0/modules.order: No such file or directory
  DEPMOD  /lib/modules/5.15.0-47-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-47-generic'
depmod -a

contents of /lib/modules/5.15.0-47-generic/updates
total 0
```

After the running the script and rebooting, the device audio is still at "dummy output" and there is no sound.
I also tried re-installing the linux-headers, but it's already at the latest version.

Here's the system info:
```
user@user-MacBookPro:/tmp/snd_hda_macbookpro$ screenfetch
                          ./+o+-       user@user-MacBookPro
                  yyyyy- -yyyyyy+      OS: Ubuntu 22.04 jammy
               ://+//////-yyyyyyo      Kernel: x86_64 Linux 5.15.0-47-generic
           .++ .:/++++++/-.+sss/`      Uptime: 15m
         .:++o:  /++++++++/:--:/-      Packages: 1749
        o:+o+:++.`..```.-/oo+++++/     Shell: bash 5.1.16
       .:+o:+o/.          `+sssoo+/    Resolution: 2560x1600
  .++/+:+oo+o:`             /sssooo.   DE: GNOME 41.7
 /+++//+:`oo+o               /::--:.   WM: Mutter
 \+/+o+++`o++o               ++////.   WM Theme: Adwaita
  .++.o+++oo+:`             /dddhhh.   GTK Theme: Yaru [GTK2/3]
       .+.o+oo:.          `oddhhhh+    Icon Theme: Yaru
        \+.++o+o``-````.:ohdhhhhh+     Font: Ubuntu 11
         `:o+++ `ohhhhhhhhyo++os:      Disk: 23G / 177G (13%)
           .o:`.syhhhhhhh/.oo++o`      CPU: Intel Core i5-7360U @ 4x 3,6GHz [49.0°C]
               /osyyyyyyo++ooo+++/     GPU: Intel Corporation Iris Plus Graphics 640 (rev 06)
                   ````` +oo+++o\:     RAM: 3502MiB / 15867MiB
                          `oo++.      
```

I'm not sure if it's an issue with the script or with my system. Would you mind giving me some pointers on how to investigate and solve the problem?

sweah82, 2022-09-05:
I've the same output and it also doesn't work for me anymore. I'm also on Ubuntu 22.04.

davidjo, 2022-09-05:
This appears to be the new error caused by the removal of a pre-existing variable (force_status_change)  in the original kernel file patch_cs8409.h at base linux kernel version 5.19.
Ubuntu as usual has backported this change to its kernel version 5.15 subversion 47 (maybe 1 or 2 earlier).

Doesnt seem to be a good solution to these sort of updates without having multiple patch files - which have to depend on differing kernel versions depending on distribution.

instructions to fix:

you need to look at patch_patch_cs8409.h.diff
1st run the install script which will fail because the patch of patch_cs8409.h with patch_patch_cs8409.h.diff fails.
hunk3 will fail because of changes in patch_cs8409.h ie removal of original kernel variable focus_status_change.
look at hunk3 (the patch part starting @@ -305,6 +328,9 @@ struct sub_codec { in patch_patch_cs8409.h.diff)
manually edit the file patch_cs8409.h and add the lines:
#ifdef APPLE_CODECS
unsigned int linein_jack_in:1;
#endif
after the line unsigned int mic_jack_in:1;
for file patch_cs8409.h

do not rerun the install script
manually run the following commands in the build/hda-.... directory (where ... is part of kernel version)
make (or sudo make if permission issues)
then if ok
sudo make install

sweah, 2022-09-05:
The following lines already are in the file patch_cs8409.h after the line unsigned int mic_jack_in:1;
  
#ifdef APPLE_CODECS
        unsigned int linein_jack_in:1;
#endif

davidjo, 2022-09-05:
I need to see the specific error messages - if those lines are in then you cant get the errors as listed by martroben.

sweah, 2022-09-06:
With the pull request of @martroben included, everyting works as expected.  Thank you so much :)

martroben, 2022-09-11:
I can also confirm, that after applying the pull request by @BenChaimberg, it starts working again.

```Shell
git clone https://github.com/davidjo/snd_hda_macbookpro
cd snd_hda_macbookpro/
curl https://patch-diff.githubusercontent.com/raw/davidjo/snd_hda_macbookpro/pull/64.diff | git apply
sudo ./install.cirrus.driver.sh
```

sainathmurali, 2022-09-13:
Thank you! This works like a charm!

ccie57654, 2022-09-19:
confirmed [this](https://github.com/davidjo/snd_hda_macbookpro/issues/63#issuecomment-1243019372) worked on clean (updated) installation of Fedora 36 after installing the dependencies in the main readme and a missing package `patch` there was a signing error during patch creation, however that doesn't seem to affect output using below

`           /:-------------:\          beastman@linux-mac
        :-------------------::        OS: Fedora 
      :-----------/shhOHbmp---:\      Kernel: x86_64 Linux 5.19.9-200.fc36.x86_64
    /-----------omMMMNNNMMD  ---:     Uptime: 7m
   :-----------sMMMMNMNMP.    ---:    Packages: 1815
  :-----------:MMMdP-------    ---\   Shell: bash 5.1.16
 ,------------:MMMd--------    ---:   Resolution: 1920x1200
 :------------:MMMd-------    .---:   DE: GNOME 42.3
 :----    oNMMMMMMMMMNho     .----:   WM: Mutter
 :--     .+shhhMMMmhhy++   .------/   WM Theme: Adwaita
 :-    -------:MMMd--------------:    GTK Theme: Adwaita [GTK2/3]
 :-   --------/MMMd-------------;     Icon Theme: Adwaita
 :-    ------/hMMMy------------:      Font: Cantarell 11
 :-- :dMNdhhdNMMNo------------;       Disk: 9.1G / 466G (2%)
 :---:sdNMMMMNds:------------:        CPU: Intel Core i5-7360U @ 4x 3.6GHz [62.0°C]
 :------:://:-------------::          GPU: Mesa Intel(R) Iris(R) Plus Graphics 640 (Kaby Lake GT3e) (KBL GT3)
 :---------------------://            RAM: 1969MiB / 7810MiB
                                     `

webxperia, 2022-09-23:
Confirm work great with ubuntu 22.04.1

ubuntu@ubuntu:~$ neofetch
            .-/+oossssoo+/-.               ubuntu@ubuntu 
        `:+ssssssssssssssssss+:`           ------------- 
      -+ssssssssssssssssssyyssss+-         OS: Ubuntu 22.04.1 LTS x86_64 
    .ossssssssssssssssssdMMMNysssso.       Host: MacBookPro14,1 1.0 
   /ssssssssssshdmmNNmmyNMMMMhssssss/      Kernel: 5.15.0-48-generic 
  +ssssssssshmydMMMMMMMNddddyssssssss+     Uptime: 1 hour 
 /sssssssshNMMMyhhyyyyhmNMMMNhssssssss/    Packages: 1854 (dpkg), 4 (flatpak), 27 (snap) 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Shell: bash 5.1.16 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   Resolution: 2560x1600 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   DE: GNOME 42.4 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   WM: Mutter 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   WM Theme: Adwaita 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Theme: Yaru-red-dark [GTK2/3] 
 /sssssssshNMMMyhhyyyyhdNMMMNhssssssss/    Icons: Yaru-red [GTK2/3] 
  +sssssssssdmydMMMMMMMMddddyssssssss+     Terminal: konsole 
   /ssssssssssshdmNNNNmyNMMMMhssssss/      CPU: Intel i5-7360U (4) @ 3.600GHz 
    .ossssssssssssssssssdMMMNysssso.       GPU: Intel Iris Plus Graphics 640 
      -+sssssssssssssssssyyyssss+-         Memory: 4574MiB / 7815MiB 
        `:+ssssssssssssssssss+:`
            .-/+oossssoo+/-.

webxperia, 2022-10-06:
I can confirm that after testing.
Patch work great with Ubuntu 22.04 in all kernel until 5.19, Linux Mint, Pop_os!, Also work perfect with Arch Linux and Garuda Linux distro with kernel 5.19. 
I have failed to work with MX and antix Linux in kernel 5.10 until kernel ahs5.18

antonis19992, 2022-12-13:
https://github.com/BenChaimberg/snd_hda_macbookpro/commits/master  h lysh einai edw mpravo sto paidi

webxperia, 2023-04-19:
With Kernel 6.1 after try to install patch

git clone https://github.com/davidjo/snd_hda_macbookpro
cd snd_hda_macbookpro/
curl https://patch-diff.githubusercontent.com/raw/davidjo/snd_hda_macbookpro/pull/64.diff | git apply
sudo ./install.cirrus.driver.sh
reboot

i take this error :

error: patch failed: patch_patch_cs8409.h.diff:420
error: patch_patch_cs8409.h.diff: patch does not apply

ppp7032, 2023-04-19:
> With Kernel 6.1 after try to install patch
>
> git clone https://github.com/davidjo/snd_hda_macbookpro cd snd_hda_macbookpro/ curl [...]

I have this exact issue on kernel 6.2.6. Help much appreciated.

davidjo, 2023-04-19:
You may not need the 64.diff patch now - I added a fixup for this I think in January.
Clear out the build directory and just run sudo ./install.cirrus.driver.sh

ppp7032, 2023-04-19:
> 

This worked for me thank you very much! I do want to ask though, as I'm not as familiar with linux as I'd like to be: this audio driver is installed as a patch onto the kernel, right? So if apt updates my kernel, will it also automatically apply the patch to the new one or will I have to reinstall the driver?

davidjo, 2023-04-19:
The linux kernel has kernel modules for adding functionality, its not really a patch. This repo creates a modified kernel module (ie it is an updated version of an existing kernel module) which is added to the kernel modules (usually stored in /lib/modules/kernel version dependent names) for a specific kernel version.
You have to rebuild the module (ie rerun  sudo ./install.cirrus.driver.sh) every time a new kernel is installed - as it will have a different kernel version number so will  go into a different /lib/modules directory.
(There are ways to do an auto update when a new kernel is loaded but there are still issues with kernel source updates causing the issues eg which were fixed by the 64.diff patch so dont think this modules kernel source is stable enough to use such techniques).

webxperia, 2023-04-20:
> The linux kernel has kernel modules for adding functionality, its not really a patch. This repo creates a modified kernel module (ie it is an updated version of an existing kernel module) which is [...]

Thanks Davidjo for you good job and for your info
This is an update message. Today and after update my Linux to Kernel 6.2.17 with Arch Garuda distro  your driver working perfect with out apply any patch.
Thanks for your help