davidjo/snd_hda_macbookpro #66: iMac19,2 No Audio

tecywiz121, 2022-09-24:
Hey! Thanks a ton for maintaining this driver. I've been trying to get it to work with an iMac19,2 (I think), but unfortunately I get no sound output. After loading the module, I do get an output device which wasn't there before, but nothing actually plays.

I had to make a few changes to get everything to compile:

 * Not present in `sub_codec`:
   https://github.com/davidjo/snd_hda_macbookpro/blob/18f0201b67d1d50f99526453e833f585b5b28321/patch_cirrus/patch_cirrus_apple.h#L158
 * Moved a bit too much:
   https://github.com/davidjo/snd_hda_macbookpro/blob/bcc755abe10efa5852c465a4b50f84132c0431d1/patch_patch_cs8409.h.diff#L427-L429

`uname -a`:

```
Linux ubuntu 5.15.0-43-generic #46-Ubuntu SMP Tue Jul 12 10:30:17 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

The only output in `dmesg` after loading the compiled module is:

```
snd_hda_codec_cs8409: module verification failed: signature and/or required key missing - tainting kernel
```

Let me know if there's anything else you need!

davidjo, 2022-09-25:
The fixups are a known issue (#63) because of changes to base linux kernel (curious as early as sub version 43).
The tainting kernel is just because  to fix that would require signing the module which means adding more dependencies in the script - however unsigned kernels work on linux - you just get that message.
No other messages is a problem - suggests the module is loading but not being activated.
Try an rmmod and a verbose modprobe of module to see if there are any messages.
Otherwise we need to find out exactly what system you are on.
Can you run OSX and get an ioreg -x -l -a (save the output) and look for IOHDACodecVendorID number.

tecywiz121, 2022-09-25:
> The fixups are a known issue (#63) because of changes to base linux kernel (curious as early as sub version 43).

Yep! I just wanted to make it clear what I was doing. It's possible that the sources I downloaded were different than the running kernel. I was on an Ubuntu LiveUSB. I did have the same behaviour on my NixOS install, but wanted to confirm using the install script.

> Can you run OSX and get an ioreg -x -l -a (save the output) and look for IOHDACodecVendorID number.

```xml
<key>IOHDACodecVendorID</key>
<integer>269714441</integer>
```

davidjo, 2022-09-26:
well that confirms you have an 8409 system (269714441 is 0x10138409 which is the 8409 codec id the base kernel module is looking for - see bottom of original kernel module file patch_cs8409.c in HDA_CODEC_ENTRY line)

Not sure what goes on on LiveUSBs - maybe it limits logging - altho dmesg should work - as that just dumps current kernel log buffer - there should be some snd_hda messages.

tecywiz121, 2022-09-27:
Even on my NixOS install (not live), I don't see any logs on `insmod`/`rmmod`.

If I run `sudo insmod ./snd-hda-codec-cs8409.ko sadf`, I do get a message:

```
[29946.636583] snd_hda_codec_cs8409: unknown parameter 'sadf' ignored
```

So it looks like logging works, at least... I've probably messed something up while patching?

davidjo, 2022-09-27:
You should get some snd_hda messages - snd_hda is a generic prefix for HDA style audio codec messages
eg I get snd_hda_intel messages because the first driver is the general driver for HDA chips (as defined by Intel).
Even if you dont patch the base kernel snd-hda-codec-cs8490.ko should load which should give some messages.
(the base cs8409 driver loads because that loading just depends on the 0x10138409 key - its just what it does then doesnt work for Apples version of the 8409 chip but it doesnt kick it out of the kernel - really make sure you are loading your version - when all else fails I move the original kernel module out of /lib/modules/../kernel/sound/pci/hda to somewhere (so can restore) (... is actual kernel version) then always do a depmod -a as root after any module changes).
Might need to set kernel log level for this module to debug.

eado, 2022-11-17:
I'm getting an error from `snd_hda_intel` saying that `0x10138409` is unknown. I followed the steps to remove the original kernel module and use the patched one, with no success. 

```
Nov 17 14:27:28 Omars-MacBook-Pro kernel: snd_hda_intel: Primary patch_cs8409
Nov 17 14:27:28 Omars-MacBook-Pro kernel: snd_hda_intel: Primary patch_cs8409 NOT FOUND trying APPLE
Nov 17 14:27:28 Omars-MacBook-Pro kernel: [drm] vm size is 64 GB, 2 levels, block size is 10-bit, fragment size is 9-bit
Nov 17 14:27:28 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0: autoconfig for CS8409: line_outs=0 (0x0/0x0/0x0/0x0/0x0) type:line
Nov 17 14:27:28 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Nov 17 14:27:28 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0:    hp_outs=0 (0x0/0x0/0x0/0x0/0x0)
Nov 17 14:27:28 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0:    mono: mono_out=0x0
Nov 17 14:27:28 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0:    inputs:
Nov 17 14:27:28 Omars-MacBook-Pro kernel: snd_hda_intel: UNKNOWN subsystem id 0x10138409
Nov 17 14:27:28 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409: probe of hdaudioC0D0 failed with error -1
Nov 17 14:27:28 Omars-MacBook-Pro kernel: snd_hda_intel: Primary patch_cs8409
Nov 17 14:27:28 Omars-MacBook-Pro kernel: snd_hda_intel: Primary patch_cs8409 NOT FOUND trying APPLE
```

eado, 2022-11-17:
I even tried to manually patch it in `patch_cirrus_apple.h` by adding `codec->core.subsystem_id == 0x10138409` on line 2537 (and anywhere `0x106b3900` is referenced since that's also the ID for MacBook Pro 14,3?) Any guidance on this matter would be greatly appreciated.

eado, 2022-11-17:
Also worth mentioning that occasionally after reboot, the patch works (no mic), and I'm able to adjust volume just fine. Didn't manage to get the `journalctl` logs for this unfortunately. After a day or so, it reverts back.

davidjo, 2022-11-18:
Unfortunately the subsystem id is completely different from the device id (0x10138409)  - you need to revert those changes.
Once you have done that if you still see UNKNOWN subsystem id then this is subsystem id I havent seen before.

eado, 2022-11-18:
I reverted the changes, and this is what I get:
```
Nov 18 10:40:50 Omars-MacBook-Pro kernel: usbcore: registered new interface driver brcmfmac
Nov 18 10:40:50 Omars-MacBook-Pro kernel: brcmfmac 0000:03:00.0: enabling device (0000 -> 0002)
Nov 18 10:40:50 Omars-MacBook-Pro kernel: snd_hda_intel 0000:00:1f.3: enabling device (0000 -> 0002)
Nov 18 10:40:50 Omars-MacBook-Pro kernel: snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
Nov 18 10:40:50 Omars-MacBook-Pro kernel: snd_hda_intel 0000:01:00.1: Force to non-snoop mode
Nov 18 10:40:50 Omars-MacBook-Pro kernel: Bluetooth: HCI UART driver ver 2.3
Nov 18 10:40:50 Omars-MacBook-Pro kernel: Bluetooth: HCI UART protocol H4 registered
--
Nov 18 10:40:52 Omars-MacBook-Pro kernel: [drm] Found UVD firmware Version: 1.130 Family ID: 16
Nov 18 10:40:52 Omars-MacBook-Pro kernel: [drm] Found VCE firmware Version: 53.26 Binary ID: 3
Nov 18 10:40:52 Omars-MacBook-Pro kernel: snd_hda_intel: Primary patch_cs8409
Nov 18 10:40:52 Omars-MacBook-Pro kernel: snd_hda_intel: Primary patch_cs8409 NOT FOUND trying APPLE
Nov 18 10:40:52 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0: UNKNOWN subsystem id 0x10138409
Nov 18 10:40:52 Omars-MacBook-Pro kernel: snd_hda_intel: Primary patch_cs8409
Nov 18 10:40:52 Omars-MacBook-Pro kernel: snd_hda_intel: Primary patch_cs8409 NOT FOUND trying APPLE
Nov 18 10:40:52 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0: UNKNOWN subsystem id 0x10138409
Nov 18 10:40:52 Omars-MacBook-Pro kernel: snd_hda_codec_generic hdaudioC0D0: autoconfig for Generic: line_outs=0 (0x0/0x0/0x0/0x0/0x0) type:line
Nov 18 10:40:52 Omars-MacBook-Pro kernel: snd_hda_codec_generic hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Nov 18 10:40:52 Omars-MacBook-Pro kernel: snd_hda_codec_generic hdaudioC0D0:    hp_outs=0 (0x0/0x0/0x0/0x0/0x0)
Nov 18 10:40:52 Omars-MacBook-Pro kernel: snd_hda_codec_generic hdaudioC0D0:    mono: mono_out=0x0
Nov 18 10:40:52 Omars-MacBook-Pro kernel: snd_hda_codec_generic hdaudioC0D0:    inputs:
Nov 18 10:40:52 Omars-MacBook-Pro kernel: [drm] Display Core initialized with v3.2.198!
Nov 18 10:40:52 Omars-MacBook-Pro kernel: snd_hda_intel 0000:01:00.1: bound 0000:01:00.0 (ops amdgpu_dm_audio_component_bind_ops [amdgpu])
Nov 18 10:40:52 Omars-MacBook-Pro kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Nov 18 10:40:52 Omars-MacBook-Pro kernel: Bluetooth: BNEP filters: protocol multicast
```

I did figure out a fix, albeit temporary. It resets after any sort of power off (shutdown, suspend, sleep, NOT reboot). Also, it depends on the fact that I'm using rEFInd as my bootloader. Possibly using GRUB would fix this issue completely; however, I haven't been able to get it to work. 
1. Turn on integrated graphics by changing EFI vars
2. Reboot. It will hang on the bootloader menu. I theorize this is because I haven't changed a rEFInd flag to spoof the boot as if I'm booting into macOS. Only in macOS are the integrated graphics enabled. 
3. Boot into any other operating system. Tried this with macOS and Windows. 
4. Modify the EFI partition to use the rEFInd flag to spoof as macOS when booting into Linux. 
5. Reboot into Linux, audio magically works:

```
Nov 18 00:10:14 Omars-MacBook-Pro kernel: amdgpu 0000:01:00.0: amdgpu: Trusted Memory Zone (TMZ) feature not supported
Nov 18 00:10:14 Omars-MacBook-Pro kernel: [drm] vm size is 64 GB, 2 levels, block size is 10-bit, fragment size is 9-bit
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_intel 0000:00:1f.3: enabling device (0000 -> 0002)
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_intel 0000:01:00.1: Force to non-snoop mode
Nov 18 00:10:14 Omars-MacBook-Pro kernel: amdgpu 0000:01:00.0: BAR 2: releasing [mem 0xc0000000-0xc01fffff 64bit pref]
Nov 18 00:10:14 Omars-MacBook-Pro kernel: amdgpu 0000:01:00.0: BAR 0: releasing [mem 0xb0000000-0xbfffffff 64bit pref]
--
Nov 18 00:10:14 Omars-MacBook-Pro kernel: input: HDA ATI HDMI HDMI/DP,pcm=11 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input15
Nov 18 00:10:14 Omars-MacBook-Pro kernel: NET: Registered PF_ALG protocol family
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_intel: Primary patch_cs8409
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_intel: Primary patch_cs8409 NOT FOUND trying APPLE
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0: autoconfig for CS8409: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0:    mono: mono_out=0x0
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0:    inputs:
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0:      Internal Mic=0x44
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0:      Internal Mic=0x45
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_codec_cs8409 hdaudioC0D0:      Mic=0x3c
Nov 18 00:10:14 Omars-MacBook-Pro kernel: usbcore: registered new interface driver usbhid
Nov 18 00:10:14 Omars-MacBook-Pro kernel: usbhid: USB HID core driver
Nov 18 00:10:14 Omars-MacBook-Pro kernel: [drm] Display Core initialized with v3.2.198!
Nov 18 00:10:14 Omars-MacBook-Pro kernel: snd_hda_intel 0000:01:00.1: bound 0000:01:00.0 (ops amdgpu_dm_audio_component_bind_ops [amdgpu])
Nov 18 00:10:14 Omars-MacBook-Pro kernel: videodev: Linux video capture interface: v2.00
Nov 18 00:10:14 Omars-MacBook-Pro kernel: [drm] UVD and UVD ENC initialized successfully.
```

davidjo, 2022-11-19:
Not sure theres much I can say here - you just should not be getting 0x10138409 as the core.subsystem_id - but all the code that sets that is from base linux kernel code (nothing in the routines Ive updated sets this just reads it - neither the original code nor my updates) so have no idea what is going on.
I also use  rEFInd (for a long time)  and dont think Ive seen this - Im not sure you can use grub as direct boot loader as Apple is pure EFI - in fact refind does use grub  to continue the linux boot after the pure EFI stage - mount 1st partition of your internal disk (the EFI partition) and you will see it has a grub.cfg file.
I have seen  boot glitches where there are some what Ill call BIOS drivers (ie Apple pre-kernel boot code)  seem to fail to load correctly (one of them I seem to remember is for the integrated graphics which has to be used before the main kernel is loaded to display the boot process) - for me tends to occur after some form of kernel issues and rebooting from Linux.
But those glitches tend to just hang at boot for me - as you have seen a power off/on or reboot into OSX seems to clear it for me also.
I have never tried to boot all 3 ie Windows as well.
I do make use of efibootmgr on the linux side a lot as my EFI boot order seems to get messed up a lot - in particular after any OSX update.
I set the EFI boot process to boot into refind boot screen first then select the actual Penguin version to boot (I use multiple Linux versions and also boot Linux from USB disks) - this seems the most reliable for me.
I do remember if I use one of the refind boot icons it fails to boot at all.

eado, 2022-11-19:
Very interesting. I think I have an idea why it's reporting a different subsystem_id--I just found a permanent fix for my solution:
1. In Startup Disk in macOS, make sure it's set to your macOS partition (not Bootcamp). Boot into Linux using the option key on startup.
2. Reset NVRAM.

Either way, it might have been some odd EFI vars configuration that caused this issue.

davidjo, 2022-11-20:
Suspend/sleep is a known problem - Ive not attempted to fix that in the current driver - there were (likely still are) lots of problems with suspend/sleep in general for linux when I started.

Its clear that these boot glitches are caused by some corruption occurring which we know is cleared by a full power off/on - Im guessing maybe some of these boot drivers/boot data may be loaded into some memory which gets corrupted under certain conditions (sleep? kernel crash?).

One guess for your issue is that maybe the linux boot process isnt detecting the subcodec chip (ie the cs42l83 chip) and sets the subsystem id to the main device id in that case - because eg the subcodec chip is powered off.

The new base kernel cs8409 module (for Dell laptops) does have some sleep handling with some specific code for the subcodec chip (in that case a cs42l42 - the non-proprietary version of the cs42l83).

KiLLeRRaT, 2024-08-27:
I'm running into the same problem on a `macbookpro14,1`.  I don't have MacOS installed at all, only GRUB/Arch kernel 6.10.x  

When I reset the NVRAM, audio works for the first boot.  Thereafter, it's dead again, until I reset the NVRAM again.

@eado does your fix permanently fix it? I'd like to avoid having to install MacOS just for this.

a000a999a, 2026-09-30:
Working configuration for **iMac19,2** (21.5" 4K, 2019), in case it helps others here, plus a small request.

**Hardware**
- `/sys/class/dmi/id/product_name`: `iMac19,2`
- Codec: Cirrus Logic CS8409, Vendor Id `0x10138409`, **Subsystem Id `0x106b0f00`**
- Distro: Manjaro, kernel 7.2.3 (previously 6.18.x), driver built from current master (`89b22ff`)

**What works**

The quirk-table entry for subsystem `0x106b0f00` is commented out in `cirrus_apple.h`:

```c
//SND_PCI_QUIRK(0x106b, 0x0f00, "Imac 18,2", CS8409_MBP143),
```

So nothing is auto-detected. Forcing the model via modprobe works well:

```
# /etc/modprobe.d/50-sound.conf
options snd-intel-dspcfg dsp_driver=1
options snd-hda-intel model=mbp143 patch=imac2019-gpio4.fw power_save=0
```

The speaker amp also needs GPIO4 enabled (bitmask `0x32`). I do that with an HDA patch firmware file:

```
# /lib/firmware/imac2019-gpio4.fw
[codec]
0x10138409 0x106b0f00 0

[init_verbs]
0x01 0x716 0x32
0x01 0x717 0x32
0x01 0x715 0x32
```

With `model=mbp143`, speaker output is clean at all volume levels and volume control works normally.

I'm now running it via the AUR package `snd-hda-macbookpro-dkms-git` (0.1-3). It builds and works on 7.2.3-2-MANJARO and builds on 6.18.49 (the 6.12.108 DKMS build fails). The `model=mbp143` override is still required.

**Warning for others:** I previously had `model=dolphin` set (a Dell quirk, `0x1028:0x0ACF`). On this iMac that produced extreme over-amplification and distortion at every volume setting, even with a clean low-level signal (about -52 dBFS measured at the output, no clipping). It's presumably applying Dell's I2C amp init sequence to the iMac's amplifier. Don't use `dolphin` on Apple hardware.

**Request:** could the `0x106b, 0x0f00` quirk entry be enabled (maybe relabelled "iMac 18,2 / 19,2", since this 19,2 reports the same subsystem ID), so these machines work without a manual `model=` override? Whether the driver should also set GPIO4 itself for this model is your call. I'm happy to test a branch.