davidjo/snd_hda_macbookpro #69: Cannot Build Drivers in Ubuntu22.04 Kernel 5.15.0-48-generic

shinidorei, 2022-10-07:
Hunk #1 succeeded at 1442 (offset 172 lines).
Hunk #2 succeeded at 1451 (offset 172 lines).
Hunk #3 succeeded at 1480 (offset 172 lines).
patching file patch_cs8409.h
Hunk #1 succeeded at 19 (offset 1 line).
Hunk #2 succeeded at 302 (offset 1 line).
Hunk #3 succeeded at 329 with fuzz 1 (offset 1 line).
Hunk #4 succeeded at 365 (offset 2 lines).
make -C /lib/modules/5.15.0-48-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/ubumac/Downloads/ubuntu drivers/snd_hda_macbookpro/build/hda-5.15.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-48-generic'
make[1]: *** No rule to make target 'drivers/snd_hda_macbookpro/build/hda-5.15.0'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-48-generic'
make: *** [Makefile:22: all] Error 2
#cp snd-hda-codec-cs8409.ko /lib/modules/5.15.0-48-generic/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.15.0-48-generic/build M=/home/ubumac/Downloads/ubuntu drivers/snd_hda_macbookpro/build/hda-5.15.0 modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-48-generic'
make[1]: *** No rule to make target 'drivers/snd_hda_macbookpro/build/hda-5.15.0'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-48-generic'
make: *** [Makefile:32: install] Error 2

contents of /lib/modules/5.15.0-48-generic/updates
total 0

davidjo, 2022-10-09:
Just realised what issue may be - I think you have a blank in the "ubuntu drivers" directory name - try removing the blank from ubuntu\ drivers - blanks in file names are very unexpected in Linux - so I didnt quote file/directory names

shinidorei, 2022-10-12:
You are right, in the path I have it escaped, I wonder why it didn't translate in the script:
Hunk #1 succeeded at 1442 (offset 172 lines).
Hunk #2 succeeded at 1451 (offset 172 lines).
Hunk #3 succeeded at 1480 (offset 172 lines).
patching file patch_cs8409.h
Hunk #1 succeeded at 19 (offset 1 line).
Hunk #2 succeeded at 302 (offset 1 line).
Hunk #3 succeeded at 329 with fuzz 1 (offset 1 line).
Hunk #4 succeeded at 365 (offset 2 lines).
make -C /lib/modules/5.15.0-48-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/ubumac/Downloads/ubuntu_drivers/snd_hda_macbookpro/build/hda-5.15.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-48-generic'
  CC [M]  /home/ubumac/Downloads/ubuntu_drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o
In file included from /home/ubumac/Downloads/ubuntu_drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1486:
/home/ubumac/Downloads/ubuntu_drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/home/ubumac/Downloads/ubuntu_drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:1191:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1191 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
  CC [M]  /home/ubumac/Downloads/ubuntu_drivers/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409-tables.o
  LD [M]  /home/ubumac/Downloads/ubuntu_drivers/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cs8409.o
  MODPOST /home/ubumac/Downloads/ubuntu_drivers/snd_hda_macbookpro/build/hda-5.15.0/Module.symvers
  CC [M]  /home/ubumac/Downloads/ubuntu_drivers/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cs8409.mod.o
  LD [M]  /home/ubumac/Downloads/ubuntu_drivers/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cs8409.ko
  BTF [M] /home/ubumac/Downloads/ubuntu_drivers/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cs8409.ko
Skipping BTF generation for /home/ubumac/Downloads/ubuntu_drivers/snd_hda_macbookpro/build/hda-5.15.0/snd-hda-codec-cs8409.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-48-generic'
#cp snd-hda-codec-cs8409.ko /lib/modules/5.15.0-48-generic/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.15.0-48-generic/build M=/home/ubumac/Downloads/ubuntu_drivers/snd_hda_macbookpro/build/hda-5.15.0 modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-48-generic'
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
  INSTALL /lib/modules/5.15.0-48-generic/updates/snd-hda-codec-cs8409.ko
  SIGN    /lib/modules/5.15.0-48-generic/updates/snd-hda-codec-cs8409.ko
At main.c:160:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: ../crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: ../crypto/bio/bss_file.c:75
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  /lib/modules/5.15.0-48-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-48-generic'
depmod -a

contents of /lib/modules/5.15.0-48-generic/updates
total 1504
-rw-r--r-- 1 root root 1538960 Oct 12 09:42 snd-hda-codec-cs8409.ko

davidjo, 2022-10-12:
Unfortunately escaping only works for one level - so every use of a file parameter in a script has also to be escaped/quoted - and also any other script called by the script etc.
your above output is expected - I havent set up module signing (a) cause not figured out how to do it (b) it would require even more additional packages to be installed to do the build.
But the module has been copied to the kernel modules directory - contents of /lib/modules.... so should be working.