davidjo/snd_hda_macbookpro #70: Make microphone driver

carrelebo, 2022-10-08:
Hello @davidjo, Can you give to me a link with your HDA build driver references for i try to make one ?

davidjo, 2022-10-08:
Sure - I couldnt really find any good HDA driver kernel docs - except of course for the code - plus the new 8409 linux kernel module gives a lot more info - wish it had been available when  I started - 
so you need to look under sound at top of your kernel source.
The HDA code is under sound/hda and sound/pci/hda.
I also used the HDA Intel specification document - which is available from intel eg https://www.intel.co.uk/content/www/uk/en/standards/high-definition-audio-specification.html
Then you can find various HDA chip specification documents - but Ive not seen an 8409 document - but the new Dell kernel module (patch_cs8409.c etc) allowed me to sort out more of how the 8409 is working.
There is a  CS42L42_F4.pdf available for the sub chip which give some pointers as to how the actual cs42l83 is working - but somethings obvioulsy not correct.

Note that the new Dell 8409 kernel module is quite different from Apple - it just seems to drive the cs42l42 chip - whereas Apples chip also drives the speaker amp chips and the internal mike chip - a headphone mike uses the cs42l83 chip (and has never worked).

You need to try and understand how the HDA widgets work and are connected first  - the internal mike just uses the 8409 widgets.

carrelebo, 2022-10-08:
thanks for all of the indications :)

davidjo, 2022-10-09:
PS one of the things you should look at is kernel function snd_kctl_jack_report (in sound/core) (which calls the prime function snd_ctl_notify (another base kernel source) - this is what I have discovered is how the kernel drivers communicate with user space - so for example when you plug something into the jack it changes what is used as the audio source/sink as seen by the user eg in the Ubuntu Audio Settings part of the Settings utility.
I only discovered this recently (as its buried in multiple function calls in the actual drivers you see in sound/pci/hda).
For example this is how you would pass back headset volume buttons to user space so the volume changes (as the 8409 chip as used by Apple does not do any volume change).
Note that the internal mike should be working  - it needs connecting and is very low volume - but havent checked this in a while.