davidjo/snd_hda_macbookpro #71: No audio input/output on Macbook Pro 14,1

mnchava, 2022-10-10:
I have tried the fixes in issue #63 an PR #64 but I still don't have any audio on speakers or headphones.
```
OS: Debian GNU/Linux 11 (bullseye) x86_64 
Host: MacBookPro14,1 1.0 
Kernel: 5.10.0-18-amd64 
Uptime: 13 mins 
Packages: 2614 (dpkg) 
Shell: bash 5.1.4 
Resolution: 2560x1600 
DE: Cinnamon 4.8.6 
WM: Mutter (Muffin) 
 WM Theme: Blue-Submarine (BlackMATE) 
Theme: Adwaita-dark [GTK2/3] 
Icons: gnome [GTK2/3] 
Terminal: gnome-terminal 
CPU: Intel i5-7360U (4) @ 3.600GHz 
GPU: Intel Iris Plus Graphics 640 
Memory: 1791MiB / 7822MiB
```
Output from the install script:
```
chava@mbp:~/snd_hda_macbookpro$ sudo ./install.cirrus.driver.sh 
--2022-10-10 16:39:28--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.10.0.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 2a04:4e42::432, 2a04:4e42:200::432, 2a04:4e42:400::432, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|2a04:4e42::432|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-10-10 16:39:29 ERROR 404: Not Found.

Failed to download linux-5.10.0.tar.xz
Trying to download base kernel version linux-5.10.tar.xz
This may lead to build failures as too old
If this is an Ubuntu-based distribution this almost certainly will fail to build

--2022-10-10 16:39:29--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.10.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 2a04:4e42::432, 2a04:4e42:400::432, 2a04:4e42:600::432, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|2a04:4e42::432|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

linux-5.10/sound/pci/hda/
linux-5.10/sound/pci/hda/Kconfig
linux-5.10/sound/pci/hda/Makefile
linux-5.10/sound/pci/hda/ca0132_regs.h
linux-5.10/sound/pci/hda/hda_auto_parser.c
linux-5.10/sound/pci/hda/hda_auto_parser.h
linux-5.10/sound/pci/hda/hda_beep.c
linux-5.10/sound/pci/hda/hda_beep.h
linux-5.10/sound/pci/hda/hda_bind.c
linux-5.10/sound/pci/hda/hda_codec.c
linux-5.10/sound/pci/hda/hda_controller.c
linux-5.10/sound/pci/hda/hda_controller.h
linux-5.10/sound/pci/hda/hda_controller_trace.h
linux-5.10/sound/pci/hda/hda_eld.c
linux-5.10/sound/pci/hda/hda_generic.c
linux-5.10/sound/pci/hda/hda_generic.h
linux-5.10/sound/pci/hda/hda_hwdep.c
linux-5.10/sound/pci/hda/hda_intel.c
linux-5.10/sound/pci/hda/hda_intel.h
linux-5.10/sound/pci/hda/hda_intel_trace.h
linux-5.10/sound/pci/hda/hda_jack.c
linux-5.10/sound/pci/hda/hda_jack.h
linux-5.10/sound/pci/hda/hda_local.h
linux-5.10/sound/pci/hda/hda_proc.c
linux-5.10/sound/pci/hda/hda_sysfs.c
linux-5.10/sound/pci/hda/hda_tegra.c
linux-5.10/sound/pci/hda/hp_x360_helper.c
linux-5.10/sound/pci/hda/patch_analog.c
linux-5.10/sound/pci/hda/patch_ca0110.c
linux-5.10/sound/pci/hda/patch_ca0132.c
linux-5.10/sound/pci/hda/patch_cirrus.c
linux-5.10/sound/pci/hda/patch_cmedia.c
linux-5.10/sound/pci/hda/patch_conexant.c
linux-5.10/sound/pci/hda/patch_hdmi.c
linux-5.10/sound/pci/hda/patch_realtek.c
linux-5.10/sound/pci/hda/patch_si3054.c
linux-5.10/sound/pci/hda/patch_sigmatel.c
linux-5.10/sound/pci/hda/patch_via.c
linux-5.10/sound/pci/hda/thinkpad_helper.c
patching file patch_cirrus.c
make -C /lib/modules/5.10.0-18-amd64/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/chava/snd_hda_macbookpro/build/hda-5.10.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.0-18-amd64'
  CC [M]  /home/chava/snd_hda_macbookpro/build/hda-5.10.0/patch_cirrus.o
  LD [M]  /home/chava/snd_hda_macbookpro/build/hda-5.10.0/snd-hda-codec-cirrus.o
  MODPOST /home/chava/snd_hda_macbookpro/build/hda-5.10.0/Module.symvers
  CC [M]  /home/chava/snd_hda_macbookpro/build/hda-5.10.0/snd-hda-codec-cirrus.mod.o
  LD [M]  /home/chava/snd_hda_macbookpro/build/hda-5.10.0/snd-hda-codec-cirrus.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-18-amd64'
#cp snd-hda-codec-cirrus.ko /lib/modules/5.10.0-18-amd64/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.10.0-18-amd64/build M=/home/chava/snd_hda_macbookpro/build/hda-5.10.0 modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.10.0-18-amd64'
  INSTALL /home/chava/snd_hda_macbookpro/build/hda-5.10.0/snd-hda-codec-cirrus.ko
  DEPMOD  5.10.0-18-amd64
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-18-amd64'
depmod -a

contents of /lib/modules/5.10.0-18-amd64/updates
total 440
-rw-r--r-- 1 root root 446944 Oct 10 16:39 snd-hda-codec-cirrus.ko
```

davidjo, 2022-10-11:
I dont know debian so what the script does is download the mainline kernel  - but debian seems to be using different versioning from the kernel - hence the message failed to download 5.10.0 because 5.10.0-18 is not a mainline kernel version - its seems to be similar to versioning Ubuntu  is using on eg Im currently on (LTS 22.04) 5.15.0-48.
Does debian have a kernel source dpkg similar to Ubuntu? - if so you should use that rather than mainline - although what little Ive learned about debian is it seems closer to mainline than Ubuntu.
However, I think the primary issue is 5.10 is very old - so you shouldnt be using #63 at all - if you check the install script you will see for kernel versions < 5.13 a totally different path is followed - and the #63 patches are ignored so wont affect the build at all.
You need to determine if you have the 8409 chip and if the module is loaded - look for snd_hda_codec_cirrus in lsmod output. Also look at lshw output and see if you what audio devices are loaded (but ignore the HDMI audio items) - or boot OSX and do ioreg -x -l -a and look for 8409 under IOHDACodec (or something similar).

mnchava, 2022-10-11:
Thanks for your response! I tried just switching to Ubuntu and audio works after applying your fixes.

closed 2022-10-11 by mnchava

Pablo12345678901, 2023-09-13:
Dear @davidjo ,

I cannot get sound output/input. It seems that input is not available on MacBook Pro 14,1 which is my device - if it is available now please let me know. Main issue here is output sound.

I was not able to find the solution neither with the scripts of "snd_hda_macbookpro" and "https://github.com/leifliddy/macbook12-audio-driver" repositories nor  by myself with your comment : I do not find  "snd_hda_codec_cirrus" module and if I load it after reboot it is not anymore loaded - sound does not work.

I check if my sound was mute - it is not.

Those are some command outputs that I hope could help to resolve my issue.

Command :
uname -a

Output :
Linux laptop-incognito 5.10.0-25-amd64 #1 SMP Debian 5.10.191-1 (2023-08-16) x86_64 GNU/Linux

Command : 
sudo dmesg | grep -i 'snd\|hda'

Output :
[   13.683726] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   13.732882] snd_hda_codec_generic hdaudioC0D0: autoconfig for Generic: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
[   13.732885] snd_hda_codec_generic hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   13.732887] snd_hda_codec_generic hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
[   13.732888] snd_hda_codec_generic hdaudioC0D0:    mono: mono_out=0x0
[   13.732888] snd_hda_codec_generic hdaudioC0D0:    inputs:
[   13.732890] snd_hda_codec_generic hdaudioC0D0:      Internal Mic=0x44
[   13.732891] snd_hda_codec_generic hdaudioC0D0:      Mic=0x3c
[   13.853808] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input8
[   13.853911] input: HDA Intel PCH HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input9
[   13.854022] input: HDA Intel PCH HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input10
[   13.854076] input: HDA Intel PCH HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input11
[   13.854127] input: HDA Intel PCH HDMI/DP,pcm=10 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input12

Command : 
sudo lsmod | grep 'snd'    

Output : 
snd_hda_codec_hdmi     77824  1
snd_hda_codec_generic    98304  1
ledtrig_audio          16384  1 snd_hda_codec_generic
snd_soc_skl           180224  0
snd_soc_hdac_hda       24576  1 snd_soc_skl
snd_hda_ext_core       36864  2 snd_soc_hdac_hda,snd_soc_skl
snd_soc_sst_ipc        20480  1 snd_soc_skl
snd_soc_sst_dsp        36864  1 snd_soc_skl
snd_soc_acpi_intel_match    45056  1 snd_soc_skl
snd_soc_acpi           16384  2 snd_soc_acpi_intel_match,snd_soc_skl
snd_hda_intel          57344  5
snd_intel_dspcfg       28672  2 snd_hda_intel,snd_soc_skl
soundwire_intel        45056  1 snd_intel_dspcfg
snd_soc_core          319488  3 soundwire_intel,snd_soc_hdac_hda,snd_soc_skl
snd_compress           32768  1 snd_soc_core
snd_hda_codec         176128  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_soc_hdac_hda
snd_hda_core          110592  7 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_soc_hdac_hda,snd_soc_skl
snd_hwdep              20480  1 snd_hda_codec
snd_pcm               143360  9 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,soundwire_intel,snd_compress,snd_soc_core,snd_soc_skl,snd_hda_core
snd_timer              49152  1 snd_pcm
snd                   110592  18 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
soundcore              16384  1 snd

Command : 
sudo lshw | grep -i audio

Output : 
             description: Audio device
             produit: Sunrise Point-LP HD Audio

Command : 
lspci | grep -i 'audio'

Output : 
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)

Command : 
sudo journalctl -b 0 -p 5 | grep -i 'audio\|snd\|sound\|firmware'

Output : 
sep 13 22:55:29 laptop-incognito kernel: ACPI: [Firmware Bug]: BIOS _OSI(Linux) query ignored
sep 13 22:55:30 laptop-incognito kernel: brcmfmac 0000:02:00.0: firmware: failed to load brcm/brcmfmac4350c2-pcie.Apple Inc.-MacBookPro14,1.txt (-2)
sep 13 22:55:30 laptop-incognito kernel: firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
sep 13 22:55:30 laptop-incognito kernel: brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.Apple Inc.-MacBookPro14,1.txt failed with error -2
sep 13 22:55:30 laptop-incognito kernel: brcmfmac 0000:02:00.0: firmware: failed to load brcm/brcmfmac4350c2-pcie.txt (-2)
sep 13 22:55:30 laptop-incognito kernel: brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.txt failed with error -2
sep 13 22:55:30 laptop-incognito kernel: brcmfmac 0000:02:00.0: firmware: failed to load brcm/brcmfmac4350c2-pcie.clm_blob (-2)
sep 13 22:55:30 laptop-incognito kernel: bluetooth hci0: firmware: failed to load brcm/BCM.hcd (-2)
sep 13 22:55:30 laptop-incognito kernel: Bluetooth: hci0: BCM: firmware Patch file not found, tried:
sep 13 22:55:43 laptop-incognito libcanberra-login-sound.desktop[1584]: Failed to play sound: File or data not found

Please ask any other missing information and I will provide it.

Thank you.