davidjo/snd_hda_macbookpro #72: Kernel 5.15.0-50 not functional

BenChaimberg, 2022-10-25:
Build script succeeds but audio module does not seem to be loaded?

Output of ``:
```
$ uname -r
5.15.0-50-generic
$ ls /lib/modules/5.15.0-50-generic/updates
snd-hda-codec-cs8409.ko
$ </var/log/syslog grep snd
Oct 25 13:21:22 ben-MacBookPro kernel: [    4.896368] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Oct 25 13:21:22 ben-MacBookPro kernel: [    4.939803] snd_hda_codec_cs8409: loading out-of-tree module taints kernel.
Oct 25 13:21:22 ben-MacBookPro kernel: [    4.939871] snd_hda_codec_cs8409: module verification failed: signature and/or required key missing - tainting kernel
Oct 25 13:21:22 ben-MacBookPro kernel: [    4.940378] snd_hda_intel: Primary patch_cs8409
Oct 25 13:21:22 ben-MacBookPro kernel: [    4.940380] snd_hda_intel: Primary patch_cs8409 NOT FOUND trying APPLE
Oct 25 13:21:22 ben-MacBookPro kernel: [    4.944258] snd_hda_codec_cs8409 hdaudioC0D0: autoconfig for CS8409: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
Oct 25 13:21:22 ben-MacBookPro kernel: [    4.944263] snd_hda_codec_cs8409 hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Oct 25 13:21:22 ben-MacBookPro kernel: [    4.944266] snd_hda_codec_cs8409 hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
Oct 25 13:21:22 ben-MacBookPro kernel: [    4.944268] snd_hda_codec_cs8409 hdaudioC0D0:    mono: mono_out=0x0
Oct 25 13:21:22 ben-MacBookPro kernel: [    4.944270] snd_hda_codec_cs8409 hdaudioC0D0:    inputs:
Oct 25 13:21:22 ben-MacBookPro kernel: [    4.944272] snd_hda_codec_cs8409 hdaudioC0D0:      Internal Mic=0x44
Oct 25 13:21:22 ben-MacBookPro kernel: [    4.944274] snd_hda_codec_cs8409 hdaudioC0D0:      Mic=0x3c
Oct 25 13:21:22 ben-MacBookPro kernel: [    5.182736] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_cs42l83_jack_unsol_event UNSOL 0xdc000003 tag 0x37
Oct 25 13:21:22 ben-MacBookPro kernel: [    5.182741] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_cs42l83_unsolicited_response - NOT UNSOL BLOCKED
Oct 25 13:21:22 ben-MacBookPro kernel: [    5.261154] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_cs42l83_jack_unsol_event UNSOL 0xdc000002 tag 0x37
Oct 25 13:21:22 ben-MacBookPro kernel: [    5.261159] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_cs42l83_unsolicited_response - NOT UNSOL BLOCKED
Oct 25 13:21:22 ben-MacBookPro kernel: [    5.305846] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_cs42l83_jack_unsol_event UNSOL 0xdc000003 tag 0x37
Oct 25 13:21:22 ben-MacBookPro kernel: [    5.305857] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_cs42l83_unsolicited_response - NOT UNSOL BLOCKED
```

BenChaimberg, 2022-10-25:
Oops, this was inaccurate. The audio is working fine but Ubuntu is broken? Audio options don't show up properly in settings and key presses don't change volume. But audio plays just fine

lbrunkho, 2023-01-13:
Confirmed this is the case for me as well on my MacBook Pro 13,1. On startup, the sound settings menu in the gnome settings appears blank but the audio output works just fine. After playing some audio the menu seems to rediscover or reinitialize the audio interface and it shows up just fine. 
Physical button control still is broken for me even after playing some audio though. I mapped the physical buttons to their corresponding action in the keyboard shortcuts window but it seems that did nothing.