davidjo/snd_hda_macbookpro #73: Error During Install Script: /lib/module/5.15.74-3-MANJARO/build: No such file or directory.

ERoels23, 2022-10-29:
I'm trying to install the drivers but the install script (or using 'make') fails because a directory doesn't exist.
Not sure if I should just create the directory and try again, or if the script needs to be directed somewhere else.

Here's the output from the install script:
```
--2022-10-28 20:57:43--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.15.74.tar.xz
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.69.176, 2a04:4e42:10::432
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.69.176|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 126457328 (121M) [application/x-xz]
Saving to: ‘build/linux-5.15.74.tar.xz’

linux-5.15.74.tar.xz                       100%[======================================================================================>] 120.60M  17.7MB/s    in 7.2s    

2022-10-28 20:57:50 (16.8 MB/s) - ‘build/linux-5.15.74.tar.xz’ saved [126457328/126457328]

linux-5.15.74/sound/pci/hda/
linux-5.15.74/sound/pci/hda/Kconfig
linux-5.15.74/sound/pci/hda/Makefile
linux-5.15.74/sound/pci/hda/ca0132_regs.h
linux-5.15.74/sound/pci/hda/hda_auto_parser.c
linux-5.15.74/sound/pci/hda/hda_auto_parser.h
linux-5.15.74/sound/pci/hda/hda_beep.c
linux-5.15.74/sound/pci/hda/hda_beep.h
linux-5.15.74/sound/pci/hda/hda_bind.c
linux-5.15.74/sound/pci/hda/hda_codec.c
linux-5.15.74/sound/pci/hda/hda_controller.c
linux-5.15.74/sound/pci/hda/hda_controller.h
linux-5.15.74/sound/pci/hda/hda_controller_trace.h
linux-5.15.74/sound/pci/hda/hda_eld.c
linux-5.15.74/sound/pci/hda/hda_generic.c
linux-5.15.74/sound/pci/hda/hda_generic.h
linux-5.15.74/sound/pci/hda/hda_hwdep.c
linux-5.15.74/sound/pci/hda/hda_intel.c
linux-5.15.74/sound/pci/hda/hda_intel.h
linux-5.15.74/sound/pci/hda/hda_intel_trace.h
linux-5.15.74/sound/pci/hda/hda_jack.c
linux-5.15.74/sound/pci/hda/hda_jack.h
linux-5.15.74/sound/pci/hda/hda_local.h
linux-5.15.74/sound/pci/hda/hda_proc.c
linux-5.15.74/sound/pci/hda/hda_sysfs.c
linux-5.15.74/sound/pci/hda/hda_tegra.c
linux-5.15.74/sound/pci/hda/hp_x360_helper.c
linux-5.15.74/sound/pci/hda/ideapad_s740_helper.c
linux-5.15.74/sound/pci/hda/patch_analog.c
linux-5.15.74/sound/pci/hda/patch_ca0110.c
linux-5.15.74/sound/pci/hda/patch_ca0132.c
linux-5.15.74/sound/pci/hda/patch_cirrus.c
linux-5.15.74/sound/pci/hda/patch_cmedia.c
linux-5.15.74/sound/pci/hda/patch_conexant.c
linux-5.15.74/sound/pci/hda/patch_cs8409-tables.c
linux-5.15.74/sound/pci/hda/patch_cs8409.c
linux-5.15.74/sound/pci/hda/patch_cs8409.h
linux-5.15.74/sound/pci/hda/patch_hdmi.c
linux-5.15.74/sound/pci/hda/patch_realtek.c
linux-5.15.74/sound/pci/hda/patch_si3054.c
linux-5.15.74/sound/pci/hda/patch_sigmatel.c
linux-5.15.74/sound/pci/hda/patch_via.c
linux-5.15.74/sound/pci/hda/thinkpad_helper.c
patching file patch_cs8409.c
patching file patch_cs8409.h
Hunk #3 FAILED at 328.
1 out of 4 hunks FAILED -- saving rejects to file patch_cs8409.h.rej
make -C /lib/modules/5.15.74-3-MANJARO/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/eric/snd_hda_macbookpro/build/hda-5.15.74 modules
make[1]: *** /lib/modules/5.15.74-3-MANJARO/build: No such file or directory.  Stop.
make: *** [Makefile:22: all] Error 2
#cp snd-hda-codec-cs8409.ko /lib/modules/5.15.74-3-MANJARO/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.15.74-3-MANJARO/build M=/home/eric/snd_hda_macbookpro/build/hda-5.15.74 modules_install
make[1]: *** /lib/modules/5.15.74-3-MANJARO/build: No such file or directory.  Stop.
make: *** [Makefile:32: install] Error 2

contents of /lib/modules/5.15.74-3-MANJARO/updates
total 0
```
inxi -A output:
```
Audio:
  Device-1: Intel Sunrise Point-LP HD Audio driver: snd_hda_intel
  Device-2: Broadcom 720p FaceTime HD Camera driver: N/A
  Sound API: ALSA v: k5.15.74-3-MANJARO running: yes
  Sound Server-1: PipeWire v: 0.3.58 running: yes
```
neofetch:
```
██████████████████  ████████   eric@macjaro 
██████████████████  ████████   ------------ 
██████████████████  ████████   OS: Manjaro Linux x86_64 
██████████████████  ████████   Host: MacBookPro14,1 1.0 
████████            ████████   Kernel: 5.15.74-3-MANJARO 
████████  ████████  ████████   Uptime: 33 mins 
████████  ████████  ████████   Packages: 916 (pacman) 
████████  ████████  ████████   Shell: zsh 5.9 
████████  ████████  ████████   Resolution: 2560x1600 
████████  ████████  ████████   WM: i3 
████████  ████████  ████████   Theme: Adapta-Nokto-Eta-Maia [GTK2/3] 
████████  ████████  ████████   Icons: Papirus-Adapta-Nokto-Maia [GTK2/3] 
████████  ████████  ████████   Terminal: termite 
████████  ████████  ████████   Terminal Font: Monospace 13 
                               CPU: Intel i5-7360U (4) @ 3.600GHz 
                               GPU: Intel Iris Plus Graphics 640 
                               Memory: 1927MiB / 7814MiB 
```

davidjo, 2022-10-30:
As I understand it Manjaro is a version of Arch linux which apparently links /lib to /usr/lib - so ensure the /lib link exists and all the components of /lib/modules/5.15.74-3-MANJARO/ exist under /usr.
Otherwise it suggests the build directory doesnt exist - I dont know how Arch packages the kernel but you need to find the kernel source/header package that creates that directory.
These names are from the standard mainline  kernel build files/Makefiles so if youve installed all the packages to enable kernel building it should work.

davidjo, 2022-10-30:
Also I think you need to check #59 for a mainline kernel update which prevents the patches from working.
You need to remove the