davidjo/snd_hda_macbookpro #75: imac18,3 Linux Mint 21 [approx=Ubuntu 22.04] kernel 15.0.52-lowlatency cs8409 compilation etc

NialMcDJ, 2022-11-06:
Firstly a heartfelt thanks to you in particular David and to all those others who over the months (and indeed years) have contributed to this work. My apologies for not having participated to date. I am now however in a position to contribute in any way I can to help carry things forward.

**No Sound**
I followed the instructions in the README re compiling the  cirrus module from the beginning a good number of times (and at several points followed the instructions in #40 -- which I believe are now somewhat out of date -- in relation to Alex Egorenkov's fork). However although both kernel and Alsa recognised the card, Alsa was adamant that the device had no controls and thus there was no sound.

 **The  Problem**
It was only when I realised that there was probably a problem with snd-hda-codec-cs8409, _and blacklisted it_, that sound miraculously appeared. 

**Sound**
I am happy to report that I now have sound -- so at this point I have the same sorts of problem that everybody else is struggling through -- RH woofer goodish -- LH tweeter there but barely present and very thin and nothing else.
I have yet to try the headphone port -- but will report on it when I do. 
Incidentally the un-reduced StereoTest32.wav file is not overly loud currently on my system.

 **compiling / pathching snd-hda-codec-cs8409
 patching this fails on my system with:**

 CC [M]  /home/nial/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o
patching file patch_cs8409.c
Hunk #1 succeeded at 1262 (offset -8 lines).
Hunk #2 succeeded at 1271 (offset -8 lines).
Hunk #3 succeeded at 1300 (offset -8 lines).
patching file patch_cs8409.h
Hunk #3 FAILED at 328.
Hunk #4 succeeded at 359 (offset -1 lines).
 CC [M]  /home/nial/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o

...

  **then**

In file included from /home/nial/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1306:
/home/nial/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:157:10: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
  157 |         .linein_jack_in = 0,
      |          ^~~~~~~~~~~~~~
      |          mic_jack_in
/home/nial/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:158:10: error: ‘struct sub_codec’ has no member named ‘force_status_change’
  158 |         .force_status_change = 1,
      |          ^~~~~~~~~~~~~~~~~~~
In file included from /home/nial/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1306:
/home/nial/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/home/nial/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:1192:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1192 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
/home/nial/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_macbook_exec_verb’:
/home/nial/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:2346:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2346 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
/home/nial/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_imac_exec_verb’:
/home/nial/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:2392:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2392 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
make[2]: *** [scripts/Makefile.build:297: /home/nial/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o] Error 1
make[1]: *** [Makefile:1900: /home/nial/snd_hda_macbookpro/build/hda-5.15.0] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-52-lowlatency'
make: *** [Makefile:22: all] Error 2

...

  **then what hasn't failed to compile, fails to install into updates with**

#cp snd-hda-codec-cs8409.ko /lib/modules/5.15.0-52-lowlatency/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.15.0-52-lowlatency/build M=/home/nial/snd_hda_macbookpro/build/hda-5.15.0 modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-52-lowlatency'
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
cat: /home/nial/snd_hda_macbookpro/build/hda-5.15.0/modules.order: No such file or directory
  DEPMOD  /lib/modules/5.15.0-52-lowlatency
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-52-lowlatency'
depmod -a

contents of /lib/modules/5.15.0-52-lowlatency/updates
total 1840
drwxr-xr-x 2 root root    4096 Oct 31 22:41 dkms
-rw-r--r-- 1 root root 1877448 Nov  5 21:18 snd-hda-codec-cirrus.ko

**-- might some of this be to do with a compiler mismatch?**

make -C /lib/modules/5.15.0-52-lowlatency/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/nial/snd_hda_macbookpro/build/hda-5.15.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-52-lowlatency'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0

**Where do I go from here?**

  **compilation**
Given that the patched ...-cirrus.ko is designed to work _with_ a patched ...cs8409.ko 
rather than _instead of_ the vanilla ...cs8409.ko -- which is what I am doing currently
has anybody any thoughts about what I need to do to get it compiled / installed?

  **Sound**
I'm about to look at all the discussion about Alsa / Pulse configuration files
I'll hopefully be in a position (ADHD allowing) to report some successes soonish

Thanks in advance for any thoughts, please let me know if there is further information you require. Also if there is anything I can do to help in general.
n

davidjo, 2022-11-07:
Yes - issue #40 is now old and not for the current version.
Interesting - the problem was that at around mainline kernel 5.13 an explicit cs8409 was introduced (for some Dell laptops - nothing to do with Apple).
So now this repo fixes up the new cs8409 module to handle Apples chips. I install this into the kernel modules updates directory which as I understand (and Im not really a kernel expert - just know whats needed) should override the now existing cs8409 module (as long as you ensure the depmod -a is done) - sounds like that override is not working for you - Im assuming the blacklisting is taking out the existing cs8409 (Dell only) module.

Your other problem is a known issue - as of around ubuntu kernel version 5.15.0-43 the mainline kernel removed a variable from the mainline kernel  8409 module sources which the patching patch files expect as context.
You need to remove any lines with force_status_change in my sources after patching plus add my new bits in the failed Hunk 3n and build manually (you could rebuild the patch files to use the script)  - example patches are in issue #64.

Im still figuring out a good way to handle these breaking mainline kernel updates which seem to happen frequently so my repo is not updated yet (Id hoped changing to patch files would reduce breakage frequency).

Channel mapping should be handled now for all Macs - except is L/R Tweeter L/R Woofer (could have Tweeter/Woofer reversed) rather than Apples L Tweeter/Woofer R Tweeter/Woofer 4 channel format.
By default the audio input is set to 2 channel which gets duplicated by my module into 4 - if you want to use 4 channels you need to manually set the 4 channel surround in Mints general user audio setttings.
 
Checkout issue #56 for recent attempt at improved quality.

NialMcDJ, 2022-11-07:
Thank you -- I'll have a look now and get back to you

NialMcDJ, 2022-11-07:
That is of course **Kernel 5.15.0-52-lowlatency** my brain has only just registered the typo
I can confirm no sound on the headphone port, perhaps unsurprisingly
nothing further to report as  ...cs8409.ko is still not arriving in /lib/modules/5.15.0-52-lowlatency/updates/ 
-- and in all likelihood still not compiling as I've seen no evidence of it anywhere

davidjo, 2022-11-09:
Have to say Ive only ever used the generic kernel  - low latency may need a special kernel source - in Ubuntu you just install linux-source to get the kernel source - I know this doesnt work for HWE kernels.
The script tries to install a mainline kernel if not ubuntu - but this may not work - and Mint I think is similar to ubuntu - the versioning is similar and as Ive discovered figuring out the mapping between versions like 5.15.0-52 and standard mainline kernel versioning eg 5.15.43 is not clear - I think in at least one case it didnt appear that 5.15.52 is the same as 5.15.0-52.

NialMcDJ, 2022-11-09:
Diolch Dafydd / thank you David.
The kernel sources are the same for generic / lowlatency (_I think_), but the headers are different -- I did wonder if perhaps this was causing a problem -- I'll see what difference it makes booting from the generic kernel. 

Yes mint is heavily Ubuntu so it does require the Debian / Ubuntu patched source code, although as  your comment in install.sh indicates,
(**75 # we need to handle Ubuntu based distributions eg Mint here**)
the code as it stands doesn't necessarily catch this -- my /etc/os-release reads like this:

****NAME="Linux Mint"
VERSION="21 (Vanessa)"
ID=linuxmint
ID_LIKE="ubuntu debian"
PRETTY_NAME="Linux Mint 21"
VERSION_ID="21"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=vanessa
UBUNTU_CODENAME=jammy**

so 

**76 isubuntu=0
 77 if [ `grep '^NAME=' /etc/os-release | grep -c Ubuntu`-eq 1 ]; then
 78         isubuntu=1
 79 fi**
(the back-ticks are there but the editor doesn't like them)

doesn't find anything...
something more like

**if [  $(grep -ic 'Ubuntu' /etc/os-release)  -ge 1 ]; then**   ...

would catch most of the seemingly ever growing number of ubuntu-bassed distributions (hopefully)

### Re ...cs8409.ko

Ive just done a **ls -l snd_hda_macbookpro/build/hda-5.15.0/ | grep cs8409** 
in case I was really going mad this time, but what that shows is:

**-rw-r--r-- 1 root root  44463 Nov  7 21:57 patch_cs8409.c
-rw-r--r-- 1 root root  43857 Oct 13 09:03 patch_cs8409.c.orig
-rw-r--r-- 1 root root  16253 Nov  7 21:57 patch_cs8409.h
-rw-r--r-- 1 root root   9721 Oct 13 09:03 patch_cs8409.h.orig
-rw-r--r-- 1 root root    290 Nov  7 21:57 patch_cs8409.h.rej
-rw-r--r-- 1 root root  24587 Oct 13 09:03 patch_cs8409-tables.c**

which is puzzling... 
I'll get back to you after I've tried recompiling from the generic kernel and had a further think
n

davidjo, 2022-11-10:
You can just blast isubuntu=1 in the script.
Have thought about using ID_LIKE in addition (I try to keep things as explicit as possible).
When Im debugging after an initial script run I tend to just go into eg build/hda.5.15.0 and do manual commands
eg make to build, sudo make install to install and make clean to clear out a previous build

NialMcDJ, 2022-11-10:
Cheers David,

Re: "you can ... **isubuntu=1**  etc" -- that's how my install script  has read since I realised that this was a problem -- my suggestion (inelegant as it was)  was an attempt to improve the script for future users (without having to resort to separate conditions for each of a dozen + ubuntu-based distributions. Of course the problem with it lies in the assumption that all of them have a an 
**os-release** file that conforms to the pattern...

Re Header files -- fortunately  I have appropriate K-headers installed for all installed kernels so I'm not having that problem.

Re "when I'm debugging ... " -- I'm currently trying to do exactly that, so it's good to know I'm approaching this in a sensible fashion.

Currently not having much success however (not even ...cirrus.ko is compiling / installing) so at the moment I'm trying to understand enough about **make** (I'm somewhat rusty), to work out where it is getting lost and about what -- there definitely seems to be a problem finding **System.map.5.15.0-52-generic** for **depmod** and also finding **/lib/modules/5.15.0-52-generic/modules.order** from **modules_install** -- but I don't understand why yet.

Anyway thanks again and I'll hopefully be able to come back to you shortly with something more sensible.
n

davidjo, 2022-11-17:
Well you have defintely got my module loaded - its not signed hence the module verifcation failure.
I suggest trying the waveforms under tests - do the low level one first StereoTest32_reduced_m24dB.wav.
command is:
aplay -D plughw StereoTest32_reduced_m24dB.wav (some audio processing)
or
aplay -D hw StereoTest32_reduced_m24dB.wav (direct to hardware)
also
aplay -D hw:0,0 StereoTest32_reduced_m24dB.wav  (fully explicit device)
The hw device bypasses all audio volume etc processing!!
You can use tail -f /var/log/syslog (in a separate terminal window) to see what is logged when you start playing - should get some messages.

NialMcDJ, 2022-11-17:
Thank you David

**aplay -D plughw StereoTest32_reduced_m24dB.wav
aplay -D hw StereoTest32_reduced_m24dB.wav
aplay -D hw:0,0 StereoTest32_reduced_m24dB.wav**

and

**aplay -D surround40 StereoTest32_reduced_m24dB.wav**

all fail to execute with:
**aplay: main:831: audio open error: No such file or directory**

nothing arrives in syslog

**aplay StereoTest32_reduced_m24dB.wav**

succeeds with
**Playing WAVE 'StereoTest32_reduced_m24dB.wav' : Signed 32 bit Little Endian, Rate 44100 Hz, Channels 4**

but of course no sound and again (possibly as you'd expect) nothing arrives in syslog

**aplay -l** gives us:
```
**** List of PLAYBACK Hardware Devices ****

card 1: HDMI [HDA ATI HDMI], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 11: HDMI 5 [HDMI 5]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 12: HDMI 6 [HDMI 6]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```
[Card 1 -- ie the hdmi sound side of the Graphics card -- which shows up as turned off in sound preferences (pulse?)]

**pacmd list-cards** 
also shows hdmi card only (active profile off)

**aplay -L**

gives us this:

```
null
    Discard all samples (playback) or generate zero samples (capture)
default
    Playback/recording through the PulseAudio sound server
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pipewire
    PipeWire Sound Server
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
front:CARD=PCH
    Front output / input
surround21:CARD=PCH
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=PCH
    4.0 Surround output to Front and Rear speakers
surround41:CARD=PCH
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=PCH
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=PCH
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=PCH
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
usbstream:CARD=PCH
    HDA Intel PCH
    USB Stream Output
hw:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 0
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=7
    HDA ATI HDMI, HDMI 1
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=8
    HDA ATI HDMI, HDMI 2
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=9
    HDA ATI HDMI, HDMI 3
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=10
    HDA ATI HDMI, HDMI 4
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=11
    HDA ATI HDMI, HDMI 5
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=12
    HDA ATI HDMI, HDMI 6
    Direct hardware device without any conversions
plughw:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 0
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=7
    HDA ATI HDMI, HDMI 1
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=8
    HDA ATI HDMI, HDMI 2
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=9
    HDA ATI HDMI, HDMI 3
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=10
    HDA ATI HDMI, HDMI 4
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=11
    HDA ATI HDMI, HDMI 5
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=12
    HDA ATI HDMI, HDMI 6
    Hardware device with all software conversions
hdmi:CARD=HDMI,DEV=0
    HDA ATI HDMI, HDMI 0
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=1
    HDA ATI HDMI, HDMI 1
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=2
    HDA ATI HDMI, HDMI 2
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 3
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=4
    HDA ATI HDMI, HDMI 4
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=5
    HDA ATI HDMI, HDMI 5
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=6
    HDA ATI HDMI, HDMI 6
    HDMI Audio Output
dmix:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 0
    Direct sample mixing device
dmix:CARD=HDMI,DEV=7
    HDA ATI HDMI, HDMI 1
    Direct sample mixing device
dmix:CARD=HDMI,DEV=8
    HDA ATI HDMI, HDMI 2
    Direct sample mixing device
dmix:CARD=HDMI,DEV=9
    HDA ATI HDMI, HDMI 3
    Direct sample mixing device
dmix:CARD=HDMI,DEV=10
    HDA ATI HDMI, HDMI 4
    Direct sample mixing device
dmix:CARD=HDMI,DEV=11
    HDA ATI HDMI, HDMI 5
    Direct sample mixing device
dmix:CARD=HDMI,DEV=12
    HDA ATI HDMI, HDMI 6
    Direct sample mixing device
usbstream:CARD=HDMI
    HDA ATI HDMI
    USB Stream Output

```
alsa seems to be attaching **hw**, **plughw** etc to the hdmi card

I hope this moves us forward a bit...

n

davidjo, 2022-11-17:
yes thats something you need to watch for - the HDMI sound device.
So what I noticed is you have a pipewire server with aplay -L - on a base ubuntu linux machine I have (not mac) I dont see it.
Current status of pipewire is weird on Ubuntu - it has some elements of pipewire but not apparently the full pipewire server - at least on 22.04 (22.10  I believe switches to pipewire fully).
You showed some syslog snd_hda_codec_cs8409 messages - were those the only messages you saw?
grep through the syslog for snd_hda messages (exclude C1D1 which should be your HDMI device - your log listed hdaudioC0D0 as the cs8409) - there should be other messages - also those with snd_hda_intel
In particular look for snd_hda_intel: Primary patch_cs8409 NOT FOUND trying APPLE

NialMcDJ, 2022-11-17:
**/var/log/syslog | grep snd_hda** 
only reveals
```
Nov 17 13:33:05 Epona systemd-modules-load[438]: Inserted module 'snd_hda_core'
Nov 17 13:33:05 Epona systemd-modules-load[438]: Inserted module 'snd_hda_codec_cs8409'
Nov 17 13:33:05 Epona kernel: [   17.433243] snd_hda_codec_cs8409: loading out-of-tree module taints kernel.
Nov 17 13:33:05 Epona kernel: [   17.434984] snd_hda_codec_cs8409: module verification failed: signature and/or required key missing - tainting kernel
Nov 17 13:33:05 Epona kernel: [   18.288438] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
Nov 17 13:33:05 Epona kernel: [   18.288496] snd_hda_intel 0000:01:00.1: Force to non-snoop mode
Nov 17 13:33:05 Epona kernel: [   18.300754] snd_hda_codec_cs8409 hdaudioC0D0: autoconfig for CS8409: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
Nov 17 13:33:05 Epona kernel: [   18.300758] snd_hda_codec_cs8409 hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Nov 17 13:33:05 Epona kernel: [   18.300760] snd_hda_codec_cs8409 hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
Nov 17 13:33:05 Epona kernel: [   18.300762] snd_hda_codec_cs8409 hdaudioC0D0:    mono: mono_out=0x0
Nov 17 13:33:05 Epona kernel: [   18.300763] snd_hda_codec_cs8409 hdaudioC0D0:    inputs:
Nov 17 13:33:05 Epona kernel: [   18.300765] snd_hda_codec_cs8409 hdaudioC0D0:      Internal Mic=0x45
Nov 17 13:33:05 Epona kernel: [   18.300766] snd_hda_codec_cs8409 hdaudioC0D0:      Mic=0x3c
Nov 17 13:33:05 Epona kernel: [   18.330620] snd_hda_intel 0000:01:00.1: bound 0000:01:00.0 (ops amdgpu_dm_audio_component_bind_ops [amdgpu])
```
repeated for each boot.  Is however the last line of any significance?

Also there are _no_ references to hdaudioC1D1 (and none incidentally in dmesg) and -- having just checked -- there have not been in the past on this system.

grepping for **sound** : this _only_ reveals references to the hdmi card of the following sort

`Nov 17 13:33:05 Epona kernel: [   18.331842] input: HDA ATI HDMI HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input10
`
again cycling in sequence per boot

grepping for **audio**

reveals messages from jack (complaining about lack of real-time facilities)
and messages from pulse
```
Nov 17 13:33:31 Epona pulseaudio[2781]: The decibel volume range for element 'PCM' (-7300 dB - -100 dB) has negative maximum. Disabling the decibel range.
Nov 17 13:33:31 Epona pulseaudio[2781]: message repeated 4 times: [ The decibel volume range for element 'PCM' (-7300 dB - -100 dB) has negative maximum. Disabling the decibel range.]
Nov 17 13:33:31 Epona pulseaudio[2781]: Failed to find a working profile.
Nov 17 13:33:31 Epona pulseaudio[2781]: Failed to load module "module-alsa-card" (argument: "device_id="0" name="pci-0000_00_1f.3" card_name="alsa_card.pci-0000_00_1f.3" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.
Nov 17 13:33:31 Epona pulseaudio[2781]: Could not find org.bluez.BatteryProviderManager1.RegisterBatteryProvider(), is bluetoothd started with experimental features enabled (-E flag)?
```
These again repeat at each boot-time 
The last line is obviously irrelevant but the first four seem to hold some significance

It begins to look as if my system is somewhat confused (new clean install on 11/10/22 -- but has only been bootable since 29/10/22 -- long story) 

I would lose very little by wiping it and reinstalling if that begins to look necessary -- but would obviously prefer not to...

NialMcDJ, 2022-11-17:
also this is of interest...

**cat /proc/asound/PCH/codec#0** 
[proc-asound-PCH-codec#0.txt](https://github.com/davidjo/snd_hda_macbookpro/files/10035209/proc-asound-PCH-codec.0.txt)

davidjo, 2022-11-17:
Look for the snd_hda_intel Primary Patch and snd_hda_intel Primary patch_cs8409 NOT FOUND trying APPLE messages.
These are standard printk messages (see patch_cs8409.c function patch_cs8409) so should appear.
If not somethings wrong.

NialMcDJ, 2022-11-18:
The only kernel messages re  snd_hda_intel in syslog to date are the three above in the output of 
cat /var/log/syslog | grep snd_hda
-- lines 5 and 6 and the last line -- there was however an extra one from the most recent boot when I ran 
**cat syslog | grep snd_hda_intel**      just now
the top one is the new one and it is probably there because I added it to modules.conf in case it made a difference loading it earlier
```
Nov 18 00:08:14 Epona systemd-modules-load[450]: Inserted module 'snd_hda_intel'
Nov 18 00:08:14 Epona kernel: [   15.968194] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
Nov 18 00:08:14 Epona kernel: [   15.971644] snd_hda_intel 0000:01:00.1: Force to non-snoop mode
Nov 18 00:08:14 Epona kernel: [   15.988468] snd_hda_intel 0000:01:00.1: bound 0000:01:00.0 (ops amdgpu_dm_audio_component_bind_ops [amdgpu])
```
the situation is the same in the archived syslogs (syslog.1 and syslog.2.gz)
there are plenty of messages containing  APPLE in these files but all in relation to disk access -- there are no references to "trying APPLE", "Primary Patch" or "patch_cs8409" in any of these logs
and the same apples to Kern.log, dmesg and their archived versions and to the systemd journal
(there are references to patch_cs8409 in the journal but only in the context of building the module not in terms of system activity)

so we have to conclude something is wrong...
also
I see the **printk** statements in 
**patch_cirrus/patch_cs8409.c** 
but not in 
**build/hda-5.15.0/snd-hda-codec-cs8409.mod.c**

NialMcDJ, 2022-11-18:
Having written that last update yesterday I was struck by a terrible thought, which was confirmed by rereading the patch man pages -- the  -v switch does not make patch more verbose, it prints a version statement to stdout and exits 0... this meant two things: 

1.  the last two hunks of the patch to cs8409. h were not being applied whilst still being treated as a success, and 
2.  although compilation occurred under these circumstances, this was only by luck, and the resultant compiled module was not the one desired. 

Please accept my apologies for the amount of time and effort of thought this mistake has wasted.

This leaves me back in a situation where patching fails for hunk #3 of  patch_ cs8409. h and compilation once again fails in the way described by my first post.  Unfortunately adding --verbose to the patch commands is not a lot of help in trying to work out why the patching is going wrong.

I'm going away to play with this for a bit...

davidjo, 2022-11-18:
Hey no problem - you have no idea how many similar errors I make.

NialMcDJ, 2022-11-18:
I still feel a little mortified re the `patch -v` gaff despite your kind words 
however...

-  `sudo make clean `
     (which missed the resident `cs8409.ko` in /updates -- so I removed by hand)
- ` sudo ./install.cirrus.driver.sh`
            patching patch_cs9409.h fails for hunk#3
            compilation fails because of this
- `cd build/hda-5.15.0/`
       hand patched 
      `patch_cs9409.h` at line 332 with the errant and rejected
```    
            #ifdef APPLE_CODECS
                 unsigned int linein_jack_in:1;
            # endif
``` 
-    removed the `force_status_change` line from 
           `build/hda-5.15.0/patch_cirrus_apple.h`
      which had somehow slithered through the earlier patching
 
- `sudo make`
 
- `sudo make install`
 
- reboot

and... 

**Sound!**    

-    dmsg contains
    ```
        [  +0.016584] snd_hda_intel: Primary patch_cs8409
        [  +0.004928] snd_hda_intel: Primary patch_cs8409 NOT FOUND trying APPLE
    ```
-    alsamixer tells me      
 ```
        Card: HDA Intell PCH
        Chip: Cirrus Logic CS8409/CS42L83
```
-    I  applied slynobody's suround 4.0 settings  (from issue 56)
     I adjusted the woofer/tweeter (= back/front) balance in sound settings slightly in favour of the woofers    
     and everything sounds pretty good 
 -   the  analogue stereo output (which presumably just doubles the stereo signal front and back) is also not at all bad            
    on my system
    
**problems**
so far
-    a little noise from sound controls both via keyboard and via the mate sound settings (pulse front-end) -- clicks and slight burrs
-    headphones work fine but don't seem keen to relinquish control back to the built-in speakers (had to reboot)

**Also**
-  I have repeated the process and the module works just as well under a low-latency kernel  -- currently writing from 5.15.0-53-lowlatency
-    I've not tried input yet

I'll keep testing and get back to you shortly. 
Thank you for all your help as well as the module itself -- when I started looking at this issue three or four years ago I despaired to the extent that I was driven back to using Mac OS exclusively for a long time (and Linux has been my operating system of choice since the late 90s).
I'm currently luxuriating in high quality linux sound on my machine -- what a relief.

nikhilss219, 2022-12-05:
Hi @NialMcDJ, I have Ubuntu 20.04 in iMac 2019 5k on dual boot. I have followed the installation using this patch but with no success, also i have no issue with the quality of sound output, because i am getting an output in sound. I am very new to this and tried my luck with this dual boot. I would love to only work with ubuntu system.I have lots of meeting to attend and i need the microphone to work somehow, is there any solution on how to achieve mic input.

davidjo, 2022-12-05:
I would suggest getting either a USB or bluetooth headset with mike or just mike - those should work out of the box.
When I first worked on this on the MacBookPro even though I got the internal mike to work it is at a very low volume - but as far as I can tell this is the input from the chip (there are no amps on the CS8409 chip) so OSX must be amplifying it at a higher level of the OSX kernel.
For testing I got a reasonable level by amplifying in Audacity.
There are amplifiers plugins available for either pulse or ALSA which you can connect to the mike input to make  the sound a reasonable volume at user level. This just needs some config files (text) writing for ALSA/pulse.
However there have also been issues with sound input on iMacs reported to me (I dont have an iMac) so dont know if this would work for you (one issue was eg the iMacs used a switched polarity for the jack plug cf. the MacBookPro but that was fixed).

nikhilss219, 2022-12-05:
Hi @davidjo , thanks for the quick response, I tried to connect the bluetooth headset to the iMac, sound output was shown as bluetooth headset and i got sounds in the headset , but input was still detected as internal microphone, is there any tweak to bring the input section to show the bluetooth headset

davidjo, 2022-12-05:
Thats sounds annoying - as far as I am aware if the bluetooth headset has a mike it should appear as an input in the user audio settings GUI panel (part of the main Settings GUI) which you should be able to set to the bluetooth headset (as it appears the auto switching may not be working for some reason - which I have had problems with in general) - I do not have USB/bluetooth headsets so cant help much more - its just something Ive seen on  sites which discuss which bits work and which dont for Mac machines.
You can also manually switch the outputs in same GUI.

spblinux, 2023-03-04:
Thanks for your work! To add build support for Linux Mint 21 (tested with kernel 5.15.0-67-generic):

Simply duplicate in installer.cirrus.driver.sh the lines
`if [ `grep '^NAME=' /etc/os-release | grep -c Ubuntu` -eq 1 ]; then`
`   isubuntu=1`
`fi`
and change Ubuntu to "Linux Mint" with double quotes
`if [ `grep '^NAME=' /etc/os-release | grep -c "Linux Mint"` -eq 1 ]; then`
`   isubuntu=1`
`fi`

Additional note: in case of my iMac 2017 the output setting analog surround 4.0 works fine. (Analog stereo works as well.)
[install.cirrus.driver.sh.diff.zip](https://github.com/davidjo/snd_hda_macbookpro/files/10889281/install.cirrus.driver.sh.diff.zip)

davidjo, 2023-03-07:
Thanks for the update - Ill add this to the main repo plus the new internal mike updates.