davidjo/snd_hda_macbookpro #77: Cannot Build Audio Driver In Kubuntu

sach491, 2022-12-05:
I am on Kubuntu 22.04. I am currently having trouble building the audio driver for my Macbook Pro 14,1. This is my first time using linux, so apologies if I am not providing enough information. This is the error I get whenever I put the 3rd command:

-----------------------------------------------------------------------------------------------------------------
kubuntu@kubuntu:~/snd_hda_macbookpro$ sudo ./install.cirrus.driver.sh
Ubuntu linux kernel source not found in /usr/src: /usr/src/linux-source-5.15.0.tar.bz2
assuming the linux kernel source package is not installed
please install the linux kernel source package:
sudo apt install linux-source-5.15.0
NOTE - This does not work for HWE kernels
-----------------------------------------------------------------------------------------------------------------

Whenever I put the "sudo apt install linux-source-5.15.0", it says it is not available.

-----------------------------------------------------------------------------------------------------------------
kubuntu@kubuntu:~/snd_hda_macbookpro$ sudo apt install linux-source-5.15.0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package linux-source-5.15.0 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'linux-source-5.15.0' has no installation candidate
-----------------------------------------------------------------------------------------------------------------

I have no clue what to do. Any help is appreciated!

davidjo, 2022-12-05:
Ive never used kubuntu so dont know what changes they have made but looks like they have.
try:
apt search linux-source
and see what it says.
you can also just try
sudo apt install linux-source
Make sure you have also installed the build tools ie compilers, make, libraries etc
- there is a package on ubuntu build-essential
try
sudo apt install build-essential

sach491, 2022-12-05:
Sorry for the late response! Was busy today. Okay, so I have run both "sudo apt install linux-source" and "sudo apt install build-essential". It seems that now 3 out of the 4 succeeded, but still one got an error. Here's the error:

----------------------------------------------------------------------------------------------------------------------
sach@sach-MacBookPro:~/snd_hda_macbookpro$ sudo ./install.cirrus.driver.sh
linux-source-5.15.0/sound/pci/hda/
linux-source-5.15.0/sound/pci/hda/patch_analog.c
linux-source-5.15.0/sound/pci/hda/ideapad_s740_helper.c
linux-source-5.15.0/sound/pci/hda/patch_cmedia.c
linux-source-5.15.0/sound/pci/hda/patch_via.c
linux-source-5.15.0/sound/pci/hda/Makefile
linux-source-5.15.0/sound/pci/hda/hda_intel.h
linux-source-5.15.0/sound/pci/hda/patch_cs8409.h
linux-source-5.15.0/sound/pci/hda/hda_beep.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.c
linux-source-5.15.0/sound/pci/hda/hda_sysfs.c
linux-source-5.15.0/sound/pci/hda/patch_sigmatel.c
linux-source-5.15.0/sound/pci/hda/hda_generic.h
linux-source-5.15.0/sound/pci/hda/hda_beep.h
linux-source-5.15.0/sound/pci/hda/hda_intel_trace.h
linux-source-5.15.0/sound/pci/hda/hda_component.h
linux-source-5.15.0/sound/pci/hda/hda_jack.h
linux-source-5.15.0/sound/pci/hda/hda_controller.h
linux-source-5.15.0/sound/pci/hda/patch_cs8409-tables.c
linux-source-5.15.0/sound/pci/hda/hda_controller.c
linux-source-5.15.0/sound/pci/hda/patch_cirrus.c
linux-source-5.15.0/sound/pci/hda/hda_local.h
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-source-5.15.0/sound/pci/hda/hda_proc.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.h
linux-source-5.15.0/sound/pci/hda/patch_ca0110.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.c
linux-source-5.15.0/sound/pci/hda/hda_jack.c
linux-source-5.15.0/sound/pci/hda/patch_si3054.c
linux-source-5.15.0/sound/pci/hda/hda_codec.c
linux-source-5.15.0/sound/pci/hda/hda_eld.c
linux-source-5.15.0/sound/pci/hda/Kconfig
linux-source-5.15.0/sound/pci/hda/patch_conexant.c
linux-source-5.15.0/sound/pci/hda/patch_realtek.c
linux-source-5.15.0/sound/pci/hda/hda_generic.c
linux-source-5.15.0/sound/pci/hda/hda_hwdep.c
linux-source-5.15.0/sound/pci/hda/hp_x360_helper.c
linux-source-5.15.0/sound/pci/hda/ca0132_regs.h
linux-source-5.15.0/sound/pci/hda/patch_cs8409.c
linux-source-5.15.0/sound/pci/hda/patch_hdmi.c
linux-source-5.15.0/sound/pci/hda/hda_controller_trace.h
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_spi.c
linux-source-5.15.0/sound/pci/hda/hda_bind.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.h
linux-source-5.15.0/sound/pci/hda/patch_ca0132.c
linux-source-5.15.0/sound/pci/hda/thinkpad_helper.c
linux-source-5.15.0/sound/pci/hda/hda_tegra.c
linux-source-5.15.0/sound/pci/hda/hda_intel.c
patching file patch_cs8409.c
Hunk #1 succeeded at 1442 (offset 172 lines).
Hunk #2 succeeded at 1451 (offset 172 lines).
Hunk #3 succeeded at 1480 (offset 172 lines).
patching file patch_cs8409.h
Hunk #1 succeeded at 19 (offset 1 line).
Hunk #2 succeeded at 302 (offset 1 line).
Hunk #3 FAILED at 328.
Hunk #4 succeeded at 362 (offset 2 lines).
1 out of 4 hunks FAILED -- saving rejects to file patch_cs8409.h.rej
make -C /lib/modules/5.15.0-56-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/sach/snd_hda_macbookpro/build/hda-5.15.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-56-generic'
  CC [M]  /home/sach/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o
In file included from /home/sach/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.c:1486:
/home/sach/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:157:10: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
  157 |         .linein_jack_in = 0,
      |          ^~~~~~~~~~~~~~
      |          mic_jack_in
[... 13 lines cut ...]
/home/sach/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_imac_exec_verb’:
/home/sach/snd_hda_macbookpro/build/hda-5.15.0/patch_cirrus_apple.h:2392:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2392 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
make[2]: *** [scripts/Makefile.build:297: /home/sach/snd_hda_macbookpro/build/hda-5.15.0/patch_cs8409.o] Error 1
make[1]: *** [Makefile:1903: /home/sach/snd_hda_macbookpro/build/hda-5.15.0] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-56-generic'
make: *** [Makefile:22: all] Error 2
#cp snd-hda-codec-cs8409.ko /lib/modules/5.15.0-56-generic/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.15.0-56-generic/build M=/home/sach/snd_hda_macbookpro/build/hda-5.15.0 modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-56-generic'
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
cat: /home/sach/snd_hda_macbookpro/build/hda-5.15.0/modules.order: No such file or directory
  DEPMOD  /lib/modules/5.15.0-56-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-56-generic'
depmod -a

contents of /lib/modules/5.15.0-56-generic/updates
total 0

-------------------------------------------------------------------------------------------------------------------------------------------------------

Also, my kernel version is 5.15.0-56-generic (64-bit). Hopefully this'll be enough information to help figure out the issue. Thanks for the quick response!

sach491, 2022-12-06:
So is there any possible solution for this? I want to set up my linux distro but am afraid all that would be worthless if the sound ends up not being able to work

davidjo, 2022-12-07:
The fixups are listed in other issues.
I working on updates to push to the repo (trying to handle how to have multiple patch files depending on kernel version).
At the moment I have a push ready but having problems updating repo - github is refusing the push login even though Ive generated a new PAT.

sach491, 2022-12-07:
Ah alright. Which issues should I be looking in for the fixes?

NialMcDJ, 2022-12-07:
For some reason there is a problem patching patch_cs8409.h -- the patch that fails
is three lines, which need to be inserted at line 328 -- the patch fails, so compilation of the module fails because of this, and there is no kernel module to install. It is beyond my current level of skill to work out why this is happening so I cannot suggest a fix -- however this is how I worked around it:

I have tried to pitch what follows  at a level suitable for someone who is new to linux -- if it is too detailed / explanatory I apologise (and please let me know).  Likewise if there's anything that I've not explained properly, please do say.

- **Firstly** make sure you haven't ended up with a broken copy of `snd-hda-codec-cs8409.ko`
  in `/lib/modules/5.15.0-56-generic/updates/`  (this is as long as you are still using the same kernel -- otherwise uname -r will tell you which modules directory you ought to look in).
  
  If there is a `snd-hda-codec-cs8409.ko` in the **updates** directory **delete it** by hand 
  ( `sudo rm -f`  ... `/updates/snd-hda-codec-cs8409.ko` ) as I found that `make clean` didn't  
  always behave and remove it as it should.

- **Secondly** `cd` into the `snd_hda_macbookpro/` directory and run 
  `sudo ./install.cirrus.driver.sh` again... 
   this will fail in exactly the same way, but it has the virtue of flushing out and rebuilding  / 
   repopulating    the `build/hda-5.15.0/` directory and freshly patching the majority of the 
   changes into the new files.

- **Step three** `cd` to `build/hda-5.15.0/`
   looking at your log from the install.sh above, the patching has missed a line which needed  
   removing from `patch_cirrus_apple.h`. Open this in a text editor (you will need something  
   which can supply line numbers and which produces only text. I use vim but I wouldn't
   recommend it for this if you haven't already spent time with it -- maybe try nano?
   In nano M-N toggles line numbering (here M=Meta=option=alt) and there's a search 
   function...
      -  so `nano patch_cirrus_apple.h` opens the file and  **<Alt-N>** turns on line numbers
      -  go to **line 158** --  you are now in a data structure called  ‘struct sub_codec’
         and line 158 says **force_status_change = 1** -- **delete** this line,  <Ctrl-O> saves your    
         changes (after asking you if you're sure) and <Ctrl-X> exits the editor.
         ok then...
     -  now for the slightly more extensive change -- firstly open a new terminal as you'll need to 
         view the contents of 2 files...  again `cd /snd_macbookpro/build/hda5.15.0/`
          -  In one terminal `nano sudo patch_cs8409.h` -- this is the file with the missing lines
          -  In the other `sudo nano patch_cs8409.h.rej` -- these are the 3 lines that  `patch` had a 
             problem with earlier.
     -  In **patch_cs8409.h** turn on line numbers and navigate to line **328**, here open a 
        new line so that you can paste into it.
     -  In **patch_cs8409.h.rej** copy these three lines starting
       ```
         +#ifdef APPLE_CODECS
         +      unsigned int linein_jack_in:1;
         +#endif
     ```
        (the simplest way to do this in nano appears to be to select the lines with your mouse and   
        right click them choosing the copy option from the contextual menu). [ Also you can copy    
        them from here if you like].
    -   Paste the lines into **patch_cs8409.h** at the empty line (again simplest is right click on  
        the line and choose paste).
        Now delete the three + signs (these are instructions to patch which we no longer need). 
        You can "tidy" up by bringing the copied material into line with the surrounding code if you  
        like, but this has no functional effect.
   -   Save and close **patch_cs8409.h**, and close **patch_cs8409.h.rej**.

               The most complicated part of this is now behind you...

-  **Finally** whilst still in the terminal  snd_macbookpro/build/hda5.15.0/
   -  `sudo make all`  This should now compile the module `snd-hda-codec-cs8409.ko`. There will 
      be warnings (notably one regarding SLL and kernal module signing), but you can safely  
      ignore all these and (hopefully) compilation will complete without errors. If all has gone to   
      plan you will have a shiny new `snd-hda-codec-cs8409.ko` sitting in the hda5.15.0 directory 
      (use `ls` to confirm).
   -  **Last step** Still in the ...  build/hda5.15.0 directory...
       **sudo make install** this moves the module to` /lib/modules/Your_Kernel_Version/updates/ `
        and runs `depmod -a` to ensure that the kernel knows about all  current modules and their  
        (inter)/dependencies. You'll get some warnings here (more re kernel module signing)       
       `depmod` fails the first time round but succeeds immediately thereafter.

        Now `ls /lib/modules/Your_Kernel_Version/updates` to make sure the module has arrived safely.
        if not you'll have to install by hand:
        run 'sudo mv ./snd-hda-codec-cs8409.ko  /lib/modules/Your_Kernel_Version/updates/`  and

NialMcDJ, 2022-12-07:
oops...
and `sudo depmod -a`

now reboot and see what happens...
n

sach491, 2022-12-07:
Wow that's a lot! Thank you! I'll try this once I get back home (which will be a while cause it's the morning for me right now), and I'll try this and update with what happens. Just to be clear, if this works, the audio on my macbook will work right?

davidjo, 2022-12-07:
Ive have just updated the repo finally (it appears I need to specifically add what permissions the PAT is for each time its generated).
if you go into your clone directory and do git pull it will update your repo to the current version and you should just be able to run the install script and it should just work.
Probably best if you sudo rm -fr build in the clone directory before running the install script to clear out any old builds.
There are no real guarantees for this repo - it is based on reverse engineering Apples OSX driver as Apple uses proprietary chips for which we have no documentation and Apple now doesnt provide source code access to OSX kernel.
If you search the issues you will see people have said it has worked for them for the 14,1 model (my model is 14,3 so its the only one I can definitively report on).

sach491, 2022-12-07:
I will try this when I arrive home, thank you! Hopefully this works! I will update with the outcome

davidjo, 2022-12-07:
Im assuming you cloned the repo using git clone .....
You then went into the directory created above for the repo - in order to run the install script - I mean in that directory.
Then just type:
git pull
Im assuming Konsole is Kubuntus terminal - so you would do this in Konsole.
This is how you update a git clone from the original repo.
Yes you can just delete everything and re-clone etc.

closed 2022-12-07 by sach491