davidjo/snd_hda_macbookpro #78: Patch failed?

slynobody, 2022-12-13:
Thank you for updating, but may it be possible you overlooked something?

patching file patch_cs8409.c
Hunk #1 succeeded at 1444 (offset 174 lines).
Hunk #2 succeeded at 1453 (offset 174 lines).
Hunk #3 succeeded at 1482 (offset 174 lines).
patching file patch_cs8409.h
Hunk #1 succeeded at 19 (offset 1 line).
Hunk #2 succeeded at 303 (offset 2 lines).
Hunk #3 FAILED at 328.
Hunk #4 succeeded at 363 (offset 3 lines).
1 out of 4 hunks FAILED -- saving rejects to file patch_cs8409.h.rej

i did not had the time to look into it and reverted to an older state with older workarounds.
Thank you for your work!

davidjo, 2022-12-13:
I can only test on my ubuntu LTS system with current kernel 5.15.0-56 at the time of commit which worked.
What is your OS and current kernel version?
Im attempting to try and handle multiple patch kernel versions depending on OS but didnt really check that.
Currently the idea is the current kernel and ubuntu LTS version (at the time of the commit) work with unversioned files
but previous versions now use different files - but that could not be working right.

antonis19992, 2022-12-13:
the same with me kali linux kernel 6.0 but before a format i do it works now the same error

antonis19992, 2022-12-13:
(6.0.10-2kali1)

antonis19992, 2022-12-13:
AFTER REMOVE HEADERS AN REINSTALL THIS PROBLEM CHANGED TO THIS
 The file is already fully retrieved; nothing to do.

linux-6.0/sound/pci/hda/
linux-6.0/sound/pci/hda/Kconfig
linux-6.0/sound/pci/hda/Makefile
linux-6.0/sound/pci/hda/ca0132_regs.h
linux-6.0/sound/pci/hda/cs35l41_hda.c
linux-6.0/sound/pci/hda/cs35l41_hda.h
linux-6.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-6.0/sound/pci/hda/cs35l41_hda_spi.c
linux-6.0/sound/pci/hda/hda_auto_parser.c
linux-6.0/sound/pci/hda/hda_auto_parser.h
linux-6.0/sound/pci/hda/hda_beep.c
linux-6.0/sound/pci/hda/hda_beep.h
linux-6.0/sound/pci/hda/hda_bind.c
linux-6.0/sound/pci/hda/hda_codec.c
linux-6.0/sound/pci/hda/hda_component.h
linux-6.0/sound/pci/hda/hda_controller.c
linux-6.0/sound/pci/hda/hda_controller.h
linux-6.0/sound/pci/hda/hda_controller_trace.h
linux-6.0/sound/pci/hda/hda_cs_dsp_ctl.c
linux-6.0/sound/pci/hda/hda_cs_dsp_ctl.h
linux-6.0/sound/pci/hda/hda_eld.c
linux-6.0/sound/pci/hda/hda_generic.c
linux-6.0/sound/pci/hda/hda_generic.h
linux-6.0/sound/pci/hda/hda_hwdep.c
linux-6.0/sound/pci/hda/hda_intel.c
linux-6.0/sound/pci/hda/hda_intel.h
linux-6.0/sound/pci/hda/hda_intel_trace.h
linux-6.0/sound/pci/hda/hda_jack.c
linux-6.0/sound/pci/hda/hda_jack.h
linux-6.0/sound/pci/hda/hda_local.h
linux-6.0/sound/pci/hda/hda_proc.c
linux-6.0/sound/pci/hda/hda_sysfs.c
linux-6.0/sound/pci/hda/hda_tegra.c
linux-6.0/sound/pci/hda/hp_x360_helper.c
linux-6.0/sound/pci/hda/ideapad_s740_helper.c
linux-6.0/sound/pci/hda/patch_analog.c
linux-6.0/sound/pci/hda/patch_ca0110.c
linux-6.0/sound/pci/hda/patch_ca0132.c
linux-6.0/sound/pci/hda/patch_cirrus.c
linux-6.0/sound/pci/hda/patch_cmedia.c
linux-6.0/sound/pci/hda/patch_conexant.c
linux-6.0/sound/pci/hda/patch_cs8409-tables.c
linux-6.0/sound/pci/hda/patch_cs8409.c
linux-6.0/sound/pci/hda/patch_cs8409.h
linux-6.0/sound/pci/hda/patch_hdmi.c
linux-6.0/sound/pci/hda/patch_realtek.c
linux-6.0/sound/pci/hda/patch_si3054.c
linux-6.0/sound/pci/hda/patch_sigmatel.c
linux-6.0/sound/pci/hda/patch_via.c
linux-6.0/sound/pci/hda/thinkpad_helper.c
patching file patch_cs8409.c
Hunk #1 succeeded at 1444 (offset 174 lines).
Hunk #2 succeeded at 1453 (offset 174 lines).
Hunk #3 succeeded at 1482 (offset 174 lines).
patching file patch_cs8409.h
Hunk #1 succeeded at 19 (offset 1 line).
Hunk #2 succeeded at 303 (offset 2 lines).
Hunk #3 FAILED at 328.
Hunk #4 succeeded at 363 (offset 3 lines).
1 out of 4 hunks FAILED -- saving rejects to file patch_cs8409.h.rej
make -C /lib/modules/6.0.0-kali5-amd64/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/antonis/snd_hda_macbookpro/build/hda-6.0.0 modules
make[1]: Entering directory '/usr/src/linux-headers-6.0.0-kali5-amd64'
  CC [M]  /home/antonis/snd_hda_macbookpro/build/hda-6.0.0/patch_cs8409.o
In file included from /home/antonis/snd_hda_macbookpro/build/hda-6.0.0/patch_cs8409.c:1488:
/home/antonis/snd_hda_macbookpro/build/hda-6.0.0/patch_cirrus_apple.h:157:10: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
  157 |         .linein_jack_in = 0,
      |          ^~~~~~~~~~~~~~
      |          mic_jack_in
/home/antonis/snd_hda_macbookpro/build/hda-6.0.0/patch_cirrus_apple.h:158:10: error: ‘struct sub_codec’ has no member named ‘force_status_change’
  158 |         .force_status_change = 1,
      |          ^~~~~~~~~~~~~~~~~~~
/home/antonis/snd_hda_macbookpro/build/hda-6.0.0/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/home/antonis/snd_hda_macbookpro/build/hda-6.0.0/patch_cirrus_apple.h:1192:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1192 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
/home/antonis/snd_hda_macbookpro/build/hda-6.0.0/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_macbook_exec_verb’:
/home/antonis/snd_hda_macbookpro/build/hda-6.0.0/patch_cirrus_apple.h:2346:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2346 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
/home/antonis/snd_hda_macbookpro/build/hda-6.0.0/patch_cirrus_apple.h: In function ‘cs8409_cs42l83_imac_exec_verb’:
/home/antonis/snd_hda_macbookpro/build/hda-6.0.0/patch_cirrus_apple.h:2392:42: error: ‘struct sub_codec’ has no member named ‘linein_jack_in’; did you mean ‘mic_jack_in’?
 2392 |                         *res = (cs42l83->linein_jack_in) ? AC_PINSENSE_PRESENCE : 0;
      |                                          ^~~~~~~~~~~~~~
      |                                          mic_jack_in
make[2]: *** [/usr/src/linux-headers-6.0.0-kali5-common/scripts/Makefile.build:254: /home/antonis/snd_hda_macbookpro/build/hda-6.0.0/patch_cs8409.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.0.0-kali5-common/Makefile:1876: /home/antonis/snd_hda_macbookpro/build/hda-6.0.0] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.0.0-kali5-amd64'
make: *** [Makefile:22: all] Error 2
#cp snd-hda-codec-cs8409.ko /lib/modules/6.0.0-kali5-amd64/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/6.0.0-kali5-amd64/build M=/home/antonis/snd_hda_macbookpro/build/hda-6.0.0 modules_install
make[1]: Entering directory '/usr/src/linux-headers-6.0.0-kali5-amd64'
cat: /home/antonis/snd_hda_macbookpro/build/hda-6.0.0/modules.order: No such file or directory
  DEPMOD  /lib/modules/6.0.0-kali5-amd64
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-6.0.0-kali5-amd64'
depmod -a

contents of /lib/modules/6.0.0-kali5-amd64/updates
total 360
-rw-r--r-- 1 root root 361776 Dec 13 06:33 hci_uart.ko
drwxr-xr-x 7 root root   4096 Dec 13 07:16 snd_hda_macbookpro

antonis19992, 2022-12-13:
https://github.com/BenChaimberg/snd_hda_macbookpro/commits/master  thankk u my friend works now

slynobody, 2022-12-14:
this could also be done with this branch by simply
git checkout "`git rev-list master  -n 1 --first-parent --before=2022-12-05`"

btw: as this branch is the 'main' so to speak, and the other is a little bit left behind, you get a few other benefits though this.

davidjo, 2022-12-16:
Sorry - as I suspected the current script doesnt handle new kernel major versions - so it dropped back to the previous kernel version code ie pre the force_status_change variable removal.
Need to add a specific test for kernel major versions greater than my current kernel major version.

antonis19992, 2022-12-19:
i bet to you to provide something like this guy here with both the changes and also one more version that works the microphone  but it not works the volume button.

antonis19992, 2022-12-19:
so much searches and we are not programmers :P

antonis19992, 2022-12-19:
u are awesome anyways witbh your script thanks

slynobody, 2023-01-08:
should work now out of the box, thank you for your changes! (can be closed - while the signing-issue remains open, at least for me (same signing-errors)

closed 2023-08-06 by slynobody