davidjo/snd_hda_macbookpro #79: Possible error during compilation

lukehayes, 2023-01-03:
Hello, firstly I just wanted to say thank you for taking the time to make this. Your work has made the audio work in my iMac and I am incredibly grateful to you.

I have been using this project successfully over the last month or so and I needed to re-install Linux and so after that, I cloned this repo and went through the instructions and I can't get seem to get the audio now.

I'm not 100% sure compilation was successful to be honest.

Here is the output:

```
linux-5.19.6/sound/pci/hda/
linux-5.19.6/sound/pci/hda/Kconfig
linux-5.19.6/sound/pci/hda/Makefile
linux-5.19.6/sound/pci/hda/ca0132_regs.h
linux-5.19.6/sound/pci/hda/cs35l41_hda.c
linux-5.19.6/sound/pci/hda/cs35l41_hda.h
linux-5.19.6/sound/pci/hda/cs35l41_hda_i2c.c
linux-5.19.6/sound/pci/hda/cs35l41_hda_spi.c
linux-5.19.6/sound/pci/hda/hda_auto_parser.c
linux-5.19.6/sound/pci/hda/hda_auto_parser.h
linux-5.19.6/sound/pci/hda/hda_beep.c
linux-5.19.6/sound/pci/hda/hda_beep.h
linux-5.19.6/sound/pci/hda/hda_bind.c
linux-5.19.6/sound/pci/hda/hda_codec.c
linux-5.19.6/sound/pci/hda/hda_component.h
linux-5.19.6/sound/pci/hda/hda_controller.c
linux-5.19.6/sound/pci/hda/hda_controller.h
linux-5.19.6/sound/pci/hda/hda_controller_trace.h
linux-5.19.6/sound/pci/hda/hda_eld.c
linux-5.19.6/sound/pci/hda/hda_generic.c
linux-5.19.6/sound/pci/hda/hda_generic.h
linux-5.19.6/sound/pci/hda/hda_hwdep.c
linux-5.19.6/sound/pci/hda/hda_intel.c
linux-5.19.6/sound/pci/hda/hda_intel.h
linux-5.19.6/sound/pci/hda/hda_intel_trace.h
linux-5.19.6/sound/pci/hda/hda_jack.c
linux-5.19.6/sound/pci/hda/hda_jack.h
linux-5.19.6/sound/pci/hda/hda_local.h
linux-5.19.6/sound/pci/hda/hda_proc.c
linux-5.19.6/sound/pci/hda/hda_sysfs.c
linux-5.19.6/sound/pci/hda/hda_tegra.c
linux-5.19.6/sound/pci/hda/hp_x360_helper.c
linux-5.19.6/sound/pci/hda/ideapad_s740_helper.c
linux-5.19.6/sound/pci/hda/patch_analog.c
linux-5.19.6/sound/pci/hda/patch_ca0110.c
linux-5.19.6/sound/pci/hda/patch_ca0132.c
linux-5.19.6/sound/pci/hda/patch_cirrus.c
linux-5.19.6/sound/pci/hda/patch_cmedia.c
linux-5.19.6/sound/pci/hda/patch_conexant.c
linux-5.19.6/sound/pci/hda/patch_cs8409-tables.c
linux-5.19.6/sound/pci/hda/patch_cs8409.c
linux-5.19.6/sound/pci/hda/patch_cs8409.h
linux-5.19.6/sound/pci/hda/patch_hdmi.c
linux-5.19.6/sound/pci/hda/patch_realtek.c
linux-5.19.6/sound/pci/hda/patch_si3054.c
linux-5.19.6/sound/pci/hda/patch_sigmatel.c
linux-5.19.6/sound/pci/hda/patch_via.c
linux-5.19.6/sound/pci/hda/thinkpad_helper.c
patching file patch_cs8409.c
Hunk #1 succeeded at 1444 (offset 174 lines).
Hunk #2 succeeded at 1453 (offset 174 lines).
Hunk #3 succeeded at 1482 (offset 174 lines).
patching file patch_cs8409.h
Hunk #2 succeeded at 303 (offset 1 line).
Hunk #3 succeeded at 330 (offset 1 line).
Hunk #4 succeeded at 366 (offset 1 line).
patching file patch_cirrus_apple.h
make -C /lib/modules/5.19.6-arch1-1-t2/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/luke/snd_hda_macbookpro/build/hda-5.19.6 modules
make[1]: Entering directory '/usr/lib/modules/5.19.6-arch1-1-t2/build'
  CC [M]  /home/luke/snd_hda_macbookpro/build/hda-5.19.6/patch_cs8409.o
In file included from /home/luke/snd_hda_macbookpro/build/hda-5.19.6/patch_cs8409.c:1488:
/home/luke/snd_hda_macbookpro/build/hda-5.19.6/patch_cirrus_apple.h: In function ���cs_8409_apple_boot_init���:
/home/luke/snd_hda_macbookpro/build/hda-5.19.6/patch_cirrus_apple.h:1191:15: warning: assignment discards ���const��� qualifier from pointer target type [-Wdiscarded-qualifiers]
 1191 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
  CC [M]  /home/luke/snd_hda_macbookpro/build/hda-5.19.6/patch_cs8409-tables.o
  LD [M]  /home/luke/snd_hda_macbookpro/build/hda-5.19.6/snd-hda-codec-cs8409.o
  MODPOST /home/luke/snd_hda_macbookpro/build/hda-5.19.6/Module.symvers
  CC [M]  /home/luke/snd_hda_macbookpro/build/hda-5.19.6/snd-hda-codec-cs8409.mod.o
  LD [M]  /home/luke/snd_hda_macbookpro/build/hda-5.19.6/snd-hda-codec-cs8409.ko
  BTF [M] /home/luke/snd_hda_macbookpro/build/hda-5.19.6/snd-hda-codec-cs8409.ko
make[1]: Leaving directory '/usr/lib/modules/5.19.6-arch1-1-t2/build'
#cp snd-hda-codec-cs8409.ko /lib/modules/5.19.6-arch1-1-t2/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.19.6-arch1-1-t2/build M=/home/luke/snd_hda_macbookpro/build/hda-5.19.6 modules_install
make[1]: Entering directory '/usr/lib/modules/5.19.6-arch1-1-t2/build'
  INSTALL /lib/modules/5.19.6-arch1-1-t2/updates/snd-hda-codec-cs8409.ko
  SIGN    /lib/modules/5.19.6-arch1-1-t2/updates/snd-hda-codec-cs8409.ko
scripts/sign-file: error while loading shared libraries: libcrypto.so.1.1: cannot open shared object file: No such file or directory
  ZSTD    /lib/modules/5.19.6-arch1-1-t2/updates/snd-hda-codec-cs8409.ko.zst
  DEPMOD  /lib/modules/5.19.6-arch1-1-t2
make[1]: Leaving directory '/usr/lib/modules/5.19.6-arch1-1-t2/build'
depmod -a

contents of /lib/modules/5.19.6-arch1-1-t2/updates
total 412
-rw-r--r-- 1 root root 418214 Jan  3 17:20 snd-hda-codec-cs8409.ko.zst
```

Is this part an error?:

```
make[1]: Entering directory '/usr/lib/modules/5.19.6-arch1-1-t2/build'
  INSTALL /lib/modules/5.19.6-arch1-1-t2/updates/snd-hda-codec-cs8409.ko
  SIGN    /lib/modules/5.19.6-arch1-1-t2/updates/snd-hda-codec-cs8409.ko
scripts/sign-file: error while loading shared libraries: libcrypto.so.1.1: cannot open shared object file: No such file or directory
  ZSTD    /lib/modules/5.19.6-arch1-1-t2/updates/snd-hda-codec-cs8409.ko.zst
  DEPMOD  /lib/modules/5.19.6-arch1-1-t2
make[1]: Leaving directory '/usr/lib/modules/5.19.6-arch1-1-t2/build'
depmod -a
```

It seems to finish compiling and pulse volume control is showing that sound is coming from Firefox but its not coming out of the speakers. I've checked if anything is muted and its not.

I'm sorry to have to ask but I'd really appreciate someone pointing out if I'm doing something wrong here.

lukehayes, 2023-01-03:
I installed `openssl-1.1` to see if that helped and now I get this error:

```
Hunk #1 succeeded at 1444 (offset 174 lines).
Hunk #2 succeeded at 1453 (offset 174 lines).
Hunk #3 succeeded at 1482 (offset 174 lines).
patching file patch_cs8409.h
Hunk #2 succeeded at 303 (offset 1 line).
Hunk #3 succeeded at 330 (offset 1 line).
Hunk #4 succeeded at 366 (offset 1 line).
patching file patch_cirrus_apple.h
make -C /lib/modules/5.19.6-arch1-1-t2/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/luke/snd_hda_macbookpro/build/hda-5.19.6 modules
make[1]: Entering directory '/usr/lib/modules/5.19.6-arch1-1-t2/build'
  CC [M]  /home/luke/snd_hda_macbookpro/build/hda-5.19.6/patch_cs8409.o
In file included from /home/luke/snd_hda_macbookpro/build/hda-5.19.6/patch_cs8409.c:1488:
/home/luke/snd_hda_macbookpro/build/hda-5.19.6/patch_cirrus_apple.h: In function ���cs_8409_apple_boot_init���:
/home/luke/snd_hda_macbookpro/build/hda-5.19.6/patch_cirrus_apple.h:1191:15: warning: assignment discards ���const��� qualifier from pointer target type [-Wdiscarded-qualifiers]
 1191 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
  CC [M]  /home/luke/snd_hda_macbookpro/build/hda-5.19.6/patch_cs8409-tables.o
  LD [M]  /home/luke/snd_hda_macbookpro/build/hda-5.19.6/snd-hda-codec-cs8409.o
  MODPOST /home/luke/snd_hda_macbookpro/build/hda-5.19.6/Module.symvers
  CC [M]  /home/luke/snd_hda_macbookpro/build/hda-5.19.6/snd-hda-codec-cs8409.mod.o
  LD [M]  /home/luke/snd_hda_macbookpro/build/hda-5.19.6/snd-hda-codec-cs8409.ko
  BTF [M] /home/luke/snd_hda_macbookpro/build/hda-5.19.6/snd-hda-codec-cs8409.ko
make[1]: Leaving directory '/usr/lib/modules/5.19.6-arch1-1-t2/build'
#cp snd-hda-codec-cs8409.ko /lib/modules/5.19.6-arch1-1-t2/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.19.6-arch1-1-t2/build M=/home/luke/snd_hda_macbookpro/build/hda-5.19.6 modules_install
make[1]: Entering directory '/usr/lib/modules/5.19.6-arch1-1-t2/build'
  INSTALL /lib/modules/5.19.6-arch1-1-t2/updates/snd-hda-codec-cs8409.ko
  SIGN    /lib/modules/5.19.6-arch1-1-t2/updates/snd-hda-codec-cs8409.ko
At main.c:167:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: ./certs/signing_key.pem: No such file or directory
  ZSTD    /lib/modules/5.19.6-arch1-1-t2/updates/snd-hda-codec-cs8409.ko.zst
  DEPMOD  /lib/modules/5.19.6-arch1-1-t2
make[1]: Leaving directory '/usr/lib/modules/5.19.6-arch1-1-t2/build'
depmod -a

```

davidjo, 2023-01-03:
The patches have worked, the compilation has worked (it is a compiler warning error which are not fatal).
(A fatal compiler error would stop before the INSTALL section).
The signing errors are known errors as I dont try and install the kernel signing system (but unsigned kernel modules still work - and actually I have no idea what needs to be done to add kernel signing - apart from the fact you need to install some crypto libraries which no distribution seems to install by default hence the missing file errors).
What I dont get is the ZSTD and the zst extension - I dont see that under Ubuntu.
Is this kernel compression? (Apparently ZST is a compression format but have no idea why its applying it to kernel modules).
What distribution is this?

davidjo, 2023-01-03:
Also can you give some details about what has changed from your initial install - which kernel version, which distribution version?

Also check your GUI settings for audio (under ubuntu this is via the Settings app) and make sure its set to Stereo Audio for speakers.

lukehayes, 2023-01-04:
> What distribution is this?

I'm using EndeavourOS but I am using the version built for T2-linux macs. I don't have a mac with a T2 chip but I belive most of the other chips/firmware are the same hence thats why I'm using it. I'm using a 2019 iMac 27 inch, I believe its model number is 19,1.

> Also can you give some details about what has changed from your initial install - which kernel version, which distribution version?

I updated to the latest version of EndeavourOS (T2 Version again) with the 6.1.1 kernel and that stopped the wifi working so I needed to completely reinstall the OS to a version where I knew that worked so I installed EndeavourOS with kernel version `5.19.6-arch1-t2` and thats what I'm currently using and the one I'm struggling to get the sound working on. Its possible that I was on kernel 6.0.* (possibly kernel 6.0.12.arch1-1) before I updated to the breaking kernelbut it was definately before 6.1.1.

I should mention that I'm trying to get the wifi working with the latest kernel version so this repo may compile successfully using that but I wont be able to test for a while.

> Also check your GUI settings for audio (under ubuntu this is via the Settings app) and make sure its set to Stereo Audio for speakers.

Pulse audio volume control is definately showing audio coming from firefox/youtube in the meter it's just not coming out of the speakers. I'm on my laptop at the moment so I can't check right now, but I will be able to in 2 hours time :+1:

lukehayes, 2023-01-04:
I have just upgraded to  kernel version `6.1.2-arch1-1-t2` to test getting the wifi working again and I also recompiled this project just to see and for some reason that has fixed the sound and its all working as it was previously!

Thanks again for your work on this project - I will close this issue if you're happy for it to be closed.

davidjo, 2023-01-05:
The very recent commits fixed handling kernel major versions greater than 5 (ie 6 or more) - my kernel version checks had an error which meant it would use the kernel version fixups for pre 5.19/515.0-47 ubuntu if the major version was greater than 5 rather than the current version - which is known to work for 5.19/5.15.0-47  - note I just have current Ubuntu LTS so I cant check later kernel versions but 6 doesnt seem to have broken anything so far based on comments from others.

Ive just discovered a possible way to prevent attempt at kernel signing so hopefully the signing errors will go away - it will still be unsigned.

Curious about wifi - I also had issues on my MacBook Pro 14,3 but I found somewhere on the linux on mac sites a fixup to the Broadcom wifi chip firmware (under /lib/firmware/brcm) which Im using - but your wifi chip is likely different.

Sure you can close it.