davidjo/snd_hda_macbookpro #80: Driver fails to load - snd_hda_codec_cs8409 hdaudioC0D0: UNKNOWN subsystem id 0x10138409

flavian112, 2023-01-10:
Hi, 
first of all thank you for your time and efforts spent on this project.
Somehow I can't get this driver to work, it all compiles and installs fine, however the sound is not working and I'm kinda stuck in finding out where the problem may lay. I'm using a **MacbookPro14,3** with kernel version: **6.1.4-arch1-1** (Arch Linux) - it didn't work for the previous few version either though. Unfortunately I can't remember when it stopped working.

Here's the output from `dmesg | grep snd_hda` and also i've attached the output from `alsa-info.sh` ([alsa-info.txt](https://github.com/davidjo/snd_hda_macbookpro/files/10381597/alsa-info.txt)):
```
[   12.599548] snd_hda_intel 0000:00:1f.3: enabling device (0000 -> 0002)
[   12.599887] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
[   12.599951] snd_hda_intel 0000:01:00.1: Force to non-snoop mode
[   14.042164] snd_hda_intel: Primary patch_cs8409
[   14.042168] snd_hda_intel: Primary patch_cs8409 NOT FOUND trying APPLE
[   14.042172] snd_hda_codec_cs8409 hdaudioC0D0: UNKNOWN subsystem id 0x10138409
[   14.059946] snd_hda_intel: Primary patch_cs8409
[   14.059949] snd_hda_intel: Primary patch_cs8409 NOT FOUND trying APPLE
[   14.059953] snd_hda_codec_cs8409 hdaudioC0D0: UNKNOWN subsystem id 0x10138409
[   14.067519] snd_hda_codec_generic hdaudioC0D0: autoconfig for Generic: line_outs=0 (0x0/0x0/0x0/0x0/0x0) type:line
[   14.067525] snd_hda_codec_generic hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   14.067528] snd_hda_codec_generic hdaudioC0D0:    hp_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   14.067530] snd_hda_codec_generic hdaudioC0D0:    mono: mono_out=0x0
[   14.067532] snd_hda_codec_generic hdaudioC0D0:    inputs:
[   14.152652] snd_hda_intel 0000:01:00.1: bound 0000:01:00.0 (ops amdgpu_dm_audio_component_bind_ops [amdgpu])
```
As it seems the driver does not recognize the subsystem id `0x10138409` (I thought this was the device id not the subsystem id). From reading through the driver code I've seen there are a few switches depending on the subsystem id. Here for example in **patch_cirrus_apple.h** on line 2537. I would assume, correct me if I'm wrong, that the correct subsystem id then should be `0x10138409` instead of `0x106b3900`.
```
if (codec->core.subsystem_id == 0x106b3900) {
                // macbook pro 14,3
                fixup_found = 1;
}
```
I wonder If anybody else has the same problem or if it is just a weird fluke on my machine. The soundcard not working in my case is not critical since I use bluetooth headphones most of the time.  I'm rather new to linux and my understanding of it's inner workings might be a bit lacking, however I'd be glad giving any (hopefully) helpful debuging info.

Flavian

davidjo, 2023-01-10:
See issue #66 which had what looks like the same issue.

Basically 0x10138409 is never a subsystem id - as you say its the device id.

The suggestion from #66 is that its some boot issue - some corruption which leads to this invalid subsystem id.

However, as you are using 6.1 there may have been some kernel change - although its clearly using the patched module - otherwise you wouldnt see trying APPLE.

flavian112, 2023-01-10:
Thank you for your reply. Seems like I have overlooked issue #66 (apologies for your inconveniences). I have reset the nvram and booting with the ALT key pressed and then selecting Linux indeed seems to work.

I've attached a dump of the efivars when booted with and without the ALT key (sound working / sound not working). The variables `BootCurrent`, `GR_CAUSE`, `HW_BOOT_DATA` and `MTC` have changed.  Guessing by the name of the variables HW_BOOT_DATA might have something to do with it but I have to do some further investigation.

[efivars_sound_working.txt](https://github.com/davidjo/snd_hda_macbookpro/files/10386513/efivars_sound_working.txt)
[efivars_sound_not_working.txt](https://github.com/davidjo/snd_hda_macbookpro/files/10386516/efivars_sound_not_working.txt)

davidjo, 2023-01-11:
Thats good - from this I pretty certain its not a kernel version issue.

KiLLeRRaT, 2024-08-27:
> Thank you for your reply. Seems like I have overlooked issue #66 (apologies for your inconveniences). I have reset the nvram and booting with the ALT key pressed and then selecting Linux indeed [...]

Hi @flavian112 did you ever find a better and permanent solution to this?

It works for me too when I use ALT/Option key on startup to boot, but it's a little less than ideal.

KiLLeRRaT, 2024-10-21:
I have since figured something out that works for me, so that I don't have to hold the option key boot anymore, see my post here: https://github.com/davidjo/snd_hda_macbookpro/issues/134#issuecomment-2426011312