davidjo/snd_hda_macbookpro #82: Lubuntu no sound - System.Map

aadcloud, 2023-02-04:
Hello,

I really really hope you can help me. I've got this working on Ubuntu before and also Fedora with some tweaking. I really want to use this on Lubuntu, just because of how lightweight the system is. I would really appreciate your help here.

I've followed on the usual setup instructions for Ubuntu, as that would be the correct thing to do for Lubuntu (I assume).

Having looked at someone who has an issue with Kubuntu (me thinking it maybe similar for Lubuntu), also ran these commands:
sudo apt install build-essential
sudo apt install linux-source

Below is the errors I'm getting. Like I said, I really would appreciate some help here :) Thank you
-------------------------------------------------------------------------------------------------
aaron@jarael-local:~/Downloads/snd_hda_macbookpro$ sudo ./install.cirrus.driver.sh
[sudo] password for aaron: 
linux-source-5.19.0/sound/pci/hda/
linux-source-5.19.0/sound/pci/hda/hda_eld.c
linux-source-5.19.0/sound/pci/hda/hda_beep.c
linux-source-5.19.0/sound/pci/hda/hda_hwdep.c
linux-source-5.19.0/sound/pci/hda/hda_auto_parser.h
linux-source-5.19.0/sound/pci/hda/hda_tegra.c
linux-source-5.19.0/sound/pci/hda/hda_bind.c
linux-source-5.19.0/sound/pci/hda/patch_analog.c
linux-source-5.19.0/sound/pci/hda/ideapad_s740_helper.c
linux-source-5.19.0/sound/pci/hda/patch_cs8409.h
linux-source-5.19.0/sound/pci/hda/hp_x360_helper.c
linux-source-5.19.0/sound/pci/hda/patch_conexant.c
linux-source-5.19.0/sound/pci/hda/hda_jack.c
linux-source-5.19.0/sound/pci/hda/Kconfig
linux-source-5.19.0/sound/pci/hda/hda_generic.h
linux-source-5.19.0/sound/pci/hda/patch_ca0110.c
linux-source-5.19.0/sound/pci/hda/patch_realtek.c
linux-source-5.19.0/sound/pci/hda/patch_ca0132.c
linux-source-5.19.0/sound/pci/hda/patch_si3054.c
linux-source-5.19.0/sound/pci/hda/patch_cs8409.c
linux-source-5.19.0/sound/pci/hda/cs35l41_hda.c
linux-source-5.19.0/sound/pci/hda/hda_auto_parser.c
linux-source-5.19.0/sound/pci/hda/patch_sigmatel.c
linux-source-5.19.0/sound/pci/hda/hda_intel.h
linux-source-5.19.0/sound/pci/hda/hda_codec.c
linux-source-5.19.0/sound/pci/hda/patch_cmedia.c
linux-source-5.19.0/sound/pci/hda/hda_intel.c
linux-source-5.19.0/sound/pci/hda/patch_via.c
linux-source-5.19.0/sound/pci/hda/hda_intel_trace.h
linux-source-5.19.0/sound/pci/hda/hda_component.h
linux-source-5.19.0/sound/pci/hda/patch_cirrus.c
linux-source-5.19.0/sound/pci/hda/cs35l41_hda.h
linux-source-5.19.0/sound/pci/hda/patch_cs8409-tables.c
linux-source-5.19.0/sound/pci/hda/hda_controller.c
linux-source-5.19.0/sound/pci/hda/patch_hdmi.c
linux-source-5.19.0/sound/pci/hda/hda_proc.c
linux-source-5.19.0/sound/pci/hda/hda_beep.h
linux-source-5.19.0/sound/pci/hda/hda_local.h
linux-source-5.19.0/sound/pci/hda/ca0132_regs.h
linux-source-5.19.0/sound/pci/hda/hda_controller.h
linux-source-5.19.0/sound/pci/hda/hda_generic.c
linux-source-5.19.0/sound/pci/hda/hda_jack.h
linux-source-5.19.0/sound/pci/hda/hda_cs_dsp_ctl.h
linux-source-5.19.0/sound/pci/hda/hda_sysfs.c
linux-source-5.19.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-source-5.19.0/sound/pci/hda/thinkpad_helper.c
linux-source-5.19.0/sound/pci/hda/Makefile
linux-source-5.19.0/sound/pci/hda/hda_cs_dsp_ctl.c
linux-source-5.19.0/sound/pci/hda/cs35l41_hda_spi.c
linux-source-5.19.0/sound/pci/hda/hda_controller_trace.h
Kernel version later than implemented version - there may be build problems
patching file patch_cs8409.c
Hunk #1 succeeded at 1444 (offset 174 lines).
Hunk #2 succeeded at 1453 (offset 174 lines).
Hunk #3 succeeded at 1482 (offset 174 lines).
patching file patch_cs8409.h
Hunk #2 succeeded at 303 (offset 1 line).
Hunk #3 succeeded at 330 (offset 1 line).
Hunk #4 succeeded at 366 (offset 1 line).
patching file patch_cirrus_apple.h
make -C /lib/modules/5.19.0-21-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/aaron/Downloads/snd_hda_macbookpro/build/hda-5.19.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-21-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.2.0-3ubuntu1) 12.2.0
  You are using:           gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0
  CC [M]  /home/aaron/Downloads/snd_hda_macbookpro/build/hda-5.19.0/patch_cs8409.o
In file included from /home/aaron/Downloads/snd_hda_macbookpro/build/hda-5.19.0/patch_cs8409.c:1488:
/home/aaron/Downloads/snd_hda_macbookpro/build/hda-5.19.0/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/home/aaron/Downloads/snd_hda_macbookpro/build/hda-5.19.0/patch_cirrus_apple.h:1191:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1191 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
  CC [M]  /home/aaron/Downloads/snd_hda_macbookpro/build/hda-5.19.0/patch_cs8409-tables.o
  LD [M]  /home/aaron/Downloads/snd_hda_macbookpro/build/hda-5.19.0/snd-hda-codec-cs8409.o
  MODPOST /home/aaron/Downloads/snd_hda_macbookpro/build/hda-5.19.0/Module.symvers
  CC [M]  /home/aaron/Downloads/snd_hda_macbookpro/build/hda-5.19.0/snd-hda-codec-cs8409.mod.o
  LD [M]  /home/aaron/Downloads/snd_hda_macbookpro/build/hda-5.19.0/snd-hda-codec-cs8409.ko
  BTF [M] /home/aaron/Downloads/snd_hda_macbookpro/build/hda-5.19.0/snd-hda-codec-cs8409.ko
Skipping BTF generation for /home/aaron/Downloads/snd_hda_macbookpro/build/hda-5.19.0/snd-hda-codec-cs8409.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-21-generic'
#cp snd-hda-codec-cs8409.ko /lib/modules/5.19.0-21-generic/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/5.19.0-21-generic/build M=/home/aaron/Downloads/snd_hda_macbookpro/build/hda-5.19.0 CONFIG_MODULE_SIG_ALL=n modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-21-generic'
  INSTALL /lib/modules/5.19.0-21-generic/updates/snd-hda-codec-cs8409.ko
  DEPMOD  /lib/modules/5.19.0-21-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-21-generic'
depmod -a

contents of /lib/modules/5.19.0-21-generic/updates
total 1524
-rw-r--r-- 1 root root 1558904 feb  4 15:37 snd-hda-codec-cs8409.ko

davidjo, 2023-02-04:
Actually there are no errors here - and the warning messages are those expected.
Kernel version later than implemented version - there may be build problems 
is a new message indicating I havent tested with your kernel version.
So it ends listing the kernel module snd-hda-codec-cs8409.ko in /lib/modules/5.19.0-21-generic/updates - it should work.
What doesnt work?

aadcloud, 2023-02-05:
I appreciate you looking into this. I've ended up just installing Lubuntu 22.04 LTS which is using the same linux kernel your using. Now I'm just getting this hunk error. Thank you so much for your help so far!

aaron@david-local:~/Downloads/snd_hda_macbookpro$ sudo ./install.cirrus.driver.sh
linux-source-5.15.0/sound/pci/hda/
linux-source-5.15.0/sound/pci/hda/hda_eld.c
linux-source-5.15.0/sound/pci/hda/hda_beep.c
linux-source-5.15.0/sound/pci/hda/hda_hwdep.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.h
linux-source-5.15.0/sound/pci/hda/hda_tegra.c
linux-source-5.15.0/sound/pci/hda/hda_bind.c
linux-source-5.15.0/sound/pci/hda/patch_analog.c
linux-source-5.15.0/sound/pci/hda/ideapad_s740_helper.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.h
linux-source-5.15.0/sound/pci/hda/hp_x360_helper.c
linux-source-5.15.0/sound/pci/hda/patch_conexant.c
linux-source-5.15.0/sound/pci/hda/hda_jack.c
linux-source-5.15.0/sound/pci/hda/Kconfig
linux-source-5.15.0/sound/pci/hda/hda_generic.h
linux-source-5.15.0/sound/pci/hda/patch_ca0110.c
linux-source-5.15.0/sound/pci/hda/patch_realtek.c
linux-source-5.15.0/sound/pci/hda/patch_ca0132.c
linux-source-5.15.0/sound/pci/hda/patch_si3054.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.c
linux-source-5.15.0/sound/pci/hda/patch_sigmatel.c
linux-source-5.15.0/sound/pci/hda/hda_intel.h
linux-source-5.15.0/sound/pci/hda/hda_codec.c
linux-source-5.15.0/sound/pci/hda/patch_cmedia.c
linux-source-5.15.0/sound/pci/hda/hda_intel.c
linux-source-5.15.0/sound/pci/hda/patch_via.c
linux-source-5.15.0/sound/pci/hda/hda_intel_trace.h
linux-source-5.15.0/sound/pci/hda/hda_component.h
linux-source-5.15.0/sound/pci/hda/patch_cirrus.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.h
linux-source-5.15.0/sound/pci/hda/patch_cs8409-tables.c
linux-source-5.15.0/sound/pci/hda/hda_controller.c
linux-source-5.15.0/sound/pci/hda/patch_hdmi.c
linux-source-5.15.0/sound/pci/hda/hda_proc.c
linux-source-5.15.0/sound/pci/hda/hda_beep.h
linux-source-5.15.0/sound/pci/hda/hda_local.h
linux-source-5.15.0/sound/pci/hda/ca0132_regs.h
linux-source-5.15.0/sound/pci/hda/hda_controller.h
linux-source-5.15.0/sound/pci/hda/hda_generic.c
linux-source-5.15.0/sound/pci/hda/hda_jack.h
linux-source-5.15.0/sound/pci/hda/hda_sysfs.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-source-5.15.0/sound/pci/hda/thinkpad_helper.c
linux-source-5.15.0/sound/pci/hda/Makefile
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_spi.c
linux-source-5.15.0/sound/pci/hda/hda_controller_trace.h
patching file patch_cs8409.c
Hunk #1 succeeded at 1442 (offset 172 lines).
Hunk #2 succeeded at 1451 (offset 172 lines).
Hunk #3 succeeded at 1480 (offset 172 lines).
patching file patch_cs8409.h
Hunk #1 succeeded at 19 (offset 1 line).
Hunk #2 succeeded at 302 (offset 1 line).
Hunk #3 FAILED at 328.
Hunk #4 succeeded at 362 (offset 2 lines).
1 out of 4 hunks FAILED -- saving rejects to file patch_cs8409.h.rej

The patch reject file has this inside
----------------------------------------
--- patch_cs8409.h
+++ patch_cs8409.h
@@ -328,6 +351,9 @@ struct sub_codec {
 
 	unsigned int hp_jack_in:1;
 	unsigned int mic_jack_in:1;
+#ifdef APPLE_CODECS
+	unsigned int linein_jack_in:1;
+#endif
 	unsigned int force_status_change:1;
 	unsigned int suspended:1;
 	unsigned int paged:1;

davidjo, 2023-02-06:
Whats the kernel version here - I need the minor number - thats the x in  5.15.0-x (so would be 21 for your original 5.19.0-21).
There was a previous that occurred around 40 minor version but the above error doesnt seem to be the same.