davidjo/snd_hda_macbookpro #84: Not working on Ubuntu 22.04.2

GreenYellowLight, 2023-02-20:
Previously I have been able to use this. I updated to the new Ubuntu version a few hours ago and when I tried to reinstall this it did not work and I got this error:

linux-source-5.19.0/sound/pci/hda/
linux-source-5.19.0/sound/pci/hda/hda_controller.h
linux-source-5.19.0/sound/pci/hda/hda_tegra.c
linux-source-5.19.0/sound/pci/hda/ideapad_s740_helper.c
linux-source-5.19.0/sound/pci/hda/patch_cs8409-tables.c
linux-source-5.19.0/sound/pci/hda/cs35l41_hda_spi.c
linux-source-5.19.0/sound/pci/hda/hda_component.h
linux-source-5.19.0/sound/pci/hda/patch_conexant.c
linux-source-5.19.0/sound/pci/hda/hp_x360_helper.c
linux-source-5.19.0/sound/pci/hda/hda_intel.h
linux-source-5.19.0/sound/pci/hda/hda_sysfs.c
linux-source-5.19.0/sound/pci/hda/hda_beep.h
linux-source-5.19.0/sound/pci/hda/patch_cs8409.h
linux-source-5.19.0/sound/pci/hda/cs35l41_hda.c
linux-source-5.19.0/sound/pci/hda/hda_local.h
linux-source-5.19.0/sound/pci/hda/hda_controller_trace.h
linux-source-5.19.0/sound/pci/hda/patch_analog.c
linux-source-5.19.0/sound/pci/hda/ca0132_regs.h
linux-source-5.19.0/sound/pci/hda/hda_jack.h
linux-source-5.19.0/sound/pci/hda/hda_beep.c
linux-source-5.19.0/sound/pci/hda/patch_cirrus.c
linux-source-5.19.0/sound/pci/hda/thinkpad_helper.c
linux-source-5.19.0/sound/pci/hda/patch_ca0132.c
linux-source-5.19.0/sound/pci/hda/patch_via.c
linux-source-5.19.0/sound/pci/hda/hda_jack.c
linux-source-5.19.0/sound/pci/hda/Makefile
linux-source-5.19.0/sound/pci/hda/cs35l41_hda.h
linux-source-5.19.0/sound/pci/hda/hda_cs_dsp_ctl.c
linux-source-5.19.0/sound/pci/hda/Kconfig
linux-source-5.19.0/sound/pci/hda/hda_intel_trace.h
linux-source-5.19.0/sound/pci/hda/hda_proc.c
linux-source-5.19.0/sound/pci/hda/hda_auto_parser.h
linux-source-5.19.0/sound/pci/hda/patch_cs8409.c
linux-source-5.19.0/sound/pci/hda/patch_realtek.c
linux-source-5.19.0/sound/pci/hda/hda_bind.c
linux-source-5.19.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-source-5.19.0/sound/pci/hda/hda_intel.c
linux-source-5.19.0/sound/pci/hda/patch_si3054.c
linux-source-5.19.0/sound/pci/hda/hda_auto_parser.c
linux-source-5.19.0/sound/pci/hda/hda_generic.c
linux-source-5.19.0/sound/pci/hda/hda_controller.c
linux-source-5.19.0/sound/pci/hda/hda_generic.h
linux-source-5.19.0/sound/pci/hda/patch_cmedia.c
linux-source-5.19.0/sound/pci/hda/hda_cs_dsp_ctl.h
linux-source-5.19.0/sound/pci/hda/hda_eld.c
linux-source-5.19.0/sound/pci/hda/hda_codec.c
linux-source-5.19.0/sound/pci/hda/patch_hdmi.c
linux-source-5.19.0/sound/pci/hda/patch_sigmatel.c
linux-source-5.19.0/sound/pci/hda/hda_hwdep.c
linux-source-5.19.0/sound/pci/hda/patch_ca0110.c
Kernel version later than implemented version - there may be build problems
patching file patch_cs8409.c
Hunk #1 succeeded at 1444 (offset 174 lines).
Hunk #2 succeeded at 1453 (offset 174 lines).
Hunk #3 succeeded at 1482 (offset 174 lines).
patching file patch_cs8409.h
Hunk #2 succeeded at 303 (offset 1 line).
Hunk #3 succeeded at 330 (offset 1 line).
Hunk #4 succeeded at 366 (offset 1 line).
patching file patch_cirrus_apple.h
make -C /lib/modules/5.19.0-32-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/linden/Documents/Drivers/snd_hda_macbookpro - Audio/build/hda-5.19.0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-32-generic'
make[1]: *** No rule to make target 'Audio/build/hda-5.19.0'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-32-generic'
make: *** [Makefile:23: all] Error 2

davidjo, 2023-02-22:
Unfortunately I think this is a blank in file name issue - you have " - " in  the downloaded repo file name - hence the error No rule to make 'Audio/build/...'  - instead of /home/linden/Documents/Drivers/snd_hda_macbookpro - Audio/build/hda-5.19.0 modules.
In general blanks in file names in linux are not handled well - and even if I fix the install script to handle blanks the kernel Makefile  scripts called by the install script cant handle them.
So all in all its just better not to use blanks in file names.

GreenYellowLight, 2023-02-22:
Thanks! Renaming the folder fixed it

closed 2023-02-22 by GreenYellowLight