davidjo/snd_hda_macbookpro #85: nixOS install

bpmct, 2023-04-09:
I'm using nixOS and running into a few issues with the install script. Has anybody else got this working?

davidjo, 2023-04-09:
What sort of issues?
There are a  number of issues to sort for a new distribution.
First is exactly what kernel version are you running (kernel version 6 not really handled),  is this a genuine mainline kernel or (eg Ubuntu) a quite modified mainline?
You need a kernel source file, kernel headers plus full build system (enough to build kernel modules) including the patch command now (eg Fedora doesnt seem to install patch as part of the main build packages).
The script needs to know where these are installed and unless the file system structure is similar to eg Ubuntu its going to need modifying, eg are kernel modules under /lib/modules.
At the moment the script only knows about Ubuntu kernel source packages and downloads a mainline kernel source for all other distributions from the main linux kernel website - which may not be correct for nixOS.
The script takes different paths depending on the kernel version, again only Ubuntu kernel versioning and mainline kernel versions are handled.

snoweuph, 2025-07-18:
@bpmct  if you still strugle with this on nixos, you could try my config 
https://github.com/davidjo/snd_hda_macbookpro/issues/153#issuecomment-3088737866